RoboStack/ros-noetic #275: Contributing page

gftabor, 2022-06-16:
Looks like the contributing page is old. 
https://github.com/RoboStack/ros-noetic/blob/master/CONTRIBUTING.md

Step 9 specifically
```
Copy the generated recipe to the current folder: cp recipes/ros-noetic-XXX.yaml recipe.yaml - note that at least one package needs to be (re)build for this folder to show up. See more info below.
```
looks like there was a reorganization from recipes/ros-noetic-XXX.yaml to recipes/ros-noetic-XXX/recipe.yaml at some point.

Also seems like there is a lot of build output files that are added inside the git repo. Can this be reorganized so all build outputs (copying vinca file, vinca outputs, boa outputs) are all in a dedicated build directory that can be added to gitignore.

gftabor, 2022-06-16:
Looks like step 9 can just be
cd to recipes folder

and step 10 would be (just change paths to go up directory)
```
boa build . -m ../.ci_support/conda_forge_pinnings.yaml -m ../conda_build_config.yaml
```
and it will run through folders in recipes and build them.

Tobias-Fischer, 2022-06-24:
Are you happy to create a PR with these changes @gftabor?

gftabor, 2022-06-24:
Ya I plan on it. Just had 1 question, is there a better mechanism than copying vinca_linux to vinca, changing stuff, testing, copying changes back?

Specifying vinca file by name? 

I want to overhaul the gitignore to make the contributing experience easier as well.

Tobias-Fischer, 2022-06-24:
I personally create a symlink between vinca.yaml and vinca_linux64.yaml 

Thanks re gitignore, great idea!

gftabor, 2022-06-24:
> I personally create a symlink between vinca.yaml and vinca_linux64.yaml

Good point. In which case I'll add that as a suggestion on contributing page, with command to run.

closed 2022-08-25 by gftabor