RoboStack/ros-noetic #276: Outdated rtabmap_ros package for python 3.8

SgtVincent, 2022-06-21:
Hi, thanks for your great framework! It really saves my life after struggling with conda and ROS for weeks. 

I had a question about rtabmap_ros versions with different versions of python. 

```
$ mamba search rtabmap-ros
...
ros-noetic-rtabmap-ros         0.20.14 py38h8fc0257_12  robostack           
ros-noetic-rtabmap-ros         0.20.18 py39h8f8ee9e_14  robostack 
```
It seems that with python 3.8 I can only install  ros-noetic-rtabmap-ros=0.20.14, which lacks some features compared to ver. 0.20.18, according to rtabmap release note: https://github.com/introlab/rtabmap/releases/tag/0.20.16. 

I also looked into the tutorial of [Adding new packages via pull requests](https://robostack.github.io/Contributing.html), but what I understand is that it will get built with python=3.9 in your CI. 

Is there anyway I can install ros-noetic-rtabmap-ros=0.20.18 with python=3.8?

Thanks a million!

Tobias-Fischer, 2022-06-21:
You can locally build the package for py38 - see https://github.com/RoboStack/ros-noetic/blob/master/CONTRIBUTING.md#testing-changes-locally

To get py38 packages, I recommend checking out the commit https://github.com/RoboStack/ros-noetic/commit/c93937ac64b6295c8c28cabc2a8efeb518a2482a which was before the change to py39

May I ask why you don't want to use the new py39 packages?

Let us know if you have any issues.

SgtVincent, 2022-06-21:
Thanks for your swift reply! I will try your suggestion and get back later. 

> May I ask why you don't want to use the new py39 packages?

I am using a simulator pre-built with python>=3.6, python<=3.8. I need to either build that simulator from the source with python=3.9 or build rtabmap_ros from the source with python 3.8. 

Thanks again!

closed 2022-06-21 by SgtVincent