RoboStack/ros-noetic #277: librviz_default_plugin.dylib Error when RVIZ runs on MacOS Catalina with Conda

zzningxp, 2022-06-23:
My issues:

Environment
OS Version: Mac Catalina
ROS Distro: Noetic

Installation:
```
conda create -n ROS python==3.8
conda activate ROS
conda config --add channels conda-forge
conda config --add channels robostack
conda config --set channel_priority strict
conda install ros-noetic-desktop-full
conda install compilers cmake pkg-config make ninja catkin_tools
conda deactivate
conda activate ROS
rosdep init
rosdep update
rviz

```

RViz, Qt, OGRE, OpenGl version as printed by rviz:

```# rviz
[ INFO] [1655954814.977154000]: rviz version 1.14.6
[ INFO] [1655954814.977211000]: compiled against Qt version 5.12.9
[ INFO] [1655954814.977221000]: compiled against OGRE version 1.10.12 (Xalafu)
[ INFO] [1655954815.017990000]: Forcing OpenGl version 0.
[ INFO] [1655954816.113320000]: Stereo is NOT SUPPORTED
[ INFO] [1655954816.113386000]: OpenGL device: AMD Radeon Pro 560X OpenGL Engine
[ INFO] [1655954816.113403000]: OpenGl version: 2.1 (GLSL 1.2).
[ERROR] [1655954816.339471000]: PluginlibFactory: The plugin for class 'rviz/Orbit' failed to load.  Error: Failed to load library /Users/d/opt/anaconda3/envs/ROS/lib/librviz_default_plugin.dylib. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = dlopen(/Users/d/opt/anaconda3/envs/ROS/lib/librviz_default_plugin.dylib, 9): Library not loaded: @rpath/liburdfdom_model.1.0.dylib
  Referenced from: /Users/d/opt/anaconda3/envs/ROS/lib/liburdf.dylib
  Reason: image not found)
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame base_footprint at time 1655954817.627902 according to authority unknown_publisher
         at line 278 in /Users/runner/miniforge3/conda-bld/ros-noetic-tf2_1615119151908/work/ros-noetic-tf2/src/work/src/buffer_core.cpp```

Tobias-Fischer, 2022-06-23:
What's the output of conda list?

zzningxp, 2022-06-23:
My conda is a brand new one.
However, my brew/system install a lot packages to try to install ROS from source but failed.

# conda env list
```
# conda environments:
#
base                  *  /Users/d/opt/anaconda3
ROS                      /Users/d/opt/anaconda3/envs/ROS
```
# conda list:
```
# packages in environment at /Users/d/opt/anaconda3/envs/ROS:
#
# Name                    Version                   Build  Channel
adwaita-icon-theme        41.0                 h694c41f_1    conda-forge
apr                       1.7.0                h0d85af4_5    conda-forge
assimp                    5.0.1                h1224e73_7    conda-forge
atk-1.0                   2.36.0               he69c4ee_4    conda-forge
bcrypt                    3.2.2            py38h0dd4459_0    conda-forge
boost                     1.74.0           py38hb0f0857_5    conda-forge
boost-cpp                 1.74.0               he5d75e3_0    conda-forge
brotli                    1.0.9                h5eb16cf_7    conda-forge
brotli-bin                1.0.9                h5eb16cf_7    conda-forge
bzip2                     1.0.8                h0d85af4_4    conda-forge
c-ares                    1.18.1               h0d85af4_0    conda-forge
c-compiler                1.4.2                had99412_0    conda-forge
ca-certificates           2022.6.15            h033912b_0    conda-forge
cairo                     1.16.0            he8db5a5_1006    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.0              pyhd8ed1ab_0    conda-forge
cctools                   973.0.1             h351d84c_10    conda-forge
cctools_osx-64            973.0.1             h3eff9a4_10    conda-forge
certifi                   2022.6.15        py38h50d1736_0    conda-forge
cffi                      1.14.4           py38h6afc60c_0    conda-forge
clang                     13.0.1               h694c41f_0    conda-forge
clang-13                  13.0.1          default_he082bbe_0    conda-forge
clang_osx-64              13.0.1               h71a8856_2    conda-forge
clangxx                   13.0.1          default_he082bbe_0    conda-forge
clangxx_osx-64            13.0.1               heae0f87_2    conda-forge
cmake                     3.20.2               h32a7a09_0    conda-forge
compiler-rt               13.0.1               he01351e_0    conda-forge
compiler-rt_osx-64        13.0.1               hd3f61c9_0    conda-forge
compilers                 1.4.2                h694c41f_0    conda-forge
console_bridge            1.0.2                hbb4e6a2_1    conda-forge
coreutils                 9.1                  h5eb16cf_0    conda-forge
cppzmq                    4.8.1                hf09c4ce_0    conda-forge
cryptography              37.0.2           py38h0f8513e_0    conda-forge
curl                      7.76.1               h06286d4_1    conda-forge
cxx-compiler              1.4.2                h1b54a9f_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
dbus                      1.13.6               h0c50699_1    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.6.0              pyhd8ed1ab_0    conda-forge
docutils                  0.18.1           py38h50d1736_1    conda-forge
eigen                     3.4.0                h940c156_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
epoxy                     1.5.10               h5eb16cf_1    conda-forge
expat                     2.4.8                h96cf925_0    conda-forge
ffmpeg                    4.3.2                hbf27d7b_3    conda-forge
flann                     1.9.1             h4541d04_1008    conda-forge
fontconfig                2.13.1            h10f422b_1005    conda-forge
fonttools                 4.33.3           py38h0dd4459_0    conda-forge
fortran-compiler          1.4.2                h373f3de_0    conda-forge
freeimage                 3.18.0               h7d96ac2_7    conda-forge
freetype                  2.10.4               h4cff582_1    conda-forge
fribidi                   1.0.10               hbcb3906_0    conda-forge
gazebo                    11.3.0               h2a9d84c_3    conda-forge
gdk-pixbuf                2.42.2               h6a27155_1    conda-forge
gettext                   0.19.8.1          haf92f58_1004    conda-forge
[... 424 lines cut ...]
xorg-libxext              1.3.4                h0d85af4_1    conda-forge
xorg-libxfixes            5.0.3             h0d85af4_1004    conda-forge
xorg-libxi                1.7.10               h0d85af4_0    conda-forge
xorg-libxinerama          1.1.4             he49afe7_1001    conda-forge
xorg-libxmu               1.1.3                h0d85af4_0    conda-forge
xorg-libxpm               3.5.13               h0d85af4_0    conda-forge
xorg-libxrandr            1.5.2                h0d85af4_1    conda-forge
xorg-libxrender           0.9.10            h0d85af4_1003    conda-forge
xorg-libxt                1.2.1                h0d85af4_2    conda-forge
xorg-libxtst              1.2.3             h0d85af4_1002    conda-forge
xorg-randrproto           1.5.0             h0d85af4_1001    conda-forge
xorg-recordproto          1.14.2            h0d85af4_1002    conda-forge
xorg-renderproto          0.11.1            h0d85af4_1002    conda-forge
xorg-util-macros          1.19.3               h35c211d_0    conda-forge
[... cut ...]
xz                        5.2.5                haf1e3a3_1    conda-forge
yaml                      0.2.5                h0d85af4_2    conda-forge
yaml-cpp                  0.6.3                hb1e8313_4    conda-forge
zeromq                    4.3.4                he49afe7_1    conda-forge
zlib                      1.2.12               hfe4f2af_1    conda-forge
zstd                      1.4.9                h582d3a0_0    conda-forge
zziplib                   0.13.69              h5dbffcc_1    conda-forge
```

Tobias-Fischer, 2022-06-23:
Alternatively, try `mamba install urdfdom=1`

zzningxp, 2022-06-23:
> 

> @zzningxp can you try with `python==3.9`? All our most recent builds are using python 3.9 and hopefully the RViz issue is fixed with that version.

3.9 is not good...

```
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: -
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed

UnsatisfiableError: The following specifications were found
to be incompatible with the existing python installation in your environment:

Specifications:

  - ros-noetic-desktop-full -> python=3.8

Your python: python==3.9

If python is on the left-most side of the chain, that's the version you've asked for.
When python appears to the right, that indicates that the thing on the left is somehow
not available for the python version you are constrained to. Note that conda will not
change your python version to a different minor version unless you explicitly specify
that.

The following specifications were found to be incompatible with your system:

  - feature:/osx-64::__osx==10.15.7=0
  - ros-noetic-desktop-full -> __osx[version='>=10.14']

Your installed version is: 10.15.7

Note that strict channel priority may have removed packages required for satisfiability.
```

wolfv, 2022-06-23:
@traversaro I think we might not yet have `ros-noetic-desktop-full` available in the rebuild for macOS. But yes, `noetic-desktop` hopefully works (and also includes rviz!).

zzningxp, 2022-06-23:
> Alternatively, try `mamba install urdfdom=1`

@Tobias-Fischer  this make no different.
I think it maybe because of that conda list includes:
```
urdfdom                   3.1.0                hb486fe8_0    conda-forge
urdfdom_headers           1.1.0                h1b54a9f_0    conda-forge
```

Tobias-Fischer, 2022-06-23:
Well do you still have version 3.1.0 or did you downgrade to 1.0.4?

zzningxp, 2022-06-24:
Now I try this method:
```
mamba create -n robostackenv ros-noetic-desktop python=3.9 -c robostack -c robostack-experimental -c conda-forge --no-channel-priority --override-channels
```
But I get this:
```
......
zziplib                  [>                   ] (00m:12s) Couldn't resolve host name
```

Tobias-Fischer, 2022-06-24:
This seems to be an issue with your internet connection.

zzningxp, 2022-06-24:
Another problem:
```
# rosdep init
ERROR: cannot download default sources list from:
https://raw.githubusercontent.com/RoboStack/rosdistro/master/rosdep/sources.list.d/20-default.list
Website may be down.
```

This web site may be alive:
```
# os-specific listings first
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml osx
yaml https://raw.githubusercontent.com/RoboStack/ros-noetic/master/conda.yaml

# generic
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
gbpdistro https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml fuerte

# newer distributions (Groovy, Hydro, ...) must not be listed anymore, they are being fetched from the rosdistro index.yaml instead
```

zzningxp, 2022-06-24:
It works !! when I use this method with miniconda:
```
mamba create -n robostackenv ros-noetic-desktop python=3.9 -c robostack -c robostack-experimental -c conda-forge --no-channel-priority --override-channels
```

zzningxp, 2022-06-24:
When RViz runs, there comes errors like:
```
[ERROR] [1656048692.945950000]: Vertex Program:rviz/glsl120/indexed_8bit_image.vert Fragment Program:rviz/glsl120/indexed_8bit_image.frag GLSL link result : Validation Failed: Sampler error:
  Samplers of different types use the same texture image unit.
   - or -
  A sampler's texture unit is out of range (greater than max allowed or negative).
```

And the resolution ratio in the RViz window is very low, like some mosaic on the map...

closed 2022-06-27 by zzningxp

Tobias-Fischer, 2022-06-27:
How did you resolve the issue?

zzningxp, 2022-07-08:
> How did you resolve the issue?

Use miniconda instead of anaconda, and all the steps follow the "Installation" doc.