RoboStack/ros-noetic #279: MacOS - Gazebo: Cannot Add Model to World

rbrobots, 2022-06-30:
Hi, it's been a while since I've used ROS and I haven't used Gazebo much. 
I've installed ROS and Gazebo (11.10.2 ) on my Mac via RoboStack installations instructions in the README.

I can't seem to add a model to the world in Gazebo. I click on the model, left-click once and the model disappears, if I click twice it crashes. The client crashes quite easily as well.

Thanks for any help in advanced! :)

Here's my PC spec:

```
MacBook Pro (16-inch, 2019)
2.4 GHz 8-Core Intel Core i9
32 GB 2667 MHz DDR4
AMD Radeon Pro 5300M 4 GB
Intel UHD Graphics 630 1536 MB
```

conda list:

```
# Name                    Version                   Build  Channel
apr                       1.7.0                h0d85af4_5    conda-forge
assimp                    5.2.3                hbcca2e3_1    conda-forge
atk-1.0                   2.36.0               he69c4ee_4    conda-forge
bcrypt                    3.2.2            py39h701faf5_0    conda-forge
blosc                     1.21.1               h97e831e_3    conda-forge
boost                     1.74.0           py39ha1f3e3e_5    conda-forge
boost-cpp                 1.74.0               hdbf7018_7    conda-forge
brotli                    1.0.9                h5eb16cf_7    conda-forge
brotli-bin                1.0.9                h5eb16cf_7    conda-forge
bullet-cpp                3.21                 hbca1734_2    conda-forge
bzip2                     1.0.8                h0d85af4_4    conda-forge
c-ares                    1.18.1               h0d85af4_0    conda-forge
c-compiler                1.4.2                had99412_0    conda-forge
ca-certificates           2022.6.15            h033912b_0    conda-forge
cairo                     1.16.0            he01c77b_1009    conda-forge
catkin_pkg                0.5.0              pyhd8ed1ab_0    conda-forge
catkin_tools              0.8.5              pyhd8ed1ab_0    conda-forge
cctools                   973.0.1             hfc3f419_10    conda-forge
cctools_osx-64            973.0.1             h2b735b3_10    conda-forge
certifi                   2022.6.15        py39hecd8cb5_0
cffi                      1.15.0           py39he338e87_0    conda-forge
cfitsio                   4.1.0                h2c97ad1_0    conda-forge
clang                     13.0.1               h694c41f_0    conda-forge
clang-13                  13.0.1          default_he082bbe_0    conda-forge
clang_osx-64              13.0.1               h71a8856_0    conda-forge
clangxx                   13.0.1          default_he082bbe_0    conda-forge
clangxx_osx-64            13.0.1               heae0f87_0    conda-forge
cmake                     3.23.1               h35a7dd9_0    conda-forge
compiler-rt               13.0.1               he01351e_0    conda-forge
compiler-rt_osx-64        13.0.1               hd3f61c9_0    conda-forge
compilers                 1.4.2                h694c41f_0    conda-forge
console_bridge            1.0.2                hbb4e6a2_1    conda-forge
coreutils                 9.1                  h5eb16cf_0    conda-forge
cppzmq                    4.8.1                hf09c4ce_0    conda-forge
cryptography              36.0.2           py39h1644bb1_1    conda-forge
curl                      7.83.1               h372c54d_0    conda-forge
cxx-compiler              1.4.2                h1b54a9f_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
dartsim                   6.12.1               h000cc95_3    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.6.0              pyhd8ed1ab_0    conda-forge
docutils                  0.18.1           py39h6e9494a_1    conda-forge
eigen                     3.4.0                h940c156_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.4.8                h96cf925_0    conda-forge
fcl                       0.7.0                h31ff308_0    conda-forge
ffmpeg                    4.3.2                hbf27d7b_3    conda-forge
flann                     1.9.1             h30321d8_1010    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
[... 441 lines cut ...]
xorg-libxau               1.0.9                h35c211d_0    conda-forge
xorg-libxaw               1.0.14               h0d85af4_1    conda-forge
xorg-libxdmcp             1.1.3                h35c211d_0    conda-forge
xorg-libxext              1.3.4                h0d85af4_1    conda-forge
xorg-libxfixes            5.0.3             h0d85af4_1004    conda-forge
xorg-libxmu               1.1.3                h0d85af4_0    conda-forge
xorg-libxpm               3.5.13               h0d85af4_0    conda-forge
xorg-libxrender           0.9.10            h0d85af4_1003    conda-forge
xorg-libxt                1.2.1                h0d85af4_2    conda-forge
xorg-renderproto          0.11.1            h0d85af4_1002    conda-forge
xorg-xextproto            7.3.0             h35c211d_1002    conda-forge
xorg-xproto               7.0.31            h35c211d_1007    conda-forge
xz                        5.2.5                haf1e3a3_1    conda-forge
yaml                      0.2.5                h0d85af4_2    conda-forge
yaml-cpp                  0.6.3                hb1e8313_4    conda-forge
zeromq                    4.3.4                he49afe7_1    conda-forge
zlib                      1.2.11            h6c3fc93_1014    conda-forge
zstd                      1.5.2                ha9df2e0_1    conda-forge
zziplib                   0.13.69              h5dbffcc_1    conda-forge
``
<img width="1536" alt="Screenshot 2022-06-30 at 11 42 44" src="https://user-images.githubusercontent.com/69148588/176658655-3841727d-02b9-4341-b602-7bc8b74bd9f1.png">
<img widt
<img width="1536" alt="Screenshot 2022-06-30 at 11 46 54" src="https://user-images.githubusercontent.com/69148588/176658932-9fdb6d13-b588-4f58-b7b7-71e07d9af559.png">
h="1536" alt="Screenshot 2022-06-30 at 11 42 48" src="https://user-images.githubusercontent.com/69148588/176658659-f6a745b6-e9d9-4de5-98e2-69c399e58d35.png">
`

rbrobots, 2022-06-30:
Hi @traversaro,

Thank you for the message. It seems to be failing on any model I try. The built-in square, cube, cylinder and a few that I have tried from the gazebo-models repo: the bus, SUV, ground plane etc.  https://github.com/osrf/gazebo_models

`(robostackenv) ➜  ~ gazebo --verbose
Gazebo multi-robot simulator, version 11.10.2
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

Gazebo multi-robot simulator, version 11.10.2
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.46.218
[Msg] Publicized address: 192.168.46.218
[Msg] Loading world file [/Users/ranvirbhogal/opt/miniconda3/envs/robostackenv/share/gazebo-11/worlds/empty.world]
[Wrn] [GuiIface.cc:120] Populating font family aliases took 282 ms. Replace uses of missing font family "Sans" with one that exists to avoid this cost.
[Err] [InsertModelWidget.cc:403] Missing model.config for model "/Users/ranvirbhogal/.gazebo/models/gazebo_models"
[Err] [TopicManager.cc:321] Shouldn't get here topic[/gazebo/default/request]
[Dbg] [ModelIter.cc:201] Got response [{"errcode":3017,"msg":"Page not found","extra":null,"errid":"4a2e8cad-dd8e-48c9-9fce-4cd31cc4dd98","route":"GET /1.0/models?page=118","user-agent":"Gazebo 11.10.2(Mac OSX)","remote-address":"82.132.228.113"}`

Tobias-Fischer, 2022-10-13:
Many many thanks @traversaro !

closed 2022-10-13 by Tobias-Fischer

troiwill, 2023-01-16:
Hello @traversaro and @Tobias-Fischer, I am experiencing the same issue (Gazebo hangs when I try to add a model to the world). I tried the solution mentioned in @traversaro [comment](https://github.com/RoboStack/ros-noetic/issues/279#issuecomment-1277926991), but I received the following error:
```
Encountered problems while solving:
  - package gazebo-11.12.0-h063ddd6_1 requires assimp >=5.2.4,<5.2.5.0a0, but none of the providers can be installed
```
when I ran `mamba update gazebo=11.12.0=h063ddd6_1`.

The problem seems to be that the `ros-noetic-rviz` package requires `assimp >=5.2.3,<5.2.4.0a0`, but `gazebo=11.12.0=*_1` requires `assimp >=5.2.4,<5.2.5.0a0` based on the following error:
```
ros-noetic-rviz-1.14.14-py39h66a3644_16 requires assimp >=5.2.3,<5.2.4.0a0
```

Do you have any follow up solutions?

Also, I am using an Intel-based Mac. But would the problem occur on a Mac with M1 or M2 chips?

Tobias-Fischer, 2023-01-16:
Hmm .. we need to rebuild packages (see #295) for this to work. I'm not sure when I'll get around to doing this, though. Unfortunately, at the moment, there is no way to install these packages in the same environment :(

troiwill, 2023-01-16:
Thanks for the quick response! Would you happen to have any recommendations on how I can force ROS (e.g., roslaunch, rosrun, gazebo_ros, etc.) in the conda environment to point to a local installation of Gazebo? I can install a standalone Gazebo 11.12.0 on my machine, and I verified that I can add models to the local installation of Gazebo as well.