RoboStack/ros-noetic #281: Conda environment becomes inconsistent, limiting the install of any other package

LemonPi, 2022-07-19:
I have a conda package with the robostack installed via `conda install -c conda-forge -c robostack ros-noetic-desktop` where everything mostly works. However, after installing robostack, I can no longer install any other package via conda, or even run `conda update --all` due to:
```
The environment is inconsistent, please check the package plan carefully
The following packages are causing the inconsistency:

  - robostack/linux-64::ros-noetic-librviz-tutorial==0.11.0=py39h27aa46a_13
  - robostack/linux-64::ros-noetic-rqt-top==0.4.10=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-rqt-image-view==0.4.16=py39h27aa46a_13
  - robostack/linux-64::ros-noetic-rqt-graph==0.4.14=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-rqt-service-caller==0.4.10=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-rqt-robot-monitor==0.5.13=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-rqt-msg==0.4.10=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-joint-state-publisher-gui==1.15.0=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-rqt-action==0.4.9=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-rqt-bag-plugins==0.5.1=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-rqt-srv==0.4.9=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-rqt-runtime-monitor==0.5.9=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-rqt-web==0.4.10=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-rqt-reconfigure==0.5.4=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-viz==1.5.0=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-rqt-shell==0.4.11=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-rqt-nav-view==0.5.7=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-rqt-py-console==0.4.10=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-rqt-rviz==0.7.0=py39h3280fdf_13
  - robostack/linux-64::ros-noetic-rqt-bag==0.5.1=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-qt-gui-cpp==0.4.2=py39h27aa46a_13
  - robostack/linux-64::ros-noetic-rqt-plot==0.4.13=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-rviz-plugin-tutorials==0.11.0=py39h27aa46a_13
  - robostack/linux-64::ros-noetic-rqt-topic==0.4.13=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-rqt-robot-plugins==0.5.8=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-rqt-gui-py==0.5.3=py39h6fdeb60_14
  - robostack/linux-64::ros-noetic-rqt-publisher==0.4.10=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-gl-dependency==1.1.2=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-rviz==1.14.14=py39h4e988fb_14
  - robostack/linux-64::ros-noetic-qt-gui-py-common==0.4.2=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-rqt-py-common==0.5.2=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-qt-gui==0.4.2=py39h27aa46a_13
  - robostack/linux-64::ros-noetic-rqt-dep==0.4.12=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-rqt-console==0.4.11=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-rqt-robot-steering==0.5.12=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-qt-dotgraph==0.4.2=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-rqt-launch==0.4.9=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-rqt-robot-dashboard==0.5.8=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-rqt-common-plugins==0.4.9=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-visualization-tutorials==0.11.0=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-rqt-pose-view==0.5.11=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-rqt-moveit==0.5.10=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-webkit-dependency==1.1.2=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-desktop==1.5.0=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-python-qt-binding==0.4.4=py39h27aa46a_13
  - robostack/linux-64::ros-noetic-rviz-python-tutorial==0.11.0=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-urdf-tutorial==0.5.0=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-rqt-gui-cpp==0.5.3=py39h27aa46a_14
  - robostack/linux-64::ros-noetic-rqt-tf-tree==0.6.2=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-rqt-logger-level==0.4.11=py39h6fdeb60_13
  - robostack/linux-64::ros-noetic-rqt-gui==0.5.3=py39h6fdeb60_14
```

```
The following specifications were found to be incompatible with each other:

Output in format: Requested package -> Available versions

Package _libgcc_mutex conflicts for:
ros-noetic-rqt-shell -> libgcc-ng[version='>=9.4.0'] -> _libgcc_mutex[version='*|0.1|0.1',build='conda_forge|main']
ros-noetic-rqt-msg -> libgcc-ng[version='>=9.4.0'] -> _libgcc_mutex[version='*|0.1|0.1',build='conda_forge|main']
ros-noetic-rqt-logger-level -> libgcc-ng[version='>=9.4.0'] -> _libgcc_mutex[version='*|0.1|0.1',build='conda_forge|main']
ros-noetic-qt-gui-py-common -> libgcc-ng[version='>=9.4.0'] -> _libgcc_mutex[version='*|0.1|0.1',build='conda_forge|main']
ros-noetic-qt-gui -> libgcc-ng[version='>=9.4.0'] -> _libgcc_mutex[version='*|0.1|0.1',build='conda_forge|main']
ros-noetic-rqt-py-common -> libgcc-ng[version='>=9.4.0'] -> _libgcc_mutex[version='*|0.1|0.1',build='conda_forge|main']
[... 777 lines cut ...]
ros-noetic-rqt-tf-tree -> ros-noetic-rospy -> numpy

Package distro conflicts for:
ros-noetic-rqt-service-caller -> rospkg -> distro
ros-noetic-rqt-bag -> rospkg -> distro
ros-noetic-rqt-shell -> rospkg -> distro
ros-noetic-rqt-plot -> rospkg -> distro
ros-noetic-rqt-py-console -> rospkg -> distro
ros-noetic-rqt-dep -> rospkg -> distro
ros-noetic-rqt-topic -> rospkg -> distro
ros-noetic-rqt-web -> rospkg -> distro
ros-noetic-rqt-publisher -> rospkg -> distro
ros-noetic-rqt-msg -> rospkg -> distro
ros-noetic-rqt-graph -> rospkg -> distro
ros-noetic-rqt-runtime-monitor -> rospkg -> distro
ros-noetic-rqt-pose-view -> rospkg -> distro
ros-noetic-qt-gui -> rospkg -> distro
ros-noetic-rqt-logger-level -> rospkg -> distro
ros-noetic-rqt-robot-steering -> rospkg -> distro
ros-noetic-qt-gui-py-common -> rospkg -> distro
ros-noetic-rqt-console -> rospkg -> distro
ros-noetic-rqt-tf-tree -> rospkg -> distro
ros-noetic-rqt-gui -> rospkg -> distro
ros-noetic-rqt-robot-monitor -> rospkg -> distro
```

LemonPi, 2022-07-20:
Ok, I want to install it in an existing environment (dev) instead of creating a new environment, so I did 
```
(dev) conda remove -n dev ros-noetic*
(dev) conda activate base
(base) conda install mamba -c conda-forge
(base) conda activate dev
(dev) mamba install ros-noetic-desktop -c robostack -c robostack-experimental -c conda-forge --no-channel-priority --override-channels
```
Everything installed with no problems and the environment is consistent again :+1: 

My use case is I have noetic installed outside of conda. I build my packages with `catkin build` when not in a conda environment, and am not planning to do any building inside the conda environment. I only need the conda environment to run python scripts that calls libraries specifically only packaged with conda (pytorch3d).

Tobias-Fischer, 2022-07-20:
That's great to hear, thanks for the update!

The readme recommends the use of `mamba` throughout - if you feel like another warning is necessary, please open a PR and we'll be happy to merge :).

closed 2022-07-20 by Tobias-Fischer