RoboStack/ros-noetic #283: Linux - Library versions when dependencies come from different channels

sadsimulation, 2022-08-21:
Hi,
I'm trying to use PCL in a robostack environment. Unfortunately I've hit some gnarly segfaults. I've seen on the ROS forums that the library versions that `ros-noetic-pcl-ros` and `pcl` have to be the same in order to work properly because of ABI incompatibility.

`ros-noetic-pcl-ros` was installed from the `robostack` channel, but `pcl` was installed from `conda-forge`. How do I know that these packages are actually compatible / the segfaults I'm seeing are not due to ABI incompatibility?

Am I SOL here and need to use a bare-metal ROS installation if I also want CPP support?

Tobias-Fischer, 2022-08-21:
We pin PCL to a specific version (currently 1.12.0), see https://github.com/RoboStack/ros-noetic/blob/9a6b854daec5ee83b5888d740fe1b6b8d625c7bd/conda_build_config.yaml#L10

So the segfaults should not be due to this. Which issues do you encounter?

sadsimulation, 2022-08-21:
Okay just wanted to make sure it's not due to that. Is there an easy way to replace the virtual environments PCL with a debug build of my own?

traversaro, 2022-08-21:
> Okay just wanted to make sure it's not due to that. Is there an easy way to replace the virtual environments PCL with a debug build of my own?

If the conda-forge package has no patches that change the ABI, it should be sufficient to configure a PCL build with the same version of the PCL package, with the same CMake options and with -DCMAKE_INSTALL_PREFIX=$CONDA_PREFIX . In this way, the manually installed  version overwrites the conda installed one. It is not a fail proof method, but od you are lucky it could work.

Tobias-Fischer, 2022-08-21:
No patches in conda-forge :)

Tobias-Fischer, 2022-08-24:
Closing here for now - feel free to reopen.

closed 2022-08-24 by Tobias-Fischer