RoboStack/ros-noetic #285: Mamba fails to resolve host name

medhap02, 2022-08-31:
Hello,

I am trying to run `mamba create -n 15-482 ros-noetic-desktop python=3.9 -c robostack -c robostack-experimental -c conda-forge --no-channel-priority --override-channels` however I am running into a `Couldn't resolve host` error for every single package that it tries to install.

`# >>>>>>>>>>>>>>>>>>>>>> ERROR REPORT <<<<<<<<<<<<<<<<<<<<<<

    Traceback (most recent call last):
      File "/Users/medhapalavalli/opt/anaconda3/lib/python3.8/site-packages/conda/exceptions.py", line 1082, in __call__
        return func(*args, **kwargs)
      File "/Users/medhapalavalli/opt/anaconda3/lib/python3.8/site-packages/mamba/mamba.py", line 814, in exception_converter
        raise e
      File "/Users/medhapalavalli/opt/anaconda3/lib/python3.8/site-packages/mamba/mamba.py", line 808, in exception_converter
        exit_code = _wrapped_main(*args, **kwargs)
      File "/Users/medhapalavalli/opt/anaconda3/lib/python3.8/site-packages/mamba/mamba.py", line 773, in _wrapped_main
        result = do_call(args, p)
      File "/Users/medhapalavalli/opt/anaconda3/lib/python3.8/site-packages/mamba/mamba.py", line 667, in do_call
        exit_code = create(args, parser)
      File "/Users/medhapalavalli/opt/anaconda3/lib/python3.8/site-packages/mamba/mamba.py", line 568, in create
        install(args, parser, 'create')
      File "/Users/medhapalavalli/opt/anaconda3/lib/python3.8/site-packages/mamba/mamba.py", line 532, in install
        downloaded = transaction.prompt(PackageCacheData.first_writable().pkgs_dir, repos)
    RuntimeError: Download error (6) Couldn't resolve host name [https://conda.anaconda.org/conda-forge/osx-64/aom-3.4.0-hb486fe8_1.tar.bz2]

`$ /Users/medhapalavalli/opt/anaconda3/bin/mamba create -n 15-482 ros-noetic-desktop python=3.9 -c robostack -c robostack-experimental -c conda-forge --no-channel-priority --override-channels`

```
  environment variables:
                 CIO_TEST=<not set>
        CONDA_DEFAULT_ENV=base
                CONDA_EXE=/Users/medhapalavalli/opt/anaconda3/bin/conda
             CONDA_PREFIX=/Users/medhapalavalli/opt/anaconda3
    CONDA_PROMPT_MODIFIER=(base)
         CONDA_PYTHON_EXE=/Users/medhapalavalli/opt/anaconda3/bin/python
               CONDA_ROOT=/Users/medhapalavalli/opt/anaconda3
              CONDA_SHLVL=1
           CURL_CA_BUNDLE=<not set>
                 INFOPATH=/opt/homebrew/share/info:
                  MANPATH=/opt/homebrew/share/man::
                     PATH=/Users/medhapalavalli/opt/anaconda3/bin:/Users/medhapalavalli/opt/anac
                          onda3/condabin:/opt/homebrew/bin:/opt/homebrew/sbin:/usr/local/bin:/us
                          r/bin:/bin:/usr/sbin:/sbin:/Library/Apple/usr/bin:/usr/local/smlnj/bin
                          /sml:/Users/medhapalavalli/CarnegieMellonUniversity/tartanhacksdashboa
                          rd/flutter/bin
       REQUESTS_CA_BUNDLE=<not set>
            SSL_CERT_FILE=<not set>
```

```
     active environment : base
    active env location : /Users/medhapalavalli/opt/anaconda3
            shell level : 1
       user config file : /Users/medhapalavalli/.condarc
 populated config files : /Users/medhapalavalli/.condarc
          conda version : 4.12.0
    conda-build version : 3.20.5
         python version : 3.8.5.final.0
       virtual packages : __osx=10.16=0
                          __unix=0=0
                          __archspec=1=x86_64
       base environment : /Users/medhapalavalli/opt/anaconda3  (writable)
      conda av data dir : /Users/medhapalavalli/opt/anaconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack/osx-64
                          https://conda.anaconda.org/robostack/noarch
                          https://conda.anaconda.org/robostack-experimental/osx-64
                          https://conda.anaconda.org/robostack-experimental/noarch
                          https://conda.anaconda.org/conda-forge/osx-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /Users/medhapalavalli/opt/anaconda3/pkgs
                          /Users/medhapalavalli/.conda/pkgs
       envs directories : /Users/medhapalavalli/opt/anaconda3/envs
                          /Users/medhapalavalli/.conda/envs
               platform : osx-64
             user-agent : conda/4.12.0 requests/2.24.0 CPython/3.8.5 Darwin/21.5.0 OSX/10.16
                UID:GID : 501:20
             netrc file : None
           offline mode : False`
```

Tobias-Fischer, 2022-08-31:
Looks like a network error. Are you sitting behind a firewall or something? Can you download https://conda.anaconda.org/conda-forge/osx-64/aom-3.4.0-hb486fe8_1.tar.bz2 from your browser?

medhap02, 2022-08-31:
I can download it from my browser. I am not sitting behind a firewall.

Tobias-Fischer, 2022-08-31:
@wolfv - any ideas what this could be?

Tobias-Fischer, 2022-09-17:
Is this still an issue @medhap02 ?

closed 2022-09-26 by Tobias-Fischer