RoboStack/ros-noetic #288: rviz undefined symbol

physar, 2022-09-27:
I have installed the ros-noetic environment on two different Linux machines: one an Ubuntu 18.04, the other an Ubuntu 22.04.

On both machines I receive the same error when I start rviz, even after a vanilla installation (only mamba create -n robostackenv, no other mamba installs)

librviz.so: undefined symbol: _ZTIN4YAML13BadConversionE 

This error seems to be related with the versions of libboost libraries that load the libyaml-cpp libraries (and expect but cannot find this symbol),

Please let me know if I have to perform any other test or give more details

Arnoud

traversaro, 2022-09-27:
Hi @physar, can you report the output of `mamba list` in the failing environment? Thanks!

physar, 2022-09-27:
Sure:
# packages in environment at ~/mambaforge/envs/robostackenv:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
alsa-lib                  1.2.3.2              h166bdaf_0    conda-forge
aom                       3.3.0                h27087fc_1    conda-forge
apr                       1.7.0                h7f98852_5    conda-forge
assimp                    5.2.4                h4c92fa2_0    conda-forge
atk-1.0                   2.36.0               h3371d22_4    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
bcrypt                    3.2.2            py39hb9d737c_0    conda-forge
boost                     1.74.0           py39h5472131_5    conda-forge
boost-cpp                 1.74.0               h6cacc03_7    conda-forge
brotli                    1.0.9                h166bdaf_7    conda-forge
brotli-bin                1.0.9                h166bdaf_7    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.18.1               h7f98852_0    conda-forge
ca-certificates           2022.9.24            ha878542_0    conda-forge
cairo                     1.16.0            ha12eb4b_1010    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
certifi                   2022.9.24          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1           py39he91dace_0    conda-forge
cmake                     3.24.2               h5432695_0    conda-forge
console_bridge            1.0.2                h924138e_1    conda-forge
contourpy                 1.0.5            py39hf939315_0    conda-forge
cryptography              37.0.4           py39hd97740a_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.6.0              pyhd8ed1ab_0    conda-forge
docutils                  0.19             py39hf3d152e_0    conda-forge
eigen                     3.4.0                h4bd325d_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.4.9                h27087fc_0    conda-forge
ffmpeg                    4.4.1                hd7ab26d_2    conda-forge
fftw                      3.3.10          nompi_hf0379b8_105    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.0               hc2a2eb6_1    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.37.3           py39hb9d737c_0    conda-forge
freeglut                  3.2.2                h9c3ff4c_1    conda-forge
freeimage                 3.18.0              h530e30e_10    conda-forge
freetype                  2.12.1               hca18f0e_0    conda-forge
fribidi                   1.0.10               h36c2ea0_0    conda-forge
gdk-pixbuf                2.42.8               hff1cb4f_1    conda-forge
gettext                   0.19.8.1          h27087fc_1009    conda-forge
giflib                    5.2.1                h36c2ea0_2    conda-forge
glib                      2.74.0               h6239696_0    conda-forge
glib-tools                2.74.0               h6239696_0    conda-forge
gmock                     1.11.0               h924138e_0    conda-forge
gmp                       6.2.1                h58526e2_0    conda-forge
gnutls                    3.6.13               h85f3911_1    conda-forge
gpgme                     1.16.0               h9c3ff4c_0    conda-forge
cffi                      1.15.1           py39he91dace_0    conda-forge
cmake                     3.24.2               h5432695_0    conda-forge
console_bridge            1.0.2                h924138e_1    conda-forge
contourpy                 1.0.5            py39hf939315_0    conda-forge
cryptography              37.0.4           py39hd97740a_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.6.0              pyhd8ed1ab_0    conda-forge
docutils                  0.19             py39hf3d152e_0    conda-forge
eigen                     3.4.0                h4bd325d_0    conda-forge
[... 375 lines cut ...]
xorg-fixesproto           5.0               h7f98852_1002    conda-forge
xorg-inputproto           2.3.2             h7f98852_1002    conda-forge
xorg-kbproto              1.0.7             h7f98852_1002    conda-forge
xorg-libice               1.0.10               h7f98852_0    conda-forge
xorg-libsm                1.2.3             hd9c2040_1000    conda-forge
xorg-libx11               1.7.2                h7f98852_0    conda-forge
xorg-libxau               1.0.9                h7f98852_0    conda-forge
[... cut ...]
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.6                h166bdaf_0    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
yaml-cpp                  0.7.0                h27087fc_1    conda-forge
zlib                      1.2.12               h166bdaf_3    conda-forge
zstd                      1.5.2                h6239696_4    conda-forge
zziplib                   0.13.69              h27826a3_1    conda-forge

traversaro, 2022-09-27:
Thanks, I was able to reproduce the issue. I do not think the problem is of a wrong version of a library, as that symbol can be found:
~~~
(robostackenv) traversaro@IITICUBLAP257:~$ nm -s /home/traversaro/mambaforge/envs/robostackenv/lib/libyaml-cpp.so  | grep YAML | grep _ZTIN4YAML13BadConversionE
00000000000759c0 V _ZTIN4YAML13BadConversionE
~~~

The problem seems that the library is not linked by `librviz.so` for some reason:
~~~
(robostackenv) traversaro@IITICUBLAP257:~$ ldd  /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/../librviz.so
        linux-vdso.so.1 (0x00007ffce8dc8000)
        libboost_filesystem.so.1.74.0 => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/.././libboost_filesystem.so.1.74.0 (0x00007fe55bdad000)
        libboost_program_options.so.1.74.0 => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/.././libboost_program_options.so.1.74.0 (0x00007fe55bd44000)
        libboost_thread.so.1.74.0 => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/.././libboost_thread.so.1.74.0 (0x00007fe55bd21000)
        libimage_transport.so => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/.././libimage_transport.so (0x00007fe55bc95000)
        libresource_retriever.so => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/.././libresource_retriever.so (0x00007fe55bc8d000)
        libtf2_ros.so => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/.././libtf2_ros.so (0x00007fe55bbed000)
        libmessage_filters.so => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/.././libmessage_filters.so (0x00007fe55bbe7000)
        libtf2.so => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/.././libtf2.so (0x00007fe55bbb4000)
        libclass_loader.so => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/.././libclass_loader.so (0x00007fe55bb8a000)
        libroslib.so => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/.././libroslib.so (0x00007fe55bb75000)
        libtinyxml2.so.9 => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/.././libtinyxml2.so.9 (0x00007fe55bb57000)
        libroscpp.so => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/.././libroscpp.so (0x00007fe55b9bc000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fe55b99c000)
        librosconsole.so => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/.././librosconsole.so (0x00007fe55b94b000)
        libroscpp_serialization.so => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/.././libroscpp_serialization.so (0x00007fe55b945000)
        librostime.so => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/.././librostime.so (0x00007fe55b925000)
        libconsole_bridge.so.1.0 => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/.././libconsole_bridge.so.1.0 (0x00007fe55b91d000)
        libQt5Widgets.so.5 => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/.././libQt5Widgets.so.5 (0x00007fe55b3d9000)
        libOgreOverlay.so.1.10.12 => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/.././libOgreOverlay.so.1.10.12 (0x00007fe55b373000)
        libOgreMain.so.1.10.12 => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/.././libOgreMain.so.1.10.12 (0x00007fe55ae07000)
        libX11.so.6 => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/.././libX11.so.6 (0x00007fe55acc4000)
        libassimp.so.5 => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/.././libassimp.so.5 (0x00007fe55a2cf000)
        libQt5Gui.so.5 => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/.././libQt5Gui.so.5 (0x00007fe559ec4000)
        libQt5Core.so.5 => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/.././libQt5Core.so.5 (0x00007fe559a72000)
        libstdc++.so.6 => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/.././libstdc++.so.6 (0x00007fe5598be000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fe5597d7000)
        libgcc_s.so.1 => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/.././libgcc_s.so.1 (0x00007fe5597be000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fe559594000)
        librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007fe55958f000)
        libcurl.so.4 => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/../././libcurl.so.4 (0x00007fe5594ef000)
        libactionlib.so => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/../././libactionlib.so (0x00007fe5594d5000)
        libPocoFoundation.so.81 => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/../././libPocoFoundation.so.81 (0x00007fe5592b4000)
        librospack.so => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/../././librospack.so (0x00007fe559268000)
        libxmlrpcpp.so => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/../././libxmlrpcpp.so (0x00007fe559242000)
        libcpp_common.so => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/../././libcpp_common.so (0x00007fe559235000)
        libboost_chrono.so.1.74.0 => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/../././libboost_chrono.so.1.74.0 (0x00007fe55922a000)
        librosconsole_log4cxx.so => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/../././librosconsole_log4cxx.so (0x00007fe559207000)
        librosconsole_backend_interface.so => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/../././librosconsole_backend_interface.so (0x00007fe559202000)
[... cut ...]
        libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007fe556155000)
        libexpat.so.1 => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/../././././libexpat.so.1 (0x00007fe556124000)
        libcrypt.so.1 => /lib/x86_64-linux-gnu/libcrypt.so.1 (0x00007fe5560e8000)
        libuuid.so.1 => /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/../././././libuuid.so.1 (0x00007fe5560df000)
(robostackenv) traversaro@IITICUBLAP257:~$ ldd  /home/traversaro/mambaforge/envs/robostackenv/lib/rviz/../librviz.so | grep yaml
~~~

traversaro, 2022-09-27:
Indeed, it seems that rviz links `YAML_CPP_LIBRARIES` (see https://github.com/ros-visualization/rviz/blob/968a654f35cc6d199c1122554fcf2f12be07f554/src/rviz/CMakeLists.txt#L142), but that variable is empty in the `yaml-cpp-config.cmake` installed by the conda-forge package:
~~~
# - Config file for the yaml-cpp package
# It defines the following variables
#  YAML_CPP_INCLUDE_DIR - include directory
#  YAML_CPP_LIBRARIES    - libraries to link against

# Compute paths
get_filename_component(YAML_CPP_CMAKE_DIR "${CMAKE_CURRENT_LIST_FILE}" PATH)
set(YAML_CPP_INCLUDE_DIR "")

# Our library dependencies (contains definitions for IMPORTED targets)
include("${YAML_CPP_CMAKE_DIR}/yaml-cpp-targets.cmake")

# These are IMPORTED targets created by yaml-cpp-targets.cmake
set(YAML_CPP_LIBRARIES "")
~~~

traversaro, 2022-09-27:
To be honest, I now wonder how this ever worked.

traversaro, 2022-09-27:
> but that variable is empty in the `yaml-cpp-config.cmake` installed by the conda-forge package:

That variable is empty due to a bug in yaml-cpp 0.7.0, that was fixed in https://github.com/jbeder/yaml-cpp/pull/1077 but the fix was never backported.

physar, 2022-09-27:
OK, thanks for confirming the issue. I will wait on the rebuild of the rviz package.

Tobias-Fischer, 2022-09-27:
I think this might have worked as I had a yaml-cop build on the robostack channel that was built with cmake where this issue does not persist, then we briefly had such a build in conda forge but it was marked as bringen soon after.

traversaro, 2022-09-29:
> Hopefully this issue will be fixed by merging [conda-forge/yaml-cpp-feedstock#14](https://github.com/conda-forge/yaml-cpp-feedstock/pull/14) and then re-building the rviz package.

The PR has been merged, a rebuild of rviz should fix this issue.

> I think this might have worked as I had a yaml-cop build on the robostack channel that was built with cmake where this issue does not persist, then we briefly had such a build in conda forge but it was marked as broken soon after.

I may be wrong, but if I remember correctly `yaml-cpp` has been always build with cmake (https://github.com/conda-forge/yaml-cpp-feedstock/blob/b21dc44c09fb0ce54ea4b117234f8ead849ec916/recipe/build.sh#L7), the one that is being built with autotools is `yaml` (https://github.com/conda-forge/yaml-feedstock/blob/1f91366b27c651714265cf09067514b7158c8fdb/recipe/build.sh#L3).

peci1, 2022-12-08:
This is still a problem for me on Steam Deck.

As a workaround, launch rviz as `LD_PRELOAD=/home/deck/mambaforge/envs/ros_env/lib/libyaml-cpp.so rviz`.

Tobias-Fischer, 2023-01-24:
Please try again with the packages from the robostack-staging channel.

closed 2023-01-24 by Tobias-Fischer

peci1, 2023-01-24:
> Please try again with the packages from the robostack-staging channel.

How can I do that? I've tried `mamba update ros-noetic-rviz -c robostack-staging` but rviz did not get any update...

Tobias-Fischer, 2023-01-24:
Please create a new environment from scratch

peci1, 2023-01-24:
Outch, really? This seems like a lot of work setting it up again... Or can I just make a temporary env for the test, and when staging goes to stable, my normal env would be able to update normally?

peci1, 2023-01-25:
Okay, I've created a new env just to verify this has been fixed, and it seems so.

physar, 2023-01-25:
I also created a new environment, but encounter another symbol lookup error. 

mamba create -n robostackenv ros-noetic-desktop python=3.9 -c robostack-staging -c conda-forge --no-channel-priority --override-channels

mamba init

mamba activate robostackenv

export ROS_HOSTNAME=localhost
export ROS_MASTER_URI=http://localhost:11311

roscore

~/mambaforge/envs/robostackenv/lib/rosout/rosout: symbol lookup error: ~/mambaforge/envs/robostackenv/lib/rosout/../libroscpp.so: undefined symbol: _ZTIN5boost6detail16thread_data_baseE

peci1, 2023-01-25:
@physar I can confirm your observation.

traversaro, 2023-01-25:
Hello @peci1 and @physar , can you include the outcome of `mamba list`? Thanks!

peci1, 2023-01-25:
Here it is:

```
# packages in environment at /home/deck/mambaforge/envs/ros_env2:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
alsa-lib                  1.2.8                h166bdaf_0    conda-forge
aom                       3.5.0                h27087fc_0    conda-forge
apr                       1.7.0                h7f98852_5    conda-forge
assimp                    5.2.5                hf40c2ba_0    conda-forge
atk-1.0                   2.38.0               hd4edc92_1    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
bcrypt                    3.2.2            py39hb9d737c_1    conda-forge
boost                     1.78.0           py39h7c9e3ff_4    conda-forge
boost-cpp                 1.78.0               h75c5d50_1    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.18.1               h7f98852_0    conda-forge
ca-certificates           2022.12.7            ha878542_0    conda-forge
cairo                     1.16.0            ha61ee94_1014    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
cffi                      1.15.1           py39he91dace_3    conda-forge
cmake                     3.25.2               h077f3f9_0    conda-forge
console_bridge            1.0.2                h924138e_1    conda-forge
cryptography              39.0.0           py39hd598818_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.6.0              pyhd8ed1ab_0    conda-forge
docutils                  0.19             py39hf3d152e_1    conda-forge
eigen                     3.4.0                h4bd325d_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.5.0                h27087fc_0    conda-forge
ffmpeg                    5.1.2           gpl_h8dda1f0_106    conda-forge
fftw                      3.3.10          nompi_hf0379b8_106    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.1               hc2a2eb6_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freeglut                  3.2.2                h9c3ff4c_1    conda-forge
freeimage                 3.18.0              hca5c823_12    conda-forge
freetype                  2.12.1               hca18f0e_1    conda-forge
fribidi                   1.0.10               h36c2ea0_0    conda-forge
gdk-pixbuf                2.42.10              h05c8ddd_0    conda-forge
gettext                   0.21.1               h27087fc_0    conda-forge
giflib                    5.2.1                h36c2ea0_2    conda-forge
glib                      2.74.1               h6239696_1    conda-forge
glib-tools                2.74.1               h6239696_1    conda-forge
gmock                     1.12.1               hf52228f_0    conda-forge
gmp                       6.2.1                h58526e2_0    conda-forge
gnutls                    3.7.8                hf3e180e_0    conda-forge
gpgme                     1.18.0               h27087fc_0    conda-forge
graphite2                 1.3.13            h58526e2_1001    conda-forge
graphviz                  6.0.2                h99bc08f_0    conda-forge
gst-plugins-base          1.21.3               h4243ec0_1    conda-forge
gstreamer                 1.21.3               h25f0c4b_1    conda-forge
gstreamer-orc             0.4.33               h166bdaf_0    conda-forge
gtest                     1.12.1               hf52228f_0    conda-forge
gtk2                      2.24.33              h90689f9_2    conda-forge
gts                       0.7.6                h64030ff_2    conda-forge
harfbuzz                  6.0.0                h8e241bc_0    conda-forge
hdf5                      1.12.2          nompi_h2386368_101    conda-forge
icu                       70.1                 h27087fc_0    conda-forge
imath                     3.1.6                h6239696_1    conda-forge
jack                      1.9.21               h583fa2b_2    conda-forge
jasper                    2.0.33               ha77e612_0    conda-forge
jpeg                      9e                   h166bdaf_2    conda-forge
[... 228 lines cut ...]
xorg-inputproto           2.3.2             h7f98852_1002    conda-forge
xorg-kbproto              1.0.7             h7f98852_1002    conda-forge
xorg-libice               1.0.10               h7f98852_0    conda-forge
xorg-libsm                1.2.3             hd9c2040_1000    conda-forge
xorg-libx11               1.7.2                h7f98852_0    conda-forge
xorg-libxau               1.0.9                h7f98852_0    conda-forge
xorg-libxaw               1.0.14               h7f98852_1    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
[... cut ...]
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.6                h166bdaf_0    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
yaml-cpp                  0.7.0                h27087fc_2    conda-forge
zlib                      1.2.13               h166bdaf_4    conda-forge
zstd                      1.5.2                h3eb15da_6    conda-forge
zziplib                   0.13.69              h27826a3_1    conda-forge
```

Tobias-Fischer, 2023-01-25:
What's the output of `ldd ~/mambaforge/envs/robostackenv/lib/rosout/../libroscpp.so`?

peci1, 2023-01-25:
```
$ ldd ~/mambaforge/envs/ros_env2/lib/libroscpp.so
	linux-vdso.so.1 (0x00007fffd9f8d000)
	librosconsole.so => /home/deck/mambaforge/envs/ros_env2/lib/./librosconsole.so (0x00007f69e36c7000)
	libroscpp_serialization.so => /home/deck/mambaforge/envs/ros_env2/lib/./libroscpp_serialization.so (0x00007f69e36c2000)
	libxmlrpcpp.so => /home/deck/mambaforge/envs/ros_env2/lib/./libxmlrpcpp.so (0x00007f69e369d000)
	librostime.so => /home/deck/mambaforge/envs/ros_env2/lib/./librostime.so (0x00007f69e3681000)
	libcpp_common.so => /home/deck/mambaforge/envs/ros_env2/lib/./libcpp_common.so (0x00007f69e3675000)
	libboost_chrono.so.1.78.0 => /home/deck/mambaforge/envs/ros_env2/lib/./libboost_chrono.so.1.78.0 (0x00007f69e3669000)
	libboost_filesystem.so.1.78.0 => /home/deck/mambaforge/envs/ros_env2/lib/./libboost_filesystem.so.1.78.0 (0x00007f69e3646000)
	libstdc++.so.6 => /home/deck/mambaforge/envs/ros_env2/lib/./libstdc++.so.6 (0x00007f69e3492000)
	libm.so.6 => /usr/lib/libm.so.6 (0x00007f69e332d000)
	libgcc_s.so.1 => /home/deck/mambaforge/envs/ros_env2/lib/./libgcc_s.so.1 (0x00007f69e3314000)
	libc.so.6 => /usr/lib/libc.so.6 (0x00007f69e3148000)
	librosconsole_log4cxx.so => /home/deck/mambaforge/envs/ros_env2/lib/././librosconsole_log4cxx.so (0x00007f69e312d000)
	librosconsole_backend_interface.so => /home/deck/mambaforge/envs/ros_env2/lib/././librosconsole_backend_interface.so (0x00007f69e3128000)
	liblog4cxx.so.11 => /home/deck/mambaforge/envs/ros_env2/lib/././liblog4cxx.so.11 (0x00007f69e2f4e000)
	libpthread.so.0 => /usr/lib/libpthread.so.0 (0x00007f69e2f2d000)
	/usr/lib64/ld-linux-x86-64.so.2 (0x00007f69e3897000)
	librt.so.1 => /usr/lib/librt.so.1 (0x00007f69e2f22000)
	libconsole_bridge.so.1.0 => /home/deck/mambaforge/envs/ros_env2/lib/././libconsole_bridge.so.1.0 (0x00007f69e2f1c000)
	libboost_atomic.so.1.78.0 => /home/deck/mambaforge/envs/ros_env2/lib/././libboost_atomic.so.1.78.0 (0x00007f69e2f11000)
	libaprutil-1.so.0 => /home/deck/mambaforge/envs/ros_env2/lib/./././libaprutil-1.so.0 (0x00007f69e2ee5000)
	libapr-1.so.0 => /home/deck/mambaforge/envs/ros_env2/lib/./././libapr-1.so.0 (0x00007f69e2ea5000)
	libexpat.so.1 => /home/deck/mambaforge/envs/ros_env2/lib/././././libexpat.so.1 (0x00007f69e2e75000)
	libcrypt.so.1 => /usr/lib/libcrypt.so.1 (0x00007f69e2e39000)
	libdl.so.2 => /usr/lib/libdl.so.2 (0x00007f69e2e32000)
```

physar, 2023-01-25:
Tested on a Ubuntu 18.04 machine
 
ldd  /home/arnoud/mambaforge/envs/robostackenv/lib/libroscpp.so 
	linux-vdso.so.1 (0x00007ffd2ea75000)
	librosconsole.so => /home/arnoud/mambaforge/envs/robostackenv/lib/././librosconsole.so (0x00007f7e1ae38000)
	libroscpp_serialization.so => /home/arnoud/mambaforge/envs/robostackenv/lib/././libroscpp_serialization.so (0x00007f7e1ae33000)
	libxmlrpcpp.so => /home/arnoud/mambaforge/envs/robostackenv/lib/././libxmlrpcpp.so (0x00007f7e1ae0e000)
	librostime.so => /home/arnoud/mambaforge/envs/robostackenv/lib/././librostime.so (0x00007f7e1adc3000)
	libcpp_common.so => /home/arnoud/mambaforge/envs/robostackenv/lib/././libcpp_common.so (0x00007f7e1adb7000)
	libboost_chrono.so.1.78.0 => /home/arnoud/mambaforge/envs/robostackenv/lib/././libboost_chrono.so.1.78.0 (0x00007f7e1adab000)
	libboost_filesystem.so.1.78.0 => /home/arnoud/mambaforge/envs/robostackenv/lib/././libboost_filesystem.so.1.78.0 (0x00007f7e1ad8a000)
	libstdc++.so.6 => /home/arnoud/mambaforge/envs/robostackenv/lib/././libstdc++.so.6 (0x00007f7e1abd6000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f7e1a7ec000)
	libgcc_s.so.1 => /home/arnoud/mambaforge/envs/robostackenv/lib/././libgcc_s.so.1 (0x00007f7e1a7d3000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f7e1a3e2000)
	librosconsole_log4cxx.so => /home/arnoud/mambaforge/envs/robostackenv/lib/./././librosconsole_log4cxx.so (0x00007f7e1a3c9000)
	librosconsole_backend_interface.so => /home/arnoud/mambaforge/envs/robostackenv/lib/./././librosconsole_backend_interface.so (0x00007f7e1a3c4000)
	liblog4cxx.so.11 => /home/arnoud/mambaforge/envs/robostackenv/lib/./././liblog4cxx.so.11 (0x00007f7e1a1ea000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f7e19fcb000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f7e1addf000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f7e19dc3000)
	libconsole_bridge.so.1.0 => /home/arnoud/mambaforge/envs/robostackenv/lib/./././libconsole_bridge.so.1.0 (0x00007f7e19dbd000)
	libboost_atomic.so.1.78.0 => /home/arnoud/mambaforge/envs/robostackenv/lib/./././libboost_atomic.so.1.78.0 (0x00007f7e19db4000)
	libaprutil-1.so.0 => /home/arnoud/mambaforge/envs/robostackenv/lib/././././libaprutil-1.so.0 (0x00007f7e19d88000)
	libapr-1.so.0 => /home/arnoud/mambaforge/envs/robostackenv/lib/././././libapr-1.so.0 (0x00007f7e19d48000)
	libexpat.so.1 => /home/arnoud/mambaforge/envs/robostackenv/lib/./././././libexpat.so.1 (0x00007f7e19d18000)
	libcrypt.so.1 => /lib/x86_64-linux-gnu/libcrypt.so.1 (0x00007f7e19ade000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f7e198da000)

Tobias-Fischer, 2023-01-25:
Thanks - I am rebuilding the affected package. Stay tuned :)

Tobias-Fischer, 2023-01-25:
Ok - this is fixed now, thanks for the bug report. Unfortunately this requires yet another re-create of the environment. You will need to pull a new ros-noetic-roscpp. You might have to `rm ~/mambaforge/pkgs/ros-noetic-roscpp-*`

physar, 2023-01-25:
Thanks, I was able to start both roscore and rviz, so this issue is solved.

So, ros-noetic is now running on a Ubuntu 18.04 machine.

A small note on the documentation. I had to set manually ROS_HOSTNAME an ROS_MASTER_URI to  localhost
I could not find in which setup script this has to be changed.

peci1, 2023-01-25:
Thanks, rviz and roscore run now. I also tested joy_node.

What I can't install now (and could before) is sound_play:

```
$ mamba install --no-banner ros-noetic-sound-play

Looking for: ['ros-noetic-sound-play']

conda-forge/linux-64                                        Using cache
conda-forge/noarch                                          Using cache
robostack-staging/linux-64                                    No change
robostack/linux-64                                            No change
robostack-experimental/linux-64                               No change
robostack-experimental/noarch                                 No change
robostack/noarch                                              No change
robostack-staging/noarch                                      No change

Pinned packages:
  - python 3.9.*

Encountered problems while solving:
  - package boost-1.78.0-py39h7c9e3ff_4 requires boost-cpp 1.78.0.*, but none of the providers can be installed
```

Tobias-Fischer, 2023-01-25:
Please open a pull request at with packages you want, it’s as simple as adding one line to the vinca*.yaml file(s) in this repository. Please see our paper for more details.

peci1, 2023-01-25:
Oh, right, I'm sorry, I had this package built from source before ;)

peci1, 2023-01-25:
When does staging get published to the stable channel?

Tobias-Fischer, 2023-01-25:
We don’t really have a timeline in mind. One option would be that staging becomes the new stable .. I’m discussion with @wolfv  what the best way forward is

physar, 2023-01-26:
Solved the issue with ROS_HOSTNAME (my fault, defined in my own ~/.bashrc to communicate with a TurtleBot3.

Added documentation how to use RoboStack for this package, to make this ROS1 package also available for a Ubuntu 22.04 machine.

https://github.com/physar/ros_hokuyo_urg04lx_description