RoboStack/ros-noetic #289: SHA256 sum validation error when running mamba create code

bugramurat, 2022-09-29:
I'm getting an error when i trying to run this code:

mamba create -n robostackenv ros-noetic-desktop python=3.9 -c robostack -c robostack-experimental -c conda-forge --no-channel-priority --override-channels

I'm using Ubuntu 20.04 on virtual machine and miniconda 3

Error report:
```

# >>>>>>>>>>>>>>>>>>>>>> ERROR REPORT <<<<<<<<<<<<<<<<<<<<<<

    Traceback (most recent call last):
      File "/home/bug/miniconda3/lib/python3.9/site-packages/conda/exceptions.py", line 1082, in __call__
        return func(*args, **kwargs)
      File "/home/bug/miniconda3/lib/python3.9/site-packages/mamba/mamba.py", line 927, in exception_converter
        raise e
      File "/home/bug/miniconda3/lib/python3.9/site-packages/mamba/mamba.py", line 921, in exception_converter
        exit_code = _wrapped_main(*args, **kwargs)
      File "/home/bug/miniconda3/lib/python3.9/site-packages/mamba/mamba.py", line 879, in _wrapped_main
        result = do_call(args, p)
      File "/home/bug/miniconda3/lib/python3.9/site-packages/mamba/mamba.py", line 761, in do_call
        exit_code = create(args, parser)
      File "/home/bug/miniconda3/lib/python3.9/site-packages/mamba/mamba.py", line 664, in create
        return install(args, parser, "create")
      File "/home/bug/miniconda3/lib/python3.9/site-packages/mamba/mamba.py", line 613, in install
        downloaded = transaction.prompt(repos)
    RuntimeError: Found incorrect download: libllvm13. Aborting

`$ /home/bug/miniconda3/bin/mamba create -n robostackenv ros-noetic-desktop python=3.9 -c robostack -c robostack-experimental -c conda-forge --no-channel-priority --override-channels`

  environment variables:
                 CIO_TEST=<not set>
        CMAKE_PREFIX_PATH=/opt/ros/noetic
        CONDA_DEFAULT_ENV=base
                CONDA_EXE=/home/bug/miniconda3/bin/conda
             CONDA_PREFIX=/home/bug/miniconda3
    CONDA_PROMPT_MODIFIER=(base)
         CONDA_PYTHON_EXE=/home/bug/miniconda3/bin/python
               CONDA_ROOT=/home/bug/miniconda3
              CONDA_SHLVL=1
           CURL_CA_BUNDLE=<not set>
          LD_LIBRARY_PATH=/opt/ros/noetic/lib
                     PATH=/home/bug/miniconda3/bin:/home/bug/miniconda3/condabin:/opt/ros/noetic
                          /bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr
                          /games:/usr/local/games:/snap/bin
          PKG_CONFIG_PATH=/opt/ros/noetic/lib/pkgconfig
               PYTHONPATH=/opt/ros/noetic/lib/python3/dist-packages
       REQUESTS_CA_BUNDLE=<not set>
         ROS_PACKAGE_PATH=/opt/ros/noetic/share
            SSL_CERT_FILE=<not set>
     TERMINATOR_DBUS_PATH=/net/tenshu/Terminator2
               WINDOWPATH=2

     active environment : base
    active env location : /home/bug/miniconda3
            shell level : 1
       user config file : /home/bug/.condarc
 populated config files : 
          conda version : 4.12.0
    conda-build version : not installed
         python version : 3.9.12.final.0
       virtual packages : __linux=5.15.0=0
                          __glibc=2.31=0
                          __unix=0=0
                          __archspec=1=x86_64
       base environment : /home/bug/miniconda3  (writable)
      conda av data dir : /home/bug/miniconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack/linux-64
                          https://conda.anaconda.org/robostack/noarch
                          https://conda.anaconda.org/robostack-experimental/linux-64
                          https://conda.anaconda.org/robostack-experimental/noarch
                          https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/bug/miniconda3/pkgs
                          /home/bug/.conda/pkgs
       envs directories : /home/bug/miniconda3/envs
                          /home/bug/.conda/envs
               platform : linux-64
             user-agent : conda/4.12.0 requests/2.27.1 CPython/3.9.12 Linux/5.15.0-48-generic ubuntu/20.04.5 glibc/2.31
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False

```

maresb, 2022-09-29:
Maybe I'm misreading something, but I'm confused that the title is "SHA256 sum validation error" but instead the error message says "Found incorrect download: libllvm13. Aborting".

Is this the complete output?

In general SHA256 sum errors are often caused by proxies interfering with downloads. Are you behind any sort of proxy?

traversaro, 2022-09-30:
Hi @bugramurat, the error seems that mamba was not able to download libllvm13 archive. Did could be a temporary network issue, did you tried after some time?

bugramurat, 2022-09-30:
> 

Actually no, i will try it soon. Thanks for reply.

bugramurat, 2022-09-30:
> Hi @bugramurat, the error seems that mamba was not able to download libllvm13 archive. Did could be a temporary network issue, did you tried after some time?

Actually no, i will try it soon. Thanks for reply.

bugramurat, 2022-09-30:
> Maybe I'm misreading something, but I'm confused that the title is "SHA256 sum validation error" but instead the error message says "Found incorrect download: libllvm13. Aborting".
>
> Is this the [...]

No its not the complete output, its just error report. It seems like im having a problem when downloading. I will try to download it again soon. Thanks for reply i will let you know.

bugramurat, 2022-10-01:
It's solved when i tried to download it again. Thanks for replys.

closed 2022-10-01 by bugramurat

wolfv, 2022-10-01:
awesome, thanks for letting us know!