RoboStack/ros-noetic #290: Compiler does not find the packages

PauMatas, 2022-10-05:
Hi, first of all, thanks for your repo. I really needed ROS1 noetic for a project to run in my mac and when I had already accepted that this would be imposible I found you. The fact is that now with all ros commands running I need to compile some code but when I run `catkin_make` I receive that all the headers are not found:

```shell
In file included from In file included from /path/to/workspace/WS/src/LIMO-Velo/include/IKFoM/use-ikfom.cpp/path/to/workspace/WS/src/LIMO-Velo/src/main.cpp::33:
:
/path/to/workspace/WS/src/LIMO-Velo/include/Headers/Common.hpp/path/to/workspace/WS/src/LIMO-Velo/include/Headers/Common.hpp::44::1010:: In file included from  /path/to/workspace/WS/src/LIMO-Velo/src/Utils/Utils.cpp:fatal error: 3fatal error: :
'ros/ros.h' file not found/path/to/workspace/WS/src/LIMO-Velo/include/Headers/Common.hpp'ros/ros.h' file not found:4

:10: fatal error: 'ros/ros.h' file not found
#include <ros/ros.h>#include <ros/ros.h>

         ^~~~~~~~~~~         ^~~~~~~~~~~
#include <ros/ros.h>

         ^~~~~~~~~~~
In file included from /path/to/workspace/WS/src/LIMO-Velo/include/ikd-Tree/ikd_Tree/ikd_Tree.cpp:1:
In file included from /path/to/workspace/WS/src/LIMO-Velo/include/ikd-Tree/ikd_Tree/ikd_Tree.h:11:
In file included from /opt/miniconda3/envs/robostackenv/include/pcl-1.12/pcl/point_types.h:354:
In file included from /opt/miniconda3/envs/robostackenv/include/pcl-1.12/pcl/impl/point_types.hpp:41:
In file included from /opt/miniconda3/envs/robostackenv/include/pcl-1.12/pcl/memory.h:46:
In file included from /opt/miniconda3/envs/robostackenv/include/pcl-1.12/pcl/type_traits.h:40:
/opt/miniconda3/envs/robostackenv/include/pcl-1.12/pcl/point_struct_traits.h:40:10: fatal error: 'boost/mpl/assert.hpp' file not found
#include <boost/mpl/assert.hpp>  // for BOOST_MPL_ASSERT_MSG
```

If I search for the `ros.h` file I can find it at `/opt/miniconda3/pkgs/ros-noetic-roscpp-1.15.14-py39h6bb9c58_15/include/ros/ros.h`. Is this correct?

I have not found any clue on how to solve this. If i had to take a guess I wolud say that the environment variable that has the path of where to find packages is not correct, but I do not know which variable is this and neither I know which is the path expected that variable to have.

If you need to know which is the code I am trying to compile, you can find it [here](https://github.com/Huguet57/LIMO-Velo). It compiles perfectly to my colleagues with linux and normal ROS.

I hope you can help me, I would be really pleased.

Thanks a lot!

Tobias-Fischer, 2022-10-05:
Change this line: 
set( CMAKE_CXX_FLAGS "-std=c++14 -O3" )

To
set( CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++14 -O3" )

PauMatas, 2022-10-05:
Thank you so much!

PauMatas, 2022-10-06:
Hi, @Tobias-Fischer 

As you saw yesterday I’m having problems compiling my colleagues’ [code](https://github.com/Huguet57/LIMO-Velo). Your answer really helped me as the problem I was facing was solved, but new problems appeared. I would be really pleased if you can solve me the new problem that I’m facing and also commenting the logic behind the solution in order to not disturb you any more. And again thank you so much for your quick and effective answer yesterday. Here goes the error:

When executing catkin_make with the mentioned repo (with the change you indicated me yesterday) in src directory inside my WS I get this response:

```shell
Base path: /path/to/workspace/WS
Source space: /path/to/workspace/WS/src
Build space: /path/to/workspace/WS/build
Devel space: /path/to/workspace/WS/devel
Install space: /path/to/workspace/WS/install
####
#### Running command: "cmake /path/to/workspace/WS/src -DCATKIN_DEVEL_PREFIX=/path/to/workspace/WS/devel -DCMAKE_INSTALL_PREFIX=/path/to/workspace/WS/install -G Unix Makefiles" in "/path/to/workspace/WS/build"
####
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Deprecation Warning at CMakeLists.txt:4 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Using CATKIN_DEVEL_PREFIX: /path/to/workspace/WS/devel
-- Using CMAKE_PREFIX_PATH: /opt/miniconda3/envs/robostackenv
-- This workspace overlays: /opt/miniconda3/envs/robostackenv
-- Using PYTHON_EXECUTABLE: /opt/miniconda3/envs/robostackenv/bin/python3.9
-- Using default Python package layout
-- Using empy: /opt/miniconda3/envs/robostackenv/lib/python3.9/site-packages/em.py
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /path/to/workspace/WS/build/test_results
-- Found gtest: gtests will be built
-- Using Python nosetests: /opt/miniconda3/envs/robostackenv/bin/nosetests-3.9
-- catkin 0.7.17
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 2 packages in topological order:
-- ~~  - as_msgs
-- ~~  - limovelo
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'as_msgs'
-- ==> add_subdirectory(as_msgs)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- as_msgs: 18 messages, 1 services
-- +++ processing catkin package: 'limovelo'
-- ==> add_subdirectory(LIMO-Velo)
CMake Deprecation Warning at LIMO-Velo/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

Current CPU archtecture: x86_64
Processer number:  4
core for MP: 2
[... 94 lines cut ...]
In file included from /path/to/workspace/WS/src/LIMO-Velo/include/ikd-Tree/ikd_Tree/ikd_Tree.cpp:1:
In file included from /path/to/workspace/WS/src/LIMO-Velo/include/ikd-Tree/ikd_Tree/ikd_Tree.h:15:
/path/to/workspace/WS/src/LIMO-Velo/include/Headers/Common.hpp:126:6: error: expected ';' after struct
    } EIGEN_ALIGN16;
     ^
     ;
/path/to/workspace/WS/src/LIMO-Velo/include/Headers/Common.hpp:126:7: error: 'alignas' attribute only applies to variables, data members and tag types
    } EIGEN_ALIGN16;
      ^
/opt/miniconda3/envs/robostackenv/include/eigen3/Eigen/src/Core/util/ConfigureVectorization.h:153:23: note: expanded from macro 'EIGEN_ALIGN16'
#define EIGEN_ALIGN16 EIGEN_ALIGN_TO_BOUNDARY(16)
                      ^
/opt/miniconda3/envs/robostackenv/include/eigen3/Eigen/src/Core/util/ConfigureVectorization.h:40:38: note: expanded from macro 'EIGEN_ALIGN_TO_BOUNDARY'
  #define EIGEN_ALIGN_TO_BOUNDARY(n) alignas(n)
                                     ^
1 warning and 2 errors generated.
make[2]: *** [LIMO-Velo/CMakeFiles/limovelo.dir/build.make:76: LIMO-Velo/CMakeFiles/limovelo.dir/include/ikd-Tree/ikd_Tree/ikd_Tree.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:3410: LIMO-Velo/CMakeFiles/limovelo.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
Invoking "make -j4 -l4" failed
```

The error seems to be a code error but the same repo for all my colleagues compiles well so I guess the problem is when including some headers. Maybe they are not in the path, but i don't know which path needs to be changed...

Thank you so much!

Tobias-Fischer, 2022-10-07:
This is an error in the code - follow the same logic as in lines 109-117 re Eigen in the following lines 119-127 and it should work

closed 2022-10-07 by Tobias-Fischer