RoboStack/ros-noetic #297: rqt_graph shows an empty nonexistent node by default

fywu85, 2022-10-21:
I installed rqt_graph via the ros-noetic conda package provided by robostack. When I launch rqt_graph after roscore, I got the an nonexistent node without name.
[image]
The rqt_graph app works normally otherwise. Any idea why this occurs?

AchmadFathoni, 2023-02-19:
I can hide with by choosing form drop-down menu `Nodes/Topics (active)`

Tobias-Fischer, 2023-02-19:
Ok so can we close here?

AchmadFathoni, 2023-02-19:
It is a bug that doesn't exist in original ros-neotic in Ubuntu 20.04. This bug also exists in Arch Linux using ros-noetic that was installed from AUR.

Tobias-Fischer, 2023-02-19:
As this is minor and easily fixed I’ll mark this as wontfix

Tobias-Fischer, 2024-02-18:
Closing as minor.

closed 2024-02-18 by Tobias-Fischer