RoboStack/ros-noetic #309: Jackal Gazebo run error

QingquanBao, 2022-11-28:
After installing ros-noetic-jackal-control, ros-noetic-jackal-description , ros-noetic-jackal-gazebo , ros-noetic-jackal-msgs , ros-noetic-jackal-navigation , ros-noetic-jackal-simulator, I was trying to run `roslaunch jackal_gazebo jackal_world.launch`. But the error comes.
```
[ERROR]: Couldn't open joystick /dev/input/ps4. Will retry every second.
```

So I intended to install `ros-noetic-joystick-drivers`, but it needs `ros-noetic-spacenav-node`. When I tried to install `ros-noetic-spacenav-node`, it showed `requires ros-distro-mutex 0.3.*, but none of the providers can be installed`. 

But when I `conda list | grep mutex`, it has `ros-distro-mutex  0.3.0 noetic robostack`.

So I am confused now, thanks for any help!!!!

Tobias-Fischer, 2022-11-28:
I don’t think this is an actual error. Do you really have a joystick attached? Gazebo should work anyhow.

QingquanBao, 2022-11-28:
Hi, I put the whole log for you. My OS is Ubuntu 20.04.5. 

Firstly I thought it was the problem of GCC (GLIBCXX_3.4.30), but even if I replace a new gcc, the system still crashed.
```
xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://system76-pc:41525/

SUMMARY
========

PARAMETERS
 * /bluetooth_teleop/joy_node/autorepeat_rate: 20
 * /bluetooth_teleop/joy_node/deadzone: 0.1
 * /bluetooth_teleop/joy_node/dev: /dev/input/ps4
 * /bluetooth_teleop/teleop_twist_joy/axis_angular: 0
 * /bluetooth_teleop/teleop_twist_joy/axis_linear: 1
 * /bluetooth_teleop/teleop_twist_joy/enable_button: 4
 * /bluetooth_teleop/teleop_twist_joy/enable_turbo_button: 5
 * /bluetooth_teleop/teleop_twist_joy/scale_angular: 1.4
 * /bluetooth_teleop/teleop_twist_joy/scale_linear: 0.4
 * /bluetooth_teleop/teleop_twist_joy/scale_linear_turbo: 2.0
 * /ekf_localization/base_link_frame: base_link
 * /ekf_localization/frequency: 50
 * /ekf_localization/imu0: /imu/data
 * /ekf_localization/imu0_config: [False, False, Fa...
 * /ekf_localization/imu0_differential: False
 * /ekf_localization/odom0: /jackal_velocity_...
 * /ekf_localization/odom0_config: [False, False, Fa...
 * /ekf_localization/odom0_differential: False
 * /ekf_localization/odom_frame: odom
 * /ekf_localization/world_frame: odom
 * /gazebo/enable_ros_network: True
 * /jackal_joint_publisher/publish_rate: 50
 * /jackal_joint_publisher/type: joint_state_contr...
 * /jackal_velocity_controller/angular/z/has_acceleration_limits: True
 * /jackal_velocity_controller/angular/z/has_velocity_limits: True
 * /jackal_velocity_controller/angular/z/max_acceleration: 25.0
 * /jackal_velocity_controller/angular/z/max_velocity: 4.0
 * /jackal_velocity_controller/cmd_vel_timeout: 0.25
 * /jackal_velocity_controller/enable_odom_tf: False
 * /jackal_velocity_controller/k_l: 0.1
 * /jackal_velocity_controller/k_r: 0.1
 * /jackal_velocity_controller/left_wheel: ['front_left_whee...
 * /jackal_velocity_controller/linear/x/has_acceleration_limits: True
 * /jackal_velocity_controller/linear/x/has_velocity_limits: True
 * /jackal_velocity_controller/linear/x/max_acceleration: 20.0
 * /jackal_velocity_controller/linear/x/max_velocity: 2.0
 * /jackal_velocity_controller/pose_covariance_diagonal: [0.001, 0.001, 10...
 * /jackal_velocity_controller/publish_rate: 50
 * /jackal_velocity_controller/right_wheel: ['front_right_whe...
 * /jackal_velocity_controller/twist_covariance_diagonal: [0.001, 0.001, 0....
 * /jackal_velocity_controller/type: diff_drive_contro...
 * /jackal_velocity_controller/wheel_radius_multiplier: 1.0
 * /jackal_velocity_controller/wheel_separation_multiplier: 1.5
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.14
 * /twist_mux/locks: [{'name': 'e_stop...
 * /twist_mux/topics: [{'name': 'joy', ...
 * /use_sim_time: True

NODES
  /
    controller_spawner (controller_manager/spawner)
    ekf_localization (robot_localization/ekf_localization_node)
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    twist_marker_server (interactive_marker_twist_server/marker_server)
    twist_mux (twist_mux/twist_mux)
[... 16 lines cut ...]
process[urdf_spawner-10]: started with pid [97322]
[ERROR] [1669658191.295569464]: Couldn't open joystick /dev/input/ps4. Will retry every second.
gzclient: /home/user/miniconda3/envs/env/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /home/user/miniconda3/envs/env/bin/../lib/./libgdal.so.30)
gzclient: /home/user/miniconda3/envs/env/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /home/user/miniconda3/envs/env/bin/../lib/././libtiledb.so.2.8)
gzserver: /home/user/miniconda3/envs/env/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /home/user/miniconda3/envs/env/bin/../lib/./libgdal.so.30)
gzserver: /home/user/miniconda3/envs/env/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /home/user/miniconda3/envs/env/bin/../lib/././libtiledb.so.2.8)
[gazebo-1] process has died [pid 97288, exit code 1, cmd /home/user/miniconda3/envs/env/lib/gazebo_ros/gzserver -e ode /home/user/miniconda3/envs/env/share/jackal_gazebo/worlds/jackal_race.world __name:=gazebo __log:=/home/user/.ros/log/efee578c-6f43-11ed-8eb4-28d0ea7c6857/gazebo-1.log].
log file: /home/user/.ros/log/efee578c-6f43-11ed-8eb4-28d0ea7c6857/gazebo-1*.log
[gazebo_gui-2] process has died [pid 97292, exit code 1, cmd /home/user/miniconda3/envs/env/lib/gazebo_ros/gzclient __name:=gazebo_gui __log:=/home/user/.ros/log/efee578c-6f43-11ed-8eb4-28d0ea7c6857/gazebo_gui-2.log].
log file: /home/user/.ros/log/efee578c-6f43-11ed-8eb4-28d0ea7c6857/gazebo_gui-2*.log
[ WARN] [1669658191.650187895]: Shutdown request received.
[ WARN] [1669658191.652057979]: Reason given for shutdown: [[/urdf_spawner] Reason: new node registered with same name]
[robot_state_publisher-3] process has finished cleanly
log file: /home/user/.ros/log/efee578c-6f43-11ed-8eb4-28d0ea7c6857/robot_state_publisher-3*.log

```

```
[ERROR] [1669665299.832174127]: Couldn't open joystick /dev/input/ps4. Will retry every second.
../src/intel/isl/isl.c:2095: FINISHME: ../src/intel/isl/isl.c:isl_surf_supports_ccs: CCS for 3D textures is disabled, but a workaround is available.
Segmentation fault (core dumped)
[gazebo_gui-2] process has died [pid 211486, exit code 139, cmd /home/user/miniconda3/envs/env/lib/gazebo_ros/gzclient __name:=gazebo_gui __log:=/home/user/.ros/log/efee578c-6f43-11ed-8eb4-28d0ea7c6857/gazebo_gui-2.log].
log file: /home/user/.ros/log/efee578c-6f43-11ed-8eb4-28d0ea7c6857/gazebo_gui-2*.log
[WARN] [1669665330.328269, 0.000000]: Controller Spawner couldn't find the expected controller_manager ROS interface.
```

Tobias-Fischer, 2022-11-28:
The interesting lines are these:
```
gzclient: /home/user/miniconda3/envs/env/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /home/user/miniconda3/envs/env/bin/../lib/./libgdal.so.30)
gzclient: /home/user/miniconda3/envs/env/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /home/user/miniconda3/envs/env/bin/../lib/././libtiledb.so.2.8)
```

Not sure what causes that - could you please provide the output of `mamba list`?

Tobias-Fischer, 2022-11-28:
Also could you please try `mamba install -c conda-forge libstdcxx-ng=12`?

QingquanBao, 2022-11-29:
> Also could you please try `mamba install -c conda-forge libstdcxx-ng=12`?

After install this, here comes the error:
```
[ERROR] [1669737203.300006416]: Couldn't open joystick /dev/input/ps4. Will retry every second.
../src/intel/isl/isl.c:2095: FINISHME: ../src/intel/isl/isl.c:isl_surf_supports_ccs: CCS for 3D textures is disabled, but a workaround is available.
Segmentation fault (core dumped)
[gazebo_gui-2] process has died [pid 253100, exit code 139, cmd /home/user/miniconda3/envs/env/lib/gazebo_ros/gzclient __name:=gazebo_gui __log:=/home/user/.ros/log/ec6c7124-6ffd-11ed-9068-28d0ea7c6857/gazebo_gui-2.log].
log file: /home/user/.ros/log/ec6c7124-6ffd-11ed-9068-28d0ea7c6857/gazebo_gui-2*.log
[WARN] [1669737233.876840, 0.000000]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[controller_spawner-4] process has finished cleanly
log file: /home/user/.ros/log/ec6c7124-6ffd-11ed-9068-28d0ea7c6857/controller_spawner-4*.log
```

Here I give the mamba list for you. Another strange thing is that I can not find the gazebo-gui log. Thank you so much 
```
on-setup-py    0.2.7                      py_0    conda-forge
colcon-recursive-crawl    0.2.1                      py_0    conda-forge
colcon-ros                0.3.23             pyhd8ed1ab_0    conda-forge
colcon-test-result        0.3.8                      py_0    conda-forge
colcon-zsh                0.4.0              pyhd8ed1ab_0    conda-forge
coloredlogs               15.0.1           py39hf3d152e_2    conda-forge
compilers                 1.5.1                ha770c72_0    conda-forge
console_bridge            1.0.2                h924138e_1    conda-forge
coreutils                 9.1                  h166bdaf_0    conda-forge
coverage                  6.5.0            py39hb9d737c_1    conda-forge
cppzmq                    4.8.1                hf7cf922_0    conda-forge
cryptography              38.0.4           py39hd97740a_0    conda-forge
curl                      7.86.0               h7bff187_1    conda-forge
cxx-compiler              1.5.1                h924138e_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
dartsim                   6.12.1               h9f176f7_3    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distlib                   0.3.6              pyhd8ed1ab_0    conda-forge
distro                    1.6.0              pyhd8ed1ab_0    conda-forge
docutils                  0.19             py39hf3d152e_1    conda-forge
double-conversion         3.2.0                h9c3ff4c_0    conda-forge
eigen                     3.4.0                h4bd325d_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
exceptiongroup            1.0.4              pyhd8ed1ab_0    conda-forge
expat                     2.4.8                h27087fc_0    conda-forge
fcl                       0.7.0                h2cbc392_0    conda-forge
ffmpeg                    4.4.1                hd7ab26d_2    conda-forge
flann                     1.9.1             hfe772e8_1010    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.0               hc2a2eb6_1    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.38.0           py39hb9d737c_1    conda-forge
fortran-compiler          1.5.1                h2a4ca65_0    conda-forge
freeglut                  3.2.2                h9c3ff4c_1    conda-forge
freeimage                 3.18.0               he1b5a44_0    conda-forge
freetype                  2.12.1               hca18f0e_1    conda-forge
freexl                    1.0.6                h166bdaf_1    conda-forge
fribidi                   1.0.10               h36c2ea0_0    conda-forge
gazebo                    11.10.2              h40dda3e_2    conda-forge
gcc                       10.4.0              hb92f740_11    conda-forge
gcc_impl_linux-64         10.4.0              h5231bdf_19    conda-forge
gcc_linux-64              10.4.0              h9215b83_11    conda-forge
gdbm                      1.18                 h0a1914f_2    conda-forge
gdk-pixbuf                2.42.8               hff1cb4f_0    conda-forge
geographiclib-cpp         1.52                 h4bd325d_1    conda-forge
geos                      3.10.2               h9c3ff4c_0    conda-forge
geotiff                   1.7.1                h509b78c_1    conda-forge
gettext                   0.21.1               h27087fc_0    conda-forge
gfortran                  10.4.0              h0c96582_11    conda-forge
[... 506 lines cut ...]
xorg-libice               1.0.10               h7f98852_0    conda-forge
xorg-libsm                1.2.3             hd9c2040_1000    conda-forge
xorg-libx11               1.7.2                h7f98852_0    conda-forge
xorg-libxau               1.0.9                h7f98852_0    conda-forge
xorg-libxaw               1.0.14               h7f98852_1    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
[... cut ...]
yaml                      0.2.5                h7f98852_2    conda-forge
yaml-cpp                  0.6.3                he1b5a44_4    conda-forge
zeromq                    4.3.4                h9c3ff4c_1    conda-forge
zipp                      3.11.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               h166bdaf_4    conda-forge
zstd                      1.5.2                h8a70e8d_1    conda-forge
zziplib                   0.13.69              h27826a3_1    conda-forge
```

Tobias-Fischer, 2023-01-24:
Please try again with packages from the robostack-staging channel. Feel free to reopen.

closed 2023-01-24 by Tobias-Fischer