RoboStack/ros-noetic #313: mamba installation tries to fiddle with non .conda/ directory

husnoo, 2022-12-10:
```
mamba create -n robostackenv ros-noetic-desktop python=3.9 -c robostack -c robostack-experimental -c conda-forge --no-channel-priority --override-channels

Executing transaction: / Failed to create file “/nix/store/gcfl4r7b73ksgdmn9s0rw7brb6cj4fcy-gdk-pixbuf-loaders.cache.84IYW1”: Read-only file system

/ Failed to create file “/nix/store/gcfl4r7b73ksgdmn9s0rw7brb6cj4fcy-gdk-pixbuf-loaders.cache.8ZUCX1”: Read-only file system
```

This is on nixos-21.11.337975.eabc3821918 (Porcupine).

Tobias-Fischer, 2022-12-10:
Is that particular to RoboStack or does it happen elsewhere, too? 

@wolfv

traversaro, 2022-12-11:
Probably this could be related to https://github.com/conda-forge/conda-forge.github.io/issues/1539 .

Tobias-Fischer, 2023-01-25:
Does this also happen when you try to `mamba create -n testenv gdk-pixbuf -c conda-forge`?

Tobias-Fischer, 2023-07-19:
Closing, as didn't hear back and seems to be a conda-wide problem, not RoboStack specific.

closed 2023-07-19 by Tobias-Fischer