RoboStack/ros-noetic #318: moveit_core exports liboctomap.so.1.9.7 as link dependency, by 1.9.8 is installed

alaurenzi, 2022-12-23:
Hi all, let me first thank you for this huge effort!
I am trying to install ROS1 inside a conda environment, on top of which I want to build some custom source code.

I have successfully installed `ros-noetic-desktop-full`, and `ros-noetic-moveit-core`.

The problem I am experiencing is the following. The cmake files exported by `moveit_core` include `liboctomap.so.1.9.7` as link dependency:
```
(ros_env) arturo@arturo-iit-laptop ~/miniconda3/envs/ros_env/share/moveit_core/cmake $ cat moveit_coreConfig.cmake | grep octomap
set(libraries "moveit_exceptions;moveit_background_processing;moveit_kinematics_base;moveit_robot_model;moveit_transforms;moveit_robot_state;moveit_robot_trajectory;moveit_planning_interface;moveit_collision_detection;moveit_collision_detection_fcl;moveit_collision_detection_bullet;moveit_kinematic_constraints;moveit_planning_scene;moveit_constraint_samplers;moveit_planning_request_adapter;moveit_profiler;moveit_python_tools;moveit_trajectory_processing;moveit_distance_field;moveit_collision_distance_field;moveit_kinematics_metrics;moveit_dynamics_solver;moveit_utils;moveit_test_utils;/home/arturo/miniconda3/envs/ros_env/lib/libboost_system.so.1.74.0;/home/arturo/miniconda3/envs/ros_env/lib/libboost_filesystem.so.1.74.0;/home/arturo/miniconda3/envs/ros_env/lib/libboost_date_time.so.1.74.0;/home/arturo/miniconda3/envs/ros_env/lib/libboost_thread.so.1.74.0;/home/arturo/miniconda3/envs/ros_env/lib/libboost_iostreams.so.1.74.0;/home/arturo/miniconda3/envs/ros_env/lib/libboost_regex.so.1.74.0;/home/arturo/miniconda3/envs/ros_env/lib/libfcl.so.0.6.1;/home/arturo/miniconda3/envs/ros_env/lib/libccd.so.2.0;/home/arturo/miniconda3/envs/ros_env/lib/liboctomap.so.1.9.7;/home/arturo/miniconda3/envs/ros_env/lib/liboctomap.so;/home/arturo/miniconda3/envs/ros_env/lib/liboctomath.so;/home/arturo/miniconda3/envs/ros_env/lib/liburdfdom_sensor.so;/home/arturo/miniconda3/envs/ros_env/lib/liburdfdom_model_state.so;/home/arturo/miniconda3/envs/ros_env/lib/liburdfdom_model.so;/home/arturo/miniconda3/envs/ros_env/lib/liburdfdom_world.so;/home/arturo/miniconda3/envs/ros_env/lib/libconsole_bridge.so.1.0;BulletSoftBody;BulletDynamics;BulletCollision;LinearMath")
set(depends "eigen_stl_containers;geometric_shapes;geometry_msgs;kdl_parser;moveit_msgs;octomap_msgs;random_numbers;sensor_msgs;shape_msgs;srdfdom;std_msgs;tf2_eigen;tf2_geometry_msgs;trajectory_msgs;visualization_msgs")
```

However, version `1.9.8` seems to be installed:
```
(ros_env) arturo@arturo-iit-laptop ~/miniconda3/envs/ros_env $ ls lib/liboctoma*
lib/liboctomap.a   lib/liboctomap.so.1.9    lib/liboctomath.a   lib/liboctomath.so.1.9
lib/liboctomap.so  lib/liboctomap.so.1.9.8  lib/liboctomath.so  lib/liboctomath.so.1.9.8
```

Leading to linking errors when I try to compile my own source code, that is supposed to link against moveit, and therefore against the non-existing `liboctomap.so.1.9.7`.

Output of `conda list`:
```
# packages in environment at /home/arturo/miniconda3/envs/ros_env:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
alsa-lib                  1.2.3.2              h166bdaf_0    conda-forge
aom                       3.3.0                h27087fc_1    conda-forge
apr                       1.7.0                h7f98852_5    conda-forge
argcomplete               2.0.0              pyhd8ed1ab_0    conda-forge
assimp                    5.2.3                h4c92fa2_1    conda-forge
atk-1.0                   2.38.0               hd4edc92_1    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     22.2.0             pyh71513ae_0    conda-forge
bcrypt                    3.2.2            py39hb9d737c_1    conda-forge
binutils                  2.39                 hdd6e379_1    conda-forge
binutils_impl_linux-64    2.39                 he00db2b_1    conda-forge
binutils_linux-64         2.39                h5fc0e48_11    conda-forge
blosc                     1.21.3               hafa529b_0    conda-forge
boost                     1.74.0           py39h5472131_5    conda-forge
boost-cpp                 1.74.0               h6cacc03_7    conda-forge
brotli                    1.0.9                h166bdaf_8    conda-forge
brotli-bin                1.0.9                h166bdaf_8    conda-forge
bullet                    3.21                 hcca971b_4    conda-forge
bullet-cpp                3.21                 h4661b88_4    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.18.1               h7f98852_0    conda-forge
c-compiler                1.5.2                h0b41bf4_0    conda-forge
ca-certificates           2022.12.7            ha878542_0    conda-forge
cairo                     1.16.0            ha12eb4b_1010    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
certifi                   2022.12.7          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1           py39he91dace_3    conda-forge
cfitsio                   4.1.0                hd9d235c_0    conda-forge
cmake                     3.25.1               h816a3e0_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.4.2              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
colcon-cmake              0.2.27             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0            py39hf3d152e_1    conda-forge
colcon-core               0.10.0             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.5                      py_0    conda-forge
colcon-devtools           0.2.2                      py_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-output             0.2.12                     py_0    conda-forge
colcon-package-information 0.3.3                      py_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
[... 561 lines cut ...]
xorg-libxau               1.0.9                h7f98852_0    conda-forge
xorg-libxaw               1.0.14               h7f98852_1    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h7f98852_1    conda-forge
xorg-libxfixes            5.0.3             h7f98852_1004    conda-forge
xorg-libxi                1.7.10               h7f98852_0    conda-forge
xorg-libxmu               1.1.3                h7f98852_0    conda-forge
xorg-libxpm               3.5.13               h7f98852_0    conda-forge
xorg-libxrender           0.9.10            h7f98852_1003    conda-forge
xorg-libxt                1.2.1                h7f98852_2    conda-forge
xorg-renderproto          0.11.1            h7f98852_1002    conda-forge
xorg-xextproto            7.3.0             h7f98852_1002    conda-forge
xorg-xproto               7.0.31            h7f98852_1007    conda-forge
xz                        5.2.6                h166bdaf_0    conda-forge
yaml                      0.2.5                h7f98852_2    conda-forge
yaml-cpp                  0.6.3                he1b5a44_4    conda-forge
zeromq                    4.3.4                h9c3ff4c_1    conda-forge
zipp                      3.11.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               h166bdaf_4    conda-forge
zstd                      1.5.2                h6239696_4    conda-forge
zziplib                   0.13.69              h27826a3_1    conda-forge
```

Thank you in advance for you kind attention,
Arturo

traversaro, 2022-12-24:
Hi @alaurenzi , thanks for the report! Just to double check, can you try installing mamba install octomap==1.9.7 and see if then that works? Clearly this is just a workaround.

Tobias-Fischer, 2022-12-24:
What do you think the issue is @traversaro? They set the SOVERSION correctly to minor, but somehow down  stream links to the patch version ..

traversaro, 2022-12-24:
> What do you think the issue is @traversaro? They set the SOVERSION correctly to minor, but somehow down stream links to the patch version ..

Yes, I was looking at this as well.

alaurenzi, 2022-12-27:
Indeed this works (meanwhile I switched to Mambaforge but I don't believe it makes any difference)

```bash
 mamba create -n ros_env python=3.9
 mamba activate ros_env
 conda config --env --add channels robostack
 mamba install ros-noetic-desktop-full ros-noetic-moveit-core octomap=1.9.7
```

traversaro, 2022-12-28:
I looked a bit into it. I think the problem is that when `moveit_coreConfig.cmake` is generated by catkin in https://github.com/ros/catkin/blob/ff31d451ce1c68d47bc058a4693aad6c0fb63a43/cmake/catkin_package.cmake#L258 the `catkin_replace_imported_library_targets` function is called, and that function uses the name with the full version names, for example for octomap (from `octomap-targets-release.cmake`):

~~~
set_property(TARGET octomap APPEND PROPERTY IMPORTED_CONFIGURATIONS RELEASE)
set_target_properties(octomap PROPERTIES
  IMPORTED_LOCATION_RELEASE "${_IMPORT_PREFIX}/lib/liboctomap.so.1.9.8"
  IMPORTED_SONAME_RELEASE "liboctomap.so.1.9"
  )
~~~

The value used is the one in `IMPORTED_LOCATION_RELEASE` and the value in `IMPORTED_SONAME_RELEASE` is ignored.

I think the upstream issue that tracks this is https://github.com/ros/catkin/issues/1141, there is also an unmerged PR in https://github.com/ros/catkin/pull/1142 that we could import in robostack.

traversaro, 2022-12-29:
I tested https://github.com/ros/catkin/pull/1142 and it works fine to avoid the problem.

closed 2023-01-02 by Tobias-Fischer