RoboStack/ros-noetic #319: ROS Noetic + gzclient crashing with SIGSEGV

alaurenzi, 2022-12-27:
Hi RoboStack mantainers, 
I am facing an issue where a plain invocation of Gazebo's (classic) GUI process `gzclient` is crashing with a segmentation fault.

Platform: Ubuntu 22.04 x86_64

[Output of `conda list`](https://pastebin.com/raw/nHxZeFQG)

Output of `gzserver --verbose`
```
(ros_env) arturo@arturo-iit-laptop ~ $ gzserver --verbose
Gazebo multi-robot simulator, version 11.10.2
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.1.51
[Msg] Loading world file [/home/arturo/mambaforge/envs/ros_env/share/gazebo-11/worlds/empty.world]
```

Output of `gdb gzclient` 
```
Starting program: /home/arturo/mambaforge/envs/ros_env/bin/gzclient 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe74d3640 (LWP 341737)]
[New Thread 0x7fffe6cd2640 (LWP 341738)]
[New Thread 0x7fffe64d1640 (LWP 341739)]
[New Thread 0x7fffe5c6d640 (LWP 341740)]
[New Thread 0x7fffe546c640 (LWP 341741)]
[New Thread 0x7fffe4c6b640 (LWP 341742)]
[New Thread 0x7fffd7fff640 (LWP 341743)]
[New Thread 0x7fffd77fe640 (LWP 341744)]
[New Thread 0x7fffd6ffd640 (LWP 341745)]
[New Thread 0x7fffd5d32640 (LWP 341746)]
[New Thread 0x7fffd5531640 (LWP 341747)]
[New Thread 0x7fffd4d30640 (LWP 341748)]
[New Thread 0x7fffcddff640 (LWP 341749)]
[New Thread 0x7fffcd5fe640 (LWP 341750)]
[New Thread 0x7fffccdfd640 (LWP 341751)]
[New Thread 0x7fffb3fff640 (LWP 341752)]
[New Thread 0x7fffb37fe640 (LWP 341753)]
[New Thread 0x7fffb2ffd640 (LWP 341754)]
[New Thread 0x7fffb27fc640 (LWP 341755)]
[New Thread 0x7fffb17fa640 (LWP 341757)]
[New Thread 0x7fffb1ffb640 (LWP 341758)]
[New Thread 0x7fff8bfff640 (LWP 341760)]
[New Thread 0x7fffb09f5640 (LWP 341759)]
[New Thread 0x7fff8b7fe640 (LWP 341762)]
[New Thread 0x7fffb05f4640 (LWP 341761)]
[New Thread 0x7fff8affc640 (LWP 341764)]
[New Thread 0x7fff8b3fd640 (LWP 341763)]
[New Thread 0x7fff8abfb640 (LWP 341765)]
[New Thread 0x7fff8a7fa640 (LWP 341766)]
[Thread 0x7fffb17fa640 (LWP 341757) exited]
[Thread 0x7fff8bfff640 (LWP 341760) exited]
[New Thread 0x7fff8bfff640 (LWP 341767)]
[Thread 0x7fff8bfff640 (LWP 341767) exited]
[New Thread 0x7fff8bfff640 (LWP 341768)]
[Thread 0x7fff8bfff640 (LWP 341768) exited]
Error while mapping shared library sections:
`/home/arturo/mambaforge/envs/ros_env/bin/../lib/.': not in executable format: file format not recognized
[New Thread 0x7fff8bfff640 (LWP 341769)]
[New Thread 0x7fffb17fa640 (LWP 341770)]
[Thread 0x7fff8bfff640 (LWP 341769) exited]
[Thread 0x7fffb17fa640 (LWP 341770) exited]

Thread 1 "gzclient" received signal SIGSEGV, Segmentation fault.
0x00007ffff5ea3361 in ?? () from /home/arturo/mambaforge/envs/ros_env/bin/../lib/./libQt5Widgets.so.5
(gdb) bt
#0  0x00007ffff5ea3361 in ?? ()
   from /home/arturo/mambaforge/envs/ros_env/bin/../lib/./libQt5Widgets.so.5
#1  0x00007ffff5ebf059 in QWidgetPrivate::showChildren(bool) ()
   from /home/arturo/mambaforge/envs/ros_env/bin/../lib/./libQt5Widgets.so.5
#2  0x00007ffff5ebf16d in QWidgetPrivate::show_helper() ()
   from /home/arturo/mambaforge/envs/ros_env/bin/../lib/./libQt5Widgets.so.5
#3  0x00007ffff5ec126b in QWidgetPrivate::setVisible(bool) ()
   from /home/arturo/mambaforge/envs/ros_env/bin/../lib/./libQt5Widgets.so.5
#4  0x00007ffff5ebf128 in QWidgetPrivate::showChildren(bool) ()
   from /home/arturo/mambaforge/envs/ros_env/bin/../lib/./libQt5Widgets.so.5
#5  0x00007ffff5ebf16d in QWidgetPrivate::show_helper() ()
   from /home/arturo/mambaforge/envs/ros_env/bin/../lib/./libQt5Widgets.so.5
#6  0x00007ffff5ec126b in QWidgetPrivate::setVisible(bool) ()
   from /home/arturo/mambaforge/envs/ros_env/bin/../lib/./libQt5Widgets.so.5
#7  0x00007ffff766d701 in gazebo::gui::MainWindow::MainWindow() ()
   from /home/arturo/mambaforge/envs/ros_env/bin/../lib/libgazebo_gui.so.11
#8  0x00007ffff763529d in gazebo::gui::load() ()
   from /home/arturo/mambaforge/envs/ros_env/bin/../lib/libgazebo_gui.so.11
#9  0x00007ffff7635445 in gazebo::gui::run(int, char**) ()
   from /home/arturo/mambaforge/envs/ros_env/bin/../lib/libgazebo_gui.so.11
#10 0x0000555555559598 in ?? ()
#11 0x00007ffff6562d90 in __libc_start_call_main (
    main=main@entry=0x555555559550, argc=argc@entry=1, 
    argv=argv@entry=0x7fffffffba28) at ../sysdeps/nptl/libc_start_call_main.h:58
#12 0x00007ffff6562e40 in __libc_start_main_impl (main=0x555555559550, argc=1, 
    argv=0x7fffffffba28, init=<optimized out>, fini=<optimized out>, 
    rtld_fini=<optimized out>, stack_end=0x7fffffffba18)
    at ../csu/libc-start.c:392
#13 0x000055555555963c in ?? ()
(gdb) 
```

Guessing it could be related to [this issue I found](https://github.com/conda-forge/gazebo-feedstock/issues/119), I also tried the following inside a fresh environment:

```
mamba install -c robostack ros-noetic-desktop-full ros-noetic-moveit-core gazebo=11.9 qt=5.12.9=hda022c4_4
```

However, the solver complains: 
```
Encountered problems while solving:
  - package ros-noetic-gazebo-ros-pkgs-2.9.2-py39h6fdeb60_13 requires ros-noetic-gazebo-dev, but none of the providers can be installed`
```

I am available to share any further information to help debugging this.

Thanks in advance for your kind attention, 

Arturo

alaurenzi, 2022-12-27:
Update (sorry for early posting)

This works: `mamba install ros-noetic-desktop-full ros-noetic-moveit-core  qt=5.12.9=hda022c4_4`

traversaro, 2022-12-29:
Yes, indeed this is the issue with qt 5.12 that we tracked in https://github.com/conda-forge/gazebo-feedstock/issues/119 . The issue is solved with qt 5.15, and given that with qt 5.12 there was the `qt=5.12.9=hda022c4_4` workaround we kind of not investigate it too much. 
However, I noticed now that in ros-noetic currently we are using (from https://github.com/RoboStack/ros-noetic/blob/8224ae4e3fdc2689eee269936a40933e3f2a5121/robostack.yaml#L334):
~~~
libqt5-core:
  robostack:
    linux: [qt 5.12]
    osx: [qt-main]
    win64: [qt 5]
~~~

5.15 (qt-main is available only for 5.15) in macOS, qt 5.12 for Linux, and I am not sure which one for Windows. @Tobias-Fischer if on the macOS side we do not have too many qt-related failure due to the use of qt 5.15, we could think of switching to 5.15 on all platfomrs?

Tobias-Fischer, 2023-01-02:
Indeed, this is the plan in #62 :)

Tobias-Fischer, 2023-01-24:
Can you try again with the new packages in robostack-staging? If error persists, please reopen.

closed 2023-01-24 by Tobias-Fischer

alaurenzi, 2023-01-24:
All good on `robostack-staging`. How can I be notified as soon as the updated packages become available on the default `robostack` channel? (like, subscribing to some PR, or similar)