RoboStack/ros-noetic #323: Cannot add models to Gazebo world on macOS with Intel chip

troiwill, 2023-01-16:
## Issue:
Hi, this issue is a follow up from another issue (#279); this post provides more detail than what I mentioned at the end of #279. Basically, I cannot add any model (including the stock cube and spheres) to a Gazebo world on macOS; I experienced the same issue as the other user. However, I can confirm that this issue *only* happens on an Intel-based Mac. I *do not* have this issue on an M1 Mac.

## Fixes Tried:
1. *Update Gazebo:*
I tried to update gazebo from `11.10.2 -> 11.12.0` (as mention [at this link](https://github.com/RoboStack/ros-noetic/issues/279#issuecomment-1277926991)). However, I get the following error when I try to update gazebo
```
Encountered problems while solving:
  - package gazebo-11.12.0-h063ddd6_1 requires assimp >=5.2.4,<5.2.5.0a0, but none of the providers can be installed
```
because `ros-noetic-rviz` requires the currently installed version of `assimp`.

2. *Remap gazebo softlinks:*
I also installed `gazebo 11.12.0` locally on my machine (via `homebrew`). Then I changed the gazebo softlinks in the `conda` environment to point to the locally installed gazebo. However, this attempt did not work.

## System Specs + Environment Setup:

Here is the specs for the Intel-based Mac.
```
MacBook Pro (13-inch, 2018)
2.3 GHz Quad-Core Intel Core i5
16 GB 2133 MHz LPDDR3
Intel Iris Plus Graphics 655 1536 MB
macOS Ventura 13.0.1
```

Here is the list of conda packages.
```
# Name                    Version                   Build  Channel
aom                       3.5.0                hf0c8a7f_0    conda-forge
apr                       1.7.0                h0d85af4_5    conda-forge
assimp                    5.2.4                hfcdac6d_0    conda-forge
bcrypt                    3.2.2            py39ha30fb19_1    conda-forge
blosc                     1.21.2               hebb52c4_0    conda-forge
boost                     1.74.0           py39ha1f3e3e_5    conda-forge
boost-cpp                 1.74.0               hdbf7018_7    conda-forge
brotli                    1.0.9                hb7f2c08_8    conda-forge
brotli-bin                1.0.9                hb7f2c08_8    conda-forge
bullet-cpp                3.21                 hc6b5a04_4    conda-forge
bzip2                     1.0.8                h0d85af4_4    conda-forge
c-ares                    1.18.1               h0d85af4_0    conda-forge
c-compiler                1.5.2                hbf74d83_0    conda-forge
ca-certificates           2022.12.7            h033912b_0    conda-forge
cairo                     1.16.0            h9e0e54b_1010    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.2              pyhd8ed1ab_0    conda-forge
cctools                   973.0.1             h76f1dac_11    conda-forge
cctools_osx-64            973.0.1             hcc6d90d_11    conda-forge
certifi                   2022.12.7        py39hecd8cb5_0  
cffi                      1.15.1           py39h131948b_3    conda-forge
cfitsio                   4.1.0                h2c97ad1_0    conda-forge
clang                     14.0.6               h694c41f_0    conda-forge
clang-14                  14.0.6          default_h55ffa42_0    conda-forge
clang_osx-64              14.0.6               h3113cd8_4    conda-forge
clangxx                   14.0.6          default_h55ffa42_0    conda-forge
clangxx_osx-64            14.0.6               h6f97653_4    conda-forge
cmake                     3.25.1               h4032537_0    conda-forge
compiler-rt               14.0.6               h613da45_0    conda-forge
compiler-rt_osx-64        14.0.6               hab78ec2_0    conda-forge
compilers                 1.5.2                h694c41f_0    conda-forge
console_bridge            1.0.2                hbb4e6a2_1    conda-forge
contourpy                 1.0.7            py39h92daf61_0    conda-forge
coreutils                 9.1                  h5eb16cf_0    conda-forge
cppzmq                    4.9.0                ha27c2bb_0    conda-forge
cryptography              39.0.0           py39h7eb6a14_0    conda-forge
curl                      7.86.0               h57eb407_1    conda-forge
cxx-compiler              1.5.2                hb8565cd_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
dartsim                   6.12.1               h4b1f789_2    conda-forge
dbus                      1.13.6               h811a1a6_3    conda-forge
[... 483 lines cut ...]
xorg-fixesproto           5.0               h0d85af4_1002    conda-forge
xorg-kbproto              1.0.7             h35c211d_1002    conda-forge
xorg-libice               1.0.10               h0d85af4_0    conda-forge
xorg-libsm                1.2.3             h0d85af4_1000    conda-forge
xorg-libx11               1.7.2                h0d85af4_0    conda-forge
xorg-libxau               1.0.9                h35c211d_0    conda-forge
xorg-libxaw               1.0.14               h0d85af4_1    conda-forge
xorg-libxdmcp             1.1.3                h35c211d_0    conda-forge
xorg-libxext              1.3.4                h0d85af4_1    conda-forge
xorg-libxfixes            5.0.3             h0d85af4_1004    conda-forge
xorg-libxmu               1.1.3                h0d85af4_0    conda-forge
xorg-libxpm               3.5.13               h0d85af4_0    conda-forge
xorg-libxrender           0.9.10            h0d85af4_1003    conda-forge
xorg-libxt                1.2.1                h0d85af4_2    conda-forge
xorg-renderproto          0.11.1            h0d85af4_1002    conda-forge
xorg-xextproto            7.3.0             h35c211d_1002    conda-forge
xorg-xproto               7.0.31            h35c211d_1007    conda-forge
xz                        5.2.8                h6c40b1e_0  
yaml                      0.2.5                h0d85af4_2    conda-forge
yaml-cpp                  0.6.3                hb1e8313_4    conda-forge
zeromq                    4.3.4                he49afe7_1    conda-forge
zlib                      1.2.13               hfd90126_4    conda-forge
zstd                      1.5.2                hfa58983_4    conda-forge
zziplib                   0.13.69              h5dbffcc_1    conda-forge
```

mora200217, 2023-01-18:
Same problem here with macOS Monterey 12.6.2. Ros Noetic, Gazebo 11.10 via Robostack

mora200217, 2023-01-18:
It also crashes when trying to pause de sim?

mora200217, 2023-01-18:
Well. First update. downgrading the python version to 3.7, there's no problem installing gazebo 11.12.0. However, still the same crashing behaviour.

traversaro, 2023-01-18:
How are you adding models to the world, via the Gazebo gui? In that case, it may be useful to check if this happens also on an environment with just gazebo, i.e. :

~~~
# In a new terminal in which the previous ros environment is not activated
mamba create -n gazebo gazebo
mamba activate gazebo
# launch gazebo and check if the problem is still there
~~~

Thanks!

troiwill, 2023-01-18:
> It also crashes when trying to pause de sim? @troiwill

@mora200217: Yes. Furthermore, the (real and simulated) times, FPS, and so on at the bottom of the gazebo window do not work at all (even if I do a simple `roslaunch gazebo_ros empty_world.launch`.

troiwill, 2023-01-18:
> How are you adding models to the world, via the Gazebo gui? In that case, it may be useful to check if this happens also on an environment with just gazebo, i.e. :
>
> ```
> # In a new terminal in which [...]

@traversaro: I tried to add models via `roslaunch` and the gazebo GUI for ROS Noetic. Both resulted in crashes.

I also tried what you suggested:
```
mamba create -n gazebo gazebo -c conda-forge
```
Gazebo does not work in this case either. I should note that the gazebo version that works on the Mac M1 is `gazebo=11.10.1` and comes from `robostack`. Could the issue stem from the channel installation source or slight version difference? As a reminder, the gazebo version for the Intel chip is `gazebo=11.10.2` and comes from `conda-forge`.

mora200217, 2023-01-20:
@troiwill Update: I've created a conda env with python=3.7 and install gazebo 11.5. It works.

traversaro, 2023-01-20:
Sorry for taking long to look at this. This is unfortunatly a known issue with `gazebo=5.12.9=h2a607e2_5`, see https://github.com/conda-forge/gazebo-feedstock/issues/119 . Once we rebuild everything with qt-main 5.15, everything should be solved, in the meanwhile for each environment in which you have `` installed, you should be able to solve this by installing:
~~~
mamba install qt=5.12.9=hda022c4_4
~~~

I guess that on macOS arm you do not have `gazebo=5.12.9=h2a607e2_5`, that why you are not having the problem.

troiwill, 2023-01-22:
> @troiwill Update: I've created a conda env with python=3.7 and install gazebo 11.5. It works.

I get errors when I try to install `ros-noetic-desktop-full` with the proposed setup. ☹️

mora200217, 2023-01-22:
> > @troiwill Update: I've created a conda env with python=3.7 and install gazebo 11.5. It works.
> 
> @mora200217, thanks for the suggestion. But I get errors when I try to install `ros-noetic-desktop-full` with the proposed setup. ☹️ 

What kind of errors? @troiwill

Tobias-Fischer, 2023-01-24:
Please try again with the robostack-staging channel.

closed 2023-01-24 by Tobias-Fischer

simonbogh, 2023-02-10:
> Sorry for taking long to look at this. This is unfortunatly a known issue with `gazebo=5.12.9=h2a607e2_5`, see [conda-forge/gazebo-feedstock#119](https://github.com/conda-forge/gazebo-feedstock/issues/119) . Once we rebuild everything with qt-main 5.15, everything should be solved, in the meanwhile for each environment in which you have `` installed, you should be able to solve this by installing:
> 
> ```
> mamba install qt=5.12.9=hda022c4_4
> ```
> 
> I guess that on macOS arm you do not have `gazebo=5.12.9=h2a607e2_5`, that why you are not having the problem.

I am not able to run `mamba install qt=5.12.9=hda022c4_4`, it gives `nothing provides requested qt ==5.12.9 hda022c4_4`. Running macOS Ventura 13.2 on Macbook Intel. I can still only install Gazebo 11.12 and ros-noetic-desktop where Gazebo will not work. Any suggestions how to fix it?

Tobias-Fischer, 2023-02-10:
Have you tried the robostack-staging channel?

simonbogh, 2023-02-15:
Isn't that the default channel used during setup? Here are my channels currently in use.

```sh
(ros_noetic_env) user ~ % conda config --show channels
channels:
  - conda-forge
  - robostack-staging
  - defaults
```

Tobias-Fischer, 2023-02-15:
Yes - what happens if you create a new env with ros-noetic-desktop and gazebo? That should work.

ps: I would recommend disabling the defaults channel

simonbogh, 2023-02-15:
Made a new clean conda environment with ros-noetic-desktop and gazebo. I still can't install that specific version of qt, it is not available. 

```
PackagesNotFoundError: The following packages are not available from current channels:

- qt==5.12.9=hda022c4_4
```

traversaro, 2023-02-15:
Hello @simonbogh, probably I was not clear in my post. The `mamba install qt=5.12.9=hda022c4_4` was a workaround necessary before the ros-noetic packages were rebuilt to use qt 5.15 that is not affected by this problem. Now that new packages are available in robostack-staging, it should be sufficient to just install gazebo and ros, without any specific qt-related package to be installed.

simonbogh, 2023-02-15:
That's what a tried first, just installing gazebo and ROS, and gazebo still behaves the same way.

traversaro, 2023-02-15:
Interesting, so you can't add any model? Is this happening also if you just install gazebo on its own, i.e. `mamba create -n gazebo gazebo` ? If that is the case, can you please open an issue at https://github.com/conda-forge/gazebo-feedstock, with all the information that the issue template ask for (mainly `mamba info`  and `mamba list`)? Thanks!

simonbogh, 2023-02-15:
Correct, no models can be added. The simulation is not running as seen in the screenshot below (RTF, SIM TIME etc. is not shown at the bottom). The GUI is not frozen, so maybe it is gzserver hanging. The same goes for 11.11 and 11.10. 

`Gazebo 11.9` (11.9.1-ha2fdce5_2) works, but the conda environment cannot be solved when trying to install `ros-noetic-desktop`
[image]
[image]

traversaro, 2023-02-15:
The problem seems to be similar to https://github.com/conda-forge/gazebo-feedstock/issues/148, that in theory has been fixed with https://github.com/conda-forge/gazebo-feedstock/pull/152, i.e. since `   11.12.0=*_1` builds. For completeness, can you try if it happens also on an environment with just gazebo, and open an issue with the info mentioned in https://github.com/RoboStack/ros-noetic/issues/323#issuecomment-1431746150  ? Thanks!

Tobias-Fischer, 2023-02-16:
Thanks for fixing this @traversaro! A new build should be available in the next few hours - could you please try again then @simonbogh?

simonbogh, 2023-02-16:
Great, thanks!

Gazebo 11.12 `hd5c0442_6` works when it is the only thing I have installed in the environment, but if I go ahead and install `ros-noetic-desktop` afterwards, then it wants change back to build `hd5c0442_5`

```
  Change:
________________
  - gazebo                                     11.12.0  hd5c0442_6          conda-forge                        
  + gazebo                                     11.12.0  hd5c0442_5          conda-forge/osx-64      
```

traversaro, 2023-02-16:
Indeed, that is what I was afraid of and the reason I am trying to mark those as broken in https://github.com/conda-forge/admin-requests/pull/677 . In the meanwhile can you try to install a build without the problem, i.e. :

~~~
mamba install gazebo=11.12.0=*_2
~~~

?

Tobias-Fischer, 2023-02-16:
Ah, do you know what pin is causing the issue @traversaro? I could rebuild the parts of noetic if needed with whatever package has been updated and causes incompatibilities

traversaro, 2023-02-16:
> Ah, do you know what pin is causing the issue @traversaro? I could rebuild the parts of noetic if needed with whatever package has been updated and causes incompatibilities

Looking at the merged PRs in gazebo-feedstock, I would guess ffmpeg=5 (ros-noetic is on 4.4) and libgdal 3.6 (ros is on 3.5), but I would guess the ffmpeg one as it is far more used (see https://index.ros.org/d/ffmpeg/ and  https://index.ros.org/d/libgdal-dev/). However, ffmpeg removed several APIs, so it is possible that the migration is not effortless.

simonbogh, 2023-02-16:
Maybe this is relevant. When trying to install `gazebo=11.12.0=*_6` with `ros-noetic-desktop` already installed I get this error:

```
$ mamba install "gazebo=11.12.0=*_6"

Encountered problems while solving:
  - package gazebo-11.12.0-hba8d896_6 requires qt-main >=5.15.8,<5.16.0a0, but none of the providers can be installed
```

traversaro, 2023-02-16:
Hello @simonbogh, were you able to check if `mamba install gazebo=11.12.0=*_2` works?

simonbogh, 2023-02-16:
Yes, and Gazebo runs now, but models can still not be spawned. Gazebo works with that build, but if I run a standard `moveit_config` example such as `demo_gazebo.launch` with a Franka, it starts Gazebo paused, which it shouldn't according to the launch-file argument, and when Gazebo is unpaused the urdf spawn service crashes.

Here are the immediate errors I found in the terminal:
```
2023-02-16 11:33:50.716 rviz[60256:4295570] IMKClient Stall detected, *please Report* your user scenario attaching a spindump (or sysdiagnose) that captures the problem - (imkxpc_bundleIdentifierWithReply:) block performed very slowly (9.54 secs).

...

[ERROR] [1676543358.522506, 10.004000]: Spawn service failed. Exiting.
[spawn_gazebo_model-4] process has died [pid 60244, exit code 1, cmd /Users/user/miniconda3/envs/ros_noetic_env/lib/gazebo_ros/spawn_model -urdf -param robot_description -model robot -unpause -x 0 -y 0 -z 0 -R 0 -P 0 -Y 0 -J panda_1_finger_joint1 0.04 -J panda_1_joint1 0 -J panda_1_joint2 0 -J panda_1_joint3 0 -J panda_1_joint4 -1.5708 -J panda_1_joint5 0 -J panda_1_joint6 1.6396 -J panda_1_joint7 0 __name:=spawn_gazebo_model __log:=/Users/user/.ros/log/5e2c9ddc-ade5-11ed-a8f7-8c85905cd4a9/spawn_gazebo_model-4.log].
```

In the end, rviz loads with the model, but no models are ever spawned in Gazebo. MoveIt `demo.launch` with rviz only works fine, it is only when Gazebo is also launched and models have to be spawned that it fails. I will test the same code in Ubuntu 20.04 and Noetic.

traversaro, 2023-02-16:
Ok, but the basic models that do not use ROS are working, right? I think the cause behind this new errors are different, can you please open a new issue in this repo with all the details (i.e. `mamba info` and `mamba list`), thanks!

traversaro, 2023-02-16:
Great, thanks!

simonbogh, 2023-02-16:
Just a quick follow-up. Tried the same code (Franka, moveit) on a computer running Ubuntu 22.04, conda/mamba, robostack, noetic, same package versions, same python version. On the Ubuntu computer it works, on my mac gazebo still pauses, and if unpaused the spawn service fails and no models are added in gazebo.

```
[INFO] [1676542770.942941, 10.001000]: Spawn status: SpawnModel: Entity pushed to spawn queue, but spawn service timed out waiting for entity to appear in simulation under the name robot
[ERROR] [1676542770.945070, 10.004000]: Spawn service failed. Exiting.
[spawn_gazebo_model-4] process has died [pid 57788, exit code 1
```

traversaro, 2023-02-16:
Perfect! Can you open a new issue at https://github.com/RoboStack/ros-noetic/issues/new/choose . Closed issue are a bit difficult to manage in the longer term.

Tobias-Fischer, 2023-02-16:
FYI - not being able to install ros-noetic-desktop and the most recent gazebo comes down to needing https://github.com/conda-forge/qt-webengine-feedstock/pull/24 (ros-noetic-desktop needs qt-webengine needs qt-main 5.15.6 which does not have the most recent migrations ..)

traversaro, 2023-02-19:
> FYI - not being able to install ros-noetic-desktop and the most recent gazebo comes down to needing [conda-forge/qt-webengine-feedstock#24](https://github.com/conda-forge/qt-webengine-feedstock/pull/24) (ros-noetic-desktop needs qt-webengine needs qt-main 5.15.6 which does not have the most recent migrations ..)

Ok! Apparently we can't mark the broken builds as broken (see https://github.com/conda-forge/admin-requests/pull/677). It is sufficient to merge https://github.com/conda-forge/admin-requests/pull/677 to avoid that gazebo _3/_4/_5 builds are installed?