RoboStack/ros-noetic #324: rviz looking for OgreDynLib.cpp  in the wrong place

gokulhari, 2023-01-19:
I installed using robostack, and I found the following issue when trying to start rviz:

[ INFO] [1674157656.417352067]: rviz version 1.14.14
[ INFO] [1674157656.417382377]: compiled against Qt version 5.12.9
[ INFO] [1674157656.417388838]: compiled against OGRE version 1.10.12 (Xalafu)
[ INFO] [1674157656.426660251]: Forcing OpenGl version 0.
[ERROR] [1674157656.651451240]: InternalErrorException: Could not load dynamic library /home/gokul/mambaforge/envs/ROS/lib/OGRE/RenderSystem_GL.  System Error: libXrandr.so.2: cannot open shared object file: No such file or directory in DynLib::load at /home/conda/feedstock_root/build_artifacts/ogre_1666481244237/work/OgreMain/src/OgreDynLib.cpp (line 107)
terminate called after throwing an instance of 'Ogre::InternalErrorException'
  what():  InternalErrorException: Could not load dynamic library /home/gokul/mambaforge/envs/ROS/lib/OGRE/RenderSystem_GL.  System Error: libXrandr.so.2: cannot open shared object file: No such file or directory in DynLib::load at /home/conda/feedstock_root/build_artifacts/ogre_1666481244237/work/OgreMain/src/OgreDynLib.cpp (line 107)
Aborted (core dumped)

The thing is that I have not installed conda, and rviz is trying to search for OgreDynLib.cpp inside a conda installation.

Tobias-Fischer, 2023-01-19:
You need to install libxrandr with apt or similar. Which system are you on?

I don’t understand why you think you do not have conda installed? RoboStack is based on conda :)

gokulhari, 2023-01-20:
Hi, thanks for the quick reply. 
I am using Ubuntu 22.04. Yes I got it running by doing a sudo apt-get install. Thank you!

closed 2023-01-24 by Tobias-Fischer