RoboStack/ros-noetic #326: Received "GLIBCXX_3.4.30 not found" error on Linux amd64 machine when running Gazebo

troiwill, 2023-01-21:
## Issue:

Hi, I installed `ros-noetic-desktop-full` on a Linux `amd64` machine. When I run `gazebo` from the terminal, I get the following error:
```
gazebo: /home/troiw/miniconda3/envs/ros-noetic/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /home/troiw/miniconda3/envs/ros-noetic/bin/../lib/./libgdal.so.30)
gazebo: /home/troiw/miniconda3/envs/ros-noetic/bin/../lib/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /home/troiw/miniconda3/envs/ros-noetic/bin/../lib/././libtiledb.so.2.8)
```

From reading other forums, it seems that the appropriate version of `libstdc++.so.6` is not installed. I verified this by running the following command
```
strings /home/troiw/miniconda3/envs/ros-noetic/bin/../lib/libstdc++.so.6 | grep GLIBCXX_3.4
```
which gave me the following output
```
GLIBCXX_3.4
GLIBCXX_3.4.1
GLIBCXX_3.4.2
GLIBCXX_3.4.3
GLIBCXX_3.4.4
GLIBCXX_3.4.5
GLIBCXX_3.4.6
GLIBCXX_3.4.7
GLIBCXX_3.4.8
GLIBCXX_3.4.9
GLIBCXX_3.4.10
GLIBCXX_3.4.11
GLIBCXX_3.4.12
GLIBCXX_3.4.13
GLIBCXX_3.4.14
GLIBCXX_3.4.15
GLIBCXX_3.4.16
GLIBCXX_3.4.17
GLIBCXX_3.4.18
GLIBCXX_3.4.19
GLIBCXX_3.4.20
GLIBCXX_3.4.21
GLIBCXX_3.4.22
GLIBCXX_3.4.23
GLIBCXX_3.4.24
GLIBCXX_3.4.25
GLIBCXX_3.4.26
GLIBCXX_3.4.27
GLIBCXX_3.4.28
GLIBCXX_3.4.29
_ZNKSt14basic_ifstreamIcSt11char_traitsIcEE7is_openEv@GLIBCXX_3.4
_ZNSt13basic_istreamIwSt11char_traitsIwEE6ignoreEv@@GLIBCXX_3.4.5
_ZNKSbIwSt11char_traitsIwESaIwEE11_M_disjunctEPKw@GLIBCXX_3.4
...
```

_This is a recent issue_ because I did not have this `GLIBCXX_3.4.30` error did not occur when I installed ROS Noetic about a week ago.

## Fixes Tried:

[ **Current Workaround** ]: I removed `ros-noetic-desktop-full`, downgraded python `python=3.9 -> python=3.8`, and reinstalled `ros-noetic-desktop-full` (version `1.5.0=py38he9ab703_6`). Now, `gazebo` runs without the `GLIBCXX_3.4.30` error. However, this means that I cannot use `python=3.9` with ROS Noetic.

Tobias-Fischer, 2023-01-21:
Which linux distro are you on?

troiwill, 2023-01-21:
> Which linux distro are you on?

Ubuntu 18.04.6 LTS

Tobias-Fischer, 2023-01-21:
Have you tried `mamba install -c conda-forge libstdcxx-ng=12`?

traversaro, 2023-01-21:
Can you report the mamba list output in the two environment, the one working and the one not working? Thanks!

troiwill, 2023-01-22:
> Can you report the mamba list output in the two environment, the one working and the one not working? Thanks!

@traversaro, unfortunately, I do not have the working conda environment any more; it was deleted. Do you still want the non-working environment list?

troiwill, 2023-01-22:
> Have you tried `mamba install -c conda-forge libstdcxx-ng=12`?

Thanks @Tobias-Fischer for the proposed fix. I no longer get the `GLIBCXX_3.4.40` error. However, gazebo crashes whenever I run `gazebo` via the terminal. Specifically, gazebo crashes just after it draws the GUI's window. I ran `gazebo --verbose` and received the following:
```
Gazebo multi-robot simulator, version 11.10.2
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
Gazebo multi-robot simulator, version 11.10.2
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 10.70.51.90
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 10.70.51.90
[Msg] Loading world file [/home/troiw/miniconda3/envs/ros-noetic/share/gazebo-11/worlds/empty.world]
[Wrn] [GuiIface.cc:120] JPEG parameter struct mismatch: library thinks size is 632, caller expects 664
```

traversaro, 2023-01-22:
The error is: 
~~~
 [Wrn] [GuiIface.cc:120] JPEG parameter struct mismatch: library thinks size is 632, caller expects 664
~~~

That means that for some reason a wrong jpeg library is loaded. Can you report the output of the commands: 
* mamba list
* env

?  

Thanks!

troiwill, 2023-01-22:
**mamba list:**
```
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
alsa-lib                  1.2.3.2              h166bdaf_0    conda-forge
aom                       3.3.0                h27087fc_1    conda-forge
apr                       1.7.0                h7f98852_5    conda-forge
assimp                    5.2.3                h4c92fa2_1    conda-forge
atk-1.0                   2.36.0               h3371d22_4    conda-forge
bcrypt                    3.2.2            py39hb9d737c_1    conda-forge
binutils                  2.38                 h1680402_1  
binutils_impl_linux-64    2.38                 h2a08ee3_1  
binutils_linux-64         2.38.0               hc2dff05_0  
blosc                     1.21.1               h83bc5f7_3    conda-forge
boost                     1.74.0           py39h5472131_5    conda-forge
boost-cpp                 1.74.0               h6cacc03_7    conda-forge
brotli                    1.0.9                h166bdaf_8    conda-forge
brotli-bin                1.0.9                h166bdaf_8    conda-forge
bullet-cpp                3.21                 hde0f152_1    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.18.1               h7f98852_0    conda-forge
c-compiler                1.5.2                h0b41bf4_0    conda-forge
ca-certificates           2022.12.7            ha878542_0    conda-forge
cairo                     1.16.0            ha12eb4b_1010    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.2              pyhd8ed1ab_0    conda-forge
certifi                   2022.12.7        py39h06a4308_0  
cffi                      1.15.1           py39he91dace_3    conda-forge
cfitsio                   4.1.0                hd9d235c_0    conda-forge
cmake                     3.23.1               h5432695_0    conda-forge
compilers                 1.5.2                ha770c72_0    conda-forge
console_bridge            1.0.2                h924138e_1    conda-forge
coreutils                 9.1                  h166bdaf_0    conda-forge
cppzmq                    4.8.1                hf7cf922_0    conda-forge
cryptography              39.0.0           py39hd598818_0    conda-forge
curl                      7.86.0               h7bff187_1    conda-forge
cxx-compiler              1.5.2                hf52228f_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
dartsim                   6.12.1               h9f176f7_3    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.6.0              pyhd8ed1ab_0    conda-forge
docutils                  0.19             py39hf3d152e_1    conda-forge
double-conversion         3.2.0                h9c3ff4c_0    conda-forge
eigen                     3.4.0                h4bd325d_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.4.8                h27087fc_0    conda-forge
fcl                       0.7.0                h2cbc392_0    conda-forge
ffmpeg                    4.4.1                h964e5f1_4    conda-forge
flann                     1.9.1             hfe772e8_1010    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.0               hc2a2eb6_1    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.38.0           py39hb9d737c_1    conda-forge
fortran-compiler          1.5.2                hdb1a99f_0    conda-forge
freeglut                  3.2.2                h9c3ff4c_1    conda-forge
freeimage                 3.18.0               he1b5a44_0    conda-forge
freetype                  2.12.1               hca18f0e_1    conda-forge
freexl                    1.0.6                h166bdaf_1    conda-forge
fribidi                   1.0.10               h36c2ea0_0    conda-forge
gazebo                    11.10.2              h40dda3e_2    conda-forge
gcc                       11.2.0              h702ea55_10    conda-forge
gcc_impl_linux-64         11.2.0               h1234567_1  
gcc_linux-64              11.2.0               h5c386dc_0  
gdbm                      1.18                 h0a1914f_2    conda-forge
[... 629 lines cut ...]
XDG_RUNTIME_DIR=/run/user/1014
XAUTHORITY=/home/troiw/.Xauthority
LDFLAGS=-Wl,-O2 -Wl,--sort-common -Wl,--as-needed -Wl,-z,relro -Wl,-z,now -Wl,--disable-new-dtags -Wl,--gc-sections -Wl,-rpath,/home/troiw/miniconda3/envs/ros-noetic/lib -Wl,-rpath-link,/home/troiw/miniconda3/envs/ros-noetic/lib -L/home/troiw/miniconda3/envs/ros-noetic/lib
TURTLEBOT_STAGE_MAP_FILE=/home/troiw/catkin_ws/src/src/turtlebot_simulator/turtlebot_stage/maps/maze.yaml
CXX_FOR_BUILD=/home/troiw/miniconda3/envs/ros-noetic/bin/x86_64-conda-linux-gnu-c++
[... cut ...]
SESSION_MANAGER=local/vision:@/tmp/.ICE-unix/24323,unix/vision:/tmp/.ICE-unix/24323
CPP=/home/troiw/miniconda3/envs/ros-noetic/bin/x86_64-conda-linux-gnu-cpp
LD=/home/troiw/miniconda3/envs/ros-noetic/bin/x86_64-conda-linux-gnu-ld
ROS_DISTRO=noetic
ENSENSO_INSTALL=/opt/ensenso
TURTLEBOT_MAP_FILE=/home/troiw/catkin_ws/src/src/turtlebot_apps/turtlebot_navigation/maps/willow-2010-02-18-0.10.yaml
GAZEBO_MODEL_DATABASE_URI=http://models.gazebosim.org
LESSOPEN=| /usr/bin/lesspipe %s
GTK_IM_MODULE=ibus
_=/home/troiw/miniconda3/envs/ros-noetic/bin/env
```

traversaro, 2023-01-22:
Can you also report the output of:
~~~
ldd $CONDA_PREFIX/lib/libfreeimage.so | grep jpeg
~~~

Anyhow, the version of qt that you are using is problematic, can you try to also run:
~~~
mamba install qt=5.12.9=*_4
~~~
?

Tobias-Fischer, 2023-01-24:
Can you please try again with the packages from the robostack-staging channel? Thanks!

closed 2023-01-24 by Tobias-Fischer