RoboStack/ros-noetic #327: Camera image messages do not appear in ROS

troiwill, 2023-01-22:
## Issue:

I installed the `turtlebot3`, `turtlebot3_simulations`, and `turtlebot3_msgs` packages within a `ROS Noetic` Robostack environment. I tried to view the camera images via Rviz, but no messages show up. In fact, I receive the following warnings:
```
[ WARN] [1674350888.265620000, 480.077000000]: The input topic '/camera/camera_info' is not yet advertised
[ WARN] [1674350948.261207000, 539.406000000]: The input topic '/camera/camera_info' is not yet advertised
[ WARN] [1674351008.262545000, 598.922000000]: The input topic '/camera/camera_info' is not yet advertised
[ WARN] [1674351068.263357000, 658.165000000]: The input topic '/camera/camera_info' is not yet advertised
[ WARN] [1674351128.260958000, 717.569000000]: The input topic '/camera/camera_info' is not yet advertised
[ WARN] [1674351188.261893000, 776.720000000]: The input topic '/camera/camera_info' is not yet advertised
```
I tried to use other ROS + Gazebo packages as well, including the Intel RealSense camera via https://github.com/rickstaa/realsense-ros-gazebo. However, I do not see the camera ROS topics using this package either.

On the other hand, I do see camera images, etc. from the turtlebot3 and Intel RealSense cameras in Gazebo via `Window > Topic Visualization` menu.

I am not sure if it matters, but I noticed that the Mac creates `*.dylib` library files. However, ROS seems to require `*.so` files (e.g., `libgazebo_ros_camera.so`). Could this be the issue?

## System Specs:

```
Mac mini M1, 2020
16 GB RAM
macOS Ventura 13.1
```

traversaro, 2023-01-22:
Can you provide the mamba list of the environment in which you are experiencing this?

troiwill, 2023-01-22:
Yes, here is the list.

```
# Name                    Version                   Build  Channel
appdirs                   1.4.4              pyh9f0ad1d_0    conda-forge
apr                       1.7.0                h9b22ae9_5    conda-forge
assimp                    5.2.4                h3287542_0    conda-forge
atk-1.0                   2.38.0               hcb7b3dd_1    conda-forge
bcrypt                    3.2.2            py39h02fc5c5_1    conda-forge
blosc                     1.21.3               h1d6ff8b_0    conda-forge
boost                     1.74.0           py39h776e71f_5    conda-forge
boost-cpp                 1.74.0               h3afdda6_7    conda-forge
brotli                    1.0.9                h1a8c8d9_8    conda-forge
brotli-bin                1.0.9                h1a8c8d9_8    conda-forge
brotlipy                  0.7.0           py39h02fc5c5_1005    conda-forge
bullet                    3.21                 h85045c0_4    conda-forge
bullet-cpp                3.21             py39hde7b980_4    conda-forge
bzip2                     1.0.8                h3422bc3_4    conda-forge
c-ares                    1.18.1               h3422bc3_0    conda-forge
c-compiler                1.5.2                h5008568_0    conda-forge
ca-certificates           2022.12.7            h4653dfc_0    conda-forge
cairo                     1.16.0            hb99c34e_1009    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.2              pyhd8ed1ab_0    conda-forge
cctools                   973.0.1             hcbb26d4_11    conda-forge
cctools_osx-arm64         973.0.1             hef52d2f_11    conda-forge
certifi                   2022.12.7        py39hca03da5_0  
cffi                      1.15.1           py39h7e6b969_3    conda-forge
cfitsio                   4.1.0                hd4f5c17_0    conda-forge
charset-normalizer        2.1.1              pyhd8ed1ab_0    conda-forge
clang                     14.0.6               hce30654_0    conda-forge
clang-14                  14.0.6          default_h81a5282_0    conda-forge
clang_osx-arm64           14.0.6               h15773ab_4    conda-forge
clangxx                   14.0.6          default_hb7ecf47_0    conda-forge
clangxx_osx-arm64         14.0.6               he29aa18_4    conda-forge
cmake                     3.25.1               hf234bd0_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
compiler-rt               14.0.6               h30b49de_0    conda-forge
compiler-rt_osx-arm64     14.0.6               h48302dc_0    conda-forge
compilers                 1.5.2                hce30654_0    conda-forge
console_bridge            1.0.2                h3e96240_1    conda-forge
contourpy                 1.0.7            py39haaf3ac1_0    conda-forge
coreutils                 9.1                  h1c322ee_0    conda-forge
cppzmq                    4.9.0                h54398fd_0    conda-forge
cryptography              39.0.0           py39haa0b8cc_0    conda-forge
curl                      7.86.0               hd538317_1    conda-forge
cxx-compiler              1.5.2                hffc8910_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
dartsim                   6.12.1               hbfbca65_2    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.6.0              pyhd8ed1ab_0    conda-forge
docutils                  0.19             py39h2804cbe_1    conda-forge
double-conversion         3.2.0                hb7217d7_1    conda-forge
eigen                     3.4.0                hc021e02_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.5.0                hb7217d7_0    conda-forge
fcl                       0.6.1                hda9dc21_3    conda-forge
ffmpeg                    4.3.2                h38cfed3_3    conda-forge
flann                     1.9.1             h71def99_1010    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.13.94              heb65262_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.38.0           py39h02fc5c5_1    conda-forge
fortran-compiler          1.5.2                h2ccabda_0    conda-forge
freeimage                 3.18.0              hb513136_10    conda-forge
freetype                  2.12.1               hd633e50_1    conda-forge
[... 491 lines cut ...]
xorg-libice               1.0.10               h27ca646_0    conda-forge
xorg-libsm                1.2.3             h27ca646_1000    conda-forge
xorg-libx11               1.7.2                h3422bc3_0    conda-forge
xorg-libxau               1.0.9                h27ca646_0    conda-forge
xorg-libxaw               1.0.14               h3422bc3_1    conda-forge
xorg-libxdmcp             1.1.3                h27ca646_0    conda-forge
xorg-libxext              1.3.4                h27ca646_1    conda-forge
[... cut ...]
yaml-cpp                  0.6.3                hc88da5d_4    conda-forge
zeromq                    4.3.4                hbdafb3b_1    conda-forge
zlib                      1.2.13               h03a7124_4    conda-forge
zstd                      1.5.2                h8128057_4    conda-forge
zziplib                   0.13.69              he1e0b03_1    conda-forge
```

I added the Intel RealSense packages via a catkin workspace, so they will not show up in the list. I got the RealSense packages from [here](https://github.com/rickstaa/realsense-ros-gazebo).

Tobias-Fischer, 2023-02-09:
Are you attempting to add a Camera display, or an Image display?

Tobias-Fischer, 2023-02-20:
Closing due to inactivity - it seems like a Camera display is used without providing all the information needed for it.

closed 2023-02-20 by Tobias-Fischer