RoboStack/ros-noetic #331: qt-gui-cpp build broken on windows

Tobias-Fischer, 2023-01-24:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Let's continue the conversation from #295 here.

Latest error (from https://github.com/RoboStack/ros-noetic/actions/runs/3991408782/jobs/6846172048):
```
link /NOLOGO /DYNAMICBASE /NXCOMPAT /OPT:REF /INCREMENTAL:NO /DLL /SUBSYSTEM:WINDOWS /OUT:release\libqt_gui_cpp_sip.dll @C:\Users\RUNNER~1\AppData\Local\Temp\nm385D.tmp
[6797](https://github.com/RoboStack/ros-noetic/actions/runs/3991408782/jobs/6846172048#step:6:6798)
   Creating library release\libqt_gui_cpp_sip.lib and object release\libqt_gui_cpp_sip.exp
[6798](https://github.com/RoboStack/ros-noetic/actions/runs/3991408782/jobs/6846172048#step:6:6799)
siplibqt_gui_cpp_sipqt_gui_cppRosPluginlibPluginProvider_ForPlugins.obj : error LNK2019: unresolved external symbol "bool ros::console::g_initialized" (?g_initialized@console@ros@@3_NA) referenced in function "public: __cdecl pluginlib::ClassLoader<class qt_gui_cpp::Plugin>::ClassLoader<class qt_gui_cpp::Plugin>(class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::vector<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> >,class std::allocator<class std::basic_string<char,struct std::char_traits<char>,class std::allocator<char> > > >)" (??0?$ClassLoader@VPlugin@qt_gui_cpp@@@pluginlib@@QEAA@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@00V?$vector@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@V?$allocator@V?$basic_string@DU?$char_traits@D@std@@V?$allocator@D@2@@std@@@2@@3@@Z)
[6799](https://github.com/RoboStack/ros-noetic/actions/runs/3991408782/jobs/6846172048#step:6:6800)
siplibqt_gui_cpp_sipqt_gui_cppRosPluginlibPluginProvider_ForPluginProviders.obj : error LNK2001: unresolved external symbol "bool ros::console::g_initialized" (?g_initialized@console@ros@@3_NA)
[6800](https://github.com/RoboStack/ros-noetic/actions/runs/3991408782/jobs/6846172048#step:6:6801)
siplibqt_gui_cpp_sipqt_gui_cppPluginBridge.obj : error LNK2001: unresolved external symbol "public: static struct QMetaObject const qt_gui_cpp::PluginBridge::staticMetaObject" (?staticMetaObject@PluginBridge@qt_gui_cpp@@2UQMetaObject@@B)
[6801](https://github.com/RoboStack/ros-noetic/actions/runs/3991408782/jobs/6846172048#step:6:6802)
siplibqt_gui_cpp_sipqt_gui_cppPlugin.obj : error LNK2001: unresolved external symbol "public: static struct QMetaObject const qt_gui_cpp::Plugin::staticMetaObject" (?staticMetaObject@Plugin@qt_gui_cpp@@2UQMetaObject@@B)
[6802](https://github.com/RoboStack/ros-noetic/actions/runs/3991408782/jobs/6846172048#step:6:6803)
release\libqt_gui_cpp_sip.dll : fatal error LNK1120: 3 unresolved externals
[6803](https://github.com/RoboStack/ros-noetic/actions/runs/3991408782/jobs/6846172048#step:6:6804)
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.29.30133\bin\HostX64\x64\link.EXE"' : return code '0x460'
[6804](https://github.com/RoboStack/ros-noetic/actions/runs/3991408782/jobs/6846172048#step:6:6805)
Stop.
[6805](https://github.com/RoboStack/ros-noetic/actions/runs/3991408782/jobs/6846172048#step:6:6806)
NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Enterprise\VC\Tools\MSVC\14.29.30133\bin\HostX64\x64\nmake.exe"' : return code '0x2'
[6806](https://github.com/RoboStack/ros-noetic/actions/runs/3991408782/jobs/6846172048#step:6:6807)
Stop.
[6807](https://github.com/RoboStack/ros-noetic/actions/runs/3991408782/jobs/6846172048#step:6:6808)
NMAKE : fatal error U1077: 'cd' : return code '0x2'
[6808](https://github.com/RoboStack/ros-noetic/actions/runs/3991408782/jobs/6846172048#step:6:6809)
Stop.
[6809](https://github.com/RoboStack/ros-noetic/actions/runs/3991408782/jobs/6846172048#step:6:6810)
Error: could not load cache
```

/cc @traversaro do you have any clue?

### Installed packages

```shell
n/a
```

### Environment info

```shell
n/a
```

Tobias-Fischer, 2023-01-24:
We can't be too far off - the rviz build succeeds (which also uses the python-qt-binding stuff)

Tobias-Fischer, 2023-01-24:
Note that https://github.com/RoboStack/ros-noetic/blob/f0385d3fb3c1ff3924f3c0eed01660cba8b35441/patch/ros-noetic-python-qt-binding.patch#L19 did not seem to have helped.

traversaro, 2023-01-24:
> Note that
> 
> https://github.com/RoboStack/ros-noetic/blob/f0385d3fb3c1ff3924f3c0eed01660cba8b35441/patch/ros-noetic-python-qt-binding.patch#L19
> did not seem to have helped.

Yes, apparently that is not actually propagating any flag. According to my local tests https://github.com/RoboStack/vinca/pull/39 should fix the problem.

closed 2023-01-24 by Tobias-Fischer

Tobias-Fischer, 2023-01-24:
It works :). Awesome work @traversaro