RoboStack/ros-noetic #335: After fresh install trying to #include<ros/ros.h> crashes in timer

Robokan, 2023-01-28:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Fresh install on macOS Monterey everything seems to work until I tried to create a package with a CPP node in it. Trying to include <ros/ros.h> crashes somewhere in the timer code.

[ 50%] Building CXX object my_robot_tutorials/CMakeFiles/node_cpp.dir/src/my_first_node.cpp.o
In file included from /Users/eric/catkin_ws/src/my_robot_tutorials/src/my_first_node.cpp:1:
In file included from /Users/eric/miniforge3/envs/robostackenv/include/ros/ros.h:38:
In file included from /Users/eric/miniforge3/envs/robostackenv/include/ros/time.h:53:
In file included from /Users/eric/miniforge3/envs/robostackenv/include/ros/platform.h:38:
In file included from /Users/eric/miniforge3/envs/robostackenv/bin/../include/c++/v1/string:525:
In file included from /Users/eric/miniforge3/envs/robostackenv/bin/../include/c++/v1/algorithm:667:
In file included from /Users/eric/miniforge3/envs/robostackenv/bin/../include/c++/v1/functional:506:
In file included from /Users/eric/miniforge3/envs/robostackenv/bin/../include/c++/v1/__functional/function.h:22:
In file included from /Users/eric/miniforge3/envs/robostackenv/bin/../include/c++/v1/__memory/shared_ptr.h:35:
In file included from /Users/eric/miniforge3/envs/robostackenv/bin/../include/c++/v1/atomic:523:
In file included from /Users/eric/miniforge3/envs/robostackenv/bin/../include/c++/v1/__thread/poll_with_backoff.h:14:
In file included from /Users/eric/miniforge3/envs/robostackenv/bin/../include/c++/v1/chrono:697:
In file included from /Users/eric/miniforge3/envs/robostackenv/bin/../include/c++/v1/__chrono/calendar.h:14:
/Users/eric/miniforge3/envs/robostackenv/bin/../include/c++/v1/__chrono/system_clock.h:37:12: error: reference to unresolved using declaration
    static time_t     to_time_t  (const time_point& __t) _NOEXCEPT;

### Installed packages

```shell
# packages in environment at /Users/eric/miniforge3/envs/robostackenv:
#
# Name                    Version                   Build  Channel
aom                       3.5.0                h7ea286d_0    conda-forge
apr                       1.7.0                h9b22ae9_5    conda-forge
assimp                    5.2.5                h276577b_0    conda-forge
atk-1.0                   2.38.0               hcb7b3dd_1    conda-forge
bcrypt                    3.2.2            py39h02fc5c5_1    conda-forge
boost                     1.78.0           py39h99de9ae_4    conda-forge
boost-cpp                 1.78.0               h1cb353e_1    conda-forge
brotli                    1.0.9                h1a8c8d9_8    conda-forge
brotli-bin                1.0.9                h1a8c8d9_8    conda-forge
bzip2                     1.0.8                h3422bc3_4    conda-forge
c-ares                    1.18.1               h3422bc3_0    conda-forge
c-compiler                1.5.2                h5008568_0    conda-forge
ca-certificates           2022.12.7            h4653dfc_0    conda-forge
cairo                     1.16.0            h73a0509_1014    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.2              pyhd8ed1ab_0    conda-forge
cctools                   973.0.1             hcbb26d4_11    conda-forge
cctools_osx-arm64         973.0.1             hef52d2f_11    conda-forge
certifi                   2022.12.7          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1           py39h7e6b969_3    conda-forge
clang                     14.0.6               hce30654_0    conda-forge
clang-14                  14.0.6          default_h81a5282_0    conda-forge
clang_osx-arm64           14.0.6               h15773ab_4    conda-forge
clangxx                   14.0.6          default_hb7ecf47_0    conda-forge
clangxx_osx-arm64         14.0.6               he29aa18_4    conda-forge
cmake                     3.25.2               hf234bd0_0    conda-forge
compiler-rt               14.0.6               h30b49de_0    conda-forge
compiler-rt_osx-arm64     14.0.6               h48302dc_0    conda-forge
compilers                 1.5.2                hce30654_0    conda-forge
console_bridge            1.0.2                h3e96240_1    conda-forge
contourpy                 1.0.7            py39haaf3ac1_0    conda-forge
coreutils                 9.1                  h1c322ee_0    conda-forge
cryptography              39.0.0           py39haa0b8cc_0    conda-forge
cxx-compiler              1.5.2                hffc8910_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.6.0              pyhd8ed1ab_0    conda-forge
docutils                  0.19             py39h2804cbe_1    conda-forge
eigen                     3.4.0                hc021e02_0    conda-forge
[... 407 lines cut ...]
            shell level : 1
       user config file : /Users/eric/.condarc
 populated config files : /Users/eric/miniforge3/.condarc
                          /Users/eric/.condarc
          conda version : 4.11.0
    conda-build version : not installed
         python version : 3.9.10.final.0
       virtual packages : __osx=12.6=0
                          __unix=0=0
                          __archspec=1=arm64
       base environment : /Users/eric/miniforge3  (writable)
      conda av data dir : /Users/eric/miniforge3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /Users/eric/miniforge3/pkgs
                          /Users/eric/.conda/pkgs
       envs directories : /Users/eric/miniforge3/envs
                          /Users/eric/.conda/envs
               platform : osx-arm64
             user-agent : conda/4.11.0 requests/2.27.1 CPython/3.9.10 Darwin/21.6.0 OSX/12.6
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2023-01-28:
Best bet: you overwrite the ${CMAKE_CXX_FLAGS} in your code, instead of appending to it.

Robokan, 2023-01-29:
What should I override it to be? I am not currently appending to CMAKE_CXX_FLAGS anywhere in my code.  It's just a simple node to test to see if ROS is working.
Any insight into why this is happening? is it some kind of library clash or version incompatibility with the header files?

Tobias-Fischer, 2023-01-29:
Can you please provide a link to your code?

Robokan, 2023-01-31:
#include <ros/ros.h>

int main (init argc, char **argv)
{
    res:init(argc, argv, "my_first_cpp_node");
    ros::NodeHandle ah;

    ROS_INFO("Node has been started");

    ros::Duration(1.0).sleep();

    ROS_INFO("exit");
}

Robokan, 2023-01-31:
one other weird thing I noticed after installing is when I source the setup file I get this error:

conda activate robostackenv
/Users/eric/miniforge3/envs/robostackenv/etc/catkin/profile.d/99.roslisp.sh:21: no such file or directory: /Users/runner/mambaforge/conda-bld/ros-noetic-roslisp-0_1674181617630/_build_env/bin/python

Somehow /Users/runner is being referenced after installation. I don't have any users on my machine with that name.

Tobias-Fischer, 2023-02-01:
I cannot replicate any of those issues locally, apologies. You have numerous typos etc. in your C++ code and CMakeLists.txt. Attached is a project that I can compile without problems.

[my_robot_tuts.zip](https://github.com/RoboStack/ros-noetic/files/10562460/my_robot_tuts.zip)

If you can't get it to work, please provide a detailed list of commands that you run.

Robokan, 2023-02-03:
Your example also crashes in the time header when I try it.

The only command I am running is: catkin_make

maybe this is just a OSX specific issue?

Tobias-Fischer, 2023-02-03:
I was trying on osx .. do you have the “compilers” package installed? I’ve only tried using catkin_tools, can you please try installing this and run “catkin build” instead of “catkin_make”?

Robokan, 2023-02-03:
I removed the build and devel  directories and  built with "catkin build" and it works! Thanks.
what is the command to install the compilers package?

The tutorial I am doing is using catkin_make. What happens when you use catkin_make on your machine?

Tobias-Fischer, 2023-02-04:
catkin_make works fine for me, too.

Tobias-Fischer, 2023-02-04:
`mamba install compilers`

Robokan, 2023-02-04:
I still have the same compile issue. As it works on your side there must be some kind of configuration issue on my side.  I have a linux machine coming soon so I will probably just try things out there which hopefully should resolve things. I will just use catkin build for now. Thanks of the help.

Tobias-Fischer, 2023-02-09:
Thanks @Robokan - closing for now, feel free to reopen.

closed 2023-02-09 by Tobias-Fischer