RoboStack/ros-noetic #338: ROS noetic conda ENV with python3.11

SSG0210, 2023-02-01:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Hi,

I have a problem when I want to install ros-noetic-desktop in a condo environment with python3.11. I executed the next commands:

- conda create -n ros python3.11
- conda activate ros
- conda config --env --add channels conda-forge
- conda config --env --add channels robostack-staging
- conda install ors-noetic-desktop.

Error:

UnsatisfiableError: The following specifications were found to be incompatible with each other:

Output in format: Requested package -> Available versionsThe following specifications were found to be incompatible with your system:

  - feature:/osx-64::__osx==10.16=0
  - ros-noetic-desktop -> __osx[version='>=10.14']

Your installed version is: 10.16

If a try te same but with python3.9 I don't have any problem, but for other reasons I need to use python3.11, Is there any solution?

Thanks

### Installed packages

```shell
At the moment nothing
```

### Environment info

```shell
python3.11
```

Tobias-Fischer, 2023-02-01:
Unfortunately we only have py39 builds at the moment - we’d like to support others, but the maintenance and CI effort is too high :(

SSG0210, 2023-02-01:
it's a shame but I understand you. Thanks!

closed 2023-02-01 by SSG0210