RoboStack/ros-noetic #339: how to contribute?

lianghongzhuo, 2023-02-01:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Thank you for the great frame work!
I have built ros linux64 in python3.11 and uploaded it to anaconda. https://anaconda.org/lianghongzhuo/repo
Do you interested in this? How can I contribute?

### Installed packages

```shell
all
```

### Environment info

```shell
conda/22.9.0 requests/2.28.1 CPython/3.10.6 Linux/5.15.0-58-generic ubuntu/20.04.5 glibc/2.31
```

Tobias-Fischer, 2023-02-01:
Awesome, thank you for this!

@wolfv - how should we deal with this?
@lianghongzhuo - a bit of background: so far we tried to limit locally built packages as much as possible, as there are no publicly available build logs etc. in this case. However, I agree that building locally is much easier and pain-free compared to our current CI solution ..

lianghongzhuo, 2023-02-03:
By the way, is there a way to set up vinca to generate recipes from the upstream source code? Many patches I made for build python3.11 are just the diff with the upstream source code. If we can build a bleed edge binary then the workload of making patches are reduced.

Tobias-Fischer, 2023-02-03:
Unfortunately not an easy one :(. @wolfv maybe you could point @lianghongzhuo to the right spot in vinca which would enable this?

@lianghongzhuo could you please open a pull request with the additional patches that you used? This would be awesome!

lianghongzhuo, 2023-02-04:
sure, done.

Tobias-Fischer, 2024-01-28:
We'll soon release Python 3.11 packages in #418

closed 2024-01-28 by Tobias-Fischer