RoboStack/ros-noetic #343: Can't spawn model in Gazebo

sylarevan, 2023-02-19:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

With a "fresh" new conda environment with robostack installed, I can't spawn any model in Gazeo from my M1 mac.

```
[ERROR] [1676790083.108568, 10.006000]: Spawn service failed. Exiting.
[spawn_urdf-4] process has died [pid 14970, exit code 1
```

Log file is not that explicit:

```
cat /Users/sylar/.ros/log/2ebee0f6-b023-11ed-b08f-ba54f82e7197/spawn_urdf-4*.log
[INFO] [1676790069.166971, 0.000000]: Loading model XML from ros parameter robot_description
[INFO] [1676790069.170384, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[INFO] [1676790072.813411, 0.000000]: Calling service /gazebo/spawn_urdf_model
[INFO] [1676790083.104997, 10.002000]: Spawn status: SpawnModel: Entity pushed to spawn queue, but spawn service timed out waiting for entity to appear in simulation under the name mybot
[rospy.client][INFO] 2023-02-19 08:01:09,055: init_node, name[/spawn_urdf], pid[14970]
[xmlrpc][INFO] 2023-02-19 08:01:09,055: XML-RPC server binding to 127.0.0.1:0
[xmlrpc][INFO] 2023-02-19 08:01:09,055: Started XML-RPC server [http://127.0.0.1:63980/]
[rospy.init][INFO] 2023-02-19 08:01:09,055: ROS Slave URI: [http://127.0.0.1:63980/]
[rospy.impl.masterslave][INFO] 2023-02-19 08:01:09,055: _ready: http://127.0.0.1:63980/
[rospy.registration][INFO] 2023-02-19 08:01:09,055: Registering with master node http://127.0.0.1:11311
[xmlrpc][INFO] 2023-02-19 08:01:09,055: xml rpc node: starting XML-RPC server
[rospy.init][INFO] 2023-02-19 08:01:09,160: registered with master
[rospy.rosout][INFO] 2023-02-19 08:01:09,160: initializing /rosout core topic
[rospy.rosout][INFO] 2023-02-19 08:01:09,164: connected to core topic /rosout
[rospy.simtime][INFO] 2023-02-19 08:01:09,164: initializing /clock core topic
[rospy.simtime][INFO] 2023-02-19 08:01:09,165: connected to core topic /clock
[rosout][INFO] 2023-02-19 08:01:09,166: Loading model XML from ros parameter robot_description
[rosout][INFO] 2023-02-19 08:01:09,170: Waiting for service /gazebo/spawn_urdf_model
[rospy.internal][INFO] 2023-02-19 08:01:09,399: topic[/rosout] adding connection to [/rosout], count 0
[rospy.internal][INFO] 2023-02-19 08:01:12,813: topic[/clock] adding connection to [http://127.0.0.1:64007/], count 0
[rosout][INFO] 2023-02-19 08:01:12,813: Calling service /gazebo/spawn_urdf_model
[rosout][INFO] 2023-02-19 08:01:23,104: Spawn status: SpawnModel: Entity pushed to spawn queue, but spawn service timed out waiting for entity to appear in simulation under the name mybot
[rosout][ERROR] 2023-02-19 08:01:23,108: Spawn service failed. Exiting.
[rospy.core][INFO] 2023-02-19 08:01:23,115: signal_shutdown [atexit]
[rospy.internal][INFO] 2023-02-19 08:01:23,118: topic[/rosout] removing connection to /rosout
[rospy.internal][INFO] 2023-02-19 08:01:23,119: topic[/clock] removing connection to http://127.0.0.1:64007/
[rospy.impl.masterslave][INFO] 2023-02-19 08:01:23,119: atexit
```

This issue seems also mentioned here: https://github.com/RoboStack/ros-noetic/issues/323#issuecomment-1433115800.
I have tried to spaw various robot model, but the issue remains the same.

### Installed packages

```shell
# packages in environment at /opt/homebrew/Caskroom/mambaforge/base/envs/ros_test:
#
# Name                    Version                   Build  Channel
aiohttp                   3.8.4            py39h02fc5c5_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
aom                       3.5.0                h7ea286d_0    conda-forge
apr                       1.7.0                h9b22ae9_5    conda-forge
assimp                    5.2.5                h276577b_0    conda-forge
async-timeout             4.0.2              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               hcb7b3dd_1    conda-forge
attrs                     22.2.0             pyh71513ae_0    conda-forge
bcrypt                    3.2.2            py39h02fc5c5_1    conda-forge
blosc                     1.21.3               h1d6ff8b_0    conda-forge
boost                     1.78.0           py39h99de9ae_4    conda-forge
boost-cpp                 1.78.0               h1cb353e_1    conda-forge
brotli                    1.0.9                h1a8c8d9_8    conda-forge
brotli-bin                1.0.9                h1a8c8d9_8    conda-forge
bullet                    3.21                 h85045c0_4    conda-forge
[... 562 lines cut ...]
                          /Users/sylar/.condarc
                          /opt/homebrew/Caskroom/mambaforge/base/envs/ros_test/.condarc
          conda version : 22.11.1
    conda-build version : not installed
         python version : 3.9.16.final.0
       virtual packages : __archspec=1=arm64
                          __osx=13.2.1=0
                          __unix=0=0
       base environment : /opt/homebrew/Caskroom/mambaforge/base  (writable)
      conda av data dir : /opt/homebrew/Caskroom/mambaforge/base/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-arm64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /opt/homebrew/Caskroom/mambaforge/base/pkgs
                          /Users/sylar/.conda/pkgs
       envs directories : /opt/homebrew/Caskroom/mambaforge/base/envs
                          /Users/sylar/.conda/envs
               platform : osx-arm64
             user-agent : conda/22.11.1 requests/2.28.2 CPython/3.9.16 Darwin/22.3.0 OSX/13.2.1
                UID:GID : 502:20
             netrc file : /Users/sylar/.netrc
           offline mode : False
```

traversaro, 2023-02-19:
This is the issue tracked in https://github.com/conda-forge/gazebo-feedstock/issues/161, it should be fixed by , until that PR is merged a workaround for now is to run:
~~~
mamba install gazebo=11.12.0=*_2
~~~

can you check if this is working?

sylarevan, 2023-02-19:
No luck, spawn service still fails, even after the gazebo downgrade.
`gazebo                    11.12.0              h5491266_2    conda-forge`

Tobias-Fischer, 2023-02-20:
Until the PR in conda-forge is merged, you can do:
```
mamba install -c "tobiasrobotics/label/cfep03" qt-webengine 
```
Which will let you install gazebo 11.12.0=*_7 which does not have the bug. Soon the PR in conda-forge should be merged and then you can just install the latest gazebo without any special instructions. Closing here, feel free to reopen if you still have the issue with the latest gazebo.

Tobias-Fischer, 2023-02-21:
PR has been merged so should all work now

simonbogh, 2023-02-21:
Installing `gazebo 11.12.0=*_7` in its own conda environment (no ROS) gives me an error and crash on startup.

`$ gazebo --versbose`
```sh
[Wrn] [GuiIface.cc:120] Populating font family aliases took 191 ms. Replace uses of missing font family "Sans" with one that exists to avoid this cost. 
[Err] [main.cc:37] Ogre Error:RuntimeAssertionException: Ogre/ShadowExtrudePointLight not found. Verify that you referenced the 'ShadowVolume' folder in your resources.cfg in initialise at /Users/runner/miniforge3/conda-bld/ogre_1666481335795/work/OgreMain/src/OgreShadowVolumeExtrudeProgram.cpp (line 71)
```

`$ mamba list`
```
(gazebo_env) sb devel % mamba list        
# packages in environment at /Users/sb/miniconda3/envs/gazebo_env:
#
# Name                    Version                   Build  Channel
aom                       3.5.0                hf0c8a7f_0    conda-forge
assimp                    5.2.5                hd9e13b6_0    conda-forge
atk-1.0                   2.38.0               h1d18e73_1    conda-forge
blosc                     1.21.2               hebb52c4_0    conda-forge
boost-cpp                 1.78.0               h8b082ac_1    conda-forge
bullet-cpp                3.21                 hc6b5a04_4    conda-forge
bzip2                     1.0.8                h0d85af4_4    conda-forge
c-ares                    1.18.1               h0d85af4_0    conda-forge
ca-certificates           2023.01.10           hecd8cb5_0  
cairo                     1.16.0            h904041c_1014    conda-forge
certifi                   2022.12.7        py39hecd8cb5_0  
cfitsio                   4.2.0                hd56cc12_0    conda-forge
console_bridge            1.0.2                hbb4e6a2_1    conda-forge
cppzmq                    4.9.0                ha27c2bb_0    conda-forge
curl                      7.88.1               h6df9250_0    conda-forge
dartsim                   6.13.0               h6e9e4ac_0    conda-forge
eigen                     3.4.0                h940c156_0    conda-forge
expat                     2.5.0                hf0c8a7f_0    conda-forge
fcl                       0.7.0                ha3e8090_2    conda-forge
ffmpeg                    5.1.2           gpl_h8b4fe81_106    conda-forge
flann                     1.9.1             h56de9e4_1011    conda-forge
fmt                       9.1.0                hb8565cd_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.2               h5bb23bf_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freeimage                 3.18.0              h4fbe5d5_12    conda-forge
freetype                  2.12.1               h3f81eb7_1    conda-forge
freexl                    1.0.6                hb7f2c08_1    conda-forge
fribidi                   1.0.10               hbcb3906_0    conda-forge
gazebo                    11.12.0              hba8d896_7    conda-forge
gdbm                      1.18                 h8a0c380_2    conda-forge
gdk-pixbuf                2.42.10              he00a9fc_0    conda-forge
geos                      3.11.1               hf0c8a7f_0    conda-forge
geotiff                   1.7.1                hd690177_6    conda-forge
gettext                   0.21.1               h8a4c099_0    conda-forge
giflib                    5.2.1                hbcb3906_2    conda-forge
glib                      2.74.1               hbc0c0cd_1    conda-forge
glib-tools                2.74.1               hbc0c0cd_1    conda-forge
gmp                       6.2.1                h2e338ed_0    conda-forge
gnutls                    3.7.8                h207c4f0_0    conda-forge
graphite2                 1.3.13            h2e338ed_1001    conda-forge
graphviz                  6.0.2                hc51f7b9_0    conda-forge
gst-plugins-base          1.22.0               h37e1711_0    conda-forge
gstreamer                 1.22.0               h1d18e73_0    conda-forge
gtk2                      2.24.33              h7c1209e_2    conda-forge
gts                       0.7.6                hccb3bdf_2    conda-forge
harfbuzz                  6.0.0                h08f8713_0    conda-forge
hdf4                      4.2.15               h7aa5921_5    conda-forge
hdf5                      1.12.2          nompi_h48135f9_101    conda-forge
icu                       70.1                 h96cf925_0    conda-forge
imath                     3.1.6                hbc0c0cd_1    conda-forge
jpeg                      9e                   hb7f2c08_3    conda-forge
[... 124 lines cut ...]
xerces-c                  3.2.4                h2007e90_1    conda-forge
xorg-fixesproto           5.0               h0d85af4_1002    conda-forge
xorg-kbproto              1.0.7             h35c211d_1002    conda-forge
xorg-libice               1.0.10               h0d85af4_0    conda-forge
xorg-libsm                1.2.3             h0d85af4_1000    conda-forge
xorg-libx11               1.7.2                h0d85af4_0    conda-forge
xorg-libxau               1.0.9                h35c211d_0    conda-forge
xorg-libxaw               1.0.14               h0d85af4_1    conda-forge
[... cut ...]
xorg-xproto               7.0.31            h35c211d_1007    conda-forge
xz                        5.2.10               h6c40b1e_1  
yaml                      0.2.5                h0d85af4_2    conda-forge
zeromq                    4.3.4                he49afe7_1    conda-forge
zlib                      1.2.13               hfd90126_4    conda-forge
zstd                      1.5.2                hbc0c0cd_6    conda-forge
zziplib                   0.13.69              h5dbffcc_1    conda-forge
```

simonbogh, 2023-02-21:
In a clean install environment `hd5c0442_7` works. Doing a `mamba update --all`will update to the following and then Gazebo crashes on launch:

```
  Package    Version  Build           Channel                  Size
─────────────────────────────────────────────────────────────────────
  Install:
─────────────────────────────────────────────────────────────────────

  + libxslt   1.1.37  h5d22bc9_0      conda-forge/osx-64     Cached
  + lxml       4.9.2  py39hfbce9ca_0  conda-forge/osx-64     Cached

  Change:
─────────────────────────────────────────────────────────────────────

  - gazebo   11.12.0  hd5c0442_7      conda-forge                  
  + gazebo   11.12.0  hba8d896_7      conda-forge/osx-64     Cached

  Upgrade:
─────────────────────────────────────────────────────────────────────

  - ogre     1.10.12  hd9b3965_10     conda-forge                  
  + ogre     1.12.13  h3b2412a_3      conda-forge/osx-64     Cached
```

traversaro, 2023-02-21:
> In a clean install environment `hd5c0442_7` works. Doing a `mamba update --all`will update to the following and then Gazebo crashes on launch. Not sure what the difference is between `hd5c0442_7` and `hba8d896_7`. Unfortunately still can't spawn URDFs into Gazebo, same error message persists:
> 
> `Spawn status: SpawnModel: Entity pushed to spawn queue, but spawn service timed out waiting for entity to appear in simulation under the name my_object. Spawn service failed. Exiting`
> 
> ```
>   Package    Version  Build           Channel                  Size
> ─────────────────────────────────────────────────────────────────────
>   Install:
> ─────────────────────────────────────────────────────────────────────
> 
>   + libxslt   1.1.37  h5d22bc9_0      conda-forge/osx-64     Cached
>   + lxml       4.9.2  py39hfbce9ca_0  conda-forge/osx-64     Cached
> 
>   Change:
> ─────────────────────────────────────────────────────────────────────
> 
>   - gazebo   11.12.0  hd5c0442_7      conda-forge                  
>   + gazebo   11.12.0  hba8d896_7      conda-forge/osx-64     Cached
> 
>   Upgrade:
> ─────────────────────────────────────────────────────────────────────
> 
>   - ogre     1.10.12  hd9b3965_10     conda-forge                  
>   + ogre     1.12.13  h3b2412a_3      conda-forge/osx-64     Cached
> ```

The problem is that the one you had before usesd ogre 1.12, that is not working: https://github.com/conda-forge/gazebo-feedstock/issues/88 .

simonbogh, 2023-02-21:
Great, thanks a lot @traversaro @Tobias-Fischer, I really appreciate your effort here! Unfortunately, I still can't spawn URDFs into Gazebo. Will do some more testing.

Tobias-Fischer, 2023-02-21:
If it happens in a gazebo-only environment, can you please open an issue in the gazebo feedstock?
Also, does the _6 build work for you?

simonbogh, 2023-02-21:
**Gazebo-only environment**
`gazebo                    11.12.0              hd5c0442_6    conda-forge`
Opens, but sometimes sim is not running and objects cannot be added from the gazebo GUI. Gazebo must be restarted a couple of times.

`gazebo                    11.12.0              hd5c0442_7    conda-forge`
Opens, sim is always running and objects can be added from the gazebo GUI.

**ros-noectic-desktop + Gazebo**
URDFs cannot be added in neither version `11.12.0=hd5c0442_6` or `11.12.0=hd5c0442_7` using `rosrun gazebo_ros spawn_model` or a launch-file.

Tobias-Fischer, 2023-02-22:
I guess the first thing to find out is on which platforms that (does not) work. Any chance you can check Linux @simonbogh?

simonbogh, 2023-02-23:
Sure @Tobias-Fischer, I tested it now on **Ubuntu 22.04**, where it is working as expected.

**Gazebo-only environment**
`gazebo                    11.12.0              h151eb85_6    conda-forge`
`gazebo                    11.12.0              h151eb85_7    conda-forge`
Both works fine. Objects can be added inside the Gazebo GUI and SIM time is running.

**ros-noectic-desktop + Gazebo**
`gazebo                    11.12.0              h151eb85_6    conda-forge`
`gazebo                    11.12.0              h151eb85_7    conda-forge`
Both works. I can spawn models using `rosrun gazebo_ros spawn_model` directly, and also using my `moveit_config` package, models are spawned in Gazebo and Moveit is working.

Tobias-Fischer, 2023-02-23:
Awesome, thanks for that. To further narrow down, do you also have a Windows machine around where you could test? Then we would know if it's an osx-only problem. It seems that osx-arm64 does not work either (tested myself).

simonbogh, 2023-02-25:
Sorry can't test on a Windows computer.

Tobias-Fischer, 2023-02-25:
No, in the `*_8` builds @traversaro removed the ogre 1.12 builds that caused issues, but no new functionality was added.

Tobias-Fischer, 2023-05-07:
@traversaro - is there any chance you could test whether this works on Windows?

traversaro, 2023-05-08:
I just tested on Windows the following commands:

~~~
roslaunch gazebo_ros empty_world.launch
~~~

and
 
~~~
curl https://gist.githubusercontent.com/traversaro/6308979ae31fb1bf6ecae2219d383f82/raw/76e5bd851744b161130f89ff9a3aca2ec37f4068/object.urdf -o object.urdf
rosrun gazebo_ros spawn_model -file %cd%\object.urdf -urdf -z 1 -model my_object
~~~

and everything works fine.

The tested env:

<details>
<summary>See environment</summary>

~~~
adwaita-icon-theme        43                        win_0    conda-forge
aom                       3.5.0                h63175ca_0    conda-forge
apr                       1.7.0                he38c35c_5    conda-forge
assimp                    5.2.5                h4dcb625_0    conda-forge
bcrypt                    3.2.2            py39ha55989b_1    conda-forge
blosc                     1.21.3               hdccc3a2_0    conda-forge
boost                     1.78.0           py39hbd792c9_4    conda-forge
boost-cpp                 1.78.0               h9f4b32c_3    conda-forge
brotli                    1.0.9                hcfcfb64_8    conda-forge
brotli-bin                1.0.9                hcfcfb64_8    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2023.5.7             h56e8100_0    conda-forge
cairo                     1.16.0            hd694305_1014    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
certifi                   2023.5.7           pyhd8ed1ab_0    conda-forge
cffi                      1.15.1           py39h68f70e3_3    conda-forge
cfitsio                   4.2.0                h9ebe7e4_0    conda-forge
cmake                     3.26.3               h1537add_0    conda-forge
console_bridge            1.0.2                h5362a0b_1    conda-forge
contourpy                 1.0.7            py39h1f6ef14_0    conda-forge
cppzmq                    4.9.0                h449d27f_0    conda-forge
cryptography              40.0.2           py39hb6bd5e6_0    conda-forge
curl                      7.88.1               h68f0423_1    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
dirent                    1.21                          0    conda-forge
distro                    1.8.0              pyhd8ed1ab_0    conda-forge
dlfcn-win32               1.3.0                h0e60522_0    conda-forge
docutils                  0.19             py39hcbf5309_1    conda-forge
eigen                     3.4.0                h2d74725_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.5.0                h63175ca_1    conda-forge
ffmpeg                    5.1.2           gpl_h5b1d025_107    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.2               hbde0cde_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.39.3           py39ha55989b_0    conda-forge
freeglut                  3.2.2                h0e60522_1    conda-forge
freeimage                 3.18.0              h30d40d2_12    conda-forge
freetype                  2.12.1               h546665d_1    conda-forge
freexl                    1.0.6                h67ca5e6_1    conda-forge
fribidi                   1.0.10               h8d14728_0    conda-forge
gazebo                    11.12.0              h0516dcc_9    conda-forge
gdk-pixbuf                2.42.10              h84213e4_0    conda-forge
geos                      3.11.1               h1537add_0    conda-forge
[... 373 lines cut ...]
wheel                     0.40.0             pyhd8ed1ab_0    conda-forge
x264                      1!164.3095           h8ffe710_2    conda-forge
x265                      3.5                  h2d74725_3    conda-forge
xerces-c                  3.2.4                h63175ca_2    conda-forge
xorg-kbproto              1.0.7             hcd874cb_1002    conda-forge
xorg-libice               1.0.10               hcd874cb_0    conda-forge
xorg-libsm                1.2.3             hcd874cb_1000    conda-forge
xorg-libx11               1.8.4                hcd874cb_0    conda-forge
xorg-libxau               1.0.9                hcd874cb_0    conda-forge
xorg-libxdmcp             1.1.3                hcd874cb_0    conda-forge
xorg-libxext              1.3.4                hcd874cb_2    conda-forge
xorg-libxpm               3.5.13               hcd874cb_0    conda-forge
xorg-libxt                1.2.1                hcd874cb_2    conda-forge
xorg-xextproto            7.3.0             hcd874cb_1003    conda-forge
xorg-xproto               7.0.31            hcd874cb_1007    conda-forge
xz                        5.2.6                h8d14728_0    conda-forge
yaml                      0.2.5                h8ffe710_2    conda-forge
yaml-cpp                  0.7.0                h63175ca_2    conda-forge
zeromq                    4.3.4                h0e60522_1    conda-forge
zipp                      3.15.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               hcfcfb64_4    conda-forge
zstd                      1.5.2                h12be248_6    conda-forge
zziplib                   0.13.69              h1d00b33_1    conda-forg
~~~
</details>

Tobias-Fischer, 2023-05-09:
Thanks a lot, @traversaro! So it indeed seems to be an osx-specific issue. Not sure how to debug this properly .. it’s odd it works when Gazebo standalone is used, but not with ROS. I can’t see relevant osx-only patches here that would cause this.

dirkpitt2050, 2023-05-11:
Is there a workaround, for example installing gazebo 11.10.1 (ha00e38d_3) and how would I do that? mamba doesn't let me do it because of incompatible dependencies.

siilats, 2023-05-31:
did you figure out how to downgrade gazebo?

dirkpitt2050, 2023-06-03:
Unfortunately no.

DevnathNair, 2023-09-18:
@Tobias-Fischer Any word on if there's any progress on this issue? Facing similar, if not the exact problem on my M1 machine (macOS 14.0 arm64).

traversaro, 2023-09-18:
> @Tobias-Fischer Any word on if there's any progress on this issue? Facing similar, if not the exact problem on my M1 machine (macOS 14.0 arm64).

Can you also provide the `mamba list` for your environment?

DevnathNair, 2023-09-18:
> > @Tobias-Fischer Any word on if there's any progress on this issue? Facing similar, if not the exact problem on my M1 machine (macOS 14.0 arm64).
> 
> Can you also provide the `mamba list` for your environment?

Sure thing!

```
# Name                    Version                   Build  Channel
adwaita-icon-theme        43                       unix_0    conda-forge
aom                       3.5.0                h7ea286d_0    conda-forge
apr                       1.7.0                h9b22ae9_5    conda-forge
argcomplete               3.1.2              pyhd8ed1ab_0    conda-forge
assimp                    5.2.5                h276577b_0    conda-forge
atk-1.0                   2.38.0               hcb7b3dd_1    conda-forge
bcrypt                    4.0.1            py39haff01fd_0    conda-forge
blosc                     1.21.5               hc338f07_0    conda-forge
boost                     1.78.0           py39h99de9ae_4    conda-forge
boost-cpp                 1.78.0               hf1d6563_2    conda-forge
brotli                    1.1.0                hb547adb_0    conda-forge
brotli-bin                1.1.0                hb547adb_0    conda-forge
bullet-cpp                3.21             py39hde7b980_4    conda-forge
bzip2                     1.0.8                h3422bc3_4    conda-forge
c-ares                    1.19.1               hb547adb_0    conda-forge
c-compiler                1.6.0                hd291e01_0    conda-forge
ca-certificates           2023.7.22            hf0a4a13_0    conda-forge
cairo                     1.16.0            h73a0509_1014    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_0    conda-forge
cctools                   973.0.1             hd1ac623_14    conda-forge
cctools_osx-arm64         973.0.1             h2a25c60_14    conda-forge
certifi                   2023.7.22          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1           py39h7e6b969_3    conda-forge
cfitsio                   4.2.0                h2f961c4_0    conda-forge
clang                     15.0.7               hce30654_3    conda-forge
clang-15                  15.0.7          default_h5dc8d65_3    conda-forge
clang_osx-arm64           15.0.7               h77e971b_3    conda-forge
clangxx                   15.0.7          default_h610c423_3    conda-forge
clangxx_osx-arm64         15.0.7               h768a7fd_3    conda-forge
cmake                     3.26.3               hf234bd0_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.4.2              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
colcon-cmake              0.2.27             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0            py39h2804cbe_1    conda-forge
colcon-core               0.12.1             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.2.2                      py_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-output             0.2.13             pyhd8ed1ab_0    conda-forge
colcon-package-information 0.3.3                      py_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
colcon-parallel-executor  0.2.4                      py_0    conda-forge
colcon-pkg-config         0.1.0                      py_0    conda-forge
colcon-powershell         0.3.6                      py_0    conda-forge
colcon-python-setup-py    0.2.7                      py_0    conda-forge
colcon-recursive-crawl    0.2.1                      py_0    conda-forge
colcon-ros                0.3.23             pyhd8ed1ab_0    conda-forge
colcon-test-result        0.3.8                      py_0    conda-forge
colcon-zsh                0.4.0              pyhd8ed1ab_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
coloredlogs               15.0.1             pyhd8ed1ab_3    conda-forge
compiler-rt               15.0.7               hf8d1dfb_1    conda-forge
compiler-rt_osx-arm64     15.0.7               hf8d1dfb_1    conda-forge
compilers                 1.6.0                hce30654_0    conda-forge
console_bridge            1.0.2                h3e96240_1    conda-forge
contourpy                 1.1.1            py39hbd775c9_0    conda-forge
coreutils                 9.4                  hb547adb_0    conda-forge
coverage                  7.3.1            py39h0f82c59_0    conda-forge
cppzmq                    4.10.0               h6c41c7e_0    conda-forge
[... 461 lines cut ...]
xorg-kbproto              1.0.7             h27ca646_1002    conda-forge
xorg-libice               1.0.10               h27ca646_0    conda-forge
xorg-libsm                1.2.3             h27ca646_1000    conda-forge
xorg-libx11               1.8.4                h1a8c8d9_0    conda-forge
xorg-libxau               1.0.11               hb547adb_0    conda-forge
xorg-libxaw               1.0.14               h3422bc3_1    conda-forge
xorg-libxdmcp             1.1.3                h27ca646_0    conda-forge
xorg-libxext              1.3.4                h1a8c8d9_2    conda-forge
[... cut ...]
yaml                      0.2.5                h3422bc3_2    conda-forge
yaml-cpp                  0.7.0                hb7217d7_2    conda-forge
zeromq                    4.3.4                hbdafb3b_1    conda-forge
zipp                      3.16.2             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               h53f4e23_5    conda-forge
zstd                      1.5.5                h4f39d0f_0    conda-forge
zziplib                   0.13.69              he1e0b03_1    conda-forge
```

DevnathNair, 2023-10-01:
@traversaro Is there anything more information I can provide to assist in identifying the issue?

saeedarabi92, 2023-10-04:
having the same issue here that I cannot spawn the robot into the Gazebo (I can manually import a model into gazebo tho).

Environment:
````
# Name                    Version                   Build  Channel
adwaita-icon-theme        43                       unix_0    conda-forge
aom                       3.5.0                hf0c8a7f_0    conda-forge
apr                       1.7.0                h0d85af4_5    conda-forge
argcomplete               3.1.2              pyhd8ed1ab_0    conda-forge
assimp                    5.2.5                hd9e13b6_0    conda-forge
atk-1.0                   2.38.0               h1d18e73_1    conda-forge
bcrypt                    4.0.1            py39hf33989b_1    conda-forge
blosc                     1.21.5               heccf04b_0    conda-forge
boost                     1.78.0           py39h4aa56a1_5    conda-forge
boost-cpp                 1.78.0               h31500c2_2    conda-forge
brotli                    1.1.0                h0dc2134_1    conda-forge
brotli-bin                1.1.0                h0dc2134_1    conda-forge
bullet-cpp                3.21                 hc6b5a04_4    conda-forge
bzip2                     1.0.8                h0d85af4_4    conda-forge
c-ares                    1.19.1               h0dc2134_0    conda-forge
c-compiler                1.6.0                h63c33a9_0    conda-forge
ca-certificates           2023.7.22            h8857fd0_0    conda-forge
cairo                     1.16.0            h904041c_1014    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_0    conda-forge
cctools                   973.0.1             hd9ad811_14    conda-forge
cctools_osx-64            973.0.1             habff3f6_14    conda-forge
certifi                   2023.7.22          pyhd8ed1ab_0    conda-forge
cffi                      1.16.0           py39h18ef598_0    conda-forge
cfitsio                   4.2.0                hd56cc12_0    conda-forge
clang                     15.0.7               h694c41f_3    conda-forge
clang-15                  15.0.7          default_hdb78580_3    conda-forge
clang_osx-64              15.0.7               h03d6864_3    conda-forge
clangxx                   15.0.7          default_hdb78580_3    conda-forge
clangxx_osx-64            15.0.7               h2133e9c_3    conda-forge
cmake                     3.26.3               h4032537_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.5.0              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
colcon-cmake              0.2.27             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0            py39h6e9494a_1    conda-forge
colcon-core               0.13.0             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.2.2                      py_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-output             0.2.13             pyhd8ed1ab_0    conda-forge
colcon-package-information 0.3.3                      py_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
colcon-parallel-executor  0.2.4                      py_0    conda-forge
colcon-pkg-config         0.1.0                      py_0    conda-forge
colcon-powershell         0.3.6                      py_0    conda-forge
colcon-python-setup-py    0.2.7                      py_0    conda-forge
colcon-recursive-crawl    0.2.1                      py_0    conda-forge
colcon-ros                0.3.23             pyhd8ed1ab_0    conda-forge
colcon-test-result        0.3.8                      py_0    conda-forge
colcon-zsh                0.4.0              pyhd8ed1ab_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
coloredlogs               15.0.1             pyhd8ed1ab_3    conda-forge
compiler-rt               15.0.7               he1888fc_1    conda-forge
compiler-rt_osx-64        15.0.7               he1888fc_1    conda-forge
compilers                 1.6.0                h694c41f_0    conda-forge
console_bridge            1.0.2                hbb4e6a2_1    conda-forge
contourpy                 1.1.1            py39h8ee36c8_1    conda-forge
coreutils                 9.4                  h0dc2134_0    conda-forge
coverage                  7.3.2            py39hdc70f33_0    conda-forge
cppzmq                    4.10.0               h478c2f3_0    conda-forge
cryptography              41.0.4           py39he8b0a07_0    conda-forge
curl                      7.88.1               h6df9250_1    conda-forge
cxx-compiler              1.6.0                h1c7c39f_0    conda-forge
[... 508 lines cut ...]
xorg-kbproto              1.0.7             h35c211d_1002    conda-forge
xorg-libice               1.0.10               h0d85af4_0    conda-forge
xorg-libsm                1.2.3             h0d85af4_1000    conda-forge
xorg-libx11               1.8.4                hb7f2c08_0    conda-forge
xorg-libxau               1.0.11               h0dc2134_0    conda-forge
xorg-libxaw               1.0.14               h0d85af4_1    conda-forge
xorg-libxdmcp             1.1.3                h35c211d_0    conda-forge
[... cut ...]
yaml                      0.2.5                h0d85af4_2    conda-forge
yaml-cpp                  0.7.0                he965462_3    conda-forge
zeromq                    4.3.4                he49afe7_1    conda-forge
zipp                      3.17.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               h8a1eda9_5    conda-forge
zstd                      1.5.5                h829000d_0    conda-forge
zziplib                   0.13.69              h5dbffcc_1    conda-forge
````

mmattamala, 2023-11-27:
I had the same problem with a Mac M1 and Gazebo 11.12.0 (`hf82afb7_9`). A workaround that allowed me to spawn models as expected was manually cloning and building [gazebo_ros_pkgs](https://github.com/ros-simulation/gazebo_ros_pkgs) within my (robostack) catkin workspace.

Perhaps the problem is in that package instead of gazebo itself.

Tobias-Fischer, 2023-11-27:
Very interesting observation @mmattamala, thanks for that!

It would be even more interesting to see which of the gazebo_ros_pkgs (it is a metapackage) causes the issue - one could try swapping out one of the contained packages after the other by installing it with `mamba`, and removing it from the catkin workspace. If we knew which individual package is the culprit, this would be even better.

mmattamala, 2023-11-28:
Thanks for the advice @Tobias-Fischer ! I tried what you suggested, just compiling `gazebo_ros` from the metapackage while keeping the other packages from mamba and I can confirm it works.

Tobias-Fischer, 2024-02-16:
Hi @mmattamala - we have recently rebuilt all packages for Noetic. Could you please check whether the problem still persists with the new packages?

mmattamala, 2024-02-16:
Hi @Tobias-Fischer, thanks for the update. I made a new environment and reinstalled RoboStack from scratch but did not succeed to make my simulation environment work - it still hangs and eventually dies trying to spawn the model.

My current `gazebo_ros` version is:
```sh
ros-noetic-gazebo-ros     2.9.2           py311h7f7f440_8    robostack-staging
```
If you could confirm this is the expected version that should fix the problems, I can dig deeper to see if there is something wrong about my environment or maybe some difference in the building process still persists.

On the other hand, I can also confirm that `rqt_image_view` works as intended now (referring to the issue I reported in https://github.com/RoboStack/ros-noetic/issues/417#issuecomment-1841827335)

Tobias-Fischer, 2024-02-17:
You have the correct version of ros-noetic-gazebo-ros. Could you please check if your gazebo versions match? Very weird.

Tobias-Fischer, 2024-02-18:
Thanks @mmattamala! Could you please provide the output of the build when compiled in a catkin_ws? We can then compare to https://github.com/RoboStack/ros-noetic/actions/runs/7732537386/job/21083181351 and see if there are any obvious differences.

Tobias-Fischer, 2024-02-28:
Closing as could not replicate and no further info provided; feel free to reopen if this is still an issue.

closed 2024-02-28 by Tobias-Fischer

GokulEdakkePuram, 2024-03-29:
> Thanks for the advice @Tobias-Fischer ! I tried what you suggested, just compiling `gazebo_ros` from the metapackage while keeping the other packages from mamba and I can confirm it works.

Could you briefly tell me how were you able to do this. I am facing the same problem and I cloned the gitrepo that you mentioned into my catkin workspace and built it, but then I had errors when starting turtlebot3 simulation. When I tried running the same, there was just an empty gazebo window with nothing populated.

GokulEdakkePuram, 2024-03-30:
> Hey @mmattamala - I just tried myself, and was able to spawn some of the standard models. Could you please let us know which model you are trying to insert / provide us with specific instructions that fail for you?

I have the same problem. I tried to spawn a turtlebot3 model for simulation using the turtlebot3 packages installed from RoboStack itself. But it is not spawning. I could add other models such as the house world from the turtlebot3_gazebo package itself, but the URDF model doesn't spawn.

sylarevan, 2024-04-05:
Hi all.
I can also confirm the issue is still present on my M1 mac, and a simple spawn of a turtlebot3 does not work after a brand new installation.
The proposed solution to add the gazebo_ros package to our own workspace solves this issue and the turtlebot3 is then correctly spawned ... but then no LDS or camera data are available on ROS (while both seems to be correctly simulated on Gazebo itself).

So far, my only solution has been to propose a UTM arm VM with a ROS Noetic native installation to my students, waiting for a solution to be found (with a spawned Turtlebot3 with LDS and camera working).

traversaro, 2024-04-05:
This is probably related to:
* https://github.com/RoboStack/ros-noetic/issues/459
* https://github.com/RoboStack/ros-noetic/issues/454

For visibility, it is easier if comments go in the active issues, as it is easy to lose them in closed issues. As mentioned in  https://github.com/RoboStack/ros-noetic/issues/459, if someone is able to run gazebo with the `--verbose` option, that would be helpful to understand what is going on.

traversaro, 2024-04-07:
> This is probably related to:
> 
>     * [Gazebo doesn't populate with URDF model on Mac (Apple M2) #459](https://github.com/RoboStack/ros-noetic/issues/459)
> 
>     * [Gazebo Spawn Fails for Turtlebot3 in ROS Noetic on M1 Mac #454](https://github.com/RoboStack/ros-noetic/issues/454)
> 
> 
> For visibility, it is easier if comments go in the active issues, as it is easy to lose them in closed issues. As mentioned in #459, if someone is able to run gazebo with the `--verbose` option, that would be helpful to understand what is going on.

A workaround was found and reported in https://github.com/RoboStack/ros-noetic/issues/459#issuecomment-2027155423 .