RoboStack/ros-noetic #344: rosrun terminal auto-complete can't find executable node

AchmadFathoni, 2023-02-19:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

After activating the ros_env and source setup.zsh, auto-complete can't show node from a package. For example:

1. Type `rosrun rqt_gra` in teminal
2. Hit tab
3. It will auto-complete to `rosrun rqt_graph `
4. Hit tab again
5. Nothing come out.

### Installed packages

```shell
List of packages in environment: "/home/toni/.mamba/envs/ros_env"

  Name                                       Version       Build                Channel          
───────────────────────────────────────────────────────────────────────────────────────────────────
  _libgcc_mutex                              0.1           conda_forge          conda-forge      
  _openmp_mutex                              4.5           2_gnu                conda-forge      
  aiohttp                                    3.8.4         py39h72bdee0_0       conda-forge      
  aiosignal                                  1.3.1         pyhd8ed1ab_0         conda-forge      
  alsa-lib                                   1.2.8         h166bdaf_0           conda-forge      
  aom                                        3.5.0         h27087fc_0           conda-forge      
  apr                                        1.7.0         h7f98852_5           conda-forge      
  assimp                                     5.2.5         hf40c2ba_0           conda-forge      
  async-timeout                              4.0.2         pyhd8ed1ab_0         conda-forge      
  atk-1.0                                    2.38.0        hd4edc92_1           conda-forge      
  attr                                       2.5.1         h166bdaf_1           conda-forge      
  attrs                                      22.2.0        pyh71513ae_0         conda-forge      
  bcrypt                                     3.2.2         py39hb9d737c_1       conda-forge      
  boost                                      1.78.0        py39h7c9e3ff_4       conda-forge      
  boost-cpp                                  1.78.0        h75c5d50_1           conda-forge      
  brotli                                     1.0.9         h166bdaf_8           conda-forge      
  brotli-bin                                 1.0.9         h166bdaf_8           conda-forge      
  bzip2                                      1.0.8         h7f98852_4           conda-forge      
  c-ares                                     1.18.1        h7f98852_0           conda-forge      
  ca-certificates                            2022.12.7     ha878542_0           conda-forge      
  cairo                                      1.16.0        ha61ee94_1014        conda-forge      
  catkin_pkg                                 0.5.2         pyhd8ed1ab_0         conda-forge      
  catkin_tools                               0.9.2         pyhd8ed1ab_0         conda-forge      
  certifi                                    2022.12.7     pyhd8ed1ab_0         conda-forge      
  cffi                                       1.15.1        py39he91dace_3       conda-forge      
  charset-normalizer                         2.1.1         pyhd8ed1ab_0         conda-forge      
  cmake                                      3.25.2        h077f3f9_0           conda-forge      
  console_bridge                             1.0.2         h924138e_1           conda-forge      
  contourpy                                  1.0.7         py39h4b4f3f3_0       conda-forge      
  coreutils                                  9.1           h166bdaf_0           conda-forge      
  cryptography                               39.0.1        py39h079d5ae_0       conda-forge      
  curl                                       7.88.0        hdc1c0ab_0           conda-forge      
  cycler                                     0.11.0        pyhd8ed1ab_0         conda-forge      
  dbus                                       1.13.6        h5008d03_3           conda-forge      
  defusedxml                                 0.7.1         pyhd8ed1ab_0         conda-forge      
  distro                                     1.8.0         pyhd8ed1ab_0         conda-forge      
  docutils                                   0.19          py39hf3d152e_1       conda-forge      
  double-conversion                          3.2.0         h27087fc_1           conda-forge      
  eigen                                      3.4.0         h4bd325d_0           conda-forge      
  eigen3                                     3.3.7         0                    omnia            
  empy                                       3.3.4         pyh9f0ad1d_1         conda-forge      
  expat                                      2.5.0         h27087fc_0           conda-forge      
  ffmpeg                                     5.1.2         gpl_h8dda1f0_106     conda-forge      
  fftw                                       3.3.10        nompi_hf0379b8_106   conda-forge      
  flann                                      1.9.1         he05ef13_1011        conda-forge      
  fmt                                        9.1.0         h924138e_0           conda-forge      
  font-ttf-dejavu-sans-mono                  2.37          hab24e00_0           conda-forge      
  font-ttf-inconsolata                       3.000         h77eed37_0           conda-forge      
  font-ttf-source-code-pro                   2.038         h77eed37_0           conda-forge      
[... 447 lines cut ...]
### Environment info

```shell
environment : ros_env (active)
           env location : /home/toni/.mamba/envs/ros_env
      user config files : /home/toni/.mambarc
 populated config files : /home/toni/.mamba/envs/ros_env/.condarc
       libmamba version : 1.3.1
     micromamba version : 1.3.1
           curl version : libcurl/7.87.0 OpenSSL/3.0.8 zlib/1.2.13 libssh2/1.10.0 nghttp2/1.47.0
     libarchive version : libarchive 3.6.2 zlib/1.2.13 bz2lib/1.0.8 libzstd/1.5.2
       virtual packages : __unix=0=0
                          __linux=6.1.12=0
                          __glibc=2.37=0
                          __archspec=1=x86_64
                          __cuda=12.0=0
               channels : https://conda.anaconda.org/robostack-staging/linux-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://conda.anaconda.org/omnia/linux-64
                          https://conda.anaconda.org/omnia/noarch
       base environment : /home/toni/.mamba
               platform : linux-64
```

Tobias-Fischer, 2023-02-20:
This works without issues for me on osx-arm64 ..

Tobias-Fischer, 2023-02-20:
Note that ros-noetic-rosbash needs to be installed, but that seems to be the case in your environment (and I am assuming you have tried opening a new terminal, as well as see if it happens without sourcing your catkin setup.bash?)

AchmadFathoni, 2023-02-20:
> Note that ros-noetic-rosbash needs to be installed, but that seems to be the case in your environment (and I am assuming you have tried opening a new terminal, as well as see if it happens without sourcing your catkin setup.bash?)

Still doesn't work in new terminal without sourcing setup.bash. Both in zsh and bash.

Tobias-Fischer, 2023-02-21:
But it does work after sourcing?!?

AchmadFathoni, 2023-02-21:
> But it does work after sourcing?!?

No, it still doesn't work.
I've tried in my ubuntu and it works. Arch Linux is so peculiar.

Tobias-Fischer, 2023-02-21:
It works in Ubuntu with RoboStack?

AchmadFathoni, 2023-02-21:
> It works in Ubuntu with RoboStack?

Yes

Tobias-Fischer, 2023-02-21:
And you have tried `pacman -S bash-completion`?

AchmadFathoni, 2023-02-21:
> And you have tried `pacman -S bash-completion`?

Already installed. I think it's not about missing package. Since the first package name completion is working but the second node name completion is not.

traversaro, 2023-02-21:
Not sure if it is relevant, but we also have a bash-completion package in conda-forge as well.

Tobias-Fischer, 2023-02-21:
Yeah, might be worth `mamba install bash-completion`. Could you please also provide the output of `echo $ROS_PACKAGE_PATH` and `echo $ROS_ROOT`, as well as of `rospack profile`?

AchmadFathoni, 2023-02-22:
> Yeah, might be worth `mamba install bash-completion`. Could you please also provide the output of `echo $ROS_PACKAGE_PATH` and `echo $ROS_ROOT`, as well as of `rospack profile`?

Sure
```
List of packages in environment: "/home/toni/.mamba/envs/ros_env"

  Name             Version  Build       Channel    
─────────────────────────────────────────────────────
  bash-completion  2.11     ha770c72_1  conda-forge
❯ echo $ROS_PACKAGE_PATH
/home/toni/.mamba/envs/ros_env/share
❯ echo $ROS_ROOT
/home/toni/.mamba/envs/ros_env/share/ros
❯ rospack profile
Full tree crawl took 0.314788 seconds.
Directories marked with (*) contain no manifest.  You may
want to delete these directories.
To get just of list of directories without manifests,
re-run the profile with --zombie-only
-------------------------------------------------------------
0.308692   /home/toni/.mamba/envs/ros_env/share
0.082386 * /home/toni/.mamba/envs/ros_env/share/qt
0.081231 * /home/toni/.mamba/envs/ros_env/share/qt/3rd_party_licenses
0.066833 * /home/toni/.mamba/envs/ros_env/share/qt/3rd_party_licenses/qtwebengine
0.066171 * /home/toni/.mamba/envs/ros_env/share/qt/3rd_party_licenses/qtwebengine/src
0.066136 * /home/toni/.mamba/envs/ros_env/share/qt/3rd_party_licenses/qtwebengine/src/3rdparty
0.065949 * /home/toni/.mamba/envs/ros_env/share/qt/3rd_party_licenses/qtwebengine/src/3rdparty/chromium
0.062755 * /home/toni/.mamba/envs/ros_env/share/qt/3rd_party_licenses/qtwebengine/src/3rdparty/chromium/third_party
0.043518 * /home/toni/.mamba/envs/ros_env/share/qt/3rd_party_licenses/qtwebengine/src/3rdparty/chromium/third_party/devtools-frontend
0.043479 * /home/toni/.mamba/envs/ros_env/share/qt/3rd_party_licenses/qtwebengine/src/3rdparty/chromium/third_party/devtools-frontend/src
0.039678 * /home/toni/.mamba/envs/ros_env/share/qt/3rd_party_licenses/qtwebengine/src/3rdparty/chromium/third_party/devtools-frontend/src/node_modules
0.039598 * /home/toni/.mamba/envs/ros_env/share/terminfo
0.034822 * /home/toni/.mamba/envs/ros_env/share/icons
0.034544 * /home/toni/.mamba/envs/ros_env/share/icons/Adwaita
0.020097 * /home/toni/.mamba/envs/ros_env/share/cmake-3.25
0.019345 * /home/toni/.mamba/envs/ros_env/share/locale
0.015581 * /home/toni/.mamba/envs/ros_env/share/doc
0.014081 * /home/toni/.mamba/envs/ros_env/share/man
0.012096 * /home/toni/.mamba/envs/ros_env/share/cmake-3.25/Help
0.009886 * /home/toni/.mamba/envs/ros_env/share/doc/gettext

```

Tobias-Fischer, 2023-05-24:
I struggled with this the other day, and installing + setting up https://github.com/conda-incubator/conda-zsh-completion solved it for me. Please give it a go - if it doesn't work, feel free to reopen.

closed 2023-05-24 by Tobias-Fischer