RoboStack/ros-noetic #346: rqt_graph Segmentation fault: 11 on MacOS

yinchuan, 2023-02-28:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Thanks for your help. I'm using ROS Noetic on MacOS and rqt_graph crashed with Segmentation fault: 11. (rqt_plot crashes as well, so I guess it's probably related to qt).

I followed the instructions in the [document](https://robostack.github.io/GettingStarted.html):
```
conda install mamba -c conda-forge
mamba create -n ros_env python=3.9
conda activate ros_env
conda config --env --add channels conda-forge
conda config --env --add channels robostack-staging
mamba install ros-noetic-desktop
```

The installations finished without error. `roscore` and `rosrun turtlesim turtlesim_node` are also good. But both `rosrun rqt_graph rqt_graph` and `rqt_graph` crashe with Segmentation fault: 11.

Please let me if you need more information. Thanks.

### Installed packages

```shell
# packages in environment at /Users/yinchuan/miniconda3/envs/robostackenv:
#
# Name                    Version                   Build  Channel
adwaita-icon-theme        41.0                 h694c41f_1    conda-forge
aom                       3.5.0                hf0c8a7f_0    conda-forge
apr                       1.7.0                h0d85af4_5    conda-forge
assimp                    5.2.5                hd9e13b6_0    conda-forge
atk-1.0                   2.38.0               h1d18e73_1    conda-forge
bcrypt                    3.2.2            py39ha30fb19_1    conda-forge
boost                     1.78.0           py39h6dc771e_4    conda-forge
boost-cpp                 1.78.0               h8b082ac_1    conda-forge
brotli                    1.0.9                hb7f2c08_8    conda-forge
brotli-bin                1.0.9                hb7f2c08_8    conda-forge
bzip2                     1.0.8                h0d85af4_4    conda-forge
c-ares                    1.18.1               h0d85af4_0    conda-forge
ca-certificates           2023.01.10           hecd8cb5_0  
cairo                     1.16.0            h904041c_1014    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
certifi                   2022.12.7        py39hecd8cb5_0  
cffi                      1.15.1           py39h131948b_3    conda-forge
cmake                     3.25.2               h4032537_0    conda-forge
console_bridge            1.0.2                hbb4e6a2_1    conda-forge
contourpy                 1.0.7            py39h92daf61_0    conda-forge
cryptography              39.0.1           py39hbeae22c_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.8.0              pyhd8ed1ab_0    conda-forge
docutils                  0.19             py39h6e9494a_1    conda-forge
eigen                     3.4.0                h940c156_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.5.0                hf0c8a7f_0    conda-forge
ffmpeg                    5.1.2           gpl_h8b4fe81_106    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.2               h5bb23bf_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.38.0           py39ha30fb19_1    conda-forge
freeimage                 3.18.0              h4fbe5d5_12    conda-forge
freetype                  2.12.1               h3f81eb7_1    conda-forge
fribidi                   1.0.10               hbcb3906_0    conda-forge
gdk-pixbuf                2.42.10              he00a9fc_0    conda-forge
gettext                   0.21.1               h8a4c099_0    conda-forge
[... 383 lines cut ...]
         python version : 3.10.9.final.0
       virtual packages : __archspec=1=x86_64
                          __osx=10.14.6=0
                          __unix=0=0
       base environment : /Users/yinchuan/miniconda3  (writable)
      conda av data dir : /Users/yinchuan/miniconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://repo.anaconda.com/pkgs/main/osx-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/osx-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /Users/yinchuan/miniconda3/pkgs
                          /Users/yinchuan/.conda/pkgs
       envs directories : /Users/yinchuan/miniconda3/envs
                          /Users/yinchuan/.conda/envs
               platform : osx-64
             user-agent : conda/23.1.0 requests/2.28.1 CPython/3.10.9 Darwin/18.7.0 OSX/10.14.6
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2023-02-28:
Hi there,

It seems like you are using some packages from the `defaults` channel, which might lead to this issue. Can you please try again using the instructions in this repository (https://github.com/RoboStack/ros-noetic)? This will avoid using the `defaults channel`. I just tried and I can run rqt_graph without issues.

Best, Tobi

yinchuan, 2023-02-28:
Thanks.

I did follow the instructions in the repo before I use the instructions in the doc. They resulted in the same issue.

I think the packages from default are installed when installing mamba because I used miniconda which has the default channel.

I'm going to try miniforge instead to see if it works.

yinchuan, 2023-02-28:
I followed the instructions in this repo. 

`rqt_graph` still crashes with Segmentation fault: 11.
```
$ rqt_graph -v
PluginManager._discover() force discovery of plugins
RospkgPluginProvider._find_plugins() crawling for plugins of type 'qt_gui'
Segmentation fault: 11
```
But `rqt_graph --qt-binding pyside` works although its takes long time to refresh.

Both `rqt_plot` and `rqt_plot -v --qt-binding pyside` crashes with Segmentation fault: 11.
```
$ rqt_plot -v --qt-binding pyside
PluginManager._discover() force discovery of plugins
RospkgPluginProvider._find_plugins() crawling for plugins of type 'qt_gui'
Segmentation fault: 11
```

Tobias-Fischer, 2023-03-01:
I checked the differences to my environment. To get it to work, please `mamba uninstall ros-noetic-qt-gui-cpp`.

Note for future self: The issue is probably linking to Python in https://github.com/RoboStack/ros-noetic/blob/master/patch/ros-noetic-qt-gui-cpp.patch on osx, which should only happen on Linux. Unfortunately I don't have capacity any time soon to fix this properly.

yinchuan, 2023-03-01:
Thanks Tobias.
I can't uninstall `ros-noetic-qt-gui-cpp` since it will also uninstall `ros-noetic-desktop` and hundreds of other packages. I think I will just use `rqt_graph --qt-binding pyside` for the moment. 

BTW, `rqt_console` and `rqt_logger_level` are working so it looks not all qt related tools have the issue.

huma0605, 2023-03-23:
I could repro the same issue with the installation instruction for ros-noetic. https://github.com/RoboStack/ros-noetic

It is working fine for ros-humble.

I am using M1 MacBook Pro.

`rqt_graph --qt-binding pyside` is not a working workaround for me as well.

Tobias-Fischer, 2023-04-19:
This was a tricky one. I just triggered new builds of a few packages which should resolve this issue :)

closed 2023-04-19 by Tobias-Fischer

yinchuan, 2023-04-21:
Thanks. I tested by creating a whole env following the instructions in README.md. `rqt_graph` works. I can confirm the issues has been fixed.