RoboStack/ros-noetic #348: Regarding installing gmapping

gokulhari, 2023-03-04:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

I am trying to installing turtlebot3 with gmapping. Platform: Ubuntu 22.04. I have been using mamba with robostack with no problems so far.

I did this to get turtlebot3 running:
```
cd ~/catkin_ws/src/
git clone https://github.com/ROBOTIS-GIT/turtlebot3_msgs.git
git clone https://github.com/ROBOTIS-GIT/turtlebot3.git
git clone git@github.com:ROBOTIS-GIT/turtlebot3_simulations.git
cd ../
catkin_make
```
So now it runs fine when I do "roslaunch turtlebot3_gazebo turtlebot3_world.launch". No problems until here. 

Now I want to get gmapping installed.  I tried "mamba install ros-neotic-gmapping". That did not work. So next, as similar to above, I cloned slam_gmapping, then it told me that I need openslam_gmapping. So I downloaded openslam gmapping, did a cmake, and make install into catkin_workspace/src/ . Now it almost compiles fine, but shows some issues. I have attached the catkin_make logs here. Any help? How do I get gmapping installed with robostack?

CMakeError.log:
```
Performing C SOURCE FILE Test CMAKE_HAVE_LIBC_PTHREAD failed with the following output:
Change Dir: /home/gokul/catkin_ws/build/CMakeFiles/CMakeScratch/TryCompile-9AijE2

Run Build Command(s):/usr/bin/gmake -f Makefile cmTC_fa6a0/fast && /usr/bin/gmake  -f CMakeFiles/cmTC_fa6a0.dir/build.make CMakeFiles/cmTC_fa6a0.dir/build
gmake[1]: Entering directory '/home/gokul/catkin_ws/build/CMakeFiles/CMakeScratch/TryCompile-9AijE2'
Building C object CMakeFiles/cmTC_fa6a0.dir/src.c.o
/home/gokul/mambaforge/envs/robostackenv/bin/x86_64-conda-linux-gnu-cc -DCMAKE_HAVE_LIBC_PTHREAD  -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /home/gokul/mambaforge/envs/robostackenv/include  -o CMakeFiles/cmTC_fa6a0.dir/src.c.o -c /home/gokul/catkin_ws/build/CMakeFiles/CMakeScratch/TryCompile-9AijE2/src.c
Linking C executable cmTC_fa6a0
/home/gokul/mambaforge/envs/robostackenv/bin/cmake -E cmake_link_script CMakeFiles/cmTC_fa6a0.dir/link.txt --verbose=1
/home/gokul/mambaforge/envs/robostackenv/bin/x86_64-conda-linux-gnu-cc -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem /home/gokul/mambaforge/envs/robostackenv/include  -Wl,-O2 -Wl,--sort-common -Wl,--as-needed -Wl,-z,relro -Wl,-z,now -Wl,--disable-new-dtags -Wl,--gc-sections -Wl,--allow-shlib-undefined -Wl,-rpath,/home/gokul/mambaforge/envs/robostackenv/lib -Wl,-rpath-link,/home/gokul/mambaforge/envs/robostackenv/lib -L/home/gokul/mambaforge/envs/robostackenv/lib  -rdynamic CMakeFiles/cmTC_fa6a0.dir/src.c.o -o cmTC_fa6a0 
/home/gokul/mambaforge/envs/robostackenv/bin/../lib/gcc/x86_64-conda-linux-gnu/11.3.0/../../../../x86_64-conda-linux-gnu/bin/ld: CMakeFiles/cmTC_fa6a0.dir/src.c.o: in function `main':
src.c:(.text.startup.main+0x24): undefined reference to `pthread_create'
/home/gokul/mambaforge/envs/robostackenv/bin/../lib/gcc/x86_64-conda-linux-gnu/11.3.0/../../../../x86_64-conda-linux-gnu/bin/ld: src.c:(.text.startup.main+0x2e): undefined reference to `pthread_detach'
/home/gokul/mambaforge/envs/robostackenv/bin/../lib/gcc/x86_64-conda-linux-gnu/11.3.0/../../../../x86_64-conda-linux-gnu/bin/ld: src.c:(.text.startup.main+0x38): undefined reference to `pthread_cancel'
/home/gokul/mambaforge/envs/robostackenv/bin/../lib/gcc/x86_64-conda-linux-gnu/11.3.0/../../../../x86_64-conda-linux-gnu/bin/ld: src.c:(.text.startup.main+0x44): undefined reference to `pthread_join'
/home/gokul/mambaforge/envs/robostackenv/bin/../lib/gcc/x86_64-conda-linux-gnu/11.3.0/../../../../x86_64-conda-linux-gnu/bin/ld: src.c:(.text.startup.main+0x50): undefined reference to `pthread_atfork'
collect2: error: ld returned 1 exit status
gmake[1]: *** [CMakeFiles/cmTC_fa6a0.dir/build.make:99: cmTC_fa6a0] Error 1
gmake[1]: Leaving directory '/home/gokul/catkin_ws/build/CMakeFiles/CMakeScratch/TryCompile-9AijE2'
gmake: *** [Makefile:127: cmTC_fa6a0/fast] Error 2

Source file was:
#include <pthread.h>

static void* test_func(void* data)
{
  return data;
}

int main(void)
{
  pthread_t thread;
  pthread_create(&thread, NULL, test_func, NULL);
  pthread_detach(thread);
  pthread_cancel(thread);
  pthread_join(thread, NULL);
  pthread_atfork(NULL, NULL, NULL);
  pthread_exit(NULL);

  return 0;
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/gokul/catkin_ws/build/CMakeFiles/CMakeScratch/TryCompile-MgfdPX

Run Build Command(s):/usr/bin/gmake -f Makefile cmTC_06426/fast && /usr/bin/gmake  -f CMakeFiles/cmTC_06426.dir/build.make CMakeFiles/cmTC_06426.dir/build
gmake[1]: Entering directory '/home/gokul/catkin_ws/build/CMakeFiles/CMakeScratch/TryCompile-MgfdPX'
[... 630 lines cut ...]
 populated config files : /home/gokul/mambaforge/.condarc
                          /home/gokul/.condarc
          conda version : 22.9.0
    conda-build version : not installed
         python version : 3.10.8.final.0
       virtual packages : __cuda=12.0=0
                          __linux=5.19.0=0
                          __glibc=2.35=0
                          __unix=0=0
                          __archspec=1=x86_64
       base environment : /home/gokul/mambaforge  (writable)
      conda av data dir : /home/gokul/mambaforge/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/gokul/mambaforge/pkgs
                          /home/gokul/.conda/pkgs
       envs directories : /home/gokul/mambaforge/envs
                          /home/gokul/.conda/envs
               platform : linux-64
             user-agent : conda/22.9.0 requests/2.28.1 CPython/3.10.8 Linux/5.19.0-32-generic ubuntu/22.04.1 glibc/2.35
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False
```

traversaro, 2023-03-04:
Interesting, I have related problems in https://github.com/conda-forge/librealsense-feedstock/issues/29 .

gokulhari, 2023-03-04:
I got this resolved.

Just one change to openslam_gmapping - they used the flag -lpthread in the place of -pthread. So I made that change there.

Still catkin_make tells me that there is some error with the steps I laid out before + this one step. But when I actually launch turtlebot3 with gmapping, everything works fine. Looking at the logs and the flag -pthread there is nothing inconsistent. Catkin_make tells me there is an error, but I guess I can live with that.

closed 2023-03-04 by gokulhari