RoboStack/ros-noetic #349: unable to install ros-noetic-visp, ros-noetic-usb-cam and zbar

barrycxg, 2023-03-13:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

I am trying to install ros-noetic-visp, ros-noetic-usb-cam and zbar with command

`mamba install ros-noetic-visp ros-noetic-usb-cam zbar`

and get error:

```
Looking for: ['ros-noetic-visp', 'ros-noetic-usb-cam', 'zbar']

conda-forge/linux-64     Using cache
conda-forge/noarch       Using cache
pkgs/main/linux-64       [====================] (00m:00s) No change
pkgs/main/noarch         [====================] (00m:00s) No change
pkgs/r/linux-64          [====================] (00m:00s) No change
pkgs/r/noarch            [====================] (00m:00s) No change
robostack-staging/noarch [====================] (00m:00s) No change
robostack-staging/linux- [====================] (00m:00s) No change
Encountered problems while solving.
Problem: nothing provides requested ros-noetic-visp
Problem: nothing provides requested ros-noetic-usb-cam
```

But according to the [documentation](https://robostack.github.io/noetic.html), `ros-noetic-visp` should be available. 
Any ideas what I might be doing wrong?

Thanks!

### Installed packages

```shell
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
adwaita-icon-theme        43                       unix_0    conda-forge
alsa-lib                  1.2.8                h166bdaf_0    conda-forge
aom                       3.5.0                h27087fc_0    conda-forge
apr                       1.7.0                h7f98852_5    conda-forge
argcomplete               2.1.1              pyhd8ed1ab_0    conda-forge
assimp                    5.2.5                hf40c2ba_0    conda-forge
atk-1.0                   2.38.0               hd4edc92_1    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     22.2.0             pyh71513ae_0    conda-forge
bcrypt                    3.2.2            py39hb9d737c_1    conda-forge
binutils                  2.39                 hdd6e379_1    conda-forge
binutils_impl_linux-64    2.39                 he00db2b_1    conda-forge
binutils_linux-64         2.39                h5fc0e48_11    conda-forge
boost                     1.78.0           py39h7c9e3ff_4    conda-forge
boost-cpp                 1.78.0               h75c5d50_1    conda-forge
brotli                    1.0.9                h166bdaf_8    conda-forge
brotli-bin                1.0.9                h166bdaf_8    conda-forge
bullet                    3.21                 hcca971b_4    conda-forge
bullet-cpp                3.21                 h4661b88_4    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.18.1               h7f98852_0    conda-forge
c-compiler                1.5.2                h0b41bf4_0    conda-forge
ca-certificates           2023.01.10           h06a4308_0  
cairo                     1.16.0            ha61ee94_1014    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.3              pyhd8ed1ab_0    conda-forge
certifi                   2022.12.7          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1           py39he91dace_3    conda-forge
cmake                     3.25.3               h077f3f9_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.4.2              pyhd8ed1ab_0    conda-forge
[... 542 lines cut ...]
                          __linux=5.8.0=0
                          __glibc=2.31=0
                          __unix=0=0
                          __archspec=1=x86_64
       base environment : /home/barry/anaconda3  (writable)
      conda av data dir : /home/barry/anaconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/linux-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://repo.anaconda.com/pkgs/main/linux-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/linux-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /home/barry/anaconda3/pkgs
                          /home/barry/.conda/pkgs
       envs directories : /home/barry/anaconda3/envs
                          /home/barry/.conda/envs
               platform : linux-64
             user-agent : conda/4.12.0 requests/2.25.1 CPython/3.8.8 Linux/5.8.0-63-generic ubuntu/20.04.5 glibc/2.31
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False
```

traversaro, 2023-03-13:
> But according to the [documentation](https://robostack.github.io/noetic.html), `ros-noetic-visp` should be available.

If you seen in the table, near `ros-noetic-visp` for all platforms it is listed ❌ , meaning that it is not available for any platform. If you want to try to add it, feel free to open a PR following the instructions in https://github.com/RoboStack/ros-noetic/blob/master/CONTRIBUTING.md,thanks!

Tobias-Fischer, 2023-06-25:
ros-noetic-usb-cam is now available for linux systems.