RoboStack/ros-noetic #350: Communication errors between nodes on MacOS

sonelu, 2023-03-24:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Using the latest packages from robostack-staging (as suggested in the new install instructions) into a fresh environment `ros-noetic-desktop-full`.

I'm using a URDF description and trying to visualise in RViz. The package is located [here](https://github.com/SYNTHesse-GIT/mh5_common).

Launching the `rviz.launch` starts the nodes but the robot fails to render:

```
(ros1) Alex3:catkin_ws Alex$ roslaunch mh5_description rviz.launch gui:=true
... logging to /Users/Alex/.ros/log/41254774-ca24-11ed-b7e2-8c8590d4cfc8/roslaunch-Alex3.local-38565.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt

started roslaunch server http://Alex3.local:56119/

SUMMARY
========

PARAMETERS
 * /joint_state_publisher/num_rows: 6
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.15

NODES
  /
    joint_state_publisher (joint_state_publisher_gui/joint_state_publisher_gui)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [38570]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 41254774-ca24-11ed-b7e2-8c8590d4cfc8
process[rosout-1]: started with pid [38571]
started core service [/rosout]
process[joint_state_publisher-2]: started with pid [38572]
process[robot_state_publisher-3]: started with pid [38573]
process[rviz-4]: started with pid [38574]
```

[image]

No errors are issued.

On closer investigation the problem seems that the `roscpp`-based nodes fail to communicate.

For instance:

```
(ros1) Alex3:catkin_ws Alex$ rosnode list
/joint_state_publisher
/robot_state_publisher
/rosout
/rviz
```

Checking the `/joint_state_publisher` node (this is a `rospy` node) seems to be fine:
```
(ros1) Alex3:catkin_ws Alex$ rosnode info /joint_state_publisher 
--------------------------------------------------------------------------------
Node [/joint_state_publisher]
Publications: 
 * /joint_states [sensor_msgs/JointState]
[... 600 lines cut ...]
       base environment : /Users/Alex/opt/miniconda3  (writable)
      conda av data dir : /Users/Alex/opt/miniconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack/osx-64
                          https://conda.anaconda.org/robostack/noarch
                          https://conda.anaconda.org/conda-forge/osx-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://conda.anaconda.org/pytorch/osx-64
                          https://conda.anaconda.org/pytorch/noarch
                          https://conda.anaconda.org/pypi/osx-64
                          https://conda.anaconda.org/pypi/noarch
                          https://repo.anaconda.com/pkgs/main/osx-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/osx-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /Users/Alex/opt/miniconda3/pkgs
                          /Users/Alex/.conda/pkgs
       envs directories : /Users/Alex/opt/miniconda3/envs
                          /Users/Alex/.conda/envs
               platform : osx-64
             user-agent : conda/23.1.0 requests/2.28.2 CPython/3.8.16 Darwin/21.6.0 OSX/10.16
                UID:GID : 502:20
             netrc file : None
           offline mode : False
```

sonelu, 2023-03-24:
Yes, it did work a few months back (it is mentioned in the text but it is drowned by all the listing...). Which other environment details would you need (the `mamba list` and `mamba info` is included above)?

I'm on Monterey 12.6.3, MacBook Pro 13  late 2018, Intel i7.

Tobias-Fischer, 2023-03-24:
The mamba list of the old environment where it worked would be great.

Is it possible to reproduce this bug with e.g. the turtlebot packages so it’s easier for me to debug?

sonelu, 2023-03-25:
I was trying to recover the previous version before the upgrade with `conda install --revision xx` but I get an error about missing `robostack-experimental/osx-64::pyqt5-sip==12.9.0=py39h66f9765_1` (probably at one point we were fixing the pyqt problems and that was left in the environment). I will try to reconstruct it from the `history` file that is in the environment folder.

In terms of replication, in my case if I just run `roscore` and then try to communicate with `/rosout` it fails:

```
(rsnew) Alex3:~ Alex$ roscore
... logging to /Users/Alex/.ros/log/5bd4d152-cb23-11ed-8cf2-8c8590d4cfc8/roslaunch-Alex3.local-53898.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt

started roslaunch server http://Alex3.local:55729/
ros_comm version 1.15.14

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES

auto-starting new master
process[master]: started with pid [53902]
ROS_MASTER_URI=http://Alex3.local:11311/

setting /run_id to 5bd4d152-cb23-11ed-8cf2-8c8590d4cfc8
process[rosout-1]: started with pid [53903]
started core service [/rosout]
```
And in another terminal (same environment activated):
```
(rsnew) Alex3:~ Alex$ rosnode list
/rosout
(rsnew) Alex3:~ Alex$ rosnode info /rosout 
--------------------------------------------------------------------------------
Node [/rosout]
Publications: 
 * /rosout_agg [rosgraph_msgs/Log]

Subscriptions: 
 * /rosout [unknown type]

Services: 
 * /rosout/get_loggers
 * /rosout/set_logger_level

contacting node http://Alex3.local:55818/ ...
ERROR: Communication with node[http://Alex3.local:55818/] failed!
(rsnew) Alex3:~ Alex$ rosnode ping /rosout 
rosnode: node is [/rosout]
pinging /rosout with a timeout of 3.0s
connection to [/rosout] failed
```

Logs after running:

<details>
  <summary>master.log</summary>
  
 ```
[rosmaster.main][INFO] 2023-03-25 15:40:29,390: initialization complete, waiting for shutdown
[rosmaster.main][INFO] 2023-03-25 15:40:29,390: Starting ROS Master Node
[xmlrpc][INFO] 2023-03-25 15:40:29,391: XML-RPC server binding to 0.0.0.0:11311
[xmlrpc][INFO] 2023-03-25 15:40:29,392: Started XML-RPC server [http://Alex3.local:11311/]
[rosmaster.master][INFO] 2023-03-25 15:40:29,392: Master initialized: port[11311], uri[http://Alex3.local:11311/]
[xmlrpc][INFO] 2023-03-25 15:40:29,392: xml rpc node: starting XML-RPC server
[rosmaster.master][INFO] 2023-03-25 15:40:29,448: +PARAM [/run_id] by /roslaunch
[rosmaster.master][INFO] 2023-03-25 15:40:29,451: +PARAM [/roslaunch/uris/host_alex3_local__55729] by /roslaunch
[rosmaster.master][INFO] 2023-03-25 15:40:30,168: +PARAM [/rosversion] by /roslaunch
[... 104 lines cut ...]
[roslaunch.pmon][INFO] 2023-03-25 15:49:44,364: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, stopped daemon 123145553358848)>
[roslaunch][INFO] 2023-03-25 15:49:44,364: ... shutting down processing monitor complete
[roslaunch][INFO] 2023-03-25 15:49:44,365: done
[roslaunch.pmon][INFO] 2023-03-25 15:49:44,365: ProcessMonitor.shutdown <ProcessMonitor(ProcessMonitor-1, stopped daemon 123145553358848)>
[rospy.core][INFO] 2023-03-25 15:49:44,365: signal_shutdown [atexit]
```
</details>

<details>
  <summary>rosout-1-stdout.log </summary>
```
logging to /Users/Alex/.ros/log/5bd4d152-cb23-11ed-8cf2-8c8590d4cfc8/rosout.log
re-publishing aggregated messages to /rosout_agg
subscribed to /rosout
```
</details>

<details>
  <summary>rosout.log </summary>
```
1679758830.324426000  Node Startup
```
</details>

If it works for you, then try also to activate the firewall and see if that makes any difference.

Tobias-Fischer, 2023-03-26:
That certainly works for me. Can you try replicating this environment?

```
(robo-new-x86) fischert@VXTGFM46Y6 ~ % cat test_env.yaml 
name: robo-new-x86
channels:
  - robostack-staging
  - conda-forge
dependencies:
  - apr=1.7.0=h0d85af4_5
  - assimp=5.2.5=hd9e13b6_0
  - atk-1.0=2.38.0=h1d18e73_1
  - bcrypt=3.2.2=py39ha30fb19_1
  - boost=1.78.0=py39h6dc771e_4
  - boost-cpp=1.78.0=h8b082ac_1
  - bzip2=1.0.8=h0d85af4_4
  - c-ares=1.18.1=h0d85af4_0
  - ca-certificates=2022.12.7=h033912b_0
  - cairo=1.16.0=h904041c_1014
  - catkin_pkg=0.5.2=pyhd8ed1ab_0
  - cffi=1.15.1=py39h131948b_3
  - cmake=3.25.2=h4032537_0
  - console_bridge=1.0.2=hbb4e6a2_1
  - cryptography=39.0.1=py39hbeae22c_0
  - defusedxml=0.7.1=pyhd8ed1ab_0
  - distro=1.8.0=pyhd8ed1ab_0
  - docutils=0.19=py39h6e9494a_1
  - eigen=3.4.0=h940c156_0
  - empy=3.3.4=pyh9f0ad1d_1
  - expat=2.5.0=hf0c8a7f_0
  - font-ttf-dejavu-sans-mono=2.37=hab24e00_0
  - font-ttf-inconsolata=3.000=h77eed37_0
  - font-ttf-source-code-pro=2.038=h77eed37_0
  - font-ttf-ubuntu=0.83=hab24e00_0
  - fontconfig=2.14.2=h5bb23bf_0
  - fonts-conda-ecosystem=1=0
  - fonts-conda-forge=1=0
  - freeimage=3.18.0=h4fbe5d5_12
  - freetype=2.12.1=h3f81eb7_1
  - fribidi=1.0.10=hbcb3906_0
  - gdk-pixbuf=2.42.10=he00a9fc_0
  - gettext=0.21.1=h8a4c099_0
  - giflib=5.2.1=hbcb3906_2
  - glib=2.74.1=hbc0c0cd_1
  - glib-tools=2.74.1=hbc0c0cd_1
  - gmock=1.12.1=hb8565cd_0
  - gmp=6.2.1=h2e338ed_0
  - gpgme=1.18.0=hbbd2c75_0
  - graphite2=1.3.13=h2e338ed_1001
  - graphviz=6.0.2=hc51f7b9_0
  - gst-plugins-base=1.22.0=h37e1711_0
  - gstreamer=1.22.0=h1d18e73_0
  - gtest=1.12.1=hb8565cd_0
  - gtk2=2.24.33=h7c1209e_2
  - gts=0.7.6=hccb3bdf_2
  - harfbuzz=6.0.0=h08f8713_0
  - icu=70.1=h96cf925_0
  - imath=3.1.6=hbc0c0cd_1
  - jpeg=9e=hb7f2c08_3
  - jxrlib=1.1=h35c211d_2
  - krb5=1.20.1=h049b76e_0
  - lcms2=2.14=h29502cd_1
  - lerc=4.0.0=hb486fe8_0
  - libapr=1.7.0=h0d85af4_5
  - libapriconv=1.2.2=h0d85af4_5
  - libaprutil=1.6.1=h664449b_5
  - libassuan=2.5.5=he49afe7_0
  - libblas=3.9.0=16_osx64_openblas
  - libcblas=3.9.0=16_osx64_openblas
  - libclang=13.0.1=default_he082bbe_0
[... 191 lines cut ...]
  - zziplib=0.13.69=h5dbffcc_1
prefix: /Users/fischert/mambaforge/envs/robo-new-x86
```

Result:
```
(base) fischert@VXTGFM46Y6 ~ % conda activate robo-new-x86 
(robo-new-x86) fischert@VXTGFM46Y6 ~ % rosnode list
/rosout
(robo-new-x86) fischert@VXTGFM46Y6 ~ % rosnode info /rosout 
--------------------------------------------------------------------------------
Node [/rosout]
Publications: 
 * /rosout_agg [rosgraph_msgs/Log]

Subscriptions: 
 * /rosout [unknown type]

Services: 
 * /rosout/get_loggers
 * /rosout/set_logger_level

contacting node http://127.0.0.1:52648/ ...
Pid: 79148
```

Tobias-Fischer, 2023-03-28:
Closing as not replicable - feel free to reopen if you have more questions.

closed 2023-03-28 by Tobias-Fischer