RoboStack/ros-noetic #354: vrpn_client_ros support

adamdai, 2023-03-30:
Hello, any plans to provide support for vrpn_client_ros (for noetic) soon? Thanks

Tobias-Fischer, 2023-03-30:
It looks like it's a matter of bringing https://github.com/vrpn/vrpn/ to conda-forge first. Would you be happy to help in the process?

adamdai, 2023-07-04:
Hi Tobias - sorry for the late response, but yes I would be willing to help. I don't have experience with this though, would it be similar to the steps as outlined in this guide?: https://blog.gishub.org/how-to-publish-a-python-package-on-conda-forge

Tobias-Fischer, 2023-07-04:
Hi - you can look at an example recipe here:
https://github.com/conda-forge/staged-recipes/blob/main/recipes/example/meta.yaml

There are lots of other examples in this repository. I would not recommend following the blog article you linked, as it specifically deals with Python packages, but vrpn is a C++ package.

adamdai, 2023-07-05:
Got it, I'll look into it. Thanks for the help!