RoboStack/ros-noetic #359: Rviz not opening

angsylvest, 2023-04-26:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

When I followed the documentation and attempted to run rviz, I get the following error. I have a MacOS Catalina Version 10.15.7.
[image]

### Installed packages

```shell
# Name                    Version                   Build  Channel
adwaita-icon-theme        43                       unix_0    conda-forge
aom                       3.5.0                hf0c8a7f_0    conda-forge
apr                       1.7.0                h0d85af4_5    conda-forge
assimp                    5.2.5                hd9e13b6_0    conda-forge
atk-1.0                   2.38.0               h1d18e73_1    conda-forge
bcrypt                    3.2.2            py39ha30fb19_1    conda-forge
boost                     1.78.0           py39h6dc771e_4    conda-forge
boost-cpp                 1.78.0               h31500c2_2    conda-forge
brotli                    1.0.9                hb7f2c08_8    conda-forge
brotli-bin                1.0.9                hb7f2c08_8    conda-forge
bzip2                     1.0.8                h0d85af4_4    conda-forge
c-ares                    1.18.1               h0d85af4_0    conda-forge
ca-certificates           2022.12.7            h033912b_0    conda-forge
cairo                     1.16.0            h904041c_1014    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.3              pyhd8ed1ab_0    conda-forge
certifi                   2022.12.7          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1           py39h131948b_3    conda-forge
cmake                     3.26.3               h4032537_0    conda-forge
console_bridge            1.0.2                hbb4e6a2_1    conda-forge
contourpy                 1.0.7            py39h92daf61_0    conda-forge
coreutils                 9.3                  hb7f2c08_0    conda-forge
cryptography              40.0.2           py39hbeae22c_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.8.0              pyhd8ed1ab_0    conda-forge
docutils                  0.19             py39h6e9494a_1    conda-forge
eigen                     3.4.0                h940c156_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.5.0                hf0c8a7f_1    conda-forge
ffmpeg                    5.1.2           gpl_h8b4fe81_106    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.2               h5bb23bf_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.39.3           py39ha30fb19_0    conda-forge
freeimage                 3.18.0              h4fbe5d5_12    conda-forge
freetype                  2.12.1               h3f81eb7_1    conda-forge
fribidi                   1.0.10               hbcb3906_0    conda-forge
gdk-pixbuf                2.42.10              he00a9fc_0    conda-forge
gettext                   0.21.1               h8a4c099_0    conda-forge
giflib                    5.2.1                hb7f2c08_3    conda-forge
glib                      2.76.2               h7d26f99_0    conda-forge
glib-tools                2.76.2               h7d26f99_0    conda-forge
gmock                     1.13.0               h694c41f_1    conda-forge
gmp                       6.2.1                h2e338ed_0    conda-forge
gnutls                    3.7.8                h207c4f0_0    conda-forge
gpgme                     1.18.0               hbbd2c75_0    conda-forge
graphite2                 1.3.13            h2e338ed_1001    conda-forge
graphviz                  6.0.2                hc51f7b9_0    conda-forge
gst-plugins-base          1.22.0               h37e1711_2    conda-forge
gstreamer                 1.22.0               h1d18e73_2    conda-forge
gtest                     1.13.0               h1c7c39f_1    conda-forge
gtk2                      2.24.33              h7c1209e_2    conda-forge
[... 371 lines cut ...]
         python version : 3.9.16.final.0
       virtual packages : __archspec=1=x86_64
                          __osx=10.15.7=0
                          __unix=0=0
       base environment : /Users/angelsylvester/miniconda3  (writable)
      conda av data dir : /Users/angelsylvester/miniconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack/osx-64
                          https://conda.anaconda.org/robostack/noarch
                          https://conda.anaconda.org/conda-forge/osx-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://repo.anaconda.com/pkgs/main/osx-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/osx-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /Users/angelsylvester/miniconda3/pkgs
                          /Users/angelsylvester/.conda/pkgs
       envs directories : /Users/angelsylvester/miniconda3/envs
                          /Users/angelsylvester/.conda/envs
               platform : osx-64
             user-agent : conda/23.3.1 requests/2.28.2 CPython/3.9.16 Darwin/19.6.0 OSX/10.15.7
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2023-04-27:
I just triggered a rebuild of `rospack` which seems the origin of this issue. Could you please try again in an hour or so, after `mamba upgrade ros-noetic-rospack`?

Tobias-Fischer, 2023-04-27:
Just tried with the latest packages - this should work fine, feel free to reopen if not.

closed 2023-04-27 by Tobias-Fischer

angsylvest, 2023-04-27:
> Just tried with the latest packages - this should work fine, feel free to reopen if not.

Yes, that fixed it! Thanks!