RoboStack/ros-noetic #360: OSX: Can't open RViz or play a rosbag file

alex-lechner, 2023-05-05:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

I'm using Mac OSX version 13.3.1 (a) and I have installed everything in accordance with the installation steps in the README.

When I'm executing the following statements in the command line: 

```sh
rviz
```

or

```sh
rosbag play -l my_bag.bag
```

I receive the following error

```sh
dyld[6102]: symbol not found in flat namespace '__Py_NoneStruct'
```

### Installed packages

```shell
mamba list

returns the following:

```
# packages in environment at /Users/me/miniconda3/envs/robostackenv:
#
# Name                    Version                   Build  Channel
adwaita-icon-theme        43                       unix_0    conda-forge
aom                       3.5.0                hf0c8a7f_0    conda-forge
apr                       1.7.0                h0d85af4_5    conda-forge
assimp                    5.2.5                hd9e13b6_0    conda-forge
atk-1.0                   2.38.0               h1d18e73_1    conda-forge
bcrypt                    3.2.2            py39ha30fb19_1    conda-forge
boost                     1.78.0           py39h6dc771e_4    conda-forge
boost-cpp                 1.78.0               h31500c2_2    conda-forge
brotli                    1.0.9                hb7f2c08_8    conda-forge
brotli-bin                1.0.9                hb7f2c08_8    conda-forge
bzip2                     1.0.8                h0d85af4_4    conda-forge
c-ares                    1.18.1               h0d85af4_0    conda-forge
ca-certificates           2023.01.10           hecd8cb5_0
cairo                     1.16.0            h904041c_1014    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
certifi                   2022.12.7          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1           py39h131948b_3    conda-forge
cmake                     3.26.3               h4032537_0    conda-forge
console_bridge            1.0.2                hbb4e6a2_1    conda-forge
contourpy                 1.0.7            py39h92daf61_0    conda-forge
cryptography              40.0.2           py39hbeae22c_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.8.0              pyhd8ed1ab_0    conda-forge
docutils                  0.19             py39h6e9494a_1    conda-forge
eigen                     3.4.0                h940c156_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.5.0                hf0c8a7f_1    conda-forge
ffmpeg                    5.1.2           gpl_h8b4fe81_106    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
[... 417 lines cut ...]
 populated config files :
          conda version : 23.3.1
    conda-build version : not installed
         python version : 3.10.10.final.0
       virtual packages : __archspec=1=x86_64
                          __osx=13.3.1=0
                          __unix=0=0
       base environment : /Users/me/miniconda3  (writable)
      conda av data dir : /Users/me/miniconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://repo.anaconda.com/pkgs/main/osx-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/osx-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /Users/me/miniconda3/pkgs
                          /Users/me/.conda/pkgs
       envs directories : /Users/me/miniconda3/envs
                          /Users/me/.conda/envs
               platform : osx-64
             user-agent : conda/23.3.1 requests/2.28.1 CPython/3.10.10 Darwin/22.4.0 OSX/13.3.1
                UID:GID : 501:20
             netrc file : /Users/me/.netrc
           offline mode : False
```
```

Tobias-Fischer, 2023-05-05:
Can you please post the full error message?

Tobias-Fischer, 2023-05-05:
Actually .. mamba upgrade ros-noetic-rospack will fix this issue. Feel free to reopen if not.

closed 2023-05-05 by Tobias-Fischer

alex-lechner, 2023-05-05:
I did `mamba upgrade ros-noetic-rospack`

```sh
                  __    __    __    __
                 /  \  /  \  /  \  /  \
                /    \/    \/    \/    \
███████████████/  /██/  /██/  /██/  /████████████████████████
              /  / \   / \   / \   / \  \____
             /  /   \_/   \_/   \_/   \    o \__,
            / _/                       \_____/  `
            |/
        ███╗   ███╗ █████╗ ███╗   ███╗██████╗  █████╗
        ████╗ ████║██╔══██╗████╗ ████║██╔══██╗██╔══██╗
        ██╔████╔██║███████║██╔████╔██║██████╔╝███████║
        ██║╚██╔╝██║██╔══██║██║╚██╔╝██║██╔══██╗██╔══██║
        ██║ ╚═╝ ██║██║  ██║██║ ╚═╝ ██║██████╔╝██║  ██║
        ╚═╝     ╚═╝╚═╝  ╚═╝╚═╝     ╚═╝╚═════╝ ╚═╝  ╚═╝

        mamba (1.4.2) supported by @QuantStack

        GitHub:  https://github.com/mamba-org/mamba
        Twitter: https://twitter.com/QuantStack

█████████████████████████████████████████████████████████████

Looking for: ['ros-noetic-rospack']

warning  libmamba Cache file "/Users/me/miniconda3/pkgs/cache/7fb2ce72.json" was modified by another program
warning  libmamba Cache file "/Users/me/miniconda3/pkgs/cache/3e39a7aa.json" was modified by another program
warning  libmamba Cache file "/Users/me/miniconda3/pkgs/cache/dd44a73b.json" was modified by another program
warning  libmamba Cache file "/Users/me/miniconda3/pkgs/cache/4ea078d6.json" was modified by another program
pkgs/main/noarch                                   822.9kB @ 377.5kB/s  2.2s
pkgs/r/osx-64                                      820.9kB @ 365.3kB/s  2.3s
```

The error is still not fixed (neither for RViz nor rosbag). Unfortunately, this is the only error message that is printed on the console:

```sh
dyld[15988]: symbol not found in flat namespace '__Py_NoneStruct'
```

Tobias-Fischer, 2023-05-05:
You did not specify the robostack-staging channel. For debugging, please add the whole output of commands.

Tobias-Fischer, 2023-05-06:
So in other words, please `mamba upgrade ros-noetic-rospack -c conda-forge -c robostack-staging`

alex-lechner, 2023-05-06:
That fixed the issue! Thank you very much!

Tobias-Fischer, 2023-05-06:
Great