RoboStack/ros-noetic #361: Problem installing ros-noetic-pcl-ros

david2611, 2023-05-12:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

There seems to be some sort of issue when trying to install the `ros-noetic-pcl-ros` package.

After running `mamba install -c robostack -c robostack-experimental -c conda-forge -y ros-noetic-pcl-ros` I get the following output.
```
Could not solve for environment specs
Encountered problems while solving:
  - package ros-noetic-viz-1.5.0-py39hac30774_15 requires ros-distro-mutex 0.4.*, but none of the providers can be installed

The environment can't be solved, aborting the operation
```

My system should already have `ros-noetic-viz` installed (found in installed packages below).

I would appreciate any advice you might have on how I should resolve this issue

### Installed packages

```shell
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
adwaita-icon-theme        43                       unix_0    conda-forge
alsa-lib                  1.2.8                h166bdaf_0    conda-forge
aom                       3.5.0                h27087fc_0    conda-forge
apr                       1.7.0                h7f98852_5    conda-forge
assimp                    5.2.5                hf40c2ba_0    conda-forge
atk-1.0                   2.38.0               hd4edc92_1    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
bcrypt                    3.2.2            py39hb9d737c_1    conda-forge
binutils                  2.39                 hdd6e379_1    conda-forge
binutils_impl_linux-64    2.39                 he00db2b_1    conda-forge
binutils_linux-64         2.39                h5fc0e48_13    conda-forge
boost                     1.78.0           py39h7c9e3ff_4    conda-forge
boost-cpp                 1.78.0               h5adbc97_2    conda-forge
brotli                    1.0.9                h166bdaf_8    conda-forge
brotli-bin                1.0.9                h166bdaf_8    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.18.1               h7f98852_0    conda-forge
c-compiler                1.5.2                h0b41bf4_0    conda-forge
ca-certificates           2023.5.7             hbcca054_0    conda-forge
cairo                     1.16.0            ha61ee94_1014    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.3              pyhd8ed1ab_0    conda-forge
certifi                   2023.5.7           pyhd8ed1ab_0    conda-forge
cffi                      1.15.1           py39he91dace_3    conda-forge
cmake                     3.26.3               h077f3f9_0    conda-forge
compilers                 1.5.2                ha770c72_0    conda-forge
console_bridge            1.0.2                h924138e_1    conda-forge
contourpy                 1.0.7            py39h4b4f3f3_0    conda-forge
coreutils                 9.3                  h0b41bf4_0    conda-forge
cryptography              40.0.2           py39h079d5ae_0    conda-forge
cxx-compiler              1.5.2                hf52228f_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.8.0              pyhd8ed1ab_0    conda-forge
docutils                  0.20             py39hf3d152e_0    conda-forge
eigen                     3.4.0                h4bd325d_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.5.0                hcb278e6_1    conda-forge
ffmpeg                    5.1.2           gpl_h8dda1f0_106    conda-forge
fftw                      3.3.10          nompi_hc118613_107    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
[... 447 lines cut ...]
       user config file : /home/hal365/.condarc
 populated config files : /home/hal365/mambaforge/.condarc
          conda version : 22.9.0
    conda-build version : not installed
         python version : 3.10.8.final.0
       virtual packages : __cuda=12.1=0
                          __linux=5.15.0=0
                          __glibc=2.31=0
                          __unix=0=0
                          __archspec=1=x86_64
       base environment : /home/hal365/mambaforge  (writable)
      conda av data dir : /home/hal365/mambaforge/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/hal365/mambaforge/pkgs
                          /home/hal365/.conda/pkgs
       envs directories : /home/hal365/mambaforge/envs
                          /home/hal365/.conda/envs
               platform : linux-64
             user-agent : conda/22.9.0 requests/2.28.1 CPython/3.10.8 Linux/5.15.0-71-generic ubuntu/20.04.6 glibc/2.31
                UID:GID : 1284263:1284263
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2023-05-12:
Hi David, please use the `robostack-staging` channel, not the old `robostack` and `robostack-experimental` channels. Feel free to reopen if issue persists.

closed 2023-05-12 by Tobias-Fischer

david2611, 2023-05-12:
Excellent! Thank you. 

As perhaps a heads up on a future bug, the command I put in this issue was copied from what was run by `rosdep install`

Tobias-Fischer, 2023-05-12:
Ah, yes - can you please open a separate issue about that? I need to fix it.

Tobias-Fischer, 2023-05-12:
See https://github.com/conda-forge/rosdep-feedstock/pull/25 - please upgrade rosdep in a few hours.