RoboStack/ros-noetic #363: issue installing ros-noetic-rviz

VModugno, 2023-05-12:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Hi everyone,

I'm trying to install ros-noetic-rviz but I get this error regarding ogre dependencies:

pkgs/main/linux-64                                            No change
pkgs/main/noarch                                              No change
pkgs/r/linux-64                                               No change
robostack/noarch                                   No change
pkgs/r/noarch                                                 No change
robostack/linux-64                                            No change

Pinned packages:
  - python 3.8.*

Encountered problems while solving:
  - nothing provides ogre >=1.10.12,<1.11.0a0 needed by ros-noetic-rviz-1.14.10-py38h641b9ce_12

thank you for your help

### Installed packages

```shell
# packages in environment at /home/robohikeuser/anaconda3/envs/hori:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
alsa-lib                  1.2.8                h166bdaf_0    conda-forge
ampl-mp                   3.1.0             h2cc385e_1006    conda-forge
anaconda-client           1.11.2             pyhd8ed1ab_0    conda-forge
anaconda-project          0.11.1           py38h06a4308_0  
anyio                     3.5.0            py38h06a4308_0  
appdirs                   1.4.4              pyhd3eb1b0_0  
apr                       1.7.0                hf178f73_5  
argon2-cffi               21.3.0             pyhd3eb1b0_0  
argon2-cffi-bindings      21.2.0           py38h7f8727e_0  
asttokens                 2.0.5              pyhd3eb1b0_0  
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     22.1.0           py38h06a4308_0  
babel                     2.11.0           py38h06a4308_0  
backcall                  0.2.0              pyhd3eb1b0_0  
bcrypt                    3.2.0            py38h5eee18b_1  
beautifulsoup4            4.12.2           py38h06a4308_0  
bleach                    4.1.0              pyhd3eb1b0_0  
boltons                   23.0.0           py38h06a4308_0  
boost                     1.74.0           py38h2b96118_5    conda-forge
boost-cpp                 1.74.0               h75c5d50_8    conda-forge
brotli                    1.0.9                h5eee18b_7  
brotli-bin                1.0.9                h5eee18b_7  
brotlipy                  0.7.0           py38h27cfd23_1003  
bzip2                     1.0.8                h7b6447c_0  
ca-certificates           2023.01.10           h06a4308_0  
cairo                     1.16.0            ha61ee94_1014    conda-forge
casadi                    3.5.5           py38h668ca38_18    conda-forge
casadi_kin_dyn            0.0.1                         0    francesco_ruscelli
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
certifi                   2022.12.7        py38h06a4308_0  
cffi                      1.15.1           py38h5eee18b_3  
chardet                   4.0.0           py38h06a4308_1003  
charset-normalizer        2.0.4              pyhd3eb1b0_0  
click                     8.0.4            py38h06a4308_0  
clyent                    1.2.2                    py38_1  
cmake                     3.12.2               h52cb24c_0  
comm                      0.1.2            py38h06a4308_0  
[... 375 lines cut ...]
          conda version : 22.9.0
    conda-build version : 3.22.0
         python version : 3.9.13.final.0
       virtual packages : __cuda=12.0=0
                          __linux=5.19.0=0
                          __glibc=2.35=0
                          __unix=0=0
                          __archspec=1=x86_64
       base environment : /home/robohikeuser/anaconda3  (writable)
      conda av data dir : /home/robohikeuser/anaconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://repo.anaconda.com/pkgs/main/linux-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/linux-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /home/robohikeuser/anaconda3/pkgs
                          /home/robohikeuser/.conda/pkgs
       envs directories : /home/robohikeuser/anaconda3/envs
                          /home/robohikeuser/.conda/envs
               platform : linux-64
             user-agent : conda/22.9.0 requests/2.28.1 CPython/3.9.13 Linux/5.19.0-41-generic ubuntu/22.04.2 glibc/2.35
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False
```

traversaro, 2023-05-12:
Hi @VModugno! I am not sure which instructions are you following, but you need to use the `robostack-staging` channel, not the `robostack` one. Furthermore, it seems that you are using an `anaconda` install, instead of a `mambaforge` install, do you have any reason for using `anaconda` ? If you do not any specific reason, I suggest you do uninstall `anaconda` and install `mambaforge`.

If for any reason you neeed to use anaconda, no problem, but pay attention that you need to install robostack using the `robostack-staging` and `conda-forge` channel, and pay attention to avoid the use of the `defaults` channel.

VModugno, 2023-05-12:
Hi Silvio,
thank you for the quick reply! I'm currently trying to use the horizon package and I have been using the environment.yml from this [link](https://github.com/ADVRHumanoids/horizon). Should I switch the channel to robostack-staging and install at least all the robostack dependencies using mamba instead of conda? is that right?
thank you for your help

traversaro, 2023-05-12:
Actually it seems that you successfully installed ros via the old `robostack` channel, so that worked out fine. So, while you could experience problems with the old packages contaned in `robostack`, the  specific problem seems something else. Can you clarify the **exact** command are you executing to install ` ros-noetic-rviz` ? From the output it seems that for some reason it is not considering the `conda-forge` channel.

VModugno, 2023-05-12:
in order to install it I just did  mamba install -y ros-noetic-rviz -c robostack. I did try with conda as well and I tried by switching the channel to robostack-staging

traversaro, 2023-05-12:
The problem is that apparently in your environment conda-forge is not configured as a channel to use (a consequence of using anaconda instead of mambaforge, hence my original comment). If you want to install robostack software, you always need to also specify the conda-forge channel, as all the non-ROS dependencies come from there.

I think that:

~~~
mamba install -y ros-noetic-rviz -c robostack -c conda-forge 
~~~

should work fine in your exising environment (for new environments, please use `robostack-staging`).

VModugno, 2023-05-12:
I see thank you Silvio, everything is clear now. that's great because robostack is an amazing tool for what we are doing now and I have just started to use it. Right now I'm in contact with Francesco and he is instructing me on how to proceed for a proper installation of their package. Thank you again for your help!

traversaro, 2023-05-12:
If everything is solved, feel free to close the issue, thanks!

closed 2023-05-12 by Tobias-Fischer