RoboStack/ros-noetic #367: issue with rviz and ogre library

VModugno, 2023-05-16:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

im trying to use rviz to do some visualization. 
after starting rviz using a roslauncher the node is executed correctly but after few seconds I get this error:

[ERROR] [1684251330.889903928]: InternalErrorException: Could not load dynamic library /home/robohikeuser/mambaforge/envs/hori/lib/OGRE/RenderSystem_GL.  System Error: /home/robohikeuser/mambaforge/envs/hori/lib/OGRE/RenderSystem_GL.so.1.10.12: cannot open shared object file: No such file or directory in DynLib::load at /home/conda/feedstock_root/build_artifacts/ogre_1573024423694/work/OgreMain/src/OgreDynLib.cpp (line 111)

and after a while rviz die with this message:
[rviz-2] process has died [pid 694424, exit code -6, cmd /home/robohikeuser/mambaforge/envs/hori/lib/rviz/rviz -d /home/robohikeuser/git/horizon/horizon/examples/replay/rviz/spot.rviz __name:=rviz __log:=/home/robohikeuser/.ros/log/807ec30c-f3df-11ed-b7b4-f057a6666322/rviz-2.log].
log file: /home/robohikeuser/.ros/log/807ec30c-f3df-11ed-b7b4-f057a6666322/rviz-2*.log

I can confirm that RenderSystem_GL.so is missing in the the OGRE folder

### Installed packages

```shell
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                  2_kmp_llvm    conda-forge
adwaita-icon-theme        43                       unix_0    conda-forge
alsa-lib                  1.2.8                h166bdaf_0    conda-forge
ampl-mp                   3.1.0             h2cc385e_1006    conda-forge
anaconda-client           1.11.2             pyhd8ed1ab_0    conda-forge
anaconda-project          0.11.1             pyhd8ed1ab_0    conda-forge
anyio                     3.6.2              pyhd8ed1ab_0    conda-forge
apr                       1.7.0                h7f98852_5    conda-forge
argon2-cffi               21.3.0             pyhd8ed1ab_0    conda-forge
argon2-cffi-bindings      21.2.0           py39hb9d737c_3    conda-forge
assimp                    5.2.5                hf40c2ba_0    conda-forge
asttokens                 2.2.1              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               hd4edc92_1    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     23.1.0             pyh71513ae_1    conda-forge
babel                     2.12.1             pyhd8ed1ab_1    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                pyhd8ed1ab_3    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
bcrypt                    3.2.2            py39hb9d737c_1    conda-forge
beautifulsoup4            4.12.2             pyha770c72_0    conda-forge
bleach                    6.0.0              pyhd8ed1ab_0    conda-forge
boltons                   23.0.0             pyhd8ed1ab_0    conda-forge
boost                     1.78.0           py39h7c9e3ff_4    conda-forge
boost-cpp                 1.78.0               h5adbc97_2    conda-forge
brotli                    1.0.9                h166bdaf_8    conda-forge
brotli-bin                1.0.9                h166bdaf_8    conda-forge
brotlipy                  0.7.0           py39hb9d737c_1005    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.18.1               h7f98852_0    conda-forge
ca-certificates           2023.5.7             hbcca054_0    conda-forge
cairo                     1.16.0            ha61ee94_1014    conda-forge
casadi                    3.5.5           py39h83d0230_18    conda-forge
casadi-horizon            0.4.5                     dev_0    <develop>
casadi-kin-dyn            0.0.1                    pypi_0    pypi
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
certifi                   2023.5.7           pyhd8ed1ab_0    conda-forge
cffi                      1.15.1           py39he91dace_3    conda-forge
chardet                   5.1.0            py39hf3d152e_0    conda-forge
charset-normalizer        3.1.0              pyhd8ed1ab_0    conda-forge
click                     8.1.3           unix_pyhd8ed1ab_2    conda-forge
clyent                    1.2.2                      py_1    conda-forge
cmake                     3.26.3               h077f3f9_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
comm                      0.1.3              pyhd8ed1ab_0    conda-forge
conda                     23.3.1           py39hf3d152e_0    conda-forge
conda-build               3.24.0           py39hf3d152e_1    conda-forge
conda-pack                0.7.0              pyh6c4a22f_0    conda-forge
[... 422 lines cut ...]
       user config file : /home/robohikeuser/.condarc
 populated config files : /home/robohikeuser/mambaforge/.condarc
          conda version : 23.1.0
    conda-build version : not installed
         python version : 3.10.10.final.0
       virtual packages : __archspec=1=x86_64
                          __cuda=12.0=0
                          __glibc=2.35=0
                          __linux=5.19.0=0
                          __unix=0=0
       base environment : /home/robohikeuser/mambaforge  (writable)
      conda av data dir : /home/robohikeuser/mambaforge/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/robohikeuser/mambaforge/pkgs
                          /home/robohikeuser/.conda/pkgs
       envs directories : /home/robohikeuser/mambaforge/envs
                          /home/robohikeuser/.conda/envs
               platform : linux-64
             user-agent : conda/23.1.0 requests/2.28.2 CPython/3.10.10 Linux/5.19.0-41-generic ubuntu/22.04.2 glibc/2.35
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2023-05-16:
Does it work if you just run `rviz` without and of the other packages?

VModugno, 2023-05-17:
hi Tobias,
if i run rviz alone i get this:

[ INFO] [1684319667.316059401]: rviz version 1.14.19
[ INFO] [1684319667.316133131]: compiled against Qt version 5.15.6
[ INFO] [1684319667.316152102]: compiled against OGRE version 1.10.12 (Xalafu)
[ INFO] [1684319667.327687442]: Forcing OpenGl version 0.
[ERROR] [1684319667.568380511]: InternalErrorException: Could not load dynamic library /home/robohikeuser/mambaforge/envs/hori/lib/OGRE/RenderSystem_GL.  System Error: /home/robohikeuser/mambaforge/envs/hori/lib/OGRE/RenderSystem_GL.so.1.10.12: cannot open shared object file: No such file or directory in DynLib::load at /home/conda/feedstock_root/build_artifacts/ogre_1573024423694/work/OgreMain/src/OgreDynLib.cpp (line 111)
terminate called after throwing an instance of 'Ogre::InternalErrorException'
  what():  InternalErrorException: Could not load dynamic library /home/robohikeuser/mambaforge/envs/hori/lib/OGRE/RenderSystem_GL.  System Error: /home/robohikeuser/mambaforge/envs/hori/lib/OGRE/RenderSystem_GL.so.1.10.12: cannot open shared object file: No such file or directory in DynLib::load at /home/conda/feedstock_root/build_artifacts/ogre_1573024423694/work/OgreMain/src/OgreDynLib.cpp (line 111)
Aborted (core dumped)

traversaro, 2023-05-17:
I just created a clean environment (listed in the following) and `rviz` starts fine, and the ogre plugin is at the correct place:
~~~
(robostackenv) traversaro@IITICUBLAP257:~$ ls $CONDA_PREFIX/lib/OGRE | grep Render
RenderSystem_GL3Plus.so
RenderSystem_GL3Plus.so.1.10.12
RenderSystem_GL.so
RenderSystem_GL.so.1.10.12
~~~

Can you check if this happens in a clean environment, and the output of `ls $CONDA_PREFIX/lib/OGRE | grep Render` in your environment?

VModugno, 2023-05-17:
I made a fresh installation following your commands and in if I do `ls $CONDA_PREFIX/lib/OGRE | grep Render` I get this:
RenderSystem_GL3Plus.so
RenderSystem_GL3Plus.so.1.10.12
RenderSystem_GL.so
RenderSystem_GL.so.1.10.12

and rviz start correctly. In my environment (hori) if run again `ls $CONDA_PREFIX/lib/OGRE | grep Render` the output is empty.
in order to get rviz in my environment I just did `mamba install ros-noetic-rviz -c robostack-staging` did i make some mistakes there?

VModugno, 2023-05-17:
using an online service i searched for differences between the package installed in my environment and the one installed in the clean installation 
you can check it at this [link](https://www.diffchecker.com/NMq4vb1t/). My environment packages are on the left and the clean environment packages from Silvio's computer are on the right. It seems that I'm missing pyopengl among others

VModugno, 2023-05-17:
in order to install the hori environment i did `mamba env create -f environment.yml` (in order to attach it to the comment i had to rename it with the .txt extension) 
Aafter that I installed a pybind python package called casadi-kin-dyn (branch collision) which is [here](https://github.com/ADVRHumanoids/casadi_kin_dyn/tree/collision) by running `pip install . ` from the repository folder (the setup.py is not the one shown on the repo i have a working version on my computer that we will pull request soon hopefully).
and finally i have installed [horizon](https://github.com/ADVRHumanoids/horizon/tree/rhc) rhc branch in development mode by doing `pip install -e .` 
[environment.txt](https://github.com/RoboStack/ros-noetic/files/11497898/environment.txt)

VModugno, 2023-05-17:
ok by updating the ogre version now it is finally working! if I do `ls $CONDA_PREFIX/lib/OGRE | grep Render` in the horizon environment i get:
RenderSystem_GL3Plus.so
RenderSystem_GL3Plus.so.1.10.12
RenderSystem_GL.so
RenderSystem_GL.so.1.10.12

So now I'll see why I ended up with the wrong ogre version but now II know what I have to check.
Guys thank you a lot for your help!

traversaro, 2023-05-17:
Creating the env with `mamba env create -f environment.yml` the ogre installed is `*_10` . Just as a final check, can you try to re-create the environment from your ` environment.yml` file, and check if the correct ogre gets installed?

traversaro, 2023-05-17:
xref: https://github.com/conda-forge/ogre-feedstock/issues/15

traversaro, 2023-05-17:
Ok, feel free to close the issue, thanks!