RoboStack/ros-noetic #368: osx-arm64, noetic - Can't start rviz

fedor-chervinskii, 2023-05-17:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

I have faced an issue similar to #187 

I tried suggestions from that discussion, but they didn't help

I found a workaround by providing 
`export ROS_PACKAGE_PATH=<my_ros_env>/share`

now the rviz is starting but with quite some errors and very limited functionality:

```
[ INFO] [1684342178.538608000]: rviz version 1.14.20
[ INFO] [1684342178.538634000]: compiled against Qt version 5.15.8
[ INFO] [1684342178.538640000]: compiled against OGRE version 1.10.12 (Xalafu)
[ INFO] [1684342178.549314000]: Forcing OpenGl version 0.
[ INFO] [1684342181.583735000]: Stereo is NOT SUPPORTED
[ INFO] [1684342181.583768000]: OpenGL device: Apple M2
[ INFO] [1684342181.583776000]: OpenGl version: 2.1 (GLSL 1.2).
[ERROR] [1684342181.622185000]: PluginlibFactory: The plugin for class 'rviz/Orbit' failed to load.  Error: Could not find library corresponding to plugin rviz/Orbit. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1684342188.900927000]: PluginlibFactory: The plugin for class 'rviz/MoveCamera' failed to load.  Error: Could not find library corresponding to plugin rviz/MoveCamera. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1684342188.901439000]: PluginlibFactory: The plugin for class 'rviz/Interact' failed to load.  Error: Could not find library corresponding to plugin rviz/Interact. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1684342188.901752000]: PluginlibFactory: The plugin for class 'rviz/Select' failed to load.  Error: Could not find library corresponding to plugin rviz/Select. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1684342188.902023000]: PluginlibFactory: The plugin for class 'rviz/SetInitialPose' failed to load.  Error: Could not find library corresponding to plugin rviz/SetInitialPose. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1684342188.902282000]: PluginlibFactory: The plugin for class 'rviz/SetGoal' failed to load.  Error: Could not find library corresponding to plugin rviz/SetGoal. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1684342188.903982000]: PluginlibFactory: The plugin for class 'rviz/Grid' failed to load.  Error: Could not find library corresponding to plugin rviz/Grid. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1684342188.904518000]: PluginlibFactory: The plugin for class 'rviz/Interact' failed to load.  Error: Could not find library corresponding to plugin rviz/Interact. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1684342188.904621000]: PluginlibFactory: The plugin for class 'rviz/MoveCamera' failed to load.  Error: Could not find library corresponding to plugin rviz/MoveCamera. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1684342188.904716000]: PluginlibFactory: The plugin for class 'rviz/Select' failed to load.  Error: Could not find library corresponding to plugin rviz/Select. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1684342188.904817000]: PluginlibFactory: The plugin for class 'rviz/FocusCamera' failed to load.  Error: Could not find library corresponding to plugin rviz/FocusCamera. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1684342188.905565000]: PluginlibFactory: The plugin for class 'rviz/Measure' failed to load.  Error: Could not find library corresponding to plugin rviz/Measure. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1684342188.906163000]: PluginlibFactory: The plugin for class 'rviz/SetInitialPose' failed to load.  Error: Could not find library corresponding to plugin rviz/SetInitialPose. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1684342188.906354000]: PluginlibFactory: The plugin for class 'rviz/SetGoal' failed to load.  Error: Could not find library corresponding to plugin rviz/SetGoal. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1684342188.906484000]: PluginlibFactory: The plugin for class 'rviz/PublishPoint' failed to load.  Error: Could not find library corresponding to plugin rviz/PublishPoint. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1684342188.907180000]: PluginlibFactory: The plugin for class 'rviz/Orbit' failed to load.  Error: Could not find library corresponding to plugin rviz/Orbit. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
```

I have found this issue described already in [this blog post](https://barelywalking.com/?p=246), however I didn't figure out how do I fix it myself 

### Installed packages

```shell
# packages in environment at /Users/theo/miniconda3/envs/ros_env:
#
# Name                    Version                   Build  Channel
adwaita-icon-theme        43                       unix_0    conda-forge
aom                       3.5.0                h7ea286d_0    conda-forge
apr                       1.7.0                h9b22ae9_5    conda-forge
assimp                    5.2.5                h276577b_0    conda-forge
atk-1.0                   2.38.0               hcb7b3dd_1    conda-forge
bcrypt                    3.2.2            py39h02fc5c5_1    conda-forge
blosc                     1.21.4               hc338f07_0    conda-forge
boost                     1.78.0           py39h99de9ae_4    conda-forge
boost-cpp                 1.78.0               hf1d6563_2    conda-forge
brotli                    1.0.9                h1a8c8d9_8    conda-forge
brotli-bin                1.0.9                h1a8c8d9_8    conda-forge
bullet-cpp                3.21             py39hde7b980_4    conda-forge
bzip2                     1.0.8                h3422bc3_4    conda-forge
c-ares                    1.19.0               hb547adb_0    conda-forge
c-compiler                1.5.2                h5008568_0    conda-forge
ca-certificates           2023.5.7             hf0a4a13_0    conda-forge
cairo                     1.16.0            h73a0509_1014    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.3              pyhd8ed1ab_0    conda-forge
cctools                   973.0.1             hcbb26d4_13    conda-forge
cctools_osx-arm64         973.0.1             hef52d2f_13    conda-forge
[... 515 lines cut ...]
         python version : 3.10.11.final.0
       virtual packages : __archspec=1=arm64
                          __osx=13.0=0
                          __unix=0=0
       base environment : /Users/theo/miniconda3  (writable)
      conda av data dir : /Users/theo/miniconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack/osx-arm64
                          https://conda.anaconda.org/robostack/noarch
                          https://conda.anaconda.org/robostack-experimental/osx-arm64
                          https://conda.anaconda.org/robostack-experimental/noarch
                          https://conda.anaconda.org/robostack-staging/osx-arm64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /Users/theo/miniconda3/pkgs
                          /Users/theo/.conda/pkgs
       envs directories : /Users/theo/miniconda3/envs
                          /Users/theo/.conda/envs
               platform : osx-arm64
             user-agent : conda/23.3.1 requests/2.28.1 CPython/3.10.11 Darwin/22.1.0 OSX/13.0
                UID:GID : 501:20
             netrc file : /Users/theo/.netrc
           offline mode : False
```

Tobias-Fischer, 2023-05-20:
This is very weird. It works perfectly for me. Something makes me think that things went wrong when you created your environment - there should be no need to set `ROS_PACKAGE_PATH` for you manually, this should happen automatically when you do `conda activate ros_env`. Could you please try and create a new environment? If it still does not work, what is the content of `ROS_PACKAGE_PATH`?

For reference, here is my conda list:
```
# packages in environment at /Users/fischert/mambaforge/envs/robo-new:
#
# Name                    Version                   Build  Channel
adwaita-icon-theme        43                       unix_0    conda-forge
aom                       3.5.0                h7ea286d_0    conda-forge
apr                       1.7.0                h9b22ae9_5    conda-forge
assimp                    5.2.5                h276577b_0    conda-forge
atk-1.0                   2.38.0               hcb7b3dd_1    conda-forge
bcrypt                    3.2.2            py39h02fc5c5_1    conda-forge
blosc                     1.21.4               hc338f07_0    conda-forge
boost                     1.78.0           py39h99de9ae_4    conda-forge
boost-cpp                 1.78.0               hf1d6563_2    conda-forge
bullet-cpp                3.21             py39hde7b980_4    conda-forge
bzip2                     1.0.8                h3422bc3_4    conda-forge
c-ares                    1.19.0               hb547adb_0    conda-forge
c-compiler                1.5.2                h5008568_0    conda-forge
ca-certificates           2023.5.7             hf0a4a13_0    conda-forge
cairo                     1.16.0            h73a0509_1014    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.3              pyhd8ed1ab_0    conda-forge
cctools                   973.0.1             hcbb26d4_13    conda-forge
cctools_osx-arm64         973.0.1             hef52d2f_13    conda-forge
cffi                      1.15.1           py39h7e6b969_3    conda-forge
cfitsio                   4.2.0                h2f961c4_0    conda-forge
clang                     14.0.6               hce30654_1    conda-forge
clang-14                  14.0.6          default_h5dc8d65_1    conda-forge
clang_osx-arm64           14.0.6               h15773ab_6    conda-forge
clangxx                   14.0.6          default_h610c423_1    conda-forge
clangxx_osx-arm64         14.0.6               he29aa18_6    conda-forge
cmake                     3.26.3               hf234bd0_0    conda-forge
colcon-cmake              0.2.27             pyhd8ed1ab_0    conda-forge
colcon-core               0.12.1             pyhd8ed1ab_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-pkg-config         0.1.0                      py_0    conda-forge
colcon-python-setup-py    0.2.7                      py_0    conda-forge
colcon-recursive-crawl    0.2.1                      py_0    conda-forge
colcon-ros                0.3.23             pyhd8ed1ab_0    conda-forge
colcon-test-result        0.3.8                      py_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
coloredlogs               15.0.1             pyhd8ed1ab_3    conda-forge
compiler-rt               14.0.6               h30b49de_0    conda-forge
compiler-rt_osx-arm64     14.0.6               h48302dc_0    conda-forge
compilers                 1.5.2                hce30654_0    conda-forge
console_bridge            1.0.2                h3e96240_1    conda-forge
coreutils                 9.3                  h1a8c8d9_0    conda-forge
coverage                  7.2.5            py39h0f82c59_0    conda-forge
cppzmq                    4.9.0                h54398fd_0    conda-forge
cryptography              40.0.2           py39he2a39a8_0    conda-forge
curl                      7.88.1               h9049daf_1    conda-forge
cxx-compiler              1.5.2                hffc8910_0    conda-forge
dartsim                   6.13.0               h1a3626c_0    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distlib                   0.3.6              pyhd8ed1ab_0    conda-forge
distro                    1.8.0              pyhd8ed1ab_0    conda-forge
docutils                  0.20.1           py39h2804cbe_0    conda-forge
eigen                     3.4.0                hc021e02_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
exceptiongroup            1.1.1              pyhd8ed1ab_0    conda-forge
expat                     2.5.0                hb7217d7_1    conda-forge
fcl                       0.7.0                h88a8a2c_2    conda-forge
ffmpeg                    5.1.2           gpl_hf318d42_106    conda-forge
flann                     1.9.1             hd3e9afc_1011    conda-forge
fmt                       9.1.0                hffc8910_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
[... 321 lines cut ...]
xorg-kbproto              1.0.7             h27ca646_1002    conda-forge
xorg-libice               1.0.10               h27ca646_0    conda-forge
xorg-libsm                1.2.3             h27ca646_1000    conda-forge
xorg-libx11               1.8.4                h1a8c8d9_0    conda-forge
[... cut ...]
yaml                      0.2.5                h3422bc3_2    conda-forge
yaml-cpp                  0.7.0                hb7217d7_2    conda-forge
zeromq                    4.3.4                hbdafb3b_1    conda-forge
zipp                      3.15.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               h03a7124_4    conda-forge
zstd                      1.5.2                hf913c23_6    conda-forge
zziplib                   0.13.69              he1e0b03_1    conda-forge
```

fedor-chervinskii, 2023-05-22:
@Tobias-Fischer thank you! It turned out, simply `conda deactivate -> activate` did the trick.

The problem is I misunderstood [these lines](https://github.com/RoboStack/robostack.github.io/blob/master/docs/GettingStarted.md?plain=1#L44) as this only needs to be done for Windows (there is no comma, and it's not clear that these are two separate sentences 😄 ) 

```
# reload environment to activate required scripts before running anything
# on Windows, please restart the Anaconda Prompt / Command Prompt!
```

Now it makes sense that I needed to reactivate env for the setup scripts to run after the installation of ROS packages.

Tobias-Fischer, 2023-05-22:
Great!

closed 2023-05-22 by Tobias-Fischer