RoboStack/ros-noetic #372: Unable to install ros-noetic-desktop on macOS

fabriziocacicia, 2023-06-13:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

I'm trying to install `ros-noetic-desktop`/`ros-noetic-desktop-full` on my Macbook using `conda` and I'm getting the following error:

```shell
~ conda install ros-noetic-desktop
Collecting package metadata (current_repodata.json): done
Solving environment: unsuccessful initial attempt using frozen solve. Retrying with flexible solve.
Solving environment: unsuccessful attempt using repodata from current_repodata.json, retrying with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: unsuccessful initial attempt using frozen solve. Retrying with flexible solve.
Solving environment: / 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                                                                                                                         

UnsatisfiableError: The following specifications were found to be incompatible with each other:

Output in format: Requested package -> Available versionsThe following specifications were found to be incompatible with your system:

  - feature:/osx-64::__osx==10.16=0
  - feature:|@/osx-64::__osx==10.16=0
  - ros-noetic-desktop -> __osx[version='>=10.14']

Your installed version is: 10.16
```

Same output if I run `conda install ros-noetic-desktop-full`

Also, a strange thing is that my MacBook is not running macOS 10.16, instead I have macOS 13.3.1.
Here is the output of the `sw_vers` command:
```shell
~ sw_vers
ProductName:		macOS
ProductVersion:		13.3.1
BuildVersion:		22E261
```

### Installed packages

```shell
# Name                    Version                   Build  Channel
boltons                   23.0.0          py311hecd8cb5_0  
brotlipy                  0.7.0           py311h6c40b1e_1002  
bzip2                     1.0.8                h1de35cc_0  
c-ares                    1.19.1               h0dc2134_0    conda-forge
ca-certificates           2023.5.7             h8857fd0_0    conda-forge
certifi                   2023.5.7           pyhd8ed1ab_0    conda-forge
cffi                      1.15.1          py311h6c40b1e_3  
charset-normalizer        2.0.4              pyhd3eb1b0_0  
conda                     23.3.1          py311h6eed73b_0    conda-forge
conda-package-handling    2.1.0           py311hecd8cb5_0  
conda-package-streaming   0.8.0           py311hecd8cb5_0  
cryptography              38.0.4          py311h61927ef_0    conda-forge
fmt                       9.1.0                hb8565cd_0    conda-forge
icu                       72.1                 h7336db1_0    conda-forge
idna                      3.4             py311hecd8cb5_0  
jsonpatch                 1.32               pyhd3eb1b0_0  
jsonpointer               2.1                pyhd3eb1b0_0  
krb5                      1.20.1               h049b76e_0    conda-forge
libarchive                3.6.2                h0b5dc4a_1    conda-forge
libcurl                   8.1.2                hbee3ae8_0    conda-forge
libcxx                    16.0.5               hd57cbcb_0    conda-forge
libedit                   3.1.20191231         h0678c8f_2    conda-forge
libev                     4.33                 haf1e3a3_1    conda-forge
libexpat                  2.5.0                hf0c8a7f_1    conda-forge
libffi                    3.4.4                hecd8cb5_0  
libiconv                  1.17                 hac89ed1_0    conda-forge
libmamba                  1.4.2                h9d281b0_0    conda-forge
libmambapy                1.4.2           py311hcc19a12_0    conda-forge
libnghttp2                1.52.0               he2ab024_0    conda-forge
libsolv                   0.7.24               hbc0c0cd_0    conda-forge
libsqlite                 3.42.0               h58db7d2_0    conda-forge
libssh2                   1.11.0               hd019ec5_0    conda-forge
libxml2                   2.11.4               hd95e348_0    conda-forge
libzlib                   1.2.13               h8a1eda9_5    conda-forge
lz4-c                     1.9.4                hf0c8a7f_0    conda-forge
lzo                       2.10              haf1e3a3_1000    conda-forge
mamba                     1.4.2           py311h8082e30_0    conda-forge
ncurses                   6.4                  hcec6c5f_0  
openssl                   3.1.1                h8a1eda9_1    conda-forge
packaging                 23.0            py311hecd8cb5_0  
pip                       23.1.2          py311hecd8cb5_0  
pluggy                    1.0.0           py311hecd8cb5_1  
pybind11-abi              4                    hd8ed1ab_3    conda-forge
pycosat                   0.6.4           py311h6c40b1e_0  
pycparser                 2.21               pyhd3eb1b0_0  
pyopenssl                 23.0.0          py311hecd8cb5_0  
pysocks                   1.7.1           py311hecd8cb5_0  
python                    3.11.4          h30d4d87_0_cpython    conda-forge
python_abi                3.11                    3_cp311    conda-forge
readline                  8.2                  hca72f7f_0  
reproc                    14.2.4               hb7f2c08_0    conda-forge
reproc-cpp                14.2.4               hf0c8a7f_0    conda-forge
requests                  2.29.0          py311hecd8cb5_0  
ruamel.yaml               0.17.21         py311h6c40b1e_0  
setuptools                67.8.0          py311hecd8cb5_0  
sqlite                    3.41.2               h6c40b1e_0  
tk                        8.6.12               h5d9f67b_0  
toolz                     0.12.0          py311hecd8cb5_0  
tqdm                      4.65.0          py311h85bffb1_0  
tzdata                    2023c                h04d1e81_0  
urllib3                   1.26.16         py311hecd8cb5_0  
wheel                     0.38.4          py311hecd8cb5_0  
xz                        5.4.2                h6c40b1e_0  
yaml-cpp                  0.7.0                hf0c8a7f_2    conda-forge
zlib                      1.2.13               h8a1eda9_5    conda-forge
zstandard                 0.19.0          py311h6c40b1e_0  
zstd                      1.5.2                hbc0c0cd_6    conda-forge
```

### Environment info

```shell
active environment : ros
    active env location : /Users/Fabrizio/opt/miniconda3/envs/ros
            shell level : 2
       user config file : /Users/Fabrizio/.condarc
 populated config files : /Users/Fabrizio/.condarc
                          /Users/Fabrizio/opt/miniconda3/envs/ros/.condarc
          conda version : 23.5.0
    conda-build version : not installed
         python version : 3.9.1.final.0
       virtual packages : __archspec=1=x86_64
                          __osx=10.16=0
                          __unix=0=0
       base environment : /Users/Fabrizio/opt/miniconda3  (writable)
      conda av data dir : /Users/Fabrizio/opt/miniconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://repo.anaconda.com/pkgs/main/osx-64
                          https://repo.anaconda.com/pkgs/main/noarch
                          https://repo.anaconda.com/pkgs/r/osx-64
                          https://repo.anaconda.com/pkgs/r/noarch
          package cache : /Users/Fabrizio/opt/miniconda3/pkgs
                          /Users/Fabrizio/.conda/pkgs
       envs directories : /Users/Fabrizio/opt/miniconda3/envs
                          /Users/Fabrizio/.conda/envs
               platform : osx-64
             user-agent : conda/23.5.0 requests/2.28.1 CPython/3.9.1 Darwin/22.4.0 OSX/10.16
                UID:GID : 501:20
             netrc file : /Users/Fabrizio/.netrc
           offline mode : False
```

Tobias-Fischer, 2023-06-13:
Could you please follow our recommendation and use mamba instead of conda? If you still have the same problem, feel free to reopen.

fabriziocacicia, 2023-06-13:
@Tobias-Fischer I just came across this [medium article](https://medium.com/robostack/cross-platform-conda-packages-for-ros-fa1974fd1de3), and I read that I could just run the command `conda create -n ros ros-noetic-desktop -conda-forge -c robostack && conda activate ros`
That's why I used conda.

Tobias-Fischer, 2023-06-13:
Yes - that one is a few years old now. Please see the readme of this repository for further instructions