RoboStack/ros-noetic #374: moveit-ros-planning-interface seems to be missing Python API

fedor-chervinskii, 2023-06-28:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Please help me understand what's the actual problem. It looks like the package has been built without Python API 

Arch: osx-arm64
Version of the package:
`ros-noetic-moveit-ros-planning-interface 1.1.12           py39hae387e0_6    robostack-staging`

Trying to call `import moveit_commander` from Python and facing this error:
`ImportError: cannot import name '_moveit_roscpp_initializer' from 'moveit_ros_planning_interface' (/Users/theo/miniconda3/envs/ros_env/lib/python3.9/site-packages/moveit_ros_planning_interface/__init__.py)`

the `__init__.py` and the folder appears to be empty

### Installed packages

```shell
# packages in environment at /Users/theo/miniconda3/envs/ros_env:
#
# Name                    Version                   Build  Channel
adwaita-icon-theme        43                       unix_0    conda-forge
aom                       3.5.0                h7ea286d_0    conda-forge
apr                       1.7.0                h9b22ae9_5    conda-forge
assimp                    5.2.5                h276577b_0    conda-forge
atk-1.0                   2.38.0               hcb7b3dd_1    conda-forge
bcrypt                    3.2.2            py39h02fc5c5_1    conda-forge
blosc                     1.21.4               hc338f07_0    conda-forge
boost                     1.78.0           py39h99de9ae_4    conda-forge
boost-cpp                 1.78.0               hf1d6563_2    conda-forge
brotli                    1.0.9                h1a8c8d9_8    conda-forge
brotli-bin                1.0.9                h1a8c8d9_8    conda-forge
bullet                    3.21                 h85045c0_4    conda-forge
bullet-cpp                3.21             py39hde7b980_4    conda-forge
bzip2                     1.0.8                h3422bc3_4    conda-forge
c-ares                    1.19.0               hb547adb_0    conda-forge
ca-certificates           2023.5.7             hf0a4a13_0    conda-forge
cairo                     1.16.0            h73a0509_1014    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
certifi                   2023.5.7           pyhd8ed1ab_0    conda-forge
cffi                      1.15.1           py39h7e6b969_3    conda-forge
cfitsio                   4.2.0                h2f961c4_0    conda-forge
cmake                     3.26.3               hf234bd0_0    conda-forge
console_bridge            1.0.2                h3e96240_1    conda-forge
contourpy                 1.0.7            py39haaf3ac1_0    conda-forge
cppzmq                    4.9.0                h54398fd_0    conda-forge
cryptography              40.0.2           py39he2a39a8_0    conda-forge
curl                      7.88.1               h9049daf_1    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
dartsim                   6.13.0               h1a3626c_0    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.8.0              pyhd8ed1ab_0    conda-forge
docutils                  0.20.1           py39h2804cbe_0    conda-forge
eigen                     3.4.0                hc021e02_0    conda-forge
eigenpy                   2.8.1            py39hb94e712_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.5.0                hb7217d7_1    conda-forge
fcl                       0.7.0                h88a8a2c_2    conda-forge
ffmpeg                    5.1.2           gpl_hf318d42_106    conda-forge
flann                     1.9.1             hd3e9afc_1011    conda-forge
fmt                       9.1.0                hffc8910_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.2               h82840c6_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
[... 515 lines cut ...]
 populated config files : /Users/theo/.condarc
                          /Users/theo/miniconda3/envs/ros_env/.condarc
          conda version : 23.3.1
    conda-build version : not installed
         python version : 3.10.11.final.0
       virtual packages : __archspec=1=arm64
                          __osx=13.0=0
                          __unix=0=0
       base environment : /Users/theo/miniconda3  (writable)
      conda av data dir : /Users/theo/miniconda3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-arm64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /Users/theo/miniconda3/pkgs
                          /Users/theo/.conda/pkgs
       envs directories : /Users/theo/miniconda3/envs
                          /Users/theo/.conda/envs
               platform : osx-arm64
             user-agent : conda/23.3.1 requests/2.28.1 CPython/3.10.11 Darwin/22.1.0 OSX/13.0
                UID:GID : 501:20
             netrc file : /Users/theo/.netrc
           offline mode : False
```

Tobias-Fischer, 2023-06-28:
Looks like https://github.com/RoboStack/ros-noetic/issues/371

fedor-chervinskii, 2023-06-28:
@Tobias-Fischer thanks for looking into it.

`mamba install --force-reinstall ros-noetic-moveit-ros-planning-interface` actually shows reinstalling the same build hash:

`mamba install --force-reinstall ros-noetic-moveit-ros-planning-interface 

                  __    __    __    __
                 /  \  /  \  /  \  /  \
                /    \/    \/    \/    \
███████████████/  /██/  /██/  /██/  /████████████████████████
              /  / \   / \   / \   / \  \____
             /  /   \_/   \_/   \_/   \    o \__,
            / _/                       \_____/  `
            |/
        ███╗   ███╗ █████╗ ███╗   ███╗██████╗  █████╗
        ████╗ ████║██╔══██╗████╗ ████║██╔══██╗██╔══██╗
        ██╔████╔██║███████║██╔████╔██║██████╔╝███████║
        ██║╚██╔╝██║██╔══██║██║╚██╔╝██║██╔══██╗██╔══██║
        ██║ ╚═╝ ██║██║  ██║██║ ╚═╝ ██║██████╔╝██║  ██║
        ╚═╝     ╚═╝╚═╝  ╚═╝╚═╝     ╚═╝╚═════╝ ╚═╝  ╚═╝

        mamba (1.4.2) supported by @QuantStack

        GitHub:  https://github.com/mamba-org/mamba
        Twitter: https://twitter.com/QuantStack

█████████████████████████████████████████████████████████████

Looking for: ['ros-noetic-moveit-ros-planning-interface']

warning  libmamba Cache file "/Users/theo/miniconda3/pkgs/cache/a850f475.json" was modified by another program
warning  libmamba Cache file "/Users/theo/miniconda3/pkgs/cache/09cdf8bf.json" was modified by another program
robostack-staging/noarch                                      No change
robostack-staging/osx-arm64                                   No change
conda-forge/osx-arm64                                6.7MB @   3.5MB/s  2.1s
conda-forge/noarch                                  13.1MB @   4.3MB/s  3.3s

Pinned packages:
  - python 3.9.*

Transaction

  Prefix: /Users/theo/miniconda3/envs/ros_env

  Updating specs:

   - ros-noetic-moveit-ros-planning-interface

  Package                                     Version  Build           Channel               Size
───────────────────────────────────────────────────────────────────────────────────────────────────
  Reinstall:
───────────────────────────────────────────────────────────────────────────────────────────────────

  o ros-noetic-moveit-ros-planning-interface   1.1.12  py39hae387e0_6  robostack-staging         

  Summary:

  Reinstall: 1 packages

  Total download: 0 B
`

Tobias-Fischer, 2023-06-28:
When did you create the environment? If it was before May 30, could you please try creating it fresh?

Tobias-Fischer, 2023-06-28:
I just checked and indeed something is wrong with this package. I'm not sure what though. The `_moveit_roscpp_initializer.dylib` gets installed to the same directory as the `__init__.py`. Any ideas @wolfv @traversaro?

traversaro, 2023-07-03:
At least `import moveit_ros_planning_interface` works fine on both Windows and Linux. Can you try if the error occurs just with `import  moveit_ros_planning_interface`? Thanks!

vkmb, 2023-07-12:
I don't have much knowledge in this but had encountered  the same problem past week with ros-noetic osx64 . Going through stackoverflow and similar git issues, I stumbled upon [this possible soln](https://github.com/MoDeNa-EUProject/MoDeNa/issues/1#issue-76779748) of soft linking the `.dylib` to `.so`. So I copied all the dylib in the path `target_prefix/lib/python3.10/site-packages/moveit_ros_planning_interface` to  `target_prefix/lib/python3.9/site-packages/moveit_ros_planning_interface` and soft linked it. After this the import issue was fixed but now stuck in seg fault. 

```
$python
Python 3.9.16 | packaged by conda-forge | (main, Feb  1 2023, 21:42:20) 
[Clang 14.0.6 ] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import sys
>>> import moveit_commander
>>> moveit_commander.roscpp_initialize(sys.argv)
[1]    52891 segmentation fault  python
```

Tobias-Fischer, 2023-07-12:
Ah cool, that's good intel. The segfault is probably because they are linking to PYTHON_LIBRARIES. That should be an easy fix.

Tobias-Fischer, 2023-07-12:
I'm not sure what the best way is to fix the library name though - @traversaro you are the CMake expert, there is probably a standard way?

Tobias-Fischer, 2024-02-09:
A quick reminder @traversaro

traversaro, 2024-02-09:
> A quick reminder @traversaro

Sorry about that, I missed the original notification. I guess this is now fixed?

Tobias-Fischer, 2024-02-09:
Yes thank you :)

vkmb, 2024-02-12:
Hi, I tried to install `ros-noetic-moveit == 1.1.13` to check the updated planning interface package. Unfortunately, there is version mismatch error regarding `ros-distro-mutex`.  

```bash
$ mamba install -c robostack-staging "ros-noetic-moveit==1.1.13"
Looking for: ['ros-noetic-moveit==1.1.13']

conda-forge/osx-64                                          Using cache
conda-forge/noarch                                          Using cache
robostack-staging/osx-64                                      No change
robostack-staging/noarch                                      No change

Pinned packages:
  - python 3.9.*

Could not solve for environment specs
The following package could not be installed
└─ ros-noetic-moveit 1.1.13  is installable and it requires
   └─ ros-distro-mutex 0.5.* , which can be installed.
```

When trying to force reinstall the `ros-distro-mutex==0.5.0`, there seems to be an unexpected error. 

```
$ mamba install --force-reinstall -c robostack-staging "ros-distro-mutex==0.5.0"

Looking for: ['ros-distro-mutex==0.5.0']

robostack-staging/osx-64                                    Using cache
robostack-staging/noarch                                    Using cache
conda-forge/osx-64                                          Using cache
conda-forge/noarch                                          Using cache
ros-distro-mutex 0.5.0: overriding channel, version and build from installed packages due to --force-reinstall.

Pinned packages:
  - python 3.9.*

Transaction

  Prefix: /Users/kunkuma/.pyenv/versions/miniconda3-latest/envs/movit

  Updating specs:

   - ros-distro-mutex==0.5.0

  Package             Version  Build   Channel               Size
───────────────────────────────────────────────────────────────────
  Reinstall:
───────────────────────────────────────────────────────────────────

  o ros-distro-mutex    0.4.0  noetic  robostack-staging         

  Summary:

  Reinstall: 1 packages

  Total download: 0 B

───────────────────────────────────────────────────────────────────

Confirm changes: [Y/n] y

# >>>>>>>>>>>>>>>>>>>>>> ERROR REPORT <<<<<<<<<<<<<<<<<<<<<<

    Traceback (most recent call last):
      File "/Users/kunkuma/.pyenv/versions/miniconda3-latest/envs/movit/lib/python3.9/site-packages/conda/exceptions.py", line 1132, in __call__
        return func(*args, **kwargs)
      File "/Users/kunkuma/.pyenv/versions/miniconda3-latest/envs/movit/lib/python3.9/site-packages/mamba/mamba.py", line 941, in exception_converter
        raise e
      File "/Users/kunkuma/.pyenv/versions/miniconda3-latest/envs/movit/lib/python3.9/site-packages/mamba/mamba.py", line 934, in exception_converter
        exit_code = _wrapped_main(*args, **kwargs)
[... 54 lines cut ...]
          conda version : 23.3.1
    conda-build version : not installed
         python version : 3.9.18.final.0
       virtual packages : __archspec=1=haswell
                          __osx=12.7.3=0
                          __unix=0=0
       base environment : /Users/kunkuma/.pyenv/versions/miniconda3-latest/envs/movit  (writable)
      conda av data dir : /Users/kunkuma/.pyenv/versions/miniconda3-latest/envs/movit/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /Users/kunkuma/.pyenv/versions/miniconda3-latest/envs/movit/pkgs
                          /Users/kunkuma/.conda/pkgs
       envs directories : /Users/kunkuma/.pyenv/versions/miniconda3-latest/envs/movit/envs
                          /Users/kunkuma/.conda/envs
               platform : osx-64
             user-agent : conda/23.3.1 requests/2.31.0 CPython/3.9.18 Darwin/21.6.0 OSX/12.7.3
                UID:GID : 501:20
             netrc file : None
           offline mode : False

An unexpected error has occurred. Conda has prepared the above report.
```

traversaro, 2024-02-12:
Can you try in a fresh environment? The new packages use Python 3.11 .

vkmb, 2024-02-12:
Ah ok, figured this could be a one of the issues. Thanks !

vkmb, 2024-02-12:
I tried the updated package in a fresh python 3.11 environment, the installation was successful.  
```
$mamba list | egrep "ros-noetic-moveit"
ros-noetic-moveit         1.1.13          py311h907a76e_20    robostack-staging
ros-noetic-moveit-commander 1.1.13          py311h907a76e_20    robostack-staging
ros-noetic-moveit-core    1.1.13          py311he536b71_20    robostack-staging
ros-noetic-moveit-fake-controller-manager 1.1.13          py311h907a76e_20    robostack-staging
ros-noetic-moveit-kinematics 1.1.13          py311h907a76e_20    robostack-staging
ros-noetic-moveit-msgs    0.11.4          py311h907a76e_20    robostack-staging
ros-noetic-moveit-planners 1.1.13          py311h907a76e_20    robostack-staging
ros-noetic-moveit-planners-chomp 1.1.13          py311h907a76e_20    robostack-staging
ros-noetic-moveit-planners-ompl 1.1.13          py311h907a76e_20    robostack-staging
ros-noetic-moveit-plugins 1.1.13          py311h907a76e_20    robostack-staging
ros-noetic-moveit-ros     1.1.13          py311h907a76e_20    robostack-staging
ros-noetic-moveit-ros-benchmarks 1.1.13          py311h907a76e_20    robostack-staging
ros-noetic-moveit-ros-control-interface 1.1.13          py311h907a76e_20    robostack-staging
ros-noetic-moveit-ros-manipulation 1.1.13          py311h907a76e_20    robostack-staging
ros-noetic-moveit-ros-move-group 1.1.13          py311h907a76e_20    robostack-staging
ros-noetic-moveit-ros-occupancy-map-monitor 1.1.13          py311h907a76e_20    robostack-staging
ros-noetic-moveit-ros-perception 1.1.13          py311hd8264e0_20    robostack-staging
ros-noetic-moveit-ros-planning 1.1.13          py311h907a76e_20    robostack-staging
ros-noetic-moveit-ros-planning-interface 1.1.13          py311h907a76e_20    robostack-staging
ros-noetic-moveit-ros-robot-interaction 1.1.13          py311h907a76e_20    robostack-staging
ros-noetic-moveit-ros-visualization 1.1.13          py311hb97643c_20    robostack-staging
ros-noetic-moveit-ros-warehouse 1.1.13          py311h907a76e_20    robostack-staging
ros-noetic-moveit-setup-assistant 1.1.13          py311h070c4d6_20    robostack-staging
ros-noetic-moveit-simple-controller-manager 1.1.13          py311h907a76e_20    robostack-staging
```
When trying to import the moveit_commander, this throws the old error related to `moveit_ros_planning_interface`
```
$ python
Python 3.11.7 | packaged by conda-forge | (main, Dec 23 2023, 14:38:52) [Clang 16.0.6 ] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import moveit_commander
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/Users/kunkuma/.pyenv/versions/miniconda3-latest/envs/movit/lib/python3.11/site-packages/moveit_commander/__init__.py", line 2, in <module>
    from .roscpp_initializer import *
  File "/Users/kunkuma/.pyenv/versions/miniconda3-latest/envs/movit/lib/python3.11/site-packages/moveit_commander/roscpp_initializer.py", line 35, in <module>
    from moveit_ros_planning_interface import _moveit_roscpp_initializer
ImportError: cannot import name '_moveit_roscpp_initializer' from 'moveit_ros_planning_interface' (/Users/kunkuma/.pyenv/versions/miniconda3-latest/envs/movit/lib/python3.11/site-packages/moveit_ros_planning_interface/__init__.py)
>>> 
```

traversaro, 2024-02-12:
I reproduced the problem, but for me the error is slightly different (Command Prompt?):

~~~
(moveit1) C:\Users\straversaro>python
Python 3.11.7 | packaged by conda-forge | (main, Dec 23 2023, 14:27:59) [MSC v.1937 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import moveit_commander
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\straversaro\AppData\Local\miniforge3\envs\moveit1\Library\lib\site-packages\moveit_commander\__init__.py", line 2, in <module>
    from .roscpp_initializer import *
  File "C:\Users\straversaro\AppData\Local\miniforge3\envs\moveit1\Library\lib\site-packages\moveit_commander\roscpp_initializer.py", line 35, in <module>
    from moveit_ros_planning_interface import _moveit_roscpp_initializer
ImportError: DLL load failed while importing _moveit_roscpp_initializer: The specified procedure could not be found.
~~~

Tobias-Fischer, 2024-02-12:
On osx-arm64, I get:
```
Python 3.11.7 | packaged by conda-forge | (main, Dec 23 2023, 14:38:07) [Clang 16.0.6 ] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import moveit_commander
Failed to import pyassimp, see https://github.com/ros-planning/moveit/issues/86 for more info
```

Tobias-Fischer, 2024-02-12:
The issue on osx-64 is that the `moveit_ros_planning_interface` libraries get installed in the `lib/python3.10` subdirectory instead of `lib/python3.11`. I am not sure why this happens; cmake seems to properly detect Python 3.10:
```
2024-02-09T09:38:49.5227030Z -- Found PythonInterp: $PREFIX/bin/python (found version "3.11.7") 
2024-02-09T09:38:49.5228150Z -- Found PythonLibs: $PREFIX/lib/libpython3.11.dylib (found suitable version "3.11.7", minimum required is "3.11") 
2024-02-09T09:38:49.5229390Z -- Found Boost: $PREFIX/lib/cmake/Boost-1.82.0/BoostConfig.cmake (found version "1.82.0")  
2024-02-09T09:38:49.5230960Z -- Found Boost: $PREFIX/lib/cmake/Boost-1.82.0/BoostConfig.cmake (found version "1.82.0") found components: date_time filesystem program_options python system thread 
2024-02-09T09:38:49.5232260Z -- eigenpy FOUND. eigenpy at $PREFIX/lib/libeigenpy.dylib
2024-02-09T09:38:49.5233330Z -- Found Python3: $PREFIX/bin/python (found version "3.11.7") found components: Interpreter Development.Module NumPy 
2024-02-09T09:38:49.5234240Z -- PythonLibraryDirs: 
2024-02-09T09:38:49.5234680Z -- PythonLibVersionString: 3.11.7
2024-02-09T09:38:49.5235420Z -- Python site lib: lib/python3.11/site-packages
2024-02-09T09:38:49.5236150Z -- Python include dirs: $PREFIX/include/python3.11
2024-02-09T09:38:49.5236780Z -- Checking for NumPy
2024-02-09T09:38:49.5237430Z --   NUMPY_INCLUDE_DIRS=$PREFIX/lib/python3.11/site-packages/numpy/core/include
2024-02-09T09:38:49.5238130Z --   NUMPY_VERSION=1.23.5
2024-02-09T09:38:49.5238800Z -- NumPy include dir: $PREFIX/lib/python3.11/site-packages/numpy/core/include
```

It seems that `CATKIN_PACKAGE_PYTHON_DESTINATION` points to the wrong directory?

vkmb, 2024-02-12:
Copying the .so files from `python3.10/site-packages/moveit_ros_planning_interface` to `python3.11/site-packages/moveit_ros_planning_interface/` and pip install `pyassimp==5.2.5` fixed the issue.

I was able to use the python interface to actuate the ur5 manipulator. 
 
[image]

Tobias-Fischer, 2024-02-12:
The issue is that `SP_DIR` points to the wrong directory. @wolfv - do you know who we should contact about this?

Tobias-Fischer, 2024-02-13:
I temporarily fixed it with https://github.com/RoboStack/vinca/commit/bebaf58ac7a7d43f42382b80227bd35bd97fc190 and rebuilt the osx-64 package (same build number, so you would need to clear your cache and force re-downloading the package - sorry for that).

Tobias-Fischer, 2024-02-16:
FYI - you can now `mamba install pyassimp`.

Tobias-Fischer, 2024-02-18:
Hi @traversaro - do you want to take a look at the Windows issue, or should I close here?

Tobias-Fischer, 2024-02-28:
Quick reminder @traversaro :)

traversaro, 2024-03-10:
## Quick workaround

TL;DR a quick workaround for anyone having this problem is to run once in the environment:
~~~
copy %CONDA_PREFIX%\Library\lib\resource_retriever.dll %CONDA_PREFIX%\Library\bin\resource_retriever.dll
copy %CONDA_PREFIX%\Library\lib\srdfdom.dll %CONDA_PREFIX%\Library\bin\srdfdom.dll
~~~

## Long explanation

Ok, this was a tricky one. Somehow:
~~~
from moveit_ros_planning_interface import _moveit_roscpp_initializer
~~~

was failing in python, but nothing was wrong when opening `%CONDA_PREFIX%\Library\lib\site-packages\moveit_ros_planning_interface\_moveit_roscpp_initializer.pyd` in [Dependencies](https://github.com/lucasg/Dependencies). So, there was something  Python-specific. I wrote a quick script to load all the the dynamic libraries of the environment using python:

~~~python
import os

prefix = "C:/Users/straversaro/AppData/Local/miniforge3/envs/ros-noetic-moveit-ros-planning-interface/Library"
libraries_to_test = []
libraries_to_test.append(prefix+"/lib/site-packages/moveit_ros_planning_interface/_moveit_move_group_interface.pyd")
libraries_to_test.append(prefix+"/lib/site-packages/moveit_ros_planning_interface/_moveit_planning_scene_interface.pyd")
libraries_to_test.append(prefix+"/lib/site-packages/moveit_ros_planning_interface/_moveit_robot_interface.pyd")
libraries_to_test.append(prefix+"/lib/site-packages/moveit_ros_planning_interface/_moveit_roscpp_initializer.pyd")

# Loop through the files in the specified directory
for filename in os.listdir(prefix+"/bin"):
    if filename.endswith(".dll"):
        libraries_to_test.append(prefix+"/bin/"+filename)

import ctypes

loaded_libraries = []
libraries_not_loaded = []

for library_name in libraries_to_test:
    try:
        # Load the library
        _ = ctypes.cdll.LoadLibrary(library_name)
        loaded_libraries.append(library_name)
        print(f"Library '{library_name}' loaded successfully.")
    except Exception as e:
        print(f"Error loading library '{library_name}': {e}")
        libraries_not_loaded.append(library_name)

# Print the list of loaded libraries
print("List of Loaded Libraries:", loaded_libraries)

print("List of non-loaded Libraries:", libraries_not_loaded)
~~~

With this, I noticed that somehow a lot of moveit libraries were listed under the "List of non-loaded Libraries" part. So there was something strange. Iterating a bit between the script and the DependenciesGui, I was able to notice that there were two .dll that were wrongly installed in `%CONDA_PREFIX%\Library\lib` instead of `%CONDA_PREFIX%\Library\bin` . This was not a problem for the DependenciesGui, as someone is adding `%CONDA_PREFIX%\Library\lib` to the PATH (probably this was added at some point as workaround for this problem), but since Python 3.8 Python on Windows ignores the `PATH` when loading dlls (see https://github.com/gazebosim/sdformat/pull/1165#issuecomment-1252967167), so those `.dll` in the wrong place were problematic. The proper fix is to ensure that srdfdom and resource_retriever are installed in the correct place.

traversaro, 2024-03-10:
> The proper fix is to ensure that srdfdom and resource_retriever are installed in the correct place.

I noticed a few more offending examples of this in a standard installation of `ros-noetic-desktop`, let me open a specific issue for those.

closed 2024-03-13 by Tobias-Fischer