RoboStack/ros-noetic #379: Unable to build ros-perception/ar_track_alvar pkg

vkmb, 2023-07-17:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Trying to build the ar_track_alvar pkg noetic branch from [here](https://github.com/ros-perception/ar_track_alvar/) leads to an compilation error with boost, specifically the make_shared<>() pointer method. 

```
ros_x_mac/ws_igmr/src/ar_track_alvar/ar_track_alvar/src/kinect_filtering.cpp:54:26: error: no viable conversion from 'typename boost::detail::sp_if_not_array<PointIndices>::type' (aka 'shared_ptr<pcl::PointIndices>') to 'pcl::PointIndices::Ptr' (aka 'shared_ptr< ::pcl::PointIndices>')
  pcl::PointIndices::Ptr inliers = boost::make_shared<pcl::PointIndices>();
```

### Installed packages

```shell
adwaita-icon-theme        43                       unix_0    conda-forge
aiofiles                  22.1.0             pyhd8ed1ab_0    conda-forge
aiosqlite                 0.19.0             pyhd8ed1ab_0    conda-forge
anyio                     3.6.2              pyhd8ed1ab_0    conda-forge
aom                       3.5.0                hf0c8a7f_0    conda-forge
appdirs                   1.4.4              pyh9f0ad1d_0    conda-forge
appnope                   0.1.3              pyhd8ed1ab_0    conda-forge
apr                       1.7.0                h0d85af4_5    conda-forge
argon2-cffi               21.3.0             pyhd8ed1ab_0    conda-forge
argon2-cffi-bindings      21.2.0           py39ha30fb19_3    conda-forge
arrow                     1.2.3                    pypi_0    pypi
assimp                    5.2.5                hd9e13b6_0    conda-forge
asttokens                 2.2.1              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               h1d18e73_1    conda-forge
attrs                     23.1.0             pyh71513ae_0    conda-forge
autobahn                  23.1.2             pyhd8ed1ab_0    conda-forge
automat                   22.10.0            pyhd8ed1ab_0    conda-forge
babel                     2.12.1             pyhd8ed1ab_1    conda-forge
backcall                  0.2.0              pyh9f0ad1d_0    conda-forge
backports                 1.0                pyhd8ed1ab_3    conda-forge
backports.functools_lru_cache 1.6.4              pyhd8ed1ab_0    conda-forge
bcrypt                    3.2.2            py39ha30fb19_1    conda-forge
beautifulsoup4            4.12.2             pyha770c72_0    conda-forge
binaryornot               0.4.4                    pypi_0    pypi
bleach                    6.0.0              pyhd8ed1ab_0    conda-forge
blosc                     1.21.2               hebb52c4_0    conda-forge
boost                     1.78.0           py39h6dc771e_4    conda-forge
boost-cpp                 1.78.0               h31500c2_2    conda-forge
bqplot                    0.12.39            pyhd8ed1ab_0    conda-forge
brotli                    1.0.9                hb7f2c08_8    conda-forge
brotli-bin                1.0.9                hb7f2c08_8    conda-forge
brotlipy                  0.7.0           py39ha30fb19_1005    conda-forge
build                     0.10.0                   pypi_0    pypi
bullet                    3.21                 h2a79d86_4    conda-forge
bullet-cpp                3.21                 hc6b5a04_4    conda-forge
bzip2                     1.0.8                h0d85af4_4    conda-forge
c-ares                    1.18.1               h0d85af4_0    conda-forge
c-compiler                1.5.2                hbf74d83_0    conda-forge
ca-certificates           2023.5.7             h8857fd0_0    conda-forge
cairo                     1.16.0            h904041c_1014    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.3              pyhd8ed1ab_0    conda-forge
cctools                   973.0.1             h76f1dac_13    conda-forge
cctools_osx-64            973.0.1             hcc6d90d_13    conda-forge
certifi                   2023.5.7           pyhd8ed1ab_0    conda-forge
cffi                      1.15.1           py39h131948b_3    conda-forge
cfitsio                   4.2.0                hd56cc12_0    conda-forge
chardet                   5.1.0                    pypi_0    pypi
charset-normalizer        3.1.0              pyhd8ed1ab_0    conda-forge
clang                     14.0.6               h694c41f_0    conda-forge
clang-14                  14.0.6          default_h55ffa42_0    conda-forge
clang_osx-64              14.0.6               h3113cd8_6    conda-forge
clangxx                   14.0.6          default_h55ffa42_0    conda-forge
clangxx_osx-64            14.0.6               h6f97653_6    conda-forge
[... 760 lines cut ...]
         python version : 3.9.12.final.0
       virtual packages : __osx=10.16=0
                          __unix=0=0
                          __archspec=1=x86_64
       base environment : /Users/kunkuma/.pyenv/versions/miniconda3-latest  (writable)
      conda av data dir : /Users/kunkuma/.pyenv/versions/miniconda3-latest/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-64
                          https://conda.anaconda.org/conda-forge/noarch
                          https://conda.anaconda.org/robostack/osx-64
                          https://conda.anaconda.org/robostack/noarch
                          https://conda.anaconda.org/robostack-experimental/osx-64
                          https://conda.anaconda.org/robostack-experimental/noarch
          package cache : /Users/kunkuma/.pyenv/versions/miniconda3-latest/pkgs
                          /Users/kunkuma/.conda/pkgs
       envs directories : /Users/kunkuma/.pyenv/versions/miniconda3-latest/envs
                          /Users/kunkuma/.conda/envs
               platform : osx-64
             user-agent : conda/4.12.0 requests/2.29.0 CPython/3.9.12 Darwin/21.6.0 OSX/10.16
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2023-07-17:
Can you try to patch the package and use std::make_shared instead?

vkmb, 2023-07-17:
Thank you for the help! Patching it solved this issue but led to undefined symbols for x86_64. Adding PCL as a dependency and linking it to the kinetic_filtering seemed to have solved it. 

I have only it tested for individual marker with no kinect and cannot do the other ones.

[image]

The patched version can be accessed [here](https://github.com/vkmb/ar_track_alvar/tree/robostack_ros_noetic).

Tobias-Fischer, 2023-07-17:
Great, I am glad that you have solved your issue. If you would like, you can open a pull request adding this package with the patch.

Feel free to reopen if you have further issues.

closed 2023-07-17 by Tobias-Fischer