RoboStack/ros-noetic #380: problems install ros-noetic-desktop

SSG0210, 2023-08-07:
Hi everyone,

I have had problem when I have wanted to install ros-noetic.

Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.5 LTS
Release:	20.04

I have run the following commands

conda create -n ros python==3.9

conda activate ros

conda config --env --add channels conda-forge

conda config --env --add channels robostack-staging

conda config --env --remove channels defaults

conda install ros-noetic-desktop

And I get this error:

**_Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible solve.
Solving environment: / 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                                                                                                                                                                                     

UnsatisfiableError: The following specifications were found to be incompatible with each other:

Output in format: Requested package -> Available versionsThe following specifications were found to be incompatible with your system:

  - feature:/linux-64::__glibc==2.31=0
  - feature:|@/linux-64::__glibc==2.31=0

Your installed version is: 2.31_**

Can anybody help me please?

Thanks

Tobias-Fischer, 2023-08-07:
Please see our official instructions at https://robostack.github.io/GettingStarted.html

feel free to reopen if still doesn’t work

closed 2023-08-07 by Tobias-Fischer

SSG0210, 2023-08-07:
Thank you Tobias for your quick response. The only two differences I see compared to the steps in the official guide are:

    - I use conda instead of mamba, is there a difference between mamba and conda?
    - I specify a specific Python version, is there any issue with specifying the Python version when creating the environment?

> Please see our official instructions at https://robostack.github.io/GettingStarted.html
> 
> feel free to reopen if still doesn’t work

Tobias-Fischer, 2023-08-07:
Yes - mamba can resolve large environments like those of RoboStack, while conda struggles (see your error ..). And yes, unfortunately RoboStack requires a particular version of Python.