RoboStack/ros-noetic #381: moveit_cpp.dll missing on Windows noetic

marcoesposito1988, 2023-08-09:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

It is impossible to start a MoveIt! instance (e.g. roslaunch moveit_resources_panda_moveit_config demo.launch) because `move_group.exe` crashes due to `moveit_cpp.dll` not being found.

The DLL [is part of moveit-ros-planning](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning/moveit_cpp/CMakeLists.txt#L1) and indeed the `moveit_cpp.lib` file is present in `..\noetic_env\Library\lib` after installing `ros-noetic-moveit-ros`.

I already deleted and recreated the conda environment from scratch today.

My guess is that either:
- (most likely) the moveit_ros CMake is not installing the DLL file because `RUNTIME DESTINATION` is missing [here](https://github.com/ros-planning/moveit/blob/master/moveit_ros/planning/moveit_cpp/CMakeLists.txt#L23) and should be added in a robostack patch
- this is already happening, but the DLL is not being copied into the installer
- moveit_cpp was made static in the robostack build, and the depending packages were not rebuilt

### Installed packages

```shell
# Name                    Version                   Build  Channel
adwaita-icon-theme        43                        win_0    conda-forge
aom                       3.5.0                h63175ca_0    conda-forge
apr                       1.7.0                he38c35c_5    conda-forge
assimp                    5.2.5                h4dcb625_0    conda-forge
bcrypt                    3.2.2            py39ha55989b_1    conda-forge
boost                     1.78.0           py39hbd792c9_4    conda-forge
boost-cpp                 1.78.0               h9f4b32c_3    conda-forge
brotli                    1.0.9                hcfcfb64_9    conda-forge
brotli-bin                1.0.9                hcfcfb64_9    conda-forge
bullet                    3.24                 h00e5391_0    conda-forge
bullet-cpp                3.24                 h2ba5b7c_0    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2023.7.22            h56e8100_0    conda-forge
cairo                     1.16.0            hd694305_1014    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
certifi                   2023.7.22          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1           py39h68f70e3_3    conda-forge
cmake                     3.26.4               h1537add_0    conda-forge
console_bridge            1.0.2                h5362a0b_1    conda-forge
contourpy                 1.1.0            py39h1f6ef14_0    conda-forge
cryptography              41.0.3           py39hb6bd5e6_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
dav1d                     1.2.1                hcfcfb64_0    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.8.0              pyhd8ed1ab_0    conda-forge
docutils                  0.20.1           py39hcbf5309_0    conda-forge
eigen                     3.4.0                h91493d7_0    conda-forge
eigenpy                   2.8.1            py39haeffccb_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.5.0                h63175ca_1    conda-forge
ffmpeg                    5.1.2           gpl_he426399_111    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.2               hbde0cde_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.42.0           py39ha55989b_0    conda-forge
freeglut                  3.2.2                h63175ca_2    conda-forge
freeimage                 3.18.0              h30d40d2_12    conda-forge
freetype                  2.12.1               h546665d_1    conda-forge
fribidi                   1.0.10               h8d14728_0    conda-forge
gdk-pixbuf                2.42.10              h84213e4_0    conda-forge
getopt-win32              0.1                  h8ffe710_0    conda-forge
gettext                   0.21.1               h5728263_0    conda-forge
glew                      2.1.0                h39d44d4_2    conda-forge
glib                      2.76.4               h12be248_0    conda-forge
glib-tools                2.76.4               h12be248_0    conda-forge
[... 391 lines cut ...]
          conda version : 23.7.2
    conda-build version : not installed
         python version : 3.10.12.final.0
       virtual packages : __archspec=1=x86_64
                          __cuda=12.2=0
                          __win=0=0
       base environment : ...\mambaforge  (writable)
      conda av data dir : ...\mambaforge\etc\conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/win-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : ...\mambaforge\pkgs
                          ...\.conda\pkgs
                          ...\AppData\Local\conda\conda\pkgs
       envs directories : ...\mambaforge\envs
                          ...\.conda\envs
                          ...\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/23.7.2 requests/2.28.1 CPython/3.10.12 Windows/10 Windows/10.0.19045
          administrator : False
             netrc file : None
           offline mode : False
```

traversaro, 2023-08-09:
Thanks @marcoesposito1988, I think that indeed your first guess is the correct one.

marcoesposito1988, 2023-08-09:
Hi @traversaro, sure, I'll give it a shot.

Could you please point me to where you list the git repositories to check out and select the branch/commit? I didn't manage to find that in the code so far

traversaro, 2023-08-09:
Details can be found in https://github.com/RoboStack/ros-noetic/blob/master/CONTRIBUTING.md . In a nutshell, you can open a PR upstream, and extract the patch related to that change by adding `.patch` to the end of the PR. Then you can add the patch (or update the existing patch) in https://github.com/RoboStack/ros-noetic/tree/master/patch .

marcoesposito1988, 2023-08-18:
PR is now open. This is the workflow I used:
- I followed CONTRIBUTING.md until the second last point (`vinca --multiple`)
- I changed the `yaml` file to only build `moveit-ros-planning`
- I let `boa build...` run, which it failed, but only after applying the patches
- I applied my change to the source code in `C:\Users\...\mambaforge\envs\robostackenv\conda-bld\ros-noetic-moveit-ros-planning-...\work\ros-noetic-moveit-ros-planning\src\work\moveit_cpp`
- I changed to the git repository directory and created a patch file with `git diff`
- I cherry-picked the hunk related to my changes and added manually to this repository to create this PR

Does it sound reasonable? Is there a more streamlined one? If so we could add it to CONTRIBUTING.md

traversaro, 2023-08-18:
> Does it sound reasonable? Is there a more streamlined one? If so we could add it to CONTRIBUTING.md

It seems reasonable to me, even if @Tobias-Fischer may have some input on this. Anyhow, it seems to me something valuable to add to CONTRIBUTING.md, thanks!

Tobias-Fischer, 2023-08-18:
That looks great, yes!

marcoesposito1988, 2023-08-18:
Cool, thanks! I will open a PR for the docs then, good thing we don't need the CI for that :see_no_evil:

marcoesposito1988, 2024-01-11:
Would it be possible to release the package again? I created a new environment, and the library is still missing so MoveIt is still unusable on Windows.. 😬

Tobias-Fischer, 2024-01-28:
Will be part of https://github.com/RoboStack/ros-noetic/pull/418

closed 2024-01-28 by Tobias-Fischer