RoboStack/ros-noetic #390: Failed to import roslz4 package

nvander, 2023-09-01:
When attempting to import the `roslz4` module from `ros-noetic-roslz4`, it fails due to a missing `libpython2.7.so.1.0` library.

Here is the error I see:
```
(base) root@0441547fb234:~# python -c "import roslz4"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/opt/conda/lib/python3.9/site-packages/roslz4/__init__.py", line 33, in <module>
    from ._roslz4 import *
ImportError: libpython2.7.so.1.0: cannot open shared object file: No such file or directory
```

Here is the version of `ros-noetic-roslz4` I have installed:
```
(base) root@0441547fb234:~# micromamba list ros-noetic-roslz4
List of packages in environment: "/opt/conda"

  Name               Version  Build            Channel          
──────────────────────────────────────────────────────────────────
  ros-noetic-roslz4  1.15.15  py39hac30774_15  robostack-staging
```

Let me know if I can provide any additional detail that may be helpful. Thanks!

Tobias-Fischer, 2023-09-01:
Could you please follow our bug report issue template? It asks for a few more details that are helpful for us for debugging. Thanks!

nvander, 2023-09-01:
Sure, sorry about that. I created this new issue from the template and will close this one:
https://github.com/RoboStack/ros-noetic/issues/391

Note that if you follow the link from the support page (https://robostack.github.io/support.html), it doesn't give you the option of selecting from a template.

closed 2023-09-01 by nvander

traversaro, 2023-09-04:
> Note that if you follow the link from the support page (https://robostack.github.io/support.html), it doesn't give you the option of selecting from a template.

Thanks for reporting the issue, I tried to fix it in https://github.com/RoboStack/robostack.github.io/pull/26 .