RoboStack/ros-noetic #391: Failed to import roslz4 package

nvander, 2023-09-01:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

When attempting to import the `roslz4` module from `ros-noetic-roslz4`, it fails due to a missing `libpython2.7.so.1.0` library.

Here is the error I see:
```
(base) root@0441547fb234:~# python -c "import roslz4"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/opt/conda/lib/python3.9/site-packages/roslz4/__init__.py", line 33, in <module>
    from ._roslz4 import *
ImportError: libpython2.7.so.1.0: cannot open shared object file: No such file or directory
```

Here is the version of `ros-noetic-roslz4` I have installed:
```
(base) root@0441547fb234:~# micromamba list ros-noetic-roslz4
List of packages in environment: "/opt/conda"

  Name               Version  Build            Channel          
──────────────────────────────────────────────────────────────────
  ros-noetic-roslz4  1.15.15  py39hac30774_15  robostack-staging
```

Let me know if I can provide any additional detail that may be helpful. Thanks!

### Installed packages

```shell
List of packages in environment: "/opt/conda"

  Name                             Version       Build                Channel          
─────────────────────────────────────────────────────────────────────────────────────────
  _libgcc_mutex                    0.1           conda_forge          conda-forge      
  _openmp_mutex                    4.5           2_gnu                conda-forge      
  apr                              1.7.0         h7f98852_5           conda-forge      
  binutils                         2.40          hdd6e379_0           conda-forge      
  binutils_impl_linux-64           2.40          hf600244_0           conda-forge      
  binutils_linux-64                2.40          hbdbef99_1           conda-forge      
  boost-cpp                        1.78.0        h2c5509c_4           conda-forge      
  bzip2                            1.0.8         h7f98852_4           conda-forge      
  c-ares                           1.19.1        hd590300_0           conda-forge      
  c-compiler                       1.6.0         hd590300_0           conda-forge      
  ca-certificates                  2023.7.22     hbcca054_0           conda-forge      
  catkin_pkg                       0.5.2         pyhd8ed1ab_0         conda-forge      
  cmake                            3.27.4        hcfe8598_4           conda-forge      
  console_bridge                   1.0.2         h924138e_1           conda-forge      
  cxx-compiler                     1.6.0         h00ab1b0_0           conda-forge      
  distro                           1.8.0         pyhd8ed1ab_0         conda-forge      
  docutils                         0.20.1        py39hf3d152e_0       conda-forge      
  empy                             3.3.4         pyh9f0ad1d_1         conda-forge      
  expat                            2.5.0         hcb278e6_1           conda-forge      
  gcc                              12.3.0        h8d2909c_1           conda-forge      
  gcc_impl_linux-64                12.3.0        he2b93b0_0           conda-forge      
  gcc_linux-64                     12.3.0        h76fc315_1           conda-forge      
  geos                             3.12.0        h59595ed_0           conda-forge      
  gettext                          0.21.1        h27087fc_0           conda-forge      
  gmock                            1.14.0        ha770c72_1           conda-forge      
  gmp                              6.2.1         h58526e2_0           conda-forge      
  gpgme                            1.18.0        h27087fc_0           conda-forge      
  gtest                            1.14.0        h00ab1b0_1           conda-forge      
  gxx                              12.3.0        h8d2909c_1           conda-forge      
  gxx_impl_linux-64                12.3.0        he2b93b0_0           conda-forge      
  gxx_linux-64                     12.3.0        h8a814eb_1           conda-forge      
  icu                              73.2          h59595ed_0           conda-forge      
  kernel-headers_linux-64          2.6.32        he073ed8_16          conda-forge      
[... 103 lines cut ...]

### Environment info

```shell
(base) root@0441547fb234:~# micromamba info

       libmamba version : 1.4.7
     micromamba version : 1.4.7
           curl version : libcurl/7.88.1 OpenSSL/3.1.1 zlib/1.2.13 zstd/1.5.2 libssh2/1.11.0 nghttp2/1.52.0
     libarchive version : libarchive 3.6.2 zlib/1.2.13 bz2lib/1.0.8 libzstd/1.5.2
       envs directories : /opt/conda/envs
          package cache : /opt/conda/pkgs
                          /root/.mamba/pkgs
            environment : base (active)
           env location : /opt/conda
      user config files : /root/.mambarc
 populated config files : 
       virtual packages : __unix=0=0
                          __linux=6.2.0=0
                          __glibc=2.31=0
                          __archspec=1=x86_64
               channels : 
       base environment : /opt/conda
               platform : linux-64
```

nvander, 2023-09-01:
OK, I may have been too quick to report this issue. It looks like the culprit may be a downstream pip install of the `bagpy` library may be clobbering some `roslz4` files. If I do an install of just `ros-noetic-roslz4` in a new environment, then the `roslz4` module imports just fine and I am able to work with lz4-compressed bag files.

Sorry for the trouble and thanks for being quick to respond!

Tobias-Fischer, 2023-09-01:
Great to hear you managed to find the culprit! Baggy should be easy to move to conda forge via grayskull, give it a go if you like and let me know if you run into issues.

closed 2023-09-02 by Tobias-Fischer