RoboStack/ros-noetic #404: Package request: ros-noetic-mavros-extras on macOS arm64

BillObamax, 2023-09-23:
### Package name

ros-noetic-mavros-extras

### Operating System(s)

- [ ] linux-64
- [ ] linux-aarch64
- [ ] osx-64
- [X] osx-arm64
- [ ] win-64

### Additional context (optional)

I found a `osx-64` build of `ros-noetic-mavros-extras` on [Ros Noetic Mavros Extras :: Anaconda.org](https://anaconda.org/RoboStack/ros-noetic-mavros-extras), but I am currently on macOS Ventura with Apple Silicon M1 Pro. Please make a `osx-arm64`'s `ros-noetic-mavros-extras` release.

### Links and references (optional)

https://anaconda.org/RoboStack/ros-noetic-mavros-extras

Tobias-Fischer, 2023-09-24:
mavros is a beast to package. I’m not sure if it ever worked as intended.

I personally won’t have the capacity to package mavros, but would be willing to help a contribution by a volunteer.

Tobias-Fischer, 2024-02-18:
Actually, I just tried and it compiles fine. I'll trigger the build. Apologies for the long wait!

closed 2024-02-18 by Tobias-Fischer