RoboStack/ros-noetic #406: qt-main 5.15.8 required but only up to 5.15.6 available in conda

joansola, 2023-09-24:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

I am following the Getting Started guide in the main RoboStack page.

My computer is a MacBook with the Apple M2 chip.

I cannot install ros-noetic-desktop. Mamba complains with this:

```
Could not solve for environment specs
The following package could not be installed
└─ ros-noetic-desktop   is uninstallable because it requires
   └─ ros-noetic-joint-state-publisher-gui  , which requires
      └─ ros-noetic-python-qt-binding  , which requires
         └─ qt-main >=5.15.8,<5.16.0a0 , which does not exist (perhaps a missing channel).
```

I tried installing qt-main with mamba. Got only 5.15.6 but not later versions:

```
  + qt-main            5.15.6  hf3dd84c_1          conda-forge/osx-arm64     Cached
```

I tried a new environment just in case I hade some things messed up. But no success.

### Installed packages

```shell
# Name                    Version                   Build  Channel
argcomplete               2.0.0              pyhd8ed1ab_0    conda-forge
attrs                     22.1.0             pyh71513ae_1    conda-forge
bzip2                     1.0.8                h3422bc3_4    conda-forge
c-ares                    1.18.1               h3422bc3_0    conda-forge
c-compiler                1.5.1                h5008568_0    conda-forge
ca-certificates           2022.9.24            h4653dfc_0    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.0              pyhd8ed1ab_0    conda-forge
cctools                   973.0.1             hcbb26d4_10    conda-forge
cctools_osx-arm64         973.0.1             he930820_10    conda-forge
clang                     14.0.6               hce30654_0    conda-forge
clang-14                  14.0.6          default_h81a5282_0    conda-forge
clang_osx-arm64           14.0.6               h15773ab_3    conda-forge
clangxx                   14.0.6          default_hb7ecf47_0    conda-forge
clangxx_osx-arm64         14.0.6               he29aa18_3    conda-forge
cmake                     3.25.0               hf234bd0_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.4.2              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
colcon-cmake              0.2.27             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0           py311h267d04e_1    conda-forge
colcon-core               0.10.0             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.5                      py_0    conda-forge
colcon-devtools           0.2.2                      py_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-output             0.2.12                     py_0    conda-forge
colcon-package-information 0.3.3                      py_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
colcon-parallel-executor  0.2.4                      py_0    conda-forge
colcon-pkg-config         0.1.0                      py_0    conda-forge
colcon-powershell         0.3.6                      py_0    conda-forge
colcon-python-setup-py    0.2.7                      py_0    conda-forge
colcon-recursive-crawl    0.2.1                      py_0    conda-forge
colcon-ros                0.3.23             pyhd8ed1ab_0    conda-forge
colcon-test-result        0.3.8                      py_0    conda-forge
colcon-zsh                0.4.0              pyhd8ed1ab_0    conda-forge
[... 97 lines cut ...]
                          /Users/jsola/mambaforge/envs/ros_env/.condarc
          conda version : 23.1.0
    conda-build version : not installed
         python version : 3.10.12.final.0
       virtual packages : __archspec=1=arm64
                          __osx=13.5.2=0
                          __unix=0=0
       base environment : /Users/jsola/mambaforge  (writable)
      conda av data dir : /Users/jsola/mambaforge/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-arm64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /Users/jsola/mambaforge/pkgs
                          /Users/jsola/.conda/pkgs
       envs directories : /Users/jsola/mambaforge/envs
                          /Users/jsola/.conda/envs
               platform : osx-arm64
             user-agent : conda/23.1.0 requests/2.31.0 CPython/3.10.12 Darwin/22.6.0 OSX/13.5.2
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```
```

traversaro, 2023-09-25:
This is strange, qt-main 5.15.8 is available in conda-forge : https://anaconda.org/conda-forge/qt-main/files . Can you try to create a clean environment with just qt-main to see which qt-main it installs? I.e. 

~~~
mamba create -n qt-main qt-main
~~~

joansola, 2023-09-27:
Thanks Silvio.

I get this list below. In particular, this line:

```
qt-main                   5.15.6               hf3dd84c_1    conda-forge
```

May I recall that I am on Mac M2, so with arm64? Or it does not matter?

and the complete list:
```
# Name                    Version                   Build  Channel
bzip2                     1.0.8                h3422bc3_4    conda-forge
ca-certificates           2022.9.24            h4653dfc_0    conda-forge
gettext                   0.21.1               h0186832_0    conda-forge
glib                      2.74.1               hb5ab8b9_1    conda-forge
glib-tools                2.74.1               hb5ab8b9_1    conda-forge
gst-plugins-base          1.21.1               h8b7775e_1    conda-forge
gstreamer                 1.21.1               hcb7b3dd_1    conda-forge
icu                       70.1                 h6b3803e_0    conda-forge
jpeg                      9e                   he4db4b2_2    conda-forge
krb5                      1.19.3               hf9b2bbe_0    conda-forge
libclang                  15.0.5          default_h9e54d93_0    conda-forge
libclang13                15.0.5          default_h7d574e7_0    conda-forge
libcxx                    14.0.6               h2692d47_0    conda-forge
libedit                   3.1.20191231         hc8eb9b7_2    conda-forge
libffi                    3.4.2                h3422bc3_5    conda-forge
libglib                   2.74.1               h4646484_1    conda-forge
libiconv                  1.17                 he4db4b2_0    conda-forge
libllvm15                 15.0.5               h62b9111_0    conda-forge
libogg                    1.3.4                h27ca646_1    conda-forge
libopus                   1.3.1                h27ca646_1    conda-forge
libpng                    1.6.38               h76d750c_0    conda-forge
libpq                     14.5                 h3f71257_1    conda-forge
libsqlite                 3.40.0               h76d750c_0    conda-forge
libvorbis                 1.3.7                h9f76cd9_0    conda-forge
libxml2                   2.10.3               h87b0503_0    conda-forge
libzlib                   1.2.13               h03a7124_4    conda-forge
mysql-common              8.0.31               hab468bb_0    conda-forge
mysql-libs                8.0.31               hea58576_0    conda-forge
ncurses                   6.3                  h07bb92c_1    conda-forge
nspr                      4.32                 hbdafb3b_1    conda-forge
nss                       3.78                 h1483a63_0    conda-forge
openssl                   1.1.1s               h03a7124_0    conda-forge
pcre2                     10.40                hb34f9b4_0    conda-forge
pip                       22.3.1             pyhd8ed1ab_0    conda-forge
python                    3.10.6          hbce4517_0_cpython    conda-forge
python_abi                3.10                    2_cp310    conda-forge
qt-main                   5.15.6               hf3dd84c_1    conda-forge
readline                  8.1.2                h46ed386_0    conda-forge
setuptools                65.5.1             pyhd8ed1ab_0    conda-forge
sqlite                    3.40.0               h2229b38_0    conda-forge
tk                        8.6.12               he1e0b03_0    conda-forge
tzdata                    2022f                h191b570_0    conda-forge
wheel                     0.38.4             pyhd8ed1ab_0    conda-forge
xz                        5.2.6                h57fd34a_0    conda-forge
zstd                      1.5.2                h8128057_4    conda-forge
```

joansola, 2023-09-27:
Also for extra context.

I did not mention this since I considered it unrelated. But I am getting this warning concerning `zstandard`. May I have a broken mamba/conda? Should I reinstall all from zero? Your comment about qt-main version makes me think so.

```
/Users/jsola/mambaforge/lib/python3.10/site-packages/conda_package_streaming/package_streaming.py:19: UserWarning: zstandard could not be imported. Running without .conda support.
  warnings.warn("zstandard could not be imported. Running without .conda support.")
/Users/jsola/mambaforge/lib/python3.10/site-packages/conda_package_handling/api.py:29: UserWarning: Install zstandard Python bindings for .conda support
  _warnings.warn("Install zstandard Python bindings for .conda support")
```

traversaro, 2023-09-27:
> May I recall that I am on Mac M2, so with arm64? Or it does not matter?

It does matter, but a recent qt-main is available on arm64 as well, see the file `https://anaconda.org/conda-forge/qt-main/5.15.8/download/osx-arm64/qt-main-5.15.8-h219f738_16.conda` from https://anaconda.org/conda-forge/qt-main/files . I also tested this locally on my M1 mac mini, and indeed 5.15.8 was installed.

> I did not mention this since I considered it unrelated. But I am getting this warning concerning zstandard. May I have a broken mamba/conda? Should I reinstall all from zero? Your comment about qt-main version makes me think so.

Thanks for mentioning, I think that is related. Basically there are two variants of conda file, .tar.bz2 and .conda files, and the error message mentions that "Running without .conda support", and indeed 5.15.6 is the newest package qt-main package available as .tar.bz2, see the end of https://anaconda.org/conda-forge/qt-main/files?page=2 .

> May I have a broken mamba/conda? Should I reinstall all from zero? Your comment about qt-main version makes me think so.

Not sure what happened. However, probably a thing that you can quickly try without reinstalling everything is to install zstandard in the base environment, i.e. : 

~~~
mamba install -n base zstandard
~~~

joansola, 2023-09-27:
OK I reinstalled Mambaforge from scratch and now I can see and install qt-main 5.15.8.

Will continue from here. thanks for the feedback.

joansola, 2023-09-27:
Not sure what happened. However, probably a thing that you can quickly try without reinstalling everything is to install zstandard in the base environment, i.e. :

```
mamba install -n base zstandard
```

This did not work -- can-t remember the outcome. But, issue solved through clean reinstall of mambaforge.

Thanks!

traversaro, 2023-09-27:
Ok, thanks! Feel free to close the issue if it is solved.

joansola, 2023-09-27:
@traversaro if you are still around.

All was going well with my new installation. However, after  compiling and installing some libs in my new environment, I wanted to proceed with the robostack material.

My first line was

```
(wolf) jsola@10-192-184-93client ~ % conda config --env --add channels robostack-staging
```

which immediately made the old problem with zstandard reappear:

```
/Users/jsola/mambaforge/lib/python3.10/site-packages/conda_package_streaming/package_streaming.py:19: UserWarning: zstandard could not be imported. Running without .conda support.
  warnings.warn("zstandard could not be imported. Running without .conda support.")
/Users/jsola/mambaforge/lib/python3.10/site-packages/conda_package_handling/api.py:29: UserWarning: Install zstandard Python bindings for .conda support
  _warnings.warn("Install zstandard Python bindings for .conda support")
```

Do you have any idea of what am I missing? I am afraid I will run into the same problems with qt-main. Indeed, now I have this:

```
(wolf) jsola@10-192-184-93client ~ % mamba install qt-main
/Users/jsola/mambaforge/lib/python3.10/site-packages/conda_package_streaming/package_streaming.py:19: UserWarning: zstandard could not be imported. Running without .conda support.
  warnings.warn("zstandard could not be imported. Running without .conda support.")
/Users/jsola/mambaforge/lib/python3.10/site-packages/conda_package_handling/api.py:29: UserWarning: Install zstandard Python bindings for .conda support
  _warnings.warn("Install zstandard Python bindings for .conda support")

                  __    __    __    __
                 /  \  /  \  /  \  /  \
                /    \/    \/    \/    \
███████████████/  /██/  /██/  /██/  /████████████████████████
              /  / \   / \   / \   / \  \____
             /  /   \_/   \_/   \_/   \    o \__,
            / _/                       \_____/  `
            |/
        ███╗   ███╗ █████╗ ███╗   ███╗██████╗  █████╗
        ████╗ ████║██╔══██╗████╗ ████║██╔══██╗██╔══██╗
        ██╔████╔██║███████║██╔████╔██║██████╔╝███████║
        ██║╚██╔╝██║██╔══██║██║╚██╔╝██║██╔══██╗██╔══██║
        ██║ ╚═╝ ██║██║  ██║██║ ╚═╝ ██║██████╔╝██║  ██║
        ╚═╝     ╚═╝╚═╝  ╚═╝╚═╝     ╚═╝╚═════╝ ╚═╝  ╚═╝

        mamba (1.4.2) supported by @QuantStack

        GitHub:  https://github.com/mamba-org/mamba
        Twitter: https://twitter.com/QuantStack

█████████████████████████████████████████████████████████████

/Users/jsola/mambaforge/lib/python3.10/site-packages/conda_package_streaming/package_streaming.py:19: UserWarning: zstandard could not be imported. Running without .conda support.
  warnings.warn("zstandard could not be imported. Running without .conda support.")
/Users/jsola/mambaforge/lib/python3.10/site-packages/conda_package_handling/api.py:29: UserWarning: Install zstandard Python bindings for .conda support
  _warnings.warn("Install zstandard Python bindings for .conda support")

Looking for: ['qt-main']

robostack-staging/noarch                           130.0 B @ 153.0 B/s  0.8s
robostack-staging/osx-arm64                        201.1kB @ 128.2kB/s  1.6s
conda-forge/osx-arm64                                7.7MB @   4.0MB/s  2.1s
conda-forge/noarch                                  14.0MB @   3.8MB/s  3.9s

Pinned packages:
  - python 3.11.*

Could not solve for environment specs
The following packages are incompatible
└─ qt-main   is installable with the potential options
   ├─ qt-main 5.15.2 would require
   │  └─ icu >=69.1,<70.0a0 , which can be installed;
   ├─ qt-main 5.15.2 would require
   │  └─ icu >=68.2,<69.0a0 , which can be installed;
   └─ qt-main [5.15.3|5.15.4|5.15.6] would require
      └─ icu >=70.1,<71.0a0 , which can be installed.

```

Last thing: Should I create a new issue?

traversaro, 2023-09-27:
> Last thing: Should I create a new issue?

I would say no, as the actual problem is still the same. 

Can you try to activate the base environment, and run `python -c "import zstandard"`? i.e. : 

~~~
mamba activate base
python -c "import zstandard"
~~~

joansola, 2023-09-27:
on the first one

```
(base) jsola@10-192-184-93client wolf-lib % python -c "import zstandard"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/jsola/mambaforge/lib/python3.10/site-packages/zstandard/__init__.py", line 39, in <module>
    from .backend_c import *  # type: ignore
ImportError: zstd C API versions mismatch; Python bindings were not compiled/linked against expected zstd version (10505 returned by the lib, 10502 hardcoded in zstd headers, 10502 hardcoded in the cext)
```

on the second one

```
(base) jsola@10-192-184-93client wolf-lib % mamba list
/Users/jsola/mambaforge/lib/python3.10/site-packages/conda_package_streaming/package_streaming.py:19: UserWarning: zstandard could not be imported. Running without .conda support.
  warnings.warn("zstandard could not be imported. Running without .conda support.")
/Users/jsola/mambaforge/lib/python3.10/site-packages/conda_package_handling/api.py:29: UserWarning: Install zstandard Python bindings for .conda support
  _warnings.warn("Install zstandard Python bindings for .conda support")
# packages in environment at /Users/jsola/mambaforge:
#
# Name                    Version                   Build  Channel
brotli-python             1.0.9           py310h0f1eb42_9    conda-forge
bzip2                     1.0.8                h3422bc3_4    conda-forge
c-ares                    1.19.1               hb547adb_0    conda-forge
ca-certificates           2023.5.7             hf0a4a13_0    conda-forge
certifi                   2023.5.7           pyhd8ed1ab_0    conda-forge
cffi                      1.15.1          py310h2399d43_3    conda-forge
charset-normalizer        3.2.0              pyhd8ed1ab_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
conda                     23.1.0          py310hbe9552e_0    conda-forge
conda-package-handling    2.0.2              pyh38be061_0    conda-forge
conda-package-streaming   0.8.0              pyhd8ed1ab_0    conda-forge
cryptography              41.0.2          py310hdd3b5e7_0    conda-forge
fmt                       9.1.0                hffc8910_0    conda-forge
icu                       72.1                 he12128b_0    conda-forge
idna                      3.4                pyhd8ed1ab_0    conda-forge
krb5                      1.21.1               h92f50d5_0    conda-forge
libarchive                3.6.2                h82b9b87_1    conda-forge
libcurl                   8.1.2                hc52a3a8_1    conda-forge
libcxx                    16.0.6               h4653b0c_0    conda-forge
libedit                   3.1.20191231         hc8eb9b7_2    conda-forge
libev                     4.33                 h642e427_1    conda-forge
libffi                    3.4.2                h3422bc3_5    conda-forge
libiconv                  1.17                 he4db4b2_0    conda-forge
libmamba                  1.4.2                h7d1d596_0    conda-forge
libmambapy                1.4.2           py310h34b6e76_0    conda-forge
libnghttp2                1.52.0               hae82a92_0    conda-forge
libsolv                   0.7.24               ha614eb4_1    conda-forge
libsqlite                 3.42.0               hb31c410_0    conda-forge
libssh2                   1.11.0               h7a5bd25_0    conda-forge
libxml2                   2.11.4               he3bdae6_0    conda-forge
libzlib                   1.2.13               h53f4e23_5    conda-forge
lz4-c                     1.9.4                hb7217d7_0    conda-forge
lzo                       2.10              h642e427_1000    conda-forge
mamba                     1.4.2           py310ha5d4528_0    conda-forge
ncurses                   6.4                  h7ea286d_0    conda-forge
openssl                   3.1.1                h53f4e23_1    conda-forge
pip                       23.1.2             pyhd8ed1ab_0    conda-forge
pluggy                    1.2.0              pyhd8ed1ab_0    conda-forge
pybind11-abi              4                    hd8ed1ab_3    conda-forge
pycosat                   0.6.4           py310h8e9501a_1    conda-forge
pycparser                 2.21               pyhd8ed1ab_0    conda-forge
pyopenssl                 23.2.0             pyhd8ed1ab_1    conda-forge
pysocks                   1.7.1              pyha2e5f31_6    conda-forge
python                    3.10.12         h01493a6_0_cpython    conda-forge
python_abi                3.10                    3_cp310    conda-forge
readline                  8.2                  h92ec313_1    conda-forge
reproc                    14.2.4               h1a8c8d9_0    conda-forge
reproc-cpp                14.2.4               hb7217d7_0    conda-forge
requests                  2.31.0             pyhd8ed1ab_0    conda-forge
ruamel.yaml               0.17.32         py310h2aa6e3c_0    conda-forge
ruamel.yaml.clib          0.2.7           py310h8e9501a_1    conda-forge
setuptools                68.0.0             pyhd8ed1ab_0    conda-forge
tk                        8.6.12               he1e0b03_0    conda-forge
toolz                     0.12.0             pyhd8ed1ab_0    conda-forge
tqdm                      4.65.0             pyhd8ed1ab_1    conda-forge
tzdata                    2023c                h71feb2d_0    conda-forge
urllib3                   2.0.3              pyhd8ed1ab_1    conda-forge
wheel                     0.40.0             pyhd8ed1ab_0    conda-forge
xz                        5.2.6                h57fd34a_0    conda-forge
yaml-cpp                  0.7.0                hb7217d7_2    conda-forge
zstandard                 0.19.0          py310h07fcf8c_2    conda-forge
zstd                      1.5.2                h4f39d0f_7    conda-forge
```

joansola, 2023-09-27:
I did install a few libs in `~/mambaforge/envs/wolf` so that I did not have to install them in `/usr/local/`. I hope this is not interfering...

I also had to install `yaml-cpp` to `envs/wolf` even though `yaml-cpp` is part of the base environment. I dont know if this is normal.

joansola, 2023-09-27:
OK I am restarting again from scratch, this time installing first the robostack ROS1-noetic packages.

Then I will install my other libs.

It seems to work fine this way. Will report results asap.

traversaro, 2023-09-27:
The key error is:
> ImportError: zstd C API versions mismatch; Python bindings were not compiled/linked against expected zstd version (10505 returned by the lib, 10502 hardcoded in zstd headers, 10502 hardcoded in the cext)

It seems that somewhere you have a zstd 1.5.5 installed that is used instead of the zstd installed by conda-forge. To debug why this is happening, in a shell were this is happening you can run:
~~~
otool -L /Users/jsola/mambaforge/lib/python3.10/site-packages/zstandard/backend_c.cpython-310-darwin.so
~~~

traversaro, 2023-09-27:
> I did install a few libs in ~/mambaforge/envs/wolf manually (not through mamba, but through make install) so that I did not have to install them in /usr/local/. I hope this is not interfering...

As long as you compiled this libs using the compilers and the dependencies from wolf environment, that is ok.

> I also had to install yaml-cpp to envs/wolf even though yaml-cpp is part of the base environment. I dont know if this is normal.

In general, yes. Nothing ensures that a library installed in the base environment is ABI compatible with libraries installed in any other environment.

joansola, 2023-09-27:
> OK I am restarting again from scratch, this time installing first the robostack ROS1-noetic packages.
> 
> Then I will install my other libs.
> 
> It seems to work fine this way. Will report results asap.

OK I got to install all ROS1 and all my libs together, and compile all my libs. Have not tried ROS1 yet but things look fine now.

I will test the `otool` thing now, will report asap. thanks

joansola, 2023-09-27:
I got this from otool on zstd:

```
(wolf) jsola@10-192-184-93client wolf-lib % otool -L /Users/jsola/mambaforge/lib/python3.10/site-packages/zstandard/backend_c.cpython-310-darwin.so 
/Users/jsola/mambaforge/lib/python3.10/site-packages/zstandard/backend_c.cpython-310-darwin.so:
	@rpath/libzstd.1.dylib (compatibility version 1.0.0, current version 1.5.2)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1292.0.0)
```

traversaro, 2023-09-27:
Is this a terminal in which `python -c "import zstandard"` fails with the error in  https://github.com/RoboStack/ros-noetic/issues/406#issuecomment-1737289941 ?

joansola, 2023-09-27:
this is what comes out of "import zstandard" on this same terminal:
```
(wolf) jsola@10-192-184-93client wolf-lib % python -c "import zstandard"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'zstandard'
```

joansola, 2023-09-27:
but: this is my new installation. Maybe I am confusing you now.

joansola, 2023-09-27:
I , however, keep receiving these kind of warnings regarding zstandard, even in the new installation.

for example, invoking `mamba list` now gives this:

```
(wolf) jsola@10-192-184-93client wolf-lib % mamba list
/Users/jsola/mambaforge/lib/python3.10/site-packages/conda_package_streaming/package_streaming.py:19: UserWarning: zstandard could not be imported. Running without .conda support.
  warnings.warn("zstandard could not be imported. Running without .conda support.")
/Users/jsola/mambaforge/lib/python3.10/site-packages/conda_package_handling/api.py:29: UserWarning: Install zstandard Python bindings for .conda support
  _warnings.warn("Install zstandard Python bindings for .conda support")
# packages in environment at /Users/jsola/mambaforge/envs/wolf:
#
# Name                    Version                   Build  Channel
adwaita-icon-theme        43                       unix_0    conda-forge
aom                       3.5.0                h7ea286d_0    conda-forge
apr                       1.7.0                h9b22ae9_5    conda-forge
...
```

traversaro, 2023-09-27:
> and so that it is all in one place, I post these two:

It seems that you are running those in the `wolf` environment. `mamba` runs in the `base` environment, you should run `python -c "import zstandard"` and `otool -L /Users/jsola/mambaforge/lib/python3.10/site-packages/zstandard/backend_c.cpython-310-darwin.so` from the base environment.

Yes, does "zstandard could not be imported. Running without .conda support" are quite problematic. Do you get the same errors also if you only activate the base environment?

joansola, 2023-09-27:
from base:

```
(base) jsola@10-192-184-93client wolf-lib % mamba list
/Users/jsola/mambaforge/lib/python3.10/site-packages/conda_package_streaming/package_streaming.py:19: UserWarning: zstandard could not be imported. Running without .conda support.
  warnings.warn("zstandard could not be imported. Running without .conda support.")
/Users/jsola/mambaforge/lib/python3.10/site-packages/conda_package_handling/api.py:29: UserWarning: Install zstandard Python bindings for .conda support
  _warnings.warn("Install zstandard Python bindings for .conda support")
# packages in environment at /Users/jsola/mambaforge:
#
# Name                    Version                   Build  Channel
brotli-python             1.0.9           py310h0f1eb42_9    conda-forge
bzip2                     1.0.8                h3422bc3_4    conda-forge

(base) jsola@10-192-184-93client wolf-lib % python -c "import zstandard"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/jsola/mambaforge/lib/python3.10/site-packages/zstandard/__init__.py", line 39, in <module>
    from .backend_c import *  # type: ignore
ImportError: zstd C API versions mismatch; Python bindings were not compiled/linked against expected zstd version (10505 returned by the lib, 10502 hardcoded in zstd headers, 10502 hardcoded in the cext)

(base) jsola@10-192-184-93client wolf-lib % otool -L /Users/jsola/mambaforge/lib/python3.10/site-packages/zstandard/backend_c.cpython-310-darwin.so
/Users/jsola/mambaforge/lib/python3.10/site-packages/zstandard/backend_c.cpython-310-darwin.so:
	@rpath/libzstd.1.dylib (compatibility version 1.0.0, current version 1.5.2)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1292.0.0)
```

traversaro, 2023-09-27:
Sorry, not a macOS expert, apparently `otool -L` is not giving us the exact library loaded (it return 1.5.2, but from the zstandard error it is clear that actually 1.5.5 is loaded: https://github.com/indygreg/python-zstandard/blob/0.19.0/c-ext/backend_c.c#L160). Can you try with:
~~~
DYLD_PRINT_LIBRARIES=1  python -c "import zstandard"
~~~

joansola, 2023-09-27:
FYI: my new installation works for ROS1 and for my libs, all in the same environment:
- I could execute roscore and rviz
- I could compile and run demos of my libs and all their non-ros dependencies, including ceres-solver, opencv, yaml-cpp and apriltag.

joansola, 2023-09-27:
```
(base) jsola@10-192-184-93client wolf-lib % DYLD_PRINT_LIBRARIES=1  python -c "import zstandard"
dyld[49577]: <0CA3BC4A-72F4-3128-98CB-4F8C9FE9B154> /Users/jsola/mambaforge/bin/python3.10
dyld[49577]: <422AACAD-6E3A-37DB-B088-17F56C545FEC> /usr/lib/libSystem.B.dylib
dyld[49577]: <FC36C110-DC4A-3131-8694-2F23E42BDA80> /usr/lib/system/libcache.dylib
dyld[49577]: <2B973778-B3AC-334C-839A-235B1B0310D8> /usr/lib/system/libcommonCrypto.dylib
dyld[49577]: <C15B1BD0-04BE-3A79-86A9-B2F3E4375878> /usr/lib/system/libcompiler_rt.dylib
dyld[49577]: <4D867984-354A-3147-95D1-4C5C0886D1AE> /usr/lib/system/libcopyfile.dylib
dyld[49577]: <4CEC1F69-3585-308B-884B-D293CEA660FA> /usr/lib/system/libcorecrypto.dylib
dyld[49577]: <9897030F-75D3-374B-8787-322D3D72E096> /usr/lib/system/libdispatch.dylib
dyld[49577]: <4B488059-5F44-3C73-9034-DAC2DEAB7D58> /usr/lib/system/libdyld.dylib
dyld[49577]: <AF5DEDE2-18F4-3C11-989D-3F394CD35FB3> /usr/lib/system/libkeymgr.dylib
dyld[49577]: <8341733F-3C14-3040-B3AD-BBD7FC557D7F> /usr/lib/system/libmacho.dylib
dyld[49577]: <568CA6AD-EAA9-3E05-B2C4-FD5AA99F375E> /usr/lib/system/libquarantine.dylib
dyld[49577]: <E91D9D9C-6063-3B1C-B76E-B0973E69E756> /usr/lib/system/libremovefile.dylib
dyld[49577]: <E04D9AFA-2C87-3C00-93DA-62B13E95610D> /usr/lib/system/libsystem_asl.dylib
dyld[49577]: <73A49FC5-CB95-397D-93AB-C8C9E651E576> /usr/lib/system/libsystem_blocks.dylib
dyld[49577]: <949943E2-52CB-3C95-AB08-7ED984333D03> /usr/lib/system/libsystem_c.dylib
dyld[49577]: <CFE1A849-2EFF-3041-9A76-7A70956A8A6D> /usr/lib/system/libsystem_collections.dylib
dyld[49577]: <2891A23A-0407-3CB6-B7DB-BDBEF413C7D7> /usr/lib/system/libsystem_configuration.dylib
dyld[49577]: <7ED744CD-3E25-36BB-8D77-7AD77644305D> /usr/lib/system/libsystem_containermanager.dylib
dyld[49577]: <96E327DE-3682-31E0-A3DD-571AD65EFFA6> /usr/lib/system/libsystem_coreservices.dylib
dyld[49577]: <0773E300-1B02-3757-A2CB-6A0B299B67D2> /usr/lib/system/libsystem_darwin.dylib
dyld[49577]: <BE8BAFA6-2637-30E2-8FBB-86DF629D8A8D> /usr/lib/system/libsystem_dnssd.dylib
dyld[49577]: <805E12B0-434D-3E9D-AA51-77EE512E608A> /usr/lib/system/libsystem_featureflags.dylib
dyld[49577]: <4CC3E383-C548-3975-B464-49E2EE7DCF4E> /usr/lib/system/libsystem_info.dylib
dyld[49577]: <006DC4B0-E134-3158-9966-A2BEE15B4F79> /usr/lib/system/libsystem_m.dylib
dyld[49577]: <8F26F714-85AE-355E-AFCB-3530B09C76CD> /usr/lib/system/libsystem_malloc.dylib
dyld[49577]: <23DF1A08-7C44-374A-90C9-F75405167766> /usr/lib/system/libsystem_networkextension.dylib
dyld[49577]: <7A0712AA-B031-3AF7-9FC3-FA63C2ED25FE> /usr/lib/system/libsystem_notify.dylib
dyld[49577]: <DFD92A1A-4E4F-3EFA-AF01-578B51F0104F> /usr/lib/system/libsystem_sandbox.dylib
dyld[49577]: <8C1FC817-62DF-396F-AD44-26DFC1442157> /usr/lib/system/libsystem_secinit.dylib
dyld[49577]: <1F30FB9A-BDF9-32DB-A709-8417666A7E45> /usr/lib/system/libsystem_pthread.dylib
dyld[49577]: <6EEAA167-2E3C-3036-B16E-975FED44A5CE> /usr/lib/system/libsystem_symptoms.dylib
dyld[49577]: <AA2DCC11-35D9-3E5B-8421-D3B4CBA0E2D9> /usr/lib/system/libsystem_trace.dylib
dyld[49577]: <F10C5B1C-4B40-3DDE-8AC3-38B4E01A674F> /usr/lib/system/libunwind.dylib
dyld[49577]: <33177E90-9BB2-36C5-9B60-21CBCA32BB70> /usr/lib/system/libxpc.dylib
dyld[49577]: <3E0CEEEE-F6BF-3497-80DA-116F0269AAD2> /usr/lib/libc++abi.dylib
dyld[49577]: <6D458354-3C39-3F35-A3BB-D01A78817B4A> /usr/lib/system/libsystem_platform.dylib
dyld[49577]: <08C5FE2A-B0BF-3AB6-BB42-460C18917D33> /usr/lib/system/libsystem_kernel.dylib
dyld[49577]: <AC12887C-D698-3627-B9D1-D2E5055A5DA4> /usr/lib/libobjc.A.dylib
dyld[49577]: <AF6B228E-26B1-32B8-82EF-B4D72EB8E79C> /usr/lib/liboah.dylib
dyld[49577]: <D9229D7D-8F7E-3E8F-A88E-15125158BCAE> /Users/jsola/mambaforge/envs/wolf/lib/libc++.1.0.dylib
dyld[49577]: <EF98C719-4D23-3563-95D7-26D29B111599> /Users/jsola/mambaforge/lib/python3.10/lib-dynload/fcntl.cpython-310-darwin.so
dyld[49577]: <ECAB6E02-F652-39C0-B908-8671C955F6EE> /Users/jsola/mambaforge/lib/python3.10/lib-dynload/_posixsubprocess.cpython-310-darwin.so
dyld[49577]: <3B3739B6-330A-3A6C-86D3-00AA84A119C6> /Users/jsola/mambaforge/lib/python3.10/lib-dynload/select.cpython-310-darwin.so
dyld[49577]: <27F1A934-355E-390C-B2E3-C2C54355C0EF> /Users/jsola/mambaforge/lib/python3.10/lib-dynload/math.cpython-310-darwin.so
dyld[49577]: <48A46CA1-C346-3C09-9D1B-BA23B0944621> /Users/jsola/mambaforge/lib/python3.10/site-packages/zstandard/backend_c.cpython-310-darwin.so
dyld[49577]: <4DBA0BB4-B15E-3BAE-8F6C-59DCAE6B6AE8> /Users/jsola/mambaforge/envs/wolf/lib/libzstd.1.5.5.dylib
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Users/jsola/mambaforge/lib/python3.10/site-packages/zstandard/__init__.py", line 39, in <module>
    from .backend_c import *  # type: ignore
ImportError: zstd C API versions mismatch; Python bindings were not compiled/linked against expected zstd version (10505 returned by the lib, 10502 hardcoded in zstd headers, 10502 hardcoded in the cext)

traversaro, 2023-09-27:
Bingo, it is loading the zstd from `wolf` environment instead of base: 

> /Users/jsola/mambaforge/envs/wolf/lib/libzstd.1.5.5.dylib

Did you by any change activated base on top of wolfv? Or modified `DYLD_LIBRARY_PATH` or similar to point to wolf?

joansola, 2023-09-27:
oh, yes. I have `DYLD_LIBRARY_PATH` pointing to wolf.

joansola, 2023-09-27:
May I use both paths, e.g. like this

```sh

export DYLD_LIBRARY_PATH=/Users/jsola/mambaforge/envs/wolf/lib:/Users/jsola/mambaforge/lib:$LD_LIBRARY_PATH
```

traversaro, 2023-09-27:
> May I use both paths, e.g. like this
> 
> ```shell
> export DYLD_LIBRARY_PATH=/Users/jsola/mambaforge/envs/wolf/lib:/Users/jsola/mambaforge/lib:$LD_LIBRARY_PATH
> ```

I am afraid that will not work, as depending on the environment, you need to either have first one directory or another. If you just want to set an environment variable in a specific conda environment, you can run `conda env config vars set my_var=value`.

traversaro, 2023-09-27:
> I should use RPATH in MacOS, but I dont know how, so I use the equivalent to linux's `LD_LIBRARY_PATH`

In our project we have being using this CMake macro: https://github.com/robotology/how-to-export-cpp-library/blob/afb21efb655e7b1cc11636c3d42aad9e02fb626f/CMakeLists.txt#L108 and it is working fine. It is quite old so it can be probably be simplified, but it works for us.

joansola, 2023-09-27:
wooo cool!

thanks so much! I would have never ever foud this alone!!

> I am afraid that will not work, as depending on the environment, you need to either have first one directory or another. If you just want to set an environment variable in a specific conda [...]

I will do like you say. The export thing worked, but I will not be able to reuse it for other envs, so it's not a valid solution.

joansola, 2023-09-27:
this worked fine

```
conda env config vars set my_var=value.
```

closing again :-)

closed 2023-09-27 by joansola