RoboStack/ros-noetic #409: ‘lnvoking make -j8 -l8 failed’

elyeesian, 2023-09-29:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Running catkin_make after initializing a catkin workspace (CMakeLists.txt created), gives the following error. I’m running ros on an M1 Mac.

 I think it has something to do with the amount of parallel computing cores? But my native OS has eight cores, so I’m not sure what the issue is.

### Installed packages

```shell
# packages in environment at /opt/homebrew/Caskroom/mambaforge/base:
#
# Name                    Version                   Build  Channel
aom                       3.5.0                h7ea286d_0    conda-forge
assimp                    5.2.5                h276577b_0    conda-forge
atk-1.0                   2.38.0               hcb7b3dd_1    conda-forge
blosc                     1.21.5               hc338f07_0    conda-forge
boltons                   23.0.0             pyhd8ed1ab_0    conda-forge
boost-cpp                 1.78.0               h00df5ed_4    conda-forge
brotli-python             1.1.0           py310h1253130_0    conda-forge
bullet-cpp                3.24            py310h2b830bf_0    conda-forge
bzip2                     1.0.8                h3422bc3_4    conda-forge
c-ares                    1.19.1               hb547adb_0    conda-forge
ca-certificates           2023.7.22            hf0a4a13_0    conda-forge
cairo                     1.16.0            hc5b65c1_1017    conda-forge
certifi                   2023.7.22          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1          py310h2399d43_3    conda-forge
cfitsio                   4.3.0                hca87796_0    conda-forge
charset-normalizer        3.2.0              pyhd8ed1ab_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
conda                     23.3.1          py310hbe9552e_0    conda-forge
conda-libmamba-solver     23.3.0             pyhd8ed1ab_0    conda-forge
conda-package-handling    2.2.0              pyh38be061_0    conda-forge
conda-package-streaming   0.9.0              pyhd8ed1ab_0    conda-forge
console_bridge            1.0.2                h3e96240_1    conda-forge
cppzmq                    4.10.0               h6c41c7e_0    conda-forge
cryptography              41.0.3          py310hdd3b5e7_0    conda-forge
dartsim                   6.13.0               h57cb2a2_1    conda-forge
dav1d                     1.2.1                hb547adb_0    conda-forge
eigen                     3.4.0                h1995070_0    conda-forge
expat                     2.5.0                hb7217d7_1    conda-forge
fcl                       0.7.0                h88a8a2c_2    conda-forge
ffmpeg                    6.0.0           gpl_hb1d784b_104    conda-forge
flann                     1.9.1             h00b795f_1014    conda-forge
fmt                       9.1.0                hffc8910_0    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.2               h82840c6_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
freeimage                 3.18.0              h0c8dc6b_16    conda-forge
freetype                  2.12.1               hadb7bae_2    conda-forge
freexl                    1.0.6                h1a8c8d9_1    conda-forge
fribidi                   1.0.10               h27ca646_0    conda-forge
gazebo                    11.13.0              h4bed19e_6    conda-forge
gdk-pixbuf                2.42.10              h1ac0d0d_2    conda-forge
geos                      3.12.0               h13dd4ca_0    conda-forge
geotiff                   1.7.1               h00e2a8a_11    conda-forge
gettext                   0.21.1               h0186832_0    conda-forge
giflib                    5.2.1                h1a8c8d9_3    conda-forge
glib                      2.78.0               ha614eb4_0    conda-forge
glib-tools                2.78.0               ha614eb4_0    conda-forge
gmp                       6.2.1                h9f76cd9_0    conda-forge
gnutls                    3.7.8                h9f1a10d_0    conda-forge
graphite2                 1.3.13            h9f76cd9_1001    conda-forge
[... 214 lines cut ...]
    active env location : /opt/homebrew/Caskroom/mambaforge/base
            shell level : 1
       user config file : /Users/goddang/.condarc
 populated config files : /opt/homebrew/Caskroom/mambaforge/base/.condarc
          conda version : 23.3.1
    conda-build version : not installed
         python version : 3.10.12.final.0
       virtual packages : __archspec=1=arm64
                          __osx=13.1=0
                          __unix=0=0
       base environment : /opt/homebrew/Caskroom/mambaforge/base  (writable)
      conda av data dir : /opt/homebrew/Caskroom/mambaforge/base/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /opt/homebrew/Caskroom/mambaforge/base/pkgs
                          /Users/goddang/.conda/pkgs
       envs directories : /opt/homebrew/Caskroom/mambaforge/base/envs
                          /Users/goddang/.conda/envs
               platform : osx-arm64
             user-agent : conda/23.3.1 requests/2.31.0 CPython/3.10.12 Darwin/22.2.0 OSX/13.1
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

elyeesian, 2023-09-29:
Forgot to add robostack-staging packages from ros_env, here’s an updated package list: 

```shell
(ros_env) goddang@wireless-10-104-81-194 ~ % mamba list
# packages in environment at /opt/homebrew/Caskroom/mambaforge/base/envs/ros_env:
#
# Name                    Version                   Build  Channel
adwaita-icon-theme        43                       unix_0    conda-forge
aom                       3.5.0                h7ea286d_0    conda-forge
apr                       1.7.0                h9b22ae9_5    conda-forge
argcomplete               3.1.2              pyhd8ed1ab_0    conda-forge
assimp                    5.2.5                h276577b_0    conda-forge
atk-1.0                   2.38.0               hcb7b3dd_1    conda-forge
bcrypt                    4.0.1            py39haff01fd_0    conda-forge
blosc                     1.21.5               hc338f07_0    conda-forge
boost                     1.78.0           py39h99de9ae_4    conda-forge
boost-cpp                 1.78.0               hf1d6563_2    conda-forge
brotli                    1.1.0                hb547adb_0    conda-forge
brotli-bin                1.1.0                hb547adb_0    conda-forge
bullet                    3.21                 h85045c0_4    conda-forge
bullet-cpp                3.21             py39hde7b980_4    conda-forge
bzip2                     1.0.8                h3422bc3_4    conda-forge
c-ares                    1.19.1               hb547adb_0    conda-forge
c-compiler                1.6.0                hd291e01_0    conda-forge
ca-certificates           2023.7.22            hf0a4a13_0    conda-forge
cairo                     1.16.0            h73a0509_1014    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_0    conda-forge
cctools                   973.0.1             hd1ac623_14    conda-forge
cctools_osx-arm64         973.0.1             h2a25c60_14    conda-forge
certifi                   2023.7.22          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1           py39h7e6b969_3    conda-forge
cfitsio                   4.2.0                h2f961c4_0    conda-forge
clang                     15.0.7               hce30654_3    conda-forge
clang-15                  15.0.7          default_h5dc8d65_3    conda-forge
clang_osx-arm64           15.0.7               h77e971b_3    conda-forge
clangxx                   15.0.7          default_h610c423_3    conda-forge
clangxx_osx-arm64         15.0.7               h768a7fd_3    conda-forge
cmake                     3.26.3               hf234bd0_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.5.0              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
colcon-cmake              0.2.27             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0            py39h2804cbe_1    conda-forge
colcon-core               0.12.1             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.2.2                      py_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-output             0.2.13             pyhd8ed1ab_0    conda-forge
colcon-package-information 0.3.3                      py_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
colcon-parallel-executor  0.2.4                      py_0    conda-forge
colcon-pkg-config         0.1.0                      py_0    conda-forge
colcon-powershell         0.3.6                      py_0    conda-forge
colcon-python-setup-py    0.2.7                      py_0    conda-forge
colcon-recursive-crawl    0.2.1                      py_0    conda-forge
colcon-ros                0.3.23             pyhd8ed1ab_0    conda-forge
colcon-test-result        0.3.8                      py_0    conda-forge
colcon-zsh                0.4.0              pyhd8ed1ab_0    conda-forge
colorama                  0.4.6              pyhd8ed1ab_0    conda-forge
coloredlogs               15.0.1             pyhd8ed1ab_3    conda-forge
compiler-rt               15.0.7               hf8d1dfb_1    conda-forge
compiler-rt_osx-arm64     15.0.7               hf8d1dfb_1    conda-forge
compilers                 1.6.0                hce30654_0    conda-forge
console_bridge            1.0.2                h3e96240_1    conda-forge
contourpy                 1.1.0            py39hbd775c9_0    conda-forge
coreutils                 9.4                  hb547adb_0    conda-forge
coverage                  7.3.1            py39h0f82c59_1    conda-forge
cppzmq                    4.10.0               h6c41c7e_0    conda-forge
[... 566 lines cut ...]
xorg-kbproto              1.0.7             h27ca646_1002    conda-forge
xorg-libice               1.0.10               h27ca646_0    conda-forge
xorg-libsm                1.2.3             h27ca646_1000    conda-forge
xorg-libx11               1.8.4                h1a8c8d9_0    conda-forge
xorg-libxau               1.0.11               hb547adb_0    conda-forge
xorg-libxaw               1.0.14               h3422bc3_1    conda-forge
xorg-libxdmcp             1.1.3                h27ca646_0    conda-forge
[... cut ...]
yaml                      0.2.5                h3422bc3_2    conda-forge
yaml-cpp                  0.7.0                hb7217d7_2    conda-forge
zeromq                    4.3.4                hbdafb3b_1    conda-forge
zipp                      3.16.2             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               h53f4e23_5    conda-forge
zstd                      1.5.5                h4f39d0f_0    conda-forge
zziplib                   0.13.69              he1e0b03_1    conda-forge
```

Tobias-Fischer, 2023-09-29:
Could you please provide the full output?

Tobias-Fischer, 2023-10-08:
Closing until we have full log.

closed 2023-10-08 by Tobias-Fischer