RoboStack/ros-noetic #412: Not able to echo topic published from mamba env on other machine

amit-420, 2023-10-02:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

I am trying to publish from a node in mamba env running on rpi 4( noetic), and subscribe it from another machine ( rpi 4 melodic).

I have set ros_hostname as ip address of corresponding machine itself and ros_master_uri of both the machines are set to master machine( which is rpi4 melodic).

But I am not able to echo the published messages from the slave machine ( rpi4 noetic)  on the master machine.

Ping to the topic from master machine give below error.
```
rosnode: node is [/force_plate_7417_1696252816221]
pinging /force_plate_7417_1696252816221 with a timeout of 3.0s
ERROR: connection refused to [http://raspberrypi:41421/]
```

### Installed packages

```shell
# packages in environment at /home/pi/miniforge3/envs/ros_env:
#
# Name                    Version                   Build  Channel
_openmp_mutex             4.5                       2_gnu    conda-forge
adwaita-icon-theme        43                       unix_0    conda-forge
alsa-lib                  1.2.8                h4e544f5_0    conda-forge
aom                       3.5.0                headf329_0    conda-forge
apr                       1.7.0                hf897c2e_5    conda-forge
assimp                    5.2.5                h3fd5656_0    conda-forge
atk-1.0                   2.38.0               hf4e84e4_1    conda-forge
attr                      2.5.1                h4e544f5_1    conda-forge
bcrypt                    4.0.1            py39hf916c31_1    conda-forge
boost                     1.78.0           py39hc2250db_5    conda-forge
boost-cpp                 1.78.0               h484ac13_2    conda-forge
brotli                    1.1.0                h31becfc_1    conda-forge
brotli-bin                1.1.0                h31becfc_1    conda-forge
bzip2                     1.0.8                hf897c2e_4    conda-forge
c-ares                    1.19.1               h31becfc_0    conda-forge
ca-certificates           2023.7.22            hcefe29a_0    conda-forge
cairo                     1.16.0            hd19fb6e_1014    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
certifi                   2023.7.22          pyhd8ed1ab_0    conda-forge
cffi                      1.16.0           py39hdf53b9e_0    conda-forge
cmake                     3.26.3               h5819af1_0    conda-forge
console_bridge            1.0.2                hdd96247_1    conda-forge
contourpy                 1.1.1            py39hd16970a_1    conda-forge
cryptography              41.0.4           py39hea2e5f7_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
dbus                      1.13.6               h12b9eeb_3    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.8.0              pyhd8ed1ab_0    conda-forge
docutils                  0.20.1           py39ha65689a_2    conda-forge
eigen                     3.4.0                h2a328a1_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.5.0                hd600fc2_1    conda-forge
ffmpeg                    5.1.2           gpl_h8bd3c30_106    conda-forge
fftw                      3.3.10          nompi_h2dcef8e_108    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.14.2               ha9a116f_0    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.43.0           py39h898b7ef_0    conda-forge
freeglut                  3.2.2                h01db608_1    conda-forge
freeimage                 3.18.0              h6af9e85_12    conda-forge
freetype                  2.12.1               hf0a5ef3_2    conda-forge
[... 417 lines cut ...]
 populated config files : /home/pi/miniforge3/.condarc
                          /home/pi/miniforge3/envs/ros_env/.condarc
          conda version : 23.3.1
    conda-build version : not installed
         python version : 3.10.12.final.0
       virtual packages : __archspec=1=aarch64
                          __glibc=2.31=0
                          __linux=6.1.21=0
                          __unix=0=0
       base environment : /home/pi/miniforge3  (writable)
      conda av data dir : /home/pi/miniforge3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/linux-aarch64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/linux-aarch64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/pi/miniforge3/pkgs
                          /home/pi/.conda/pkgs
       envs directories : /home/pi/miniforge3/envs
                          /home/pi/.conda/envs
               platform : linux-aarch64
             user-agent : conda/23.3.1 requests/2.31.0 CPython/3.10.12 Linux/6.1.21-v8+ debian/11 glibc/2.31
                UID:GID : 1000:1000
             netrc file : None
```

traversaro, 2023-10-03:
Does this work if you use Noetic on both sides?

amit-420, 2023-10-03:
No, I have checked it with another system with noetic but the same issue.

traversaro, 2023-10-03:
Did you try with a apt-based installation of ROS Noetic? I am not sure if this error is related to the specific conda packaging of ROS.

Tobias-Fischer, 2023-10-03:
To me this looks like an issue with http://wiki.ros.org/ROS/NetworkSetup

amit-420, 2023-10-05:
Now, it is working, I am not sure why. I have made no changes in the code or ros_hostname and ros_master_uri. The only thing, I think, I changed was starting wps setting of my router( which was because i restarted my router).

Tobias-Fischer, 2023-10-05:
Great

closed 2023-10-05 by Tobias-Fischer