RoboStack/ros-noetic #414: Cannot find GL/gl.h

aravindsiv, 2023-10-04:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

Hi, 

I am trying to build a package that depends on `glfw3`. I was able to install this using `mamba install glfw.` However, when I run `catkin_make,` I get the following error:

```
/home/aravind/mambaforge/envs/ros_env/include/GLFW/glfw3.h:241:12: fatal error: GL/gl.h: No such file or directory
  241 |   #include <GL/gl.h>
```

### Installed packages

```shell
# packages in environment at /home/aravind/mambaforge/envs/ros_env:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
adwaita-icon-theme        43                       unix_0    conda-forge
alsa-lib                  1.2.8                h166bdaf_0    conda-forge
aom                       3.5.0                h27087fc_0    conda-forge
apr                       1.7.0                h7f98852_5    conda-forge
argcomplete               3.1.2              pyhd8ed1ab_0    conda-forge
assimp                    5.2.5                hf40c2ba_0    conda-forge
atk-1.0                   2.38.0               hd4edc92_1    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
bcrypt                    4.0.1            py39h9fdd4d6_1    conda-forge
binutils                  2.40                 hdd6e379_0    conda-forge
binutils_impl_linux-64    2.40                 hf600244_0    conda-forge
binutils_linux-64         2.40                 hbdbef99_2    conda-forge
boost                     1.78.0           py39hda80f44_5    conda-forge
boost-cpp                 1.78.0               h5adbc97_2    conda-forge
brotli                    1.1.0                hd590300_1    conda-forge
brotli-bin                1.1.0                hd590300_1    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.19.1               hd590300_0    conda-forge
c-compiler                1.6.0                hd590300_0    conda-forge
ca-certificates           2023.7.22            hbcca054_0    conda-forge
cairo                     1.16.0            ha61ee94_1014    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_0    conda-forge
certifi                   2023.7.22          pyhd8ed1ab_0    conda-forge
cffi                      1.15.1           py39h7a31438_5    conda-forge
cmake                     3.26.3               h077f3f9_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.5.0              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
colcon-cmake              0.2.27             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0            py39hf3d152e_1    conda-forge
colcon-core               0.12.1             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.2.2                      py_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-output             0.2.13             pyhd8ed1ab_0    conda-forge
colcon-package-information 0.3.3                      py_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
colcon-parallel-executor  0.2.4                      py_0    conda-forge
colcon-pkg-config         0.1.0                      py_0    conda-forge
colcon-powershell         0.3.6                      py_0    conda-forge
colcon-python-setup-py    0.2.7                      py_0    conda-forge
colcon-recursive-crawl    0.2.1                      py_0    conda-forge
colcon-ros                0.3.23             pyhd8ed1ab_0    conda-forge
colcon-test-result        0.3.8                      py_0    conda-forge
colcon-zsh                0.4.0              pyhd8ed1ab_0    conda-forge
[... 492 lines cut ...]
          conda version : 23.3.1
    conda-build version : not installed
         python version : 3.10.12.final.0
       virtual packages : __archspec=1=x86_64
                          __cuda=12.2=0
                          __glibc=2.35=0
                          __linux=6.2.0=0
                          __unix=0=0
       base environment : /home/aravind/mambaforge  (writable)
      conda av data dir : /home/aravind/mambaforge/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/linux-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/aravind/mambaforge/pkgs
                          /home/aravind/.conda/pkgs
       envs directories : /home/aravind/mambaforge/envs
                          /home/aravind/.conda/envs
               platform : linux-64
             user-agent : conda/23.3.1 requests/2.31.0 CPython/3.10.12 Linux/6.2.0-33-generic ubuntu/22.04.3 glibc/2.35
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False
```

traversaro, 2023-10-04:
You need to install the packages that provide OpenGL, it should be sufficient to install:
~~~
mamba install mesa-libgl-devel-cos7-x86_64 mesa-libgl-devel-cos7-x86_64 libselinux-cos7-x86_64 libxdamage-cos7-x86_64  libxxf86vm-cos7-x86_64  libxext-cos7-x86_64  xorg-libxfixes
~~~

See https://github.com/RoboStack/vinca/blob/f1d8bd35c705323faa620a6d5d59d18f8abb9d43/vinca/main.py#L494 and https://conda-forge.org/docs/maintainer/knowledge_base.html#libgl .

traversaro, 2023-10-04:
Probably by adding REQUIRE_GL in https://github.com/RoboStack/ros-noetic/blob/b4a46111611b36c8619c9a750ae9250d172a51b8/robostack.yaml#L259 this would be handled automatically by rosdep ?

aravindsiv, 2023-10-05:
Thx @traversaro . I tried the `mamba install` you suggested and got the following error message:

```
Could not solve for environment specs
The following packages are incompatible
├─ libselinux-cos7-x86_64   is installable and it requires
│  └─ sysroot_linux-64 2.17.*  with the potential options
│     ├─ sysroot_linux-64 2.17 would require
│     │  └─ kernel-headers_linux-64 [3.10.0 h4a8ded7_12|3.10.0 h4a8ded7_13] with the potential options
│     │     ├─ kernel-headers_linux-64 3.10.0 would require
│     │     │  └─ sysroot_linux-64 2.17 , which can be installed;
│     │     └─ kernel-headers_linux-64 3.10.0 would require
│     │        └─ sysroot_linux-64 2.17 , which can be installed;
│     ├─ sysroot_linux-64 2.17 would require
│     │  └─ kernel-headers_linux-64 3.10.0 h4a8ded7_11, which conflicts with any installable versions previously reported;
│     └─ sysroot_linux-64 2.17 would require
│        └─ kernel-headers_linux-64 3.10.0 h77966d4_10, which conflicts with any installable versions previously reported;
├─ libxdamage-cos7-x86_64   is installable and it requires
│  └─ sysroot_linux-64 2.17.*  with the potential options
│     ├─ sysroot_linux-64 2.17, which can be installed (as previously explained);
│     ├─ sysroot_linux-64 2.17, which cannot be installed (as previously explained);
│     └─ sysroot_linux-64 2.17, which cannot be installed (as previously explained);
├─ libxext-cos7-x86_64   is installable with the potential options
│  ├─ libxext-cos7-x86_64 1.3.3 would require
│  │  └─ libx11-cos7-x86_64 >=1.5.99.902 *_1102, which can be installed;
│  ├─ libxext-cos7-x86_64 1.3.3 would require
│  │  └─ sysroot_linux-64 2.17.*  with the potential options
│  │     ├─ sysroot_linux-64 2.17, which can be installed (as previously explained);
│  │     ├─ sysroot_linux-64 2.17, which cannot be installed (as previously explained);
│  │     └─ sysroot_linux-64 2.17, which cannot be installed (as previously explained);
│  ├─ libxext-cos7-x86_64 1.3.3 would require
│  │  ├─ libx11-cos7-x86_64 >=1.5.99.902 *_1104, which conflicts with any installable versions previously reported;
│  │  └─ sysroot_linux-64 2.17.*  with the potential options
│  │     ├─ sysroot_linux-64 2.17, which can be installed (as previously explained);
│  │     ├─ sysroot_linux-64 2.17, which cannot be installed (as previously explained);
│  │     └─ sysroot_linux-64 2.17, which cannot be installed (as previously explained);
│  └─ libxext-cos7-x86_64 1.3.3 would require
│     ├─ libx11-cos7-x86_64 >=1.5.99.902 *_1103, which conflicts with any installable versions previously reported;
│     └─ sysroot_linux-64 2.17.*  with the potential options
│        ├─ sysroot_linux-64 2.17, which can be installed (as previously explained);
│        ├─ sysroot_linux-64 2.17, which cannot be installed (as previously explained);
│        └─ sysroot_linux-64 2.17, which cannot be installed (as previously explained);
├─ libxxf86vm-cos7-x86_64   is installable with the potential options
│  ├─ libxxf86vm-cos7-x86_64 1.1.4 would require
│  │  └─ sysroot_linux-64 2.17.*  with the potential options
│  │     ├─ sysroot_linux-64 2.17, which can be installed (as previously explained);
│  │     ├─ sysroot_linux-64 2.17, which cannot be installed (as previously explained);
│  │     └─ sysroot_linux-64 2.17, which cannot be installed (as previously explained);
│  ├─ libxxf86vm-cos7-x86_64 1.1.4 would require
│  │  └─ libx11-cos7-x86_64 >=1.5.99.902 *_1105, which conflicts with any installable versions previously reported;
│  ├─ libxxf86vm-cos7-x86_64 1.1.4 would require
│  │  └─ libx11-cos7-x86_64 >=1.5.99.902 *_1104, which conflicts with any installable versions previously reported;
│  └─ libxxf86vm-cos7-x86_64 1.1.4 would require
│     └─ libx11-cos7-x86_64 >=1.5.99.902 *_1103, which conflicts with any installable versions previously reported;
└─ mesa-libgl-devel-cos7-x86_64   is installable and it requires
   └─ sysroot_linux-64 2.17.*  with the potential options
      ├─ sysroot_linux-64 2.17, which can be installed (as previously explained);
      ├─ sysroot_linux-64 2.17, which cannot be installed (as previously explained);
      └─ sysroot_linux-64 2.17, which cannot be installed (as previously explained).
```

traversaro, 2023-10-05:
Sorry, I think I forgot to specify the channels, can you try now?

traversaro, 2023-10-05:
Sorry, I made an error, cos6 should be used in place of cos7 .

traversaro, 2023-10-05:
Sorry, I made an error, cos6 should be used in place of cos7 .

~~~
mamba install -c conda-forge -c robostack-staging mesa-libgl-devel-cos6-x86_64 mesa-libgl-devel-cos6-x86_64 libselinux-cos6-x86_64 libxdamage-cos6-x86_64  libxxf86vm-cos6-x86_64  libxext-cos6-x86_64  xorg-libxfixes
~~~