RoboStack/ros-noetic #415: Fail to load gazebo ros plugins due to undefined symbol

alaurenzi, 2023-10-04:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

I am on a fresh ros1 environment. Running `rosrun gazebo_ros gazebo --verbose` gives
```
ros(ros1) alaurenzi@alaurenzi-iit-yoga:~$ rosrun gazebo_ros gazebo --verbose
Gazebo multi-robot simulator, version 11.12.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Err] [Plugin.hh:212] Failed to load plugin /home/iit.local/alaurenzi/miniforge3/envs/ros1/lib/libgazebo_ros_paths_plugin.so: /home/iit.local/alaurenzi/miniforge3/envs/ros1/lib/libgazebo_ros_paths_plugin.so: undefined symbol: _ZN6gazebo6common11SystemPaths8InstanceEv
[Err] [Plugin.hh:212] Failed to load plugin /home/iit.local/alaurenzi/miniforge3/envs/ros1/lib/libgazebo_ros_api_plugin.so: /home/iit.local/alaurenzi/miniforge3/envs/ros1/lib/libgazebo_ros_api_plugin.so: undefined symbol: _ZN6gazebo9transport12TopicManager8InstanceEv
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 10.240.29.84
[Msg] Loading world file [/home/iit.local/alaurenzi/miniforge3/envs/ros1/share/gazebo-11/worlds/empty.world]
```

Some additional info: those symbols are actually missing, "similar" ones exist maybe due to version mismatch involving changes in how gazebo deals with singletons ?
```
(ros1) alaurenzi@alaurenzi-iit-yoga:~/miniforge3/envs/ros1/lib$ nm -D libgazebo_transport.so | grep TopicManager | grep Instance
00000000000a2148 u _ZGVZN10SingletonTIN6gazebo9transport12TopicManagerEE11GetInstanceEvE1t
000000000005fe80 W _ZN10SingletonTIN6gazebo9transport12TopicManagerEE11GetInstanceEv
000000000005fde0 W _ZN10SingletonTIN6gazebo9transport12TopicManagerEE8InstanceEv
00000000000a2160 u _ZZN10SingletonTIN6gazebo9transport12TopicManagerEE11GetInstanceEvE1t

(ros1) alaurenzi@alaurenzi-iit-yoga:~/miniforge3/envs/ros1/lib$ nm -D libgazebo_common.so | grep SystemPaths | grep Instance
000000000012c768 u _ZGVZN10SingletonTIN6gazebo6common11SystemPathsEE11GetInstanceEvE1t
000000000006d8d0 W _ZN10SingletonTIN6gazebo6common11SystemPathsEE11GetInstanceEv
000000000006d830 W _ZN10SingletonTIN6gazebo6common11SystemPathsEE8InstanceEv
00000000000ded70 T _ZNK6gazebo6common11SystemPaths15TmpInstancePathB5cxx11Ev
000000000012c780 u _ZZN10SingletonTIN6gazebo6common11SystemPathsEE11GetInstanceEvE1t
``` 

Do not hesitate to ask for further tests!

Thanks,
Arturo

### Installed packages

```shell
(ros1) alaurenzi@alaurenzi-iit-yoga:~$ mamba list
# packages in environment at /home/iit.local/alaurenzi/miniforge3/envs/ros1:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
_sysroot_linux-64_curr_repodata_hack 3                   h69a702a_13    conda-forge
adwaita-icon-theme        43                       unix_0    conda-forge
aiohttp                   3.8.5            py39hd1e30aa_0    conda-forge
aiosignal                 1.3.1              pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.8                h166bdaf_0    conda-forge
anytree                   2.9.0                    pypi_0    pypi
aom                       3.5.0                h27087fc_0    conda-forge
apr                       1.7.0                h7f98852_5    conda-forge
argcomplete               3.1.2              pyhd8ed1ab_0    conda-forge
assimp                    5.2.5                hf40c2ba_0    conda-forge
async-timeout             4.0.3              pyhd8ed1ab_0    conda-forge
atk-1.0                   2.38.0               hd4edc92_1    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
attrs                     23.1.0             pyh71513ae_1    conda-forge
bash-completion           2.11                 ha770c72_1    conda-forge
bcrypt                    4.0.1            py39h9fdd4d6_1    conda-forge
binutils                  2.40                 hdd6e379_0    conda-forge
binutils_impl_linux-64    2.40                 hf600244_0    conda-forge
[... 714 lines cut ...]
          conda version : 23.3.1
    conda-build version : not installed
         python version : 3.10.12.final.0
       virtual packages : __archspec=1=x86_64
                          __cuda=12.2=0
                          __glibc=2.35=0
                          __linux=6.2.0=0
                          __unix=0=0
       base environment : /home/iit.local/alaurenzi/miniforge3  (writable)
      conda av data dir : /home/iit.local/alaurenzi/miniforge3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/linux-64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/iit.local/alaurenzi/miniforge3/pkgs
                          /home/iit.local/alaurenzi/.conda/pkgs
       envs directories : /home/iit.local/alaurenzi/miniforge3/envs
                          /home/iit.local/alaurenzi/.conda/envs
               platform : linux-64
             user-agent : conda/23.3.1 requests/2.31.0 CPython/3.10.12 Linux/6.2.0-33-generic ubuntu/22.04.3 glibc/2.35
                UID:GID : 1021419905:1021400513
             netrc file : None
           offline mode : False
```

Tobias-Fischer, 2023-10-04:
Any idea @traversaro?

alaurenzi, 2023-10-04:
Just adding that compiling gazebo_ros_pkgs from source works ok (as expected)

traversaro, 2023-10-04:
> Any idea @traversaro?

Fun fact: when you did this comment I was actually sitting side by side with @alaurenzi for a total unrelated reason.

As a first step, in the following environment `rosrun gazebo_ros gazebo --verbose`  is working fine for me:
~~~
(ros1gazebo) traversaro@IITICUBLAP257:~$ rosrun gazebo_ros gazebo --verbose
Gazebo multi-robot simulator, version 11.12.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[ INFO] [1696456303.682239229]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1696456303.684034729]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 172.24.171.139
[Msg] Loading world file [/home/traversaro/miniforge3/envs/ros1gazebo/share/gazebo-11/worlds/empty.world]
[ INFO] [1696456305.281463619]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1696456305.317375412, 0.014000000]: Physics dynamic reconfigure ready.
~~~

env: 
~~~
(ros1gazebo) traversaro@IITICUBLAP257:~$ mamba list
# packages in environment at /home/traversaro/miniforge3/envs/ros1gazebo:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
adwaita-icon-theme        43                       unix_0    conda-forge
alsa-lib                  1.2.8                h166bdaf_0    conda-forge
aom                       3.5.0                h27087fc_0    conda-forge
apr                       1.7.0                h7f98852_5    conda-forge
assimp                    5.2.5                hf40c2ba_0    conda-forge
atk-1.0                   2.38.0               hd4edc92_1    conda-forge
attr                      2.5.1                h166bdaf_1    conda-forge
bcrypt                    4.0.1            py39h9fdd4d6_1    conda-forge
blosc                     1.21.5               h0f2a231_0    conda-forge
boost                     1.78.0           py39hda80f44_5    conda-forge
boost-cpp                 1.78.0               h5adbc97_2    conda-forge
brotli                    1.1.0                hd590300_1    conda-forge
brotli-bin                1.1.0                hd590300_1    conda-forge
bullet-cpp                3.21                 h4661b88_4    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.19.1               hd590300_0    conda-forge
ca-certificates           2023.7.22            hbcca054_0    conda-forge
cairo                     1.16.0            ha61ee94_1014    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
certifi                   2023.7.22          pyhd8ed1ab_0    conda-forge
cffi                      1.16.0           py39h7a31438_0    conda-forge
cfitsio                   4.2.0                hd9d235c_0    conda-forge
cmake                     3.26.3               h077f3f9_0    conda-forge
console_bridge            1.0.2                h924138e_1    conda-forge
contourpy                 1.1.1            py39h7633fee_1    conda-forge
cppzmq                    4.10.0               h7e20d1c_0    conda-forge
cryptography              41.0.4           py39hd4f0224_0    conda-forge
curl                      7.88.1               hdc1c0ab_1    conda-forge
cycler                    0.12.0             pyhd8ed1ab_0    conda-forge
dartsim                   6.13.0               hbbc9fda_0    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
defusedxml                0.7.1              pyhd8ed1ab_0    conda-forge
distro                    1.8.0              pyhd8ed1ab_0    conda-forge
docutils                  0.20.1           py39hf3d152e_2    conda-forge
eigen                     3.4.0                h00ab1b0_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.5.0                hcb278e6_1    conda-forge
fcl                       0.7.0                hdb826dc_3    conda-forge
ffmpeg                    5.1.2           gpl_h8dda1f0_106    conda-forge
fftw                      3.3.10          nompi_hc118613_108    conda-forge
[... 452 lines cut ...]
xorg-libice               1.0.10               h7f98852_0    conda-forge
xorg-libsm                1.2.3             hd9c2040_1000    conda-forge
xorg-libx11               1.8.4                h0b41bf4_0    conda-forge
xorg-libxau               1.0.11               hd590300_0    conda-forge
xorg-libxaw               1.0.14               h7f98852_1    conda-forge
xorg-libxdmcp             1.1.3                h7f98852_0    conda-forge
xorg-libxext              1.3.4                h0b41bf4_2    conda-forge
xorg-libxfixes            5.0.3             h7f98852_1004    conda-forge
xorg-libxi                1.7.10               h7f98852_0    conda-forge
xorg-libxmu               1.1.3                h7f98852_0    conda-forge
xorg-libxpm               3.5.16               hd590300_0    conda-forge
xorg-libxrender           0.9.10            h7f98852_1003    conda-forge
xorg-libxt                1.3.0                hd590300_0    conda-forge
xorg-renderproto          0.11.1            h7f98852_1002    conda-forge
[... cut ...]
yaml                      0.2.5                h7f98852_2    conda-forge
yaml-cpp                  0.7.0                h59595ed_3    conda-forge
zeromq                    4.3.4                h9c3ff4c_1    conda-forge
zipp                      3.17.0             pyhd8ed1ab_0    conda-forge
zlib                      1.2.13               hd590300_5    conda-forge
zstd                      1.5.5                hfc55251_0    conda-forge
zziplib                   0.13.69              h27826a3_1    conda-forge
~~~

traversaro, 2023-10-04:
Manually installing the older gazebo I can reproduce the problem:
~~~
mamba install gazebo=11.12.0=*_5
─────────────────────────────────────────────────────────────────
  Change:
─────────────────────────────────────────────────────────────────

  - gazebo   11.12.0  h327f067_9  conda-forge
  + gazebo   11.12.0  h151eb85_5  conda-forge/linux-64     58MB

  Downgrade:
─────────────────────────────────────────────────────────────────

  - dartsim   6.13.0  hbbc9fda_0  conda-forge
  + dartsim   6.12.2  h0a4c592_3  conda-forge/linux-64     16MB
~~~

Log:
~~~
(ros1gazebo) traversaro@IITICUBLAP257:~$ rosrun gazebo_ros gazebo --verbose
Gazebo multi-robot simulator, version 11.12.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Err] [Plugin.hh:212] Failed to load plugin /home/traversaro/miniforge3/envs/ros1gazebo/lib/libgazebo_ros_paths_plugin.so: /home/traversaro/miniforge3/envs/ros1gazebo/lib/libgazebo_ros_paths_plugin.so: undefined symbol: _ZN6gazebo6common11SystemPaths8InstanceEv
[Err] [Plugin.hh:212] Failed to load plugin /home/traversaro/miniforge3/envs/ros1gazebo/lib/libgazebo_ros_api_plugin.so: /home/traversaro/miniforge3/envs/ros1gazebo/lib/libgazebo_ros_api_plugin.so: undefined symbol: _ZN6gazebo9transport12TopicManager8InstanceEv
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 172.24.171.139
[Msg] Loading world file [/home/traversaro/miniforge3/envs/ros1gazebo/share/gazebo-11/worlds/empty.world]
~~~

traversaro, 2023-10-04:
So, I guess this is related to https://github.com/gazebosim/gazebo-classic/pull/3269 and https://github.com/conda-forge/gazebo-feedstock/pull/162, but I am not sure how. I will come back to this tomorrow.

traversaro, 2023-10-05:
> [!IMPORTANT]  
> **TL;DR:** Hotfix to get @alaurenzi's environment to work: 
> `mamba install -c conda-forge -c robostack-staging ros-noetic-gazebo-ros=2.9.2=*_15`

I think I got a clear picture on this. The root cause is a combination of a some bugs:
* Around 2023/02/15 in https://github.com/conda-forge/gazebo-feedstock/pull/162 (that changed gazebo version from `11.12.0=*_5` to `11.12.0=*_6`) I added symbols to the ABI of a package, without changing its version (only changing its build number). I never realized this, but this can lead to problems, as any downstream package that  builds against `11.12.0=*_6` will only get a run_constraint of `gazebo >= 11.12.0`, and nothing prevents that the package get installed with an earlier version of gazebo that does not have the symbols added in `11.12.0=*_6`.
* On 2023/09/23, a version of ros-noetic-gazebo-ros `2.9.2=*_16` was uploaded (see https://anaconda.org/robostack-staging/ros-noetic-gazebo-ros/files), and that version of ros-noetic-gazebo-ros was compiled against `11.12.0=*_11`, while the earlier version ros-noetic-gazebo-ros `2.9.2=*_15` was compiled against gazebo `11.12.0=*_5`
* (Minor): Even if we had a way to account for the build version of gazebo, the metadata of ros-noetic-gazebo-ros does not state that gazebo is a dependency, so boa has no idea that it should had a run constraint on ros-noetic-gazebo-ros. Perhaps vinca should be fixed to handle correctly `build_export_depend` in https://github.com/ros-simulation/gazebo_ros_pkgs/blob/398cde933cba5d72921055d57d4faa47c4445654/gazebo_dev/package.xml#L23C1-L23C61 ?

traversaro, 2023-10-05:
I updated the last comment with some request for further info from @alaurenzi .

alaurenzi, 2023-10-05:
@traversaro thanks for promptly looking into this

```
(ros1) alaurenzi@alaurenzi-iit-yoga:~ $ mamba env export --from-history
name: ros1
channels:
  - robostack-staging
  - conda-forge
dependencies:
  - ros-noetic-desktop-full
  - compilers
  - cmake
  - pkg-config
  - make
  - ninja
  - colcon-common-extensions
  - catkin_tools
  - ca-certificates
  - certifi
  - openssl
  - bash-completion
  - ros-noetic-moveit
  - mesa-libgl-devel-cos7-x86_64
  - ros-noetic-realsense2-description
  - ros-noetic-rosmon
  - ros-noetic-velodyne-description
  - ros-noetic-velodyne-gazebo-plugins
  - ros-noetic-rviz-visual-tools
prefix: /home/iit.local/alaurenzi/miniforge3/envs/ros1
```
Just `scipy` from pip IIRC

traversaro, 2023-10-05:
I found a minimal reproducer:
~~~
traversaro@IITICUBLAP257:~/robostack415$ mamba create -n ros1 -c conda-forge -c robostack-staging ros-noetic-desktop-full ros-noetic-velodyne-gazebo-plugins gazebo=11.12.0=*_6

                  __    __    __    __
                 /  \  /  \  /  \  /  \
                /    \/    \/    \/    \
███████████████/  /██/  /██/  /██/  /████████████████████████
              /  / \   / \   / \   / \  \____
             /  /   \_/   \_/   \_/   \    o \__,
            / _/                       \_____/  `
            |/
        ███╗   ███╗ █████╗ ███╗   ███╗██████╗  █████╗
        ████╗ ████║██╔══██╗████╗ ████║██╔══██╗██╔══██╗
        ██╔████╔██║███████║██╔████╔██║██████╔╝███████║
        ██║╚██╔╝██║██╔══██║██║╚██╔╝██║██╔══██╗██╔══██║
        ██║ ╚═╝ ██║██║  ██║██║ ╚═╝ ██║██████╔╝██║  ██║
        ╚═╝     ╚═╝╚═╝  ╚═╝╚═╝     ╚═╝╚═════╝ ╚═╝  ╚═╝

        mamba (1.4.2) supported by @QuantStack

        GitHub:  https://github.com/mamba-org/mamba
        Twitter: https://twitter.com/QuantStack

█████████████████████████████████████████████████████████████

Looking for: ['ros-noetic-desktop-full', 'ros-noetic-velodyne-gazebo-plugins', 'gazebo==11.12.0[build=*_6]']

warning  libmamba Cache file "/home/traversaro/miniforge3/pkgs/cache/497deca9.json" was modified by another program
warning  libmamba Cache file "/home/traversaro/miniforge3/pkgs/cache/09cdf8bf.json" was modified by another program
robostack-staging/noarch                                      No change
robostack-staging/linux-64                                    No change
conda-forge/noarch                                  14.1MB @   4.9MB/s  3.2s
conda-forge/linux-64                                34.5MB @   6.5MB/s  6.0s
Could not solve for environment specs
The following packages are incompatible
├─ gazebo 11.12.0 *_6 is installable with the potential options
│  ├─ gazebo 11.12.0 would require
│  │  ├─ libgdal >=3.6.2,<3.7.0a0  with the potential options
│  │  │  ├─ libgdal 3.6.2 would require
│  │  │  │  └─ libtiff >=4.5.0,<4.6.0a0  with the potential options
│  │  │  │     ├─ libtiff 4.5.0 would require
│  │  │  │     │  └─ libdeflate >=1.14,<1.15.0a0 , which can be installed;
│  │  │  │     ├─ libtiff [4.5.0|4.5.1] would require
│  │  │  │     │  ├─ jpeg <0.0.0a , which can be installed;
│  │  │  │     │  └─ libjpeg-turbo >=2.1.5.1,<3.0a0 , which requires
│  │  │  │     │     └─ jpeg <0.0.0a , which can be installed;
│  │  │  │     ├─ libtiff 4.5.0, which can be installed;
│  │  │  │     └─ libtiff 4.5.0 would require
│  │  │  │        └─ libdeflate >=1.16,<1.17.0a0 , which can be installed;
│  │  │  ├─ libgdal 3.6.2 would require
│  │  │  │  ├─ lerc >=4.0.0,<5.0a0 , which can be installed;
│  │  │  │  └─ libdeflate >=1.17,<1.18.0a0 , which conflicts with any installable versions previously reported;
│  │  │  ├─ libgdal [3.6.2|3.6.3] would require
│  │  │  │  └─ libjpeg-turbo >=2.1.5.1,<3.0a0 , which can be installed (as previously explained);
│  │  │  ├─ libgdal 3.6.2 would require
│  │  │  │  ├─ lerc >=4.0.0,<5.0a0 , which can be installed;
│  │  │  │  ├─ libdeflate >=1.17,<1.18.0a0 , which conflicts with any installable versions previously reported;
│  │  │  │  └─ poppler >=23.1.0,<23.2.0a0 , which requires
│  │  │  │     └─ libtiff >=4.5.0,<4.6.0a0  with the potential options
│  │  │  │        ├─ libtiff 4.5.0, which can be installed (as previously explained);
│  │  │  │        ├─ libtiff [4.5.0|4.5.1], which can be installed (as previously explained);
│  │  │  │        ├─ libtiff 4.5.0, which can be installed;
│  │  │  │        └─ libtiff 4.5.0, which can be installed (as previously explained);
│  │  │  ├─ libgdal [3.6.3|3.6.4] would require
│  │  │  │  └─ icu >=72.1,<73.0a0 , which can be installed;
│  │  │  ├─ libgdal 3.6.4 would require
│  │  │  │  └─ libcurl >=8.3.0,<9.0a0 , which can be installed;
│  │  │  ├─ libgdal 3.6.4 would require
│  │  │  │  └─ libcurl >=8.2.1,<9.0a0 , which can be installed;
[... 96 lines cut ...]
   │  │           └─ python_abi 3.9.* *_cp39, which conflicts with any installable versions previously reported;
   │  └─ ros-noetic-vision-opencv  , which requires
   │     └─ ros-noetic-image-geometry  , which requires
   │        └─ libopencv >=4.6.0,<4.6.1.0a0  but there are no viable options
   │           ├─ libopencv 4.6.0 would require
   │           │  └─ harfbuzz [>=4.4.1,<5.0a0 |>=5.0.1,<6.0a0 |>=5.1.0,<6.0a0 |>=5.3.0,<6.0a0 ], which requires
   │           │     └─ icu >=70.1,<71.0a0 , which conflicts with any installable versions previously reported;
   │           └─ libopencv 4.6.0 would require
   │              ├─ harfbuzz >=6.0.0,<7.0a0  with the potential options
   │              │  ├─ harfbuzz 6.0.0, which can be installed (as previously explained);
   │              │  └─ harfbuzz 6.0.0, which cannot be installed (as previously explained);
   │              └─ jpeg >=9e,<10a , which conflicts with any installable versions previously reported;
   └─ ros-noetic-urdf-sim-tutorial  , which requires
      ├─ ros-noetic-rqt-robot-steering  , which requires
      │  └─ ros-noetic-python-qt-binding  , which requires
      │     └─ qt-main >=5.15.8,<5.16.0a0  with the potential options
      │        ├─ qt-main 5.15.8, which can be installed (as previously explained);
      │        ├─ qt-main 5.15.8, which cannot be installed (as previously explained);
      │        ├─ qt-main 5.15.8, which cannot be installed (as previously explained);
      │        ├─ qt-main 5.15.8, which cannot be installed (as previously explained);
      │        └─ qt-main 5.15.8, which cannot be installed (as previously explained);
      └─ ros-noetic-rviz  , which requires
         └─ ros-noetic-resource-retriever  , which requires
            └─ libcurl >=7.87.0,<8.0a0 , which conflicts with any installable versions previously reported.
~~~

traversaro, 2023-10-05:
An easier fix perhaps is to pin gazebo in to `11.12.0=*_5` in https://github.com/RoboStack/ros-noetic/blob/master/conda_build_config.yaml, and recompile `ros-noetic-gazebo-ros`. This should ensure that a `ros-noetic-gazebo-ros` build that does not use any symbol introduced in `11.12.0=*_6` is the one installed, and so ensure that it can run fine with gazebo version >=11.12.0 , regardless of the build version.

alaurenzi, 2024-01-22:
Hi maintainers! Any news on this?

closed 2024-01-29 by Tobias-Fischer