RoboStack/ros-noetic #417: ros-noetic-cv-bridge does not work on osx-arm64 although the official docs say otherwise

ChristianEllis, 2023-10-14:
### Solution to issue cannot be found in the documentation.

- [X] I checked the documentation.

### Issue

According to the [official documentation](https://robostack.github.io/noetic.html) listing ros-noetic supported packages, the package `ros-noetic-cv-bridge` does not work on `osx-arm64`.  Namely, the boost package seems to be missing. I get the following output when trying to use this package.
```
miniforge3/envs/ros_env/lib/python3.9/site-packages/cv_bridge/core.py", line 91, in encoding_to_cvtype2
    from cv_bridge.boost.cv_bridge_boost import getCvType
ModuleNotFoundError: No module named 'cv_bridge.boost'
```

After introspection, I can confirm that the boost module does not exist in the official robostack staging conda package for `osx-arm64`, found [here](https://anaconda.org/robostack-staging/ros-noetic-cv-bridge).

Can we please update this package to include boost?

### Installed packages

```shell
# packages in environment at /Users/christian/miniforge3/envs/ros_env:
#
# Name                    Version                   Build  Channel
adwaita-icon-theme        43                       unix_0    conda-forge
aom                       3.5.0                h7ea286d_0    conda-forge
apr                       1.7.0                h9b22ae9_5    conda-forge
argcomplete               3.1.2              pyhd8ed1ab_0    conda-forge
assimp                    5.2.5                h276577b_0    conda-forge
atk-1.0                   2.38.0               hcb7b3dd_1    conda-forge
bcrypt                    4.0.1            py39haff01fd_1    conda-forge
boost                     1.78.0           py39h819cc4c_5    conda-forge
boost-cpp                 1.78.0               hf1d6563_2    conda-forge
brotli                    1.1.0                hb547adb_1    conda-forge
brotli-bin                1.1.0                hb547adb_1    conda-forge
bzip2                     1.0.8                h3422bc3_4    conda-forge
c-ares                    1.20.1               h93a5062_0    conda-forge
c-compiler                1.6.0                hd291e01_0    conda-forge
ca-certificates           2023.7.22            hf0a4a13_0    conda-forge
cairo                     1.16.0            h73a0509_1014    conda-forge
catkin_pkg                1.0.0              pyhd8ed1ab_0    conda-forge
catkin_tools              0.9.4              pyhd8ed1ab_0    conda-forge
cctools                   973.0.1             hd1ac623_15    conda-forge
cctools_osx-arm64         973.0.1             h2a25c60_15    conda-forge
certifi                   2023.7.22          pyhd8ed1ab_0    conda-forge
cffi                      1.16.0           py39he153c15_0    conda-forge
clang                     15.0.7               hce30654_3    conda-forge
clang-15                  15.0.7          default_h5dc8d65_3    conda-forge
clang_osx-arm64           15.0.7               h77e971b_3    conda-forge
clangxx                   15.0.7          default_h610c423_3    conda-forge
clangxx_osx-arm64         15.0.7               h768a7fd_3    conda-forge
cmake                     3.26.3               hf234bd0_0    conda-forge
colcon-argcomplete        0.3.3              pyhd8ed1ab_0    conda-forge
colcon-bash               0.5.0              pyhd8ed1ab_0    conda-forge
colcon-cd                 0.1.1              pyhd8ed1ab_0    conda-forge
colcon-cmake              0.2.28             pyhd8ed1ab_0    conda-forge
colcon-common-extensions  0.3.0            py39h2804cbe_1    conda-forge
colcon-core               0.15.0             pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.8              pyhd8ed1ab_0    conda-forge
colcon-devtools           0.2.5              pyhd8ed1ab_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-output             0.2.13             pyhd8ed1ab_0    conda-forge
colcon-package-information 0.3.3                      py_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
colcon-parallel-executor  0.2.4                      py_0    conda-forge
colcon-pkg-config         0.1.0                      py_0    conda-forge
colcon-powershell         0.4.0              pyhd8ed1ab_0    conda-forge
colcon-python-setup-py    0.2.7                      py_0    conda-forge
colcon-recursive-crawl    0.2.3              pyhd8ed1ab_0    conda-forge
colcon-ros                0.4.1              pyhd8ed1ab_0    conda-forge
[... 443 lines cut ...]
                          /Users/USER/.condarc
                          /Users/USER/miniforge3/envs/ros_env/.condarc
          conda version : 23.3.1
    conda-build version : not installed
         python version : 3.10.12.final.0
       virtual packages : __archspec=1=arm64
                          __osx=14.0=0
                          __unix=0=0
       base environment : /Users/USER/miniforge3  (writable)
      conda av data dir : /Users/USER/miniforge3/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-staging/osx-arm64
                          https://conda.anaconda.org/robostack-staging/noarch
                          https://conda.anaconda.org/conda-forge/osx-arm64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /Users/USER/miniforge3/pkgs
                          /Users/USER/.conda/pkgs
       envs directories : /Users/USER/miniforge3/envs
                          /Users/USER/.conda/envs
               platform : osx-arm64
             user-agent : conda/23.3.1 requests/2.31.0 CPython/3.10.12 Darwin/23.0.0 OSX/14.0
                UID:GID : 501:20
             netrc file : None
           offline mode : False
```

ChristianEllis, 2023-10-14:
To add to this, after downloading the `ros-noetic-cv-bridge-1.16.2-py39h099e9b2_5.tar.bz2` from this [link](https://anaconda.org/robostack-staging/ros-noetic-cv-bridge/files) for `osx-arm64`, in `lib/python3.9/site-packages/cv_bridge` is core.py, and in `lib/python3.10/site-packages/cv_bridge` is a directory called boost. Simlinking this directory into the directory `lib/python3.9/site-packages/cv_bridge` with the name "boost" has fixed the issue.  Is there a way to make this symlink somehow through conda?

traversaro, 2023-10-14:
> To add to this, after downloading the `ros-noetic-cv-bridge-1.16.2-py39h099e9b2_5.tar.bz2` from this [link](https://anaconda.org/robostack-staging/ros-noetic-cv-bridge/files) for `osx-arm64`, in `lib/python3.9/site-packages/cv_bridge` is core.py, and in `lib/python3.10/site-packages/cv_bridge` is a directory called `boost`. Symlinking this directory into the directory `lib/python3.9/site-packages/cv_bridge` with the name "boost" has fixed the issue. Is there a way to make this symlink somehow through conda?

Thanks for the inspection! Probably the error is exactly that, for some reason the boost directory got installed in `python3.10` for some reason, even if it should be installed `python3.9` instead.

traversaro, 2023-10-14:
I tested and the problem seems to be only in osx-arm64, all other architectures work fine.

mmattamala, 2023-12-05:
Related to this problem in osx-arm64, I also observed that `rqt_image_view` was not working due to similar issues which I solved by inspecting the `lib` directories.
Particularly, I was getting `ImportError: cannot import name 'libqt_gui_cpp_sip' from 'qt_gui_cpp'`. By symlinking `lib/python3.10/site_packages/qt_gui_cpp/libqt_gui_cpp_sip.cpython-39-darwin.so` into the corresponding python3.9 directory the problem also disappeared and rqt_image_view was opening as intended.

FYI, all the packages that seem to have components living in the python3.10 directory are:
- cv_bridge
- qt_gui_cpp
- rviz
- turtle_actionlib
- turtlesim
- xcbgen

Tobias-Fischer, 2024-02-03:
Hi @mmattamala - any chance you could please try this with the recently build packages?

Tobias-Fischer, 2024-02-06:
Should be all good :)

closed 2024-02-06 by Tobias-Fischer